syssi/esphome-jk-bms #5: Add BLE support

syssi, 2021-09-26:
https://forum.drbacke.de/viewtopic.php?f=8&t=982&start=10

```
# service
0xffe0
# characteristic
0xffe1

# request device info
{0xaa, 0x55, 0x90, 0xeb, 0x97, 0x00, 0xdf, 0x52, 0x88, 0x67, 0x9d, 0x0a, 0x09, 0x6b, 0x9a, 0xf6, 0x70, 0x9a, 0x17, 0xfd}

# request status
{0xaa, 0x55, 0x90, 0xeb, 0x96, 0x00, 0x79, 0x62, 0x96, 0xed, 0xe3, 0xd0, 0x82, 0xa1, 0x9b, 0x5b, 0x3c, 0x9c, 0x4b, 0x5d}

# data response
{0x55, 0xAA, 0xEB, 0x90, 0x02, ...}
```

syssi, 2022-02-26:
Todos:

- Combine & parse notifications
- Publish values
- Make sure there is a delay between the write commands

JohnGalt1717, 2022-04-16:
It would be really awesome if you could pull data from multiple over BLE in home assistant. I have 4 and really don't want to have 4 esp32s or something running. I'd love to be able to do all 4 in one shot. Obviously I could use a rs484 breakout chip, but not sure that that is supported with ESPhome yet.

Alternatively CanBus would be awesome with the ability to read multiples.

syssi, 2022-04-16:
@JohnGalt1717 The modbus interface / firmware of the JK-BMS doesn't support device addresses. In other words: You cannot attach multiple devices to the same (mod)bus. A ESP32 has 3 hardware UARTs so you can attach 3 BMS to 1 ESP without much effort.

In feature I will improve/add this BLE support so you will be able to query the data without connecting any cables.

supiiik, 2022-04-17:
> In feature I will improve/add this BLE support so you will be able to query the data without connecting any cables.

It would be great. At the moment I afraid (little bit) to connect BMS by wire because of the potential to destroy the board (https://github.com/syssi/esphome-jk-bms/issues/28). The BLE connection should be totally safe from this perspective.

syssi, 2022-04-17:
To be careful is a good idea but don't be too scared the RS485 (TTL) port is read-only if we can trust the seller.

supiiik, 2022-04-17:
Is it possible to connect only GND and RX (on ESP) to read data? Will this work? Or the ESP is sending data requests to BMS over TX?

syssi, 2022-04-17:
This doesn't work because the status frame must be requested.

JohnGalt1717, 2022-04-17:
FYI, one of the esp32 uarts is read only so you’ll get only 2 per esp32.

JohnGalt1717, 2022-04-17:
But it seems that as long as you can use 9600 baud you can have as many software uarts as pins…

https://esphome.io/components/uart.html

Does this work here to control many units?

syssi, 2022-04-17:
You are mixing things up here. The ESP32 has 3 hardware UARTs and a GPIO matrix. Because of the GPIO matrix you can use any GPIO as RX/TX pin (if the GPIO supports input and output). In summary: You can connect 3 BMS to the 3 hardware UARTs. The JK-BMS uses 115200 baud (in general). Some old versions use slower baudrates.

If you want to use a ESP8266 there is just one accessible hardware UART and **no** GPIO matrix. You must use GPIO1 and GPIO3 if you want to use the hardware UART. The ESPHome platforms provides a software serial implementation (ESP8266 only!). So you can use any GPIO as software UART but please be careful: Software serial and high baudrates aren't recommended.

Long story short: BLE is your best bet. ;-)

syssi, 2022-04-18:
The (reading) BLE support is ready. Please give it a try:

1. Download this example config https://github.com/syssi/esphome-jk-bms/blob/770ca59eec22fc300d67b4d4a3f5bbcb30aec25b/esp32-ble-example.yaml
2. Change the `mac_address` to the MAC address of your BMS
3. Compile and flash: `esphome run esp32-ble-example.yaml`

closed 2022-04-19 by syssi

JohnGalt1717, 2022-04-20:
Thanks!

What's the best/easiest way to get the mac addresses required for this?

syssi, 2022-04-20:
@JohnGalt1717 Flash the yaml with some random address, start the ESP32 and take a look at the logs. The BLE tracker will report every device in the air. You should find the mac address next to the name of your BMS here.

syssi, 2022-04-20:
It looks like this:

```
[14:24:01][D][esp32_ble_tracker:723]: Found device C8:47:8C:F2:7E:03 RSSI=-90
[14:24:01][D][esp32_ble_tracker:744]:   Address Type: PUBLIC
[14:24:01][D][esp32_ble_tracker:746]:   Name: 'JK-BD6A17S6P'
```

JohnGalt1717, 2022-04-20:
Ok, I got it working with 1 and 2 BMSes. But as soon as you add a 3rd which should be supported (BLE uses the Uarts underneath so you'll get 3 max) it errors on compilation.  2 works fine...

syssi, 2022-04-20:
Could you try to reduce the number of sensors just for testing. Try somehting like this:

```
substitutions:
  name: jk-bms
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address0: C8:47:FF:FF:FF:FF
  mac_address1: C8:47:FF:FF:FF:FF
  mac_address2: C8:47:FF:FF:FF:FF

esphome:
  name: ${name}

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf
    version: latest

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
logger:

api:

esp32_ble_tracker:

ble_client:
  - mac_address: ${mac_address0}
    id: client0
  - mac_address: ${mac_address1}
    id: client1
  - mac_address: ${mac_address2}
    id: client2

jk_bms_ble:
  - ble_client_id: client0
    throttle: 5s
    id: bms0
  - ble_client_id: client1
    throttle: 5s
    id: bms1
  - ble_client_id: client2
    throttle: 5s
    id: bms2

sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:

  - platform: jk_bms_ble
    jk_bms_ble_id: bms1
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:

    - platform: jk_bms_ble
    jk_bms_ble_id: bms2
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:
```

I don't think the number of BLE connections is limited by the number of hardware UARTs. I think your ESP32 crashs because of a stack overflow. If the "reduced sensor example" works for you I would provide some instructions how to increase the stack size to have room for a lot of sensors.

JohnGalt1717, 2022-04-20:
I have things hooked in home assistant and didn't want it to go nuts so I manually updated the stack size, and it seems to have solved it.

On another note, the Balancing Enabled switch doesn't appear to be working with BLE.

syssi, 2022-04-20:
I don't have any docs about the BLE traffic. I've used a traffic capture (btsnoop) to reverse engineer the communication plus some existing code from different open source projects (f.e. `mpp-solar`).

JohnGalt1717, 2022-04-20:
Ok, I'll see what I can come up with.

syssi, 2022-04-20:
Please monitor the "Unknown" messages at the log. Stop the ESP, change a setting via the App, close the app and let the ESP reconnect. Check the "Unknown" log lines again. May be you can identify some status bits or bytes.

syssi, 2022-04-21:
@JohnGalt1717 I've pushed a breaking change. Please adapt your configuration yaml:

https://github.com/syssi/esphome-jk-bms/pull/83

1. Remove the `balancing_switch`.
2. Rename `charging_switch` to `charging`
3. Rename `discharging_switch` to `discharging`
4. Add the binary_sensor `balancing`

All binary sensors are indicators of the current status of the BMS. They don't represent switch states.

JohnGalt1717, 2022-04-21:
Seemed to work!

There's a bunch of items I'd love to be able to change if changing is possible (then I wouldn't need to ever use the app)

1. OVCP (over voltage cell protection) (i.e. 3.65 or 3.55 etc.)
2. OVCR (over voltage cell recovery) (i.e. 3.6 or 3.5 or whatever it falls back to before turning charge on again automatically)
3. UVDP (under voltage discharge protection)
4. UVCP (under voltage discharge recovery)
5. Enable/Disable Charging, Balancing and Discharging.

Not sure if this is possible. Maybe an option to pass in the pass code in the yaml configuration and then be able to setup switches for No. 5 and then text fields for the rest?

JohnGalt1717, 2022-04-21:
Awesome! Thanks!

tssparky, 2024-08-07:
Hello, I’ve been trying to use BLE to get data from jk BMS for days, I didn’t want to use esphome as I want to just print to screen or serial for now. 
so I used the logic and methods of this code to connect and get data but. 
System finds bms, connects, gets a bit of info from FF01, and it’s just jk model, user password (hmm not so good probably) and software version. If I try to connect to 180F or 180A, it won’t connect, can’t find service. 
at one point I thought I got it going and was just a parsing issue but the data appears static. 
has the data structure changed? Since this esphome version was created? 
I can post my code if there is anyone here to read it and hopefully point me in the right direction. 
any advice will be appreciated 
Cheers
Nick 

Been using VScode with esp32-s3 / arduino ide.

syssi, 2024-08-07:
Please create a new issue. Please keep in mind ESPHome is very flexible. If you remove the `wifi` section and the other components requiring network access (`ota` f.e.) you can also reach your goal without writing a second implementation.

PR #10 (2022-04-19): Add basic BLE support
Closes: #5