syssi/esphome-jk-bms #6: About RS485/GPS

ortegafernando, 2021-10-01:
Hi, do you know if we can connect directly to the GPS/485 port our rx/tx pins of an esp8266/arduino/esp32?

I mean. What is the protocol that have that port? I suppose that it is TTL232 or RS232, as they have a RS485 converter.
Otherwise, what is the 485 converter converting? What is it converting "from"?

May be more photos of the RS485 converter (ICs) could be significant to understand what it is doing and which "protocol" has the raw GPS/485 port.

Thanks a lot.

syssi, 2021-10-01:
Sorry for the confusion. I tried to make it as clear as possible.

> Hi, do you know if we can connect directly to the GPS/485 port our rx/tx pins of an esp8266/arduino/esp32?

Yes! I connected the ESP32 directly to the BMS using the `GPS/485` port.

> I mean. What is the protocol that have that port? I suppose that it is TTL232 or RS232, as they have a RS485 converter.
> Otherwise, what is the 485 converter converting? What is it converting "from"?

IMO the `GPS/485` jack has a TTL level of 0V to 3.3V. So we are safe to connect the ESP directly.

> May be more photos of the RS485 converter (ICs) could be significant to understand what it is doing and which "protocol" has the raw GPS/485 port.

Some photos to inspect the RS485 converter: https://github.com/syssi/esphome-jk-bms/tree/main/images/rs485-adapter

> You say: "The RS485-TTL jack of the BMS can be attached to any UART pins of the ESP. A hardware UART should be 
> preferred because of the baudrate (115200 baud)"
> But, sorry I am wrong, does RS485 be TTL? it is not a -12 +12v communication?

The RS485 adapter converts the TTL signal to RS485. The voltage shift (0V/3V to -12V/+12V or -7V/+12V) happens here. We aren't interested in a -12V/+12V signal so we don't use the RS485 adapter. If your setup requires long wires you could use the RS485 adapter plus some RS485 arduino module to shift the levels back to 3.3V:

```
JK-BMS <--> RS485 adapter of the manufacturer <---- long cable ----> RS485 to TTL module <---->  ESP32
```

> May be a photo of your connections (or schema) could clarify a bit more the readme.md file.

I tried to make it clear by this schematics:

https://github.com/syssi/esphome-jk-bms#schematics

I will add a photo too :-)

mihaigsm2003, 2021-12-30:
/*
 *                 RS485-TTL
┌──────────┐                            ┌─────┐
│                        3│<-TX--------RX->│             │
│  JK-BMS          2│<-RX--------TX->│ ESP32/  │
│                       1 │<----- GND ---->│ ESP8266│
│                          │                            │              │<-- 3.3V
└──────────┘                               ─────┘

# RS485-TTL jack (4 Pin, JST 1.25mm pinch)
┌─────────-
│ 1   2   3   4      │
│ O   O   O   O  │
│GND  RX  TX VBAT│
└──────────
i use this schematics, work for me.

trippfam07, 2022-01-21:
I am new to home assistant and I am trying to get this working.  I have attached the jk bms to my esp32 but I am not sure what I need to do in the program to make it work? Any help would be appreciated.

syssi, 2022-01-21:
If you use Home Assistant use the `esp32-example.yaml` and replace the mqtt component/section:

```
mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: mqtt_client
```

By the `api` component:

```
api:
```

If you flash the yaml to your esp32 by `esphome run esp32-example.yaml` the device should be auto-discovered by Home Assistant. There will be a notification about the discovered device.

trippfam07, 2022-01-21:
So this is not possible through esp home integration?

syssi, 2022-01-21:
Do you mean the Esphome Dashboard? https://esphome.io/guides/getting_started_hassio.html

Should be possible. I don't use the dashboard so I cannot help here much. Just follow the official esphome docs. As soon you are able to flash/install a generic esphome device you are also able to use this component.

trippfam07, 2022-01-21:
I dont have to use esp home. I am not sure how to go about it doing it the other way. Is it something I do in the configuration.yaml?

syssi, 2022-01-21:
This implementation requires esphome. Could you try to flash a basic configuration.yaml to an ESP? Just to get a better understanding what esphome is and how it works. You will see there is no additional code required. Esphome plus the yaml file generates/assembles everything.

trippfam07, 2022-01-21:
Ok thanks. I will try it tonight.

trippfam07, 2022-01-21:
Does this look right to you?

esphome:
  name: jk-bms-test

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "21c34a069e5274678297d9fba4227c74"
  
mqtt:
  broker: core-mosquitto
  username: mqtt-user
  password: *****
  port: 1883

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Jk-Bms-Test Fallback Hotspot"
    password: "n9f9vKiLt01H"

captive_portal:

syssi, 2022-01-21:
Please don't use `mqtt` if you use the `api` or vice versa.

trippfam07, 2022-01-21:
I would use mqtt and not api. So would i just remove the api and everything else looks good?

syssi, 2022-01-21:
If you use Home Assistant the `api` is the better choice. If you use `!secret wifi_ssid` you've to create a `secrets.yaml`. If you don't want to care about a secrets file just replace `!secret wifi_ssid` and `!secret wifi_password` with valid values.

trippfam07, 2022-01-22:
So leave out the mqtt portion and just do api?

trippfam07, 2022-01-22:
So this is my configuration.
And I have installed it on the ESP 8266 and see it in Home assistant but no sensors show up

esphome:
  name: jk-bms-test-8266

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:

#mqtt:
  #broker: 192.168.1.40
  #username: mqtt-user
  #password: ******
  #port: 1883

ota:
  password: "99849608f78d892990a87e32dae261ad"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Jk-Bms-Test-8266"
    password: "nc2CN4dkH1wn"

captive_portal:

syssi, 2022-01-22:
This is fine because your configuration doesn't have a `sensor` section. In other words: Your code/configuration doesn't expose any sensor. Some example: If your ESP8266 board has a LED you could expose the LED as `light` now:

```
output:
  - platform: esp8266_pwm
    id: output_blue_led
    pin:
      number: GPIO2
      inverted: true

light:
  - platform: monochromatic
    id: blue_led
    internal: true
    name: "${name} blue led"
    output: output_blue_led
```

You could wire/solder some sensor (f.e. a BH1750) to your ESP and expose the values as `sensor`: https://esphome.io/components/sensor/bh1750.html

If this is working pick your ESP32 or stick to the ESP8266 and connect some GPIO to `RX` and `TX` to your BMS and extend/adopt your configuration.yaml with this minimal setup of the custom component:

```
external_components:
  - source: github://syssi/esphome-jk-bms@main
    refresh: 0s

uart:
  id: uart0
  baud_rate: 115200
  rx_buffer_size: 384
  tx_pin: GPIO14
  rx_pin: GPIO4

jk_modbus:
  id: modbus0
  uart_id: uart0

jk_bms:
  id: bms0
  jk_modbus_id: modbus0

sensor:
  - platform: jk_bms
    min_cell_voltage:
      name: "${name} min cell voltage"
```

trippfam07, 2022-01-22:
That was awesome. Thanks. I took all the sensors examples and plugged them in and configured. I am getting a lot of data back now. I do see a couple of issues though. 1. actual battery capacity show 512 and on the bms app I have it 560.
2. The remaining battery capacity shows up as a percentage and no amp hours. 3. the current is showing up as a negative value on home assistant and positive value in bluetooth app.
Do I have something wrong in my configuration?

syssi, 2022-01-22:
I don't think so. The firmware of the BMS isn't perfect and this custom component publishes values received from the BMS most of the time 1:1 (untouched). Let's talk about specific sensors (names) and let's try to compare the device response with the sensor value. It possible to enable some debug output to make the UART traffic visible. Do you like to dig deeper into it? I will provide some instructions. Please create a new issue per problem.

trippfam07, 2022-01-22:
I would love to do some debug what do i need to do?

syssi, 2022-01-22:
Please create new issue first. Do not longer abuse this issue. This issue is about RS485/RS232 and TTLs. ;-)

trippfam07, 2022-01-22:
ok Thanks

Ching858, 2025-08-14:
Every one can you send the picture wiring esp to jk bms I use for reference thank you