syssi/esphome-jk-bms #15: Jibms esphome

jouking, 2021-11-01:
I have a jkbms with hardware 8.x and soft 8.0.2H, version 3.7.0.  I have installed the esphome and I have put the code that you have put minus the mqtt instructions that give me an error.  I have connected the rs485 module, with an rs485 to TTL 3.3v adapter and an esp8266 but it does not work.  Is the version of my JKBms compatible ??

syssi, 2021-11-01:
Which model of the JKBMS do you own? Where did you buy the BMS? Could you ask the seller for some specifications about the RS485 of your model? Unfortunately I do not know the differences of the different models / software / hardware versions.

jouking, 2021-11-01:
>
[JK-BMS-bluetooth-Manual.pdf](https://github.com/syssi/esphome-jk-bms/files/7455216/JK-BMS-bluetooth-Manual.pdf)
B2A24S15P

syssi, 2021-11-01:
Could you ask the seller about some details of the RS485 protocol? In my case the seller (HankzorBMS Store) was able to provide some infos about the protocol. How did you wire the modules? Does your rs485 ttl module has a RX/TX led? Does the TX led blink periodically? Did you try to swap the RS485 wires (A & B)  once?

KlausLi, 2021-11-04:
hello syssi, thanks for your effort! It's a shame that people don't want to help :-(.

KlausLi, 2021-11-04:
i have the unsupported one :-)   JK-B2A24S20P HW 8.X SW 8.1.2H version 3.7.1 Manuf date: 2021
its from IC GOGOGO (aliexpress)
I can not ask him things , its closed that way !
i had a problem on begin with connection , now soldering the rx tx dírectly to the microcontroller .
now i become a response on Hterm but not the right one 

4E 57 00 13 00 00 00 00 06 03 00 00 00 00 00 00 68 00 00 01 29
moment i made a screenshot from hterm terminal

KlausLi, 2021-11-04:
[image]

on try again its other response and so on

KlausLi, 2021-11-04:
normal the response are over 150 signs but today desoldering the digital isolator :-) *ggg  , he mades only 43 signs as response :-=) ... good quality toy BMS:-)

KlausLi, 2021-11-04:
i now desoldering the rest of bms and sale on ebay !

syssi, 2021-11-04:
@KlausLi I've added some additional wires to the RX/TX pin of the bluetooth module. My idea was to sniff some traffic there and to get rid of the bluetooth module some day. The HankzorBMS Store guy told me the protocol can be customized if I buy more than 100 pieces. It's possible the IC GOGOGO versions are a bit different.

syssi, 2021-11-04:
Do you like to test another payload (the old protocol) with different baud rates: `0xDD 0xA5 0x03 0x00 0xFF 0xFD 0x77`

KlausLi, 2021-11-04:
ok 100 pieces , very good seller , YES we must have that protokoll ! *g  

other payloads do not work, i try it all 
only that 4E 57 00 13 00 00 00 00 06 03 00 00 00 00 00 00 68 00 00 01 29  
has a response

KlausLi, 2021-11-04:
i think , we can drill a hole in the MCU and pin it on the wall !  shity bms , i will try with seplos 200A 16s BMS , i becom from alibaba Seller the protokol, its a very robust and really smart bms .
the most solarstorages has that inside :-)

KlausLi, 2021-11-04:
https://cn1524225498fqks.en.alibaba.com/product/1600246930684-904757264/Seplos_16S_200A_48V_Lifepo4_lithium_battery_protection_board_LFP_Cell_Balance_Integrated_Circuits_BMS_with_LCD_screen.html?spm=a2700.shop_plgr.41413.16.65bc37f5mlSdgv

KlausLi, 2021-11-04:
> bluetooth

Can you made a picture from your board , high resolution  from folowing area :
i want no that smd resistors  what value there have 

[image]

KlausLi, 2021-11-04:
and have you  an IC on CAN place?

syssi, 2021-11-04:
This is my BMS (the small / 60A version): https://github.com/syssi/esphome-jk-bms/tree/main/images/jk-bd6a17s6p

KlausLi, 2021-11-04:
> This is my BMS (the small / 60A version): https://github.com/syssi/esphome-jk-bms/tree/main/images/jk-bd6a17s6p

122u31 is the same but the rest is not

KlausLi, 2021-11-04:
> This is my BMS (the small / 60A version): https://github.com/syssi/esphome-jk-bms/tree/main/images/jk-bd6a17s6p

OK , hab nun den 8pin wieder eingelötet , weil der gut ist aber ich hab ne kaputte diode gefunden, flux vom can bus gemopst und siehe da v8.x funktioniert doch , hab nun einen korrekten  response 👍 
(erst nach wechsel zum schnelleren Laptop)

wieso willst du den bluetooth anzapfen wann du eh 485 hast ?

4E 57 01 18 00 00 00 00 06 00 01 79 27 01 0E BE 02 0E BB 03 0E BA 04 0E BA 05 0E BA 06 0E BA 07 0E B9 08 0E BB 09 0E 88 0A 0E 85 0B 0E B6 0C 0E B6 0D 0D ED 80 00 13 81 00 2C 82 00 2C 83 13 06 84 00 00 85 44 86 02 87 00 00 89 00 00 00 00 8A 00 0D 8B 00 00 8C 00 03 8E 15 54 8F 0E 38 90 10 68 91 10 04 92 00 05 93 0A F0 94 0C E4 95 00 05 96 01 2C 97 00 05 98 01 2C 99 00 05 9A 00 1E 9B 0D AC 9C 00 14 9D 01 9E 00 5A 9F 00 46 A0 00 64 A1 00 64 A2 00 14 A3 00 46 A4 00 46 A5 FF EC A6 FF F6 A7 FF EC A8 FF F6 A9 0D AA 00 00 00 05 AB 01 AC 01 AD 03 F4 AE 01 AF 01 B0 00 0A B1 14 B2 31 32 33 34 35 36 00 00 00 00 B3 00 B4 49 6E 70 75 74 20 55 73 B5 32 31 30 37 B6 00 00 05 00 B7 48 38 2E 58 5F 5F 53 38 2E 31 2E 32 48 5F 5F B8 00 B9 00 00 00 00 BA 42 54 33 30 37 32 30 32 30 31 32 30 30 30 30 32 30 30 35 32 31 30 30 31 C0 01 00 00 00 00 68 00 00 4E 21

syssi, 2021-11-05:
Du bist super! Respekt! Die Response vom RS485 Protokoll hat weniger Infos enthalten als die via Bluetooth. Im RS485 Protokoll fehlt der Balancer-Strom und die Innenwiderstände der Zellen etc.

Kannst du dein Setup noch einmal beschreiben? Antwortet dein BMS nun an der regulären Buchse? Sprichst du sofort mit dem ESP oder befinden sich echte RS485-Bauelemente direkt auf dem Mainboard des BMS, so dass es sich nicht mehr um ein UART-TTL Signal handelt, wie bei meiner Version?

syssi, 2021-11-05:
Ich sehe gerade, dass du bereits alle Fragen im Edit beantwortet hattest. :-)

KlausLi, 2021-11-05:
...echte RS485-Bauelemente direkt auf dem Mainboard...  
no keine 485 wie bei dir vorhanden
alles wird über den 122U31 geschleift , wie bei dir 

balancerstrom und widerstände sind mir wurscht! :-) , is eh grad a bastler BMS ,ich trau dem teil ned !
Naechste Woche kommt mein Scheini  Scheini Profi BMS :-)  
Batterie kommen erst .... nachdem Sie wohl mit Spacex eine Rundreise gemacht haben, erst dann zu mir.

syssi, 2021-11-05:
Alles klar. Das hilft mir weiter!

syssi, 2021-11-05:
Conclusion: The device is supported if your firmware version is `>=6.0` and can be connected directly to the ESP.

closed 2021-11-05 by syssi