syssi/esphome-jk-bms #17: Is my Active Balancer JK-DZ11-B2A24S hw6.0 sw6.0.5 supported?

bobwho69, 2021-11-05:
Hi Syssi,
First off, thanks for your efforts so far!

I have the JK-DZ11-B2A24S 2A RS485 active balancer, bought from ICGOGOGO store a year ago.  It is hardware version 6.0., software version 6.0.5 running the 3.7.3 app for android.  The information for this balancer comes from its QR code pointing to h5.clewm.net for documentation and software download.  The pdf documents that matter are in chinese but I have a crude translation and it states it uses the V2 protocol with hardware 6 and up. 

I am connecting to it via a WT32-ETH01 over ethernet and a POE splitter for 5V, the pins for this board are GPIO5 and 17.  in ESPHome it is running the ESP32Dev board.   Like many others, I removed the RS485 converter when getting nowhere, both RX and TX leds flash.  Swapping the RX and TX wires hasnt helped, the garbage data looks the same, but as its not the full BMS, I'm not sure if it's formatted the same.

How can I help?

bobwho69, 2021-11-06:
[Aurora Active Balance Protection Board RS485 interface general protocol V2.0 (applicable to firmware version V6.x and above).zh-CN.en.pdf](https://github.com/syssi/esphome-jk-bms/files/7489487/Aurora.Active.Balance.Protection.Board.RS485.interface.general.protocol.V2.0.applicable.to.firmware.version.V6.x.and.above.zh-CN.en.pdf)
Here is above mentioned translated document

syssi, 2021-11-06:
Please increase the log level to very verbose. At this log level the uart_esp32 components writes every transmitted/received byte to the log:

```
logger:
  level: VERY_VERBOSE
  logs:
    api.service: WARN
    ota: WARN
    sensor: WARN
    jk_bms_ble: VERY_VERBOSE
    uart_esp32: VERY_VERBOSE
```

Please provide some responses of your BMS.

bobwho69, 2021-11-06:
Hi, thanks, results below

```
[22:09:22][VV][uart.arduino_esp32:121]:     Wrote 0b01001110 (0x4E)
[22:09:22][VV][uart.arduino_esp32:121]:     Wrote 0b01010111 (0x57)
[22:09:22][VV][uart.arduino_esp32:121]:     Wrote 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:121]:     Wrote 0b00010011 (0x13)
[22:09:22][VV][uart.arduino_esp32:121]:     Wrote 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:121]:     Wrote 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:121]:     Wrote 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:121]:     Wrote 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:121]:     Wrote 0b00000110 (0x06)
[22:09:22][VV][uart.arduino_esp32:121]:     Wrote 0b00000011 (0x03)
[22:09:22][VV][uart.arduino_esp32:121]:     Wrote 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:121]:     Wrote 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:121]:     Wrote 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:121]:     Wrote 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:121]:     Wrote 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:121]:     Wrote 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:121]:     Wrote 0b01101000 (0x68)
[22:09:22][VV][uart.arduino_esp32:121]:     Wrote 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:121]:     Wrote 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:121]:     Wrote 0b00000001 (0x01)
[22:09:22][VV][uart.arduino_esp32:121]:     Wrote 0b00101001 (0x29)
[22:09:22][VV][uart.arduino_esp32:142]:     Flushing...
[22:09:22][V][component:186]: Component esphome.coroutine took a long time for an operation (0.13 s).
[22:09:22][V][component:187]: Components should block for at most 20-30ms.
[22:09:22][VV][uart.arduino_esp32:135]:     Read 0b01001110 (0x4E)
[22:09:22][VV][uart.arduino_esp32:135]:     Read 0b01010111 (0x57)
[22:09:22][VV][uart.arduino_esp32:135]:     Read 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:135]:     Read 0b00010011 (0x13)
[22:09:22][VV][uart.arduino_esp32:135]:     Read 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:135]:     Read 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:135]:     Read 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:135]:     Read 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:135]:     Read 0b00000110 (0x06)
[22:09:22][VV][uart.arduino_esp32:135]:     Read 0b00000011 (0x03)
[22:09:22][VV][uart.arduino_esp32:135]:     Read 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:135]:     Read 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:135]:     Read 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:135]:     Read 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:135]:     Read 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:135]:     Read 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:135]:     Read 0b01101000 (0x68)
[22:09:22][VV][uart.arduino_esp32:135]:     Read 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:135]:     Read 0b00000000 (0x00)
[22:09:22][VV][uart.arduino_esp32:135]:     Read 0b00000001 (0x01)
[22:09:22][VV][uart.arduino_esp32:135]:     Read 0b00101001 (0x29)

```

syssi, 2021-11-06:
This doesn't look like a valid response of the BMS. The bytes of the written request (`0x4E 0x57 0x00 0x13...`) and 100% equal to the read bytes: `0x4E 0x57 0x00 0x13...`

Please unplug the BMS and try again. Is the ESP still able to read something?

bobwho69, 2021-11-06:
I powered off, unplugged and fired it up, now only TXD lights up and the logs show it only writing and not reading.

syssi, 2021-11-06:
Okay. In this case your wiring isn't the root cause. ;-) I will compare the request now to the protocol documentation of your device. My be the request must be slightly different. Could you change the baudrate to 9600 baud an give it another try?

syssi, 2021-11-06:
If your balancer doesn't respond to the BMS status request please try this protocol / request:

https://github.com/syssi/esphome-jk-bms/pull/18

syssi, 2021-11-06:
I've checked the protocol specifications you did provide at one of your first posts. There is no difference to the implemented protocol.

syssi, 2022-04-23:
Let's close this issue in favor of #16. I would like to bundle all further discussion/info there.

closed 2022-04-23 by syssi

syssi, 2025-02-11:
The device is supported now. See https://github.com/syssi/esphome-jk-bms/pull/18