syssi/esphome-jk-bms #19: Issue with ESP32 Node-MCU board

rem72, 2021-11-15:
Hi Syssi, I'm haveing issues with your code, the board is constantly rebooting, I just installed your example with this result:

[14:16:14]
[14:16:14]Rebooting...
[14:16:14]ets Jun  8 2016 00:22:57
[14:16:14]
[14:16:14]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[14:16:14]configsip: 0, SPIWP:0xee
[14:16:14]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[14:16:14]mode:DIO, clock div:2
[14:16:14]load:0x3fff0018,len:4
[14:16:14]load:0x3fff001c,len:1044
[14:16:14]load:0x40078000,len:10124
[14:16:14]load:0x40080400,len:5828
[14:16:14]entry 0x400806a8
[14:16:14]Guru Meditation Error: Core  0 panic'ed (LoadStoreAlignment). Exception was unhandled.
[14:16:14]Core 0 register dump:
[14:16:14]PC      : 0x4008ad9a  PS      : 0x00060933  A0      : 0x8008cdc6  A1      : 0x3ffbc030
←[33mWARNING Decoded 0x4008ad9a: uxPortCompareSet at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vTaskEnterCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4201←[0m
[14:16:14]A2      : 0x20766f4e  A3      : 0x0000cdcd  A4      : 0xb33fffff  A5      : 0x00000001
[14:16:14]A6      : 0x00060720  A7      : 0x0000abab  A8      : 0x0000cdcd  A9      : 0x3ffbc120
[14:16:14]A10     : 0x3ffbbc74  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x80000020
[14:16:14]A14     : 0x00000003  A15     : 0x00060c23  SAR     : 0x00000000  EXCCAUSE: 0x00000009
[14:16:14]EXCVADDR: 0x20766f4e  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
[14:16:14]
[14:16:14]ELF file SHA256: 0000000000000000
[14:16:14]
[14:16:14]Backtrace: 0x4008ad9a:0x3ffbc030 0x4008cdc3:0x3ffbc060 0x4008d43d:0x3ffbc080 0x40081f6e:0x3ffbc0a0 0x40086c19:0x3ffbc0c0 0x4000bec7:0x3ffbc0e0 0x4008a759:0x3ffbc100 0x4008b522:0x3ffbc120 0x4008b537:0x3ffbc140 0x40089d8e:0x3ffbc160
←[33mWARNING Found stack trace! Trying to decode it←[0m
←[33mWARNING Decoded 0x4008ad9a: uxPortCompareSet at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vTaskEnterCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4201←[0m
←[33mWARNING Decoded 0x4008cdc3: multi_heap_internal_lock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:380←[0m
←[33mWARNING Decoded 0x4008d43d: multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:321←[0m
←[33mWARNING Decoded 0x40081f6e: heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:232←[0m
←[33mWARNING Decoded 0x40086c19: _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c:42←[0m
←[33mWARNING Decoded 0x4008a759: prvDeleteTCB at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507←[0m
←[33mWARNING Decoded 0x4008b522: prvCheckTasksWaitingTermination at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507←[0m
←[33mWARNING Decoded 0x4008b537: prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507←[0m
←[33mWARNING Decoded 0x40089d8e: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)←[0m
[14:16:15]
[14:16:15]Rebooting...

The only thing I've changes is the board to nodemcu-32s to fit my board, ¿can you help me?

syssi, 2021-11-15:
Could you remove most of the sensors from your configuration? Something like this:

```
substitutions:
  name: jk-bms

esphome:
  name: ${name}
  platform: ESP32
  board: esp-wrover-kit

external_components:
  - source: github://syssi/esphome-jk-bms@main
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
logger:

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: mqtt_client

uart:
  id: uart0
  baud_rate: 115200
  rx_buffer_size: 384
  tx_pin: GPIO14
  rx_pin: GPIO4

jk_modbus:
  id: modbus0
  uart_id: uart0

jk_bms:
  id: bms0
  jk_modbus_id: modbus0

sensor:
  - platform: jk_bms
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
```

Does the device still crash? Which Esphome version do you use?

rem72, 2021-11-15:
Now perfect I reduced the number of sensors, ¿Is there any limitation? 
The same code in a ESP 8266 works well.
It's weird, I'll try to find which is the maximum.

esphome version
Version: 2021.10.3

rem72, 2021-11-15:
One more question.
Are the switches functional or they only show what is configured at the bms?
Can you disable or enable balance? It' s not working for me.

syssi, 2021-11-16:
The default stack size of a ESP32 build is a bit small. In future it will be possible (at esphome) to increase the stack size a bit. At the moment the maximum number of sensors is a bit limited IMO and hard to predict.

Switches: You are right. The switches are read-only at the moment because writing a new state isn't implemented. I have seen a few voices telling it's not possible to write registers / change states via the serial connection. I haven't tried it yet.

Which BMS model and firmware version do you own? I would like to add tested devices to the README.

rem72, 2021-11-16:
Thank you so much for your help and comprehension with me. 
You're  doing a great job.

I own a 
jk-b2a24s20p 
hardware version 8 
software 8.1.2H
Version 3.7.4

Tell me if you want me to test something. I'm a member of a telegram group in Spain and we can test new improvements.

Thank you again.

syssi, 2021-11-16:
Thanks for your support! Feel free to create new issues if there is something wrong, unclear or if you see room for improvements.

closed 2021-11-16 by syssi

diybateriaslifepo4, 2021-11-16:
> The default stack size of a ESP32 build is a bit small. In future it will be possible (at esphome) to increase the stack size a bit. At the moment the maximum number of sensors is a bit limited IMO [...]

Hello syssi, I am Juanmi, administrator of a Spanish DIY group about LiFePO4 batteries for solar energy in homes and motorhomes, we have been able to communicate different bms (such as DALY, JK, ANT, JBD) with inverters (such as Victron, Voltronic, Deye for a long time , Growatt, etc.)

I have learned about your work thanks to rem72, who is a colleague of our group, your project is very interesting, thanks to the instructions you gave him yesterday I tested the code successfully on an esp32, I tested a bms jk 8s 200a with hardware 9. x and communicates perfectly.

I am in contact with a JK engineer through WECHAT, but they are hard to extract information ... but he told me that the GPS module they have is capable of controlling the BMS remotely, as a remote control for robberies, from what I understand that it should be possible to change parameters through the serial port.

In our group there are several colleagues collaborating on different projects, little by little we are achieving great things for the benefit of the DIY community, especially thanks to people like you

anything I can help with I will be delighted.