syssi/esphome-jk-bms #20: Control the BMS via serial interface

syssi, 2021-11-16:
Follow-up: https://github.com/syssi/esphome-jk-bms/issues/19

@diybateriaslifepo4 @rem72

> I am in contact with a JK engineer through WECHAT, but they are hard to extract information ... but he told me that the GPS module 
> they have is capable of controlling the BMS remotely, as a remote control for robberies, from what I understand that it should be 
> possible to change parameters through the serial port.

syssi, 2021-11-16:
@diybateriaslifepo4 We are talking about this value:

https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_modbus/jk_modbus.cpp#L135

If we change the value from `0x03` to `0x02` we simulate a communication between the BMS and a GPS device. It's possible to BMS response differs a bit in this case. In best case the response is the same and commands we issue are executed properly (if the frame source is GPS).

diybateriaslifepo4, 2021-11-17:
I am trying to get jk me information about this communication, but they insist that I must use their adapter to rs485 and their documentation for communication by rs485, I do not know if you already have it, how could I send it to you?

syssi, 2021-11-17:
You could attach the docs here by drag&drop.

diybateriaslifepo4, 2021-11-17:
[BMS CAN instruction.pdf](https://github.com/syssi/esphome-jk-bms/files/7553289/BMS.CAN.instruction.pdf)
[BMS 485 instruction.pdf](https://github.com/syssi/esphome-jk-bms/files/7553290/BMS.485.instruction.pdf)

syssi, 2021-11-17:
The "BMS 485 instruction.pdf" describes the implemented protocol and doesn't contain new details.

ortegafernando, 2021-12-12:
hi, how to write settings via serial? Thanks

syssi, 2021-12-12:
This compoent doesn't support write operations right now. It's also unconfirmed it's possible or not.

diybateriaslifepo4, 2021-12-12:
[image]
in the protocol it says RW

> @diybateriaslifepo4 We are talking about this value:
>
> https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_modbus/jk_modbus.cpp#L135
>
> If we change the value from `0x03` to `0x02` we [...]

have you been able to try this? I would do it but I don't have much idea about programming... Thanks for your job

syssi, 2021-12-12:
I will give it a try but my powerwall isn't assembled at the moment so my feedback / testing will take a few weeks.

diybateriaslifepo4, 2021-12-12:
ok, there is no rush, in case you don't know you can turn on the bms with a power supply, with about 40v, only putting the negative in b0 & B- and positive in b+, without cells

SeByDocKy, 2022-02-18:
Any update for such control ? :) ... .Now my JK BMS runs like a charm thanks your component, I would like to see these switch really working.... would be also amazing :)

diybateriaslifepo4, 2022-02-18:
> Any update for such control ? :) ... .Now my JK BMS runs like a charm thanks your component, I would like to see these switch really working.... would be also amazing :)

I have tried to do it and it has not worked, I have tried creating an external component, changing the value to bms, bluetooth, gps and computer, and not able to change values in any way
But maybe I haven't done it correctly, let's hope that Sysssi can test it one of these days.

SeByDocKy, 2022-02-26:
> The switch entities are now able to send some instructions to the BMS:
> 
> ```
> [switch:017]: 'jk-bms charging' Turning OFF.
> [uart_debug:114]: >>> 4E:57:00:14:00:00:00:00:02:02:02:AB:00:00:00:00:00:68:00:00:01:6A
> [switch:013]: 'jk-bms charging' Turning ON.
> [uart_debug:114]: >>> 4E:57:00:14:00:00:00:00:02:02:02:AB:01:00:00:00:00:68:00:00:01:6B
> 
> [switch:017]: 'jk-bms discharging' Turning OFF.
> [uart_debug:114]: >>> 4E:57:00:14:00:00:00:00:02:02:02:AC:00:00:00:00:00:68:00:00:01:6B
> [switch:013]: 'jk-bms discharging' Turning ON.
> [uart_debug:114]: >>> 4E:57:00:14:00:00:00:00:02:02:02:AC:01:00:00:00:00:68:00:00:01:6C
> 
> [switch:017]: 'jk-bms balancing' Turning OFF.
> [uart_debug:114]: >>> 4E:57:00:14:00:00:00:00:02:02:02:9D:00:00:00:00:00:68:00:00:01:5C
> [switch:013]: 'jk-bms balancing' Turning ON.
> [uart_debug:114]: >>> 4E:57:00:14:00:00:00:00:02:02:02:9D:01:00:00:00:00:68:00:00:01:5D
> 
> [switch:013]: 'jk-bms dedicated charger' Turning ON.
> [uart_debug:114]: >>> 4E:57:00:14:00:00:00:00:02:02:02:B3:01:00:00:00:00:68:00:00:01:73
> [switch:017]: 'jk-bms dedicated charger' Turning OFF.
> [uart_debug:114]: >>> 4E:57:00:14:00:00:00:00:02:02:02:B3:00:00:00:00:00:68:00:00:01:72
> ```
> 
> Is somebody willing to test the new feature?

Sure I wil ltest right now

SeByDocKy, 2022-02-26:
Ok ... seems to not working for me :( .... I can see the OFF command sent ... but the switch is returning to its initial state (ON)

rob-oravec, 2022-02-27:
Same for me.

SeByDocKy, 2022-02-27:
I tried whatever the password is entered before or not .... switch control does not work for me...

syssi, 2022-03-01:
I talked to the seller of my BMS and he told me the serial interface is read only. :-( I fear it is the truth.

SeByDocKy, 2022-03-01:
> I talked to the seller of my BMS and he told me the serial interface is read only. :-( I fear it is the truth.

Damned :(

SeByDocKy, 2022-03-02:
I just checked the protocol pdf I got, effectively, we car read

R 0 X 84 The current data 2 Hex 10000 (10000-11000) * 0.01 = 10.00), a (discharge)
(10000-9500) *0.01= 5.00A (charging) P r e c i s i o n 1 0 mAunit:0 1 A
Note: C 0:0 x 01 redefines the 0 x 84 current data in 10 mA, with the highest bit being0"1" means "discharge" and "1" means charge
If the discharge is 20 A, the transmitted data is 2000 (0x 07 D0)
If the charge is 20 A, the transmission data is 34768 (0x 87 D0

in Read only :( :( .... what's a pity. Maybe time to change them into binary sensor so ?

syssi, 2022-03-02:
I cannot follow / I didn't understand. Could you rephrase your statement? :-)

SeByDocKy, 2022-03-02:
I mean if it's only in read only .... no need to define them in switch but more in binary sensors ?

syssi, 2022-03-02:
Yes. I will convert the switches into binary sensors soon and yes the sign of the `current` sensor is inverted for my BMS software/hardware version too. If we are doing breaking changes we can fix/improve this as well.

syssi, 2022-03-02:
FYI: https://github.com/syssi/esphome-jk-bms/pull/50

diybateriaslifepo4, 2022-03-02:
The strange thing is that they told me that the mosfets could be controlled from the gps, and the gps uses the same serial interface...

syssi, 2022-03-02:
Do you know the exact product (GPS) I've to buy? As soon we have to devices talking to each other the communication can be captured easily. I've captured the traffic between the bluetooth module and the MCU already. I had the hope the same protocol is used here but it looks like this isn't the case: https://github.com/syssi/esphome-jk-bms/issues/49

Funny thing: The bluetooth password (123456) isn't passed to the MCU at all. It's used to unlock the bluetooth module but it's not part of the traffic between the bluetooth module and the MCU of the BMS.

diybateriaslifepo4, 2022-03-02:
> Do you know the exact product (GPS) I've to buy? As soon we have two devices talking to each other the communication can be captured easily. I've captured the traffic between the bluetooth module and the MCU already. I had the hope the same protocol is used here but it looks like this isn't the case: #49
> 
> Funny thing: The bluetooth password (123456) isn't passed to the MCU at all. It's used to unlock the bluetooth module but it's not part of the traffic between the bluetooth module and the MCU of the BMS.

tomorrow I ask the distributor, to see if I can ensure the functions of the GPS, I was going to send one pars test, but he told me that not all the functions would work in Spain, because it is designed to work in China

syssi, 2022-03-02:
I asked my seller (HankzorBMS Store) about connecting an GPS/GSM device and he told me I need a customized version of the BMS.

SeByDocKy, 2022-03-02:
> I asked my seller (HankzorBMS Store) about connecting an GPS/GSM device and he told me I need a customized version of the BMS.

Custom firmware ? or also hardware modified ?

syssi, 2022-03-02:
In general the seller wants to sell 1000 pieces of a special BMS version. The difference will be the firmware only. I don't think we will receive an alternative/updated firmware ever.

diybateriaslifepo4, 2022-03-03:
> Do you know the exact product (GPS) I've to buy? As soon we have two devices talking to each other the communication can be captured easily. I've captured the traffic between the bluetooth module [...]

sure enough, my dealer is now telling me that the port is read only, sorry 😭

syssi, 2022-03-03:
The minimum quantity for customized devices is 100.

SeByDocKy, 2022-03-03:
> The minimum quantity for customized devices is 100.

It's 99 more that I need :) :)  ... Any way are the UVP? UVPR, LVP, LVPR writable ? Coz we could fake  charging/discharging by this way ?

SeByDocKy, 2022-03-03:
> I think UVP, UVPR, LVP and LVPR can be written/controlled via BLE. Whats the goal of faking charging/discharging? I think I didn't get the point.

By setting UVP=UVPR=LVP=LVPR is disabling charging/discharing de facto

syssi, 2022-03-03:
No write operations are possible via the serial interface. We have to use BLE to control the device.

rem72, 2022-03-03:
Hello Sebastian, will you try to connect the Voltronic inverter to an esp32
that in turn speaks via bluetooth with the jk? Perhaps it's the way to talk
between them.

syssi, 2022-04-23:
I've implemented a BLE verison of this component a few days ago:

https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-example.yaml

The switches (`charging`, `discharging`, `balancer`) can be controlled already.

syssi, 2022-04-23:
I will close this issue because it cannot be solved. I assume we must accept the serial connection is "read-only".

closed 2022-04-23 by syssi

Wiebo, 2022-10-04:
> The switch entities are now able to send some instructions to the BMS:
>
> ```
> [switch:017]: 'jk-bms charging' Turning OFF.
> [uart_debug:114]: >>> [...]
The CRC is calculated wrong...

syssi, 2022-10-04:
@Wiebo Why do you think so? :-)

Wiebo, 2022-10-04:
The checksum of this line:
4E:57:00:14:00:00:00:00:02:02:02:9D:00:00:00:00:00:68:00:00:01:5C
does not give me a response from the BMS. Also adding the END marker 0x68 gives checksum 0x01 0xC4
So,
4E:57:00:14:00:00:00:00:02:02:02:9D:00:00:00:00:00:68:00:00:01:C4
does give me a response from the BMS

syssi, 2022-10-04:
Thanks for your feedback! Too bad it's not working anyway. Did you try different frame sources (f.e. GPS)?

Wiebo, 2022-10-05:
Yes, but also looking into command code 0x5, but don't really understand what they mean with that chinglish that's put into the manual...
"Pair code instruction - To modify the parameters, the first command is correct before the change"

If i make any progress i'll post it here. The manual clearly states these registers as R/W

humpataa, 2024-12-29:
> I've implemented a BLE verison of this component a few days ago:
> 
> https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-example.yaml
> 
> The switches (`charging`, `discharging`, `balancer`) can be controlled already.

Hi and sorry for bringing this up again. Reading this thread I understand that JKBMS (not inverter versions) cannot be controlled by RS485 but via Bluetooth only. Correct?

The mentioned switches mean switching Charging / Discharging and Emergency mode on / off, correct? Why does it named "balancer" then? There are two more functions (set_emergency_switch and set_heating_switch) defined but not used (yet)?

And after checking the code ... I see the switch-script sends 00 or 01. But how does it differ between the three switches? Please advise.

syssi, 2024-12-29:
> The mentioned switches mean switching Charging / Discharging and Emergency mode on / off, correct?

There is not much known about write instructions via the GPS port (UART-TTL). `charging` and `discharging` are the only well tested registers. AFAIK the balancing switch cannot be changed via the GPS port. 

https://github.com/syssi/esphome-jk-bms/blob/main/esp32-example.yaml#L244-L249

Here you can find the different registers of the different switches wich gets disabled/enabled (`0/1`). 

https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms/switch/__init__.py#L17-L22

May be there are more writable registers. However, trying them out blindly is too dangerous for me.

Do these details provide some clarity? Feel free to ask more questions.

humpataa, 2024-12-29:
Thank you for the quick reply!

Yes indeed it sheds some light. I just didn't know that the BLE code makes use of the BMS code as well and therefor didn't check the init file.

So basically sending these bytes will ask for info reply:
AA
55
90
EB
97 <—
...

And to switch discharging ON/OFF it needs to send AC instead of the 97 followed by 0 or 1, right?
I am not eager to try values blindly either, just want to remotely turn on / off discharging via Bluetooth without having to fumble around with the app all the time.

humpataa, 2025-01-15:
So I have tried to send the following bytes to set Discharging on:

`AA 55 90 EB AC 4 1 0 0 0 0 0 0 0 0 0 0 0 0 2B`

But the switch doesn't work. The response is this ... looks like a confirmation or maybe an error. Any idea?

`AA 55 90 EB C8 1 0 0 0 0 0 0 0 0 0 0 0 0 0 43`

syssi, 2025-01-15:
It you use this project and increase the log level to `VERY_VERBOSE` you can see the raw traffic at the log.

humpataa, 2025-01-15:
Thanks, unfortunately I don't know how to use all this. I don't have a display, no mqtt and uart socket and whatelse is needed, just an ESP32 and Arduino IDE to write code for it. I have found capture log data in the docs folder, but it seems to be encoded and I doubt it contains sending switch data anyway. All "looks" right referring to your BLE protocol docs, using the exactly the crc-function ... you don't happen to have a log flying around? :innocent:

Wiebo, 2025-01-15:
Controlling the switches is never gonna work with the normal serial commands. The switches can be controlled by the wired serial connection using "AT"-Commands. Open the BMS and attach a spy at the bluetooth module. Switch the switches on and off and log the data.

AlbertWete, 2025-01-16:
@humpataa Try sending the commands in Hex not ASCI

"Turn ON Charge": [0x4E, 0x57, 0X00, 0x14, 0x00, 0x00, 0x00, 0x00, 0X02, 0X03,0X00, 0XAB, 0X01, 0X00, 0X00, 0X00, 0X00, 0X68, 0X00, 0X00, 0X01, 0XD2]);

"Turn OFF Charge":[0x4E, 0x57, 0X00, 0x14, 0x00, 0x00, 0x00, 0x00, 0X02, 0X03, 0X00, 0XAB, 0X00, 0X00, 0X00, 0X00, 0X00, 0X68, 0X00,0X00, 0X01, 0XD1]);

"Turn ON Discharge":[0x4E, 0x57, 0X00, 0x14, 0x00, 0x00, 0x00, 0x00, 0X02, 0X03, 0X00, 0XAC, 0X01, 0X00, 0X00, 0X00, 0X00, 0X68, 0X00, 0X00, 0X01, 0XD3]);

"Turn OFF Discharge":[0x4E, 0x57, 0X00, 0x14, 0x00, 0x00, 0x00, 0x00, 0X02, 0X03, 0X00, 0XAC, 0X00, 0X00, 0X00, 0X00, 0X00, 0X68, 0X00, 0X00, 0X01, 0XD2]);

syssi, 2025-01-16:
@humpataa Could you create a new issue? This one is a mixture of serial and BLE commands now.

syssi, 2025-01-16:
@Wiebo It's possible to control the Mosfets via serial connection using the GPS protocol already. Also the BLE payload is well known. No further reverse engineering required.

syssi, 2025-01-16:
@Wiebo See https://github.com/syssi/esphome-jk-bms/pull/343

syssi, 2025-01-16:
@humpataa Please create a new issue. I will provide the raw payloads and could offer some guidance how to flash this project onto your ESP32 without hassle.