syssi/esphome-jk-bms #21: Date: Delta Cell Vol??

jouking, 2021-11-21:
I don't see the delta cell vol data in the commands for esphome, doesn't that value exist?

[image]

syssi, 2021-11-21:
The value isn't part of the BMS response. The resistance and balancing current is missing too (and only available via Bluetooth). It would be possible to calculate min/max and the delta voltage on component level and expose it as sensor measurement.

jouking, 2021-11-21:
and what would the sensor be like?  a template in Home Assistant?

syssi, 2021-11-21:
My idea is to extend this component by the computed sensor.

jouking, 2021-11-21:
ok it would be great if you could have that sensor

syssi, 2021-11-22:
Could you try the feature branch by changing your configuration slightly:

```
external_components:
  - source: github://syssi/esphome-jk-bms@add-min-max-delta-sensors
    refresh: 0s

# [...]

sensor:
  - platform: jk_bms
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
```

closed 2021-11-25 by syssi

jouking, 2021-11-26:
perfect, thank you very much

commit 2f7d178b (2021-11-22): Add min/max/delta cell voltage sensor (Closes: #21)
PR #22 (2021-11-25): Add min/max/delta cell voltage sensor
```
external_components:
  - source: github://syssi/esphome-jk-bms@add-min-max-delta-sensors
    refresh: 0s

# [...]

sensor:
  - platform: jk_bms
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
```

Closes: #21