syssi/esphome-jk-bms #23: get_temperature_ and get_current_ incorrect

kscholty, 2021-11-27:
Hi,

i belive that both get_current and get_temperature in  `components/jk_bms/jk_bms.h` are incorrect.

for teh temperatur the manual say s´tat values > 100 are negative, e.g. 101 is in fact -1
Your calculation would make it `-1*101 - 100 = -201`
Instead you should calculate `100-(int16_t)value` which would give `100-101 = -1`

For the current the documentation says that a set sign bit indicates a negative value.
Your calculation gets it the wrong way around. 
if `value & 0x8000 == 0x8000`    then sign is set and the result should be nagative. Otherwise not.

syssi, 2021-12-05:
Good catch and thanks for point this out!

closed 2021-12-05 by syssi

syssi, 2022-03-02:
FYI: https://github.com/syssi/esphome-jk-bms/pull/50

PR #24 (2021-12-05): Fix temperature and current calculation
Closes: #23
commit 4a02b76e (2021-12-05): Fix temperature and current calculation (Closes: #23)