syssi/esphome-jk-bms #25: Read data with ESP8266

mihaigsm2003, 2021-12-22:
Hi, 
I need to read data voltage for each cell and send to mqtt via ESP
I see code for esphome but for esp8266 i not find eny ino file.

syssi, 2021-12-22:
Please read the documentation of ESPHome. The is the framework required to generate, compile and upload the firmware from a Yaml file. It's pretty easy and powerful!

mihaigsm2003, 2021-12-30:
Thanks, i solve with software version from Scotty89, now i have data to grafana.

[image]

syssi, 2021-12-30:
If you want to try this project nevertheless please follow the installation guide: https://github.com/syssi/esphome-jk-bms#installation

The framework/command line tool will be used the generate, compile and flash the firmware. No .ino files required. The Yaml configuration describes the full setup / is used to generate the code.

closed 2021-12-30 by syssi