syssi/esphome-jk-bms #28: Bricked BMS after OTA using ESP32

tmarkson, 2022-01-16:
Whatsup @syssi I was trying your code via ESPhome and managed to brick my JKBMS 8s20p model.  Now the buzzer beeps repeatedly and connecting to bluetooth doesn't show any battery info. 

I have another one in hand as of Jan 16 so I'm sticking to software UART, still trying your code. 

I used an ESP32 on the hardware UART.  The first time with that connected, I used OTA to send the changed firmware.  After the short waiting period for the OTA firmware to move from flash to program memory, it bricked.  The fast beeping started and I was unable to revive the BMS.  
 
That was after a soft reset by the ESP to initialize the new firmware.  My guess is there is some UART data sent during boot that caused a bad state to the BMS. 

Here is my config that did the deed. ...

```
### esp32-jkbms-bridge.yaml
# ESP32 dev board connected to JK-BMS via hardware UART, pins IO1 and IO3.
#
# Included project: https://github.com/syssi/esphome-jk-bms
# Target hardware is JK-B2A8S20P

substitutions:
  devicename: "esp32-jkbms-bridge"
  deviceid: "esp32_jkbms_bridge"

esphome:
  name: "${devicename}"
  platform: ESP32
  board: nodemcu-32s

external_components:
  - source: github://syssi/esphome-jk-bms@main
    refresh: 0s

wifi:
  ssid: !secret ssid_home
  password: !secret pass_home
  fast_connect: true

ota:
logger:
  baud_rate: 0
  level: DEBUG

api:

uart:
  id: uart0
  baud_rate: 115200
  rx_buffer_size: 384
  tx_pin: TX
  rx_pin: RX

jk_modbus:
  id: modbus0
  uart_id: uart0

jk_bms:
  id: bms0
  jk_modbus_id: modbus0

sensor:
  - platform: jk_bms
    min_cell_voltage:
      name: "${devicename} min cell voltage"
```

syssi, 2022-01-17:
That is annoying. Unfortunately I cannot help here much. You are right this custom component polls the device periodically.

syssi, 2022-01-17:
Wait...  it took me some time to understand. Did I get you correctly:

1. You updated the esphome device via OTA (`esphome run your.yaml`) 
2. After the update of the esphome node you pressed the reset button of your dev board.
3. If you try to connect via bluetooth to the BMS it just beeps once per try or always multiple times?

tmarkson, 2022-01-17:
> Wait... it took me some time to understand. Did I get you correctly:
> 
> 1. You updated the esphome device via OTA (`esphome run your.yaml`)
> 2. After the update of the esphome node you pressed the reset button of your dev board
> 3. If you try to connect via bluetooth to the BMS now it just beeps once per try or always multiple times?

Here's what happened. 

1. connected esp32 via some IO pins, not hardware UART. 
2. used OTA multiple times with your component, tried different baud rates, etc. 
3. re-read all the issues and your docs a few times.  Found your suggestion to use hardware UART on the ESP.
4. changed wiring to use hardware UART pins -- `TX` and `RX` in the config. Those are same as `GPIO1` and `GPIO3`, maybe not in that order.
5. Used OTA from a laptop, like you say `esphome run...`
6. heard a fast beeping from the BMS.  see this video. https://photos.app.goo.gl/VYEGpG5vmUUJ9dcq8
7. I submitted to JK for help from their tech team.  They asked me to return it.  I asked for the firmware file to flash the Mind motion MM32F103 processor to save shipping -- they did not seem amused. 
8. Bought a new BMS to keep using this battery. The new BMS has arrived last week and I'm trying to connect again with software UART only.

syssi, 2022-03-05:
Summary: Please be careful!! If you send random data to the BMS it's possible to destroy the device. The bootloader of the ESP outputs some debug messages on bootup (`TXD`). If you connect the BMS to `TXD` (`GPIO1`) and `RXD` (`GPIO3`)  the BMS receives the boot-up messages every time you restart the ESP.

Conclusion: Try to avoid using `TXD` (`GIO1`) to send data to the BMS. I will update the esp8266 examples to use the second hardware uart.

TODO / Note to myself: Evaluate the different UARTs
```
UART0: GPIO1 (TX), GPIO3 (RX)
UART1: GPIO2 (TX), GPIO8 (RX)
UART0 Serial.swap(): GPIO15 (TX), GPIO13 (RX)
```

tmarkson, 2022-03-06:
Thanks for looking Sebo
I agree with your summary! 
Be careful of wayward bits.
I have since ordered a new BMS and installed it on location without telemetry.

syssi, 2022-05-10:
@amagr0 Yes it is. See https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

GPIO16 and GPIO17 aren't in use per default.

tmarkson, 2022-05-10:
> Hello, First of all, congratulations for this amazing work. I have the B2A24S20P JKBMS unit and I intent to get the data with an ESP32 via RS485. Just to clarify this "bricked issue", it is safe to use the ports in esp32-example.yaml? [image]
> 
tell us how you get on! @amagr0 
I'd still like to setup mine but haven't had success on bluetooth or wired

closed 2023-04-16 by syssi