syssi/esphome-jk-bms #31: Data Published from Bms not accurate.

trippfam07, 2022-01-22:
Talking to BMS and getting Data back but some stuff does not come back correct.

syssi, 2022-01-22:
1. Could you provide the model number, hardware version and software version of your BMS?
2. Could you name the sensors which doesn't provide correct measurements/values? Please try to be explicit by using the config values f.e. `capacity_remaining`.

trippfam07, 2022-01-22:
Model JK-B1A20S15P
Hardware V =8.x
Software V=8.14U
ESP8266

trippfam07, 2022-01-22:
I am not sure what you mean by using the config values f.e.

syssi, 2022-01-22:
Could you provide your configuration.yaml and mark the "faulty" sensors somehow?

For example:

```
sensor:
  - platform: jk_bms
    min_cell_voltage: # <-- is empty
      name: "${name} min cell voltage"
    max_cell_voltage: # <-- should be 5 is 10
      name: "${name} max cell voltage"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
```

trippfam07, 2022-01-22:
actual_battery_capacity:
      name: "${name} actual battery capacity"

Should be 560 and it is showing 512

trippfam07, 2022-01-22:
I can send you a copy of the logs if you like?

syssi, 2022-01-22:
Please enable the debug mode of the uart component by adding:

```
uart:
  ...
  debug:
    direction: BOTH
```

The full `uart` section will look like this:

```
uart:
  id: uart0
  baud_rate: 115200
  rx_buffer_size: 384
  tx_pin: GPIO14
  rx_pin: GPIO4
  debug:
    direction: BOTH
```

Please flash again and check the logs of the esphome node by:

```
esphome logs your-device.yaml
```

Esphome prints the log after flashing automatically:

```
esphome run your-device.yaml
```

I'm interested in this lines:

```
[20:29:14][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
[20:29:19][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
```

syssi, 2022-01-22:
Let's first focus on `actual_battery_capacity`: https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms/jk_bms.cpp#L331-L332

trippfam07, 2022-01-22:
[13:43:12][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
[13:43:12][D][uart_debug:114]: <<< 4E:57:01:21:00:00:00:00:06:00:01:79:30:01:0D:38:02:0D:2F:03:0D:34:04:0D:31:05:0D:2F:06:0D:2C:07:0D:3C:08:0D:2C:09:0D:33:0A:0D:2A:0B:0D:3C:0C:0D:2A:0D:0D:3E:0E:0D:2D:58:58:33:AC:18:19:00:15:81:00:11:82:00:08:83:15:1A:84:85:A5:85:4F:86:02:87:00:00:89:00:00:00:00:8A:00:10:8B:00:00:8C:00:03:8E:16:D0:8F:10:40:90:0E:42:91:0D:DE:92:00:05:93:0A:28:94:0B:B8:95:00:05:96:01:2C:97:00:96:98:01:2C:99:00:64:9A:00:1E:9B:0D:48:9C:00:07:9D:01:9E:00:5A:9F:00:46:A0
[13:43:12][D][uart_debug:114]: <<< 00:64:A1:00:64:A2:00:14:A3:00:46:A4:00:46:A5:FF:EC:A6:FF:F6:A7:FF:EC:A8:FF:F6:A9:10:AA:00:00:02:30:AB:01:AC:01:AD:04:C9:AE:01:AF:01:D8:00:0A:B1:14:B2:31:32:33:34:35:36:00:00:00:00:B3:C0:BB:9A:1B:AB:A3:20:55:73:B5:32:31:31:30:B6:00:01:F6:00:B7:48:38:2E:58:5F:5F:53:38:2E:31:34:55:5F:5F:5F:B8:00:B9:00:00:02:00:BA:42:54:33:30:37:32:30:32:30:31:32:30:30:30:30:32:30:30:35:32:31:30:30:31:C0:01:00:00:00:00:68:00:00:4B:68

trippfam07, 2022-01-22:
Is that what you wanted?

trippfam07, 2022-01-22:
I looked at the logs and also found this.

[13:55:32][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
[13:55:32][D][uart_debug:114]: <<< 4E:57:01:21:00:00:00:00:06:00:01:79:30:01:0D:2A:02:0D:2D:03:0D:2D:04:0D:2A:05:0D:37:06:0D:28:07:0D:37:08:0D:29:09:0D:27:0A:0D:2D:0B:0D:2E:0C:0D:2A:0D:0D:37:0E:0D:2A:0F:0D:32:10:0D:25:80:00:15:81:00:11:82:00:11:83:15:15:84:93:86:85:50:86:02:87:00:00:89:00:00:00:00:8A:00:10:8B:00:00:8C:00:03:8E:16:D0:8F:10:40:90:0E:42:91:0D:DE:92:00:05:93:0A:28:94:0B:B8:95:00:05:96:01:2C:97:00:96:98:01:2C:99:00:64:9A:00:1E:9B:0D:48:9C:00:07:9D:01:9E:80:5A:9F:00:46
[13:55:32][W][jk_modbus:073]: JkModbus CRC Check failed! 4B8A!=4B0A
[13:55:32][D][uart_debug:114]: <<< A0:00:64:A1:00:64:A2:00:14:A3:00:46:A4:00:46:A5:FF:EC:A6:FF:F6:A7:FF:EC:A8:FF:F6:A9:10:AA:00:00:02:30:AB:01:AC:01:AD:04:C9:AE:01:AF:01:B0:00:0A:B1:14:B2:31:32:33:34:35:36:00:00:00:00:B3:00:B4:49:6E:70:75:74:20:55:73:B5:32:31:31:30:B6:00:01:F6:00:B7:48:38:2E:58:5F:5F:53:38:2E:31:34:55:5F:5F:5F:B8:00:B9:00:00:02:00:BA:42:54:33:30:37:32:30:32:30:31:32:30:30:30:30:32:30:30:35:32:31:30:30:31:C0:01:00:00:00:00:68:00:00:4B:0A

syssi, 2022-01-22:
The first line is the request (TX). Line 2 and 3 is the response from the BMS. If we add some linebreaks the response is better readable:

```
4E:57:01:21:00:00:00:00:06:00:01:
79:30:
01:0D:38:
02:0D:2F:
03:0D:34:
04:0D:31:
05:0D:2F:
06:0D:2C:
07:0D:3C:
08:0D:2C:
09:0D:33:
0A:0D:2A:
0B:0D:3C:
0C:0D:2A:
0D:0D:3E:
0E:0D:2D:
58:58:33:AC:18:19:00:15:
81:00:11:
82:00:08:
83:15:1A:
84:85:A5:
85:4F:
86:02:
87:00:00:
89:00:00:00:00:
8A:00:10:
8B:00:00:
8C:00:03:
8E:16:D0:
8F:10:40:
90:0E:42:
91:0D:DE:
92:00:05:
93:0A:28:
94:0B:B8:
95:00:05:
96:01:2C:
97:00:96:
98:01:2C:
99:00:64:
9A:00:1E:
9B:0D:48:
9C:00:07:
9D:01:
9E:00:5A:
9F:00:46:
A0:00:64:
A1:00:64:
A2:00:14:
A3:00:46:
A4:00:46:
A5:FF:EC:
A6:FF:F6:
A7:FF:EC:
A8:FF:F6:
A9:10:
AA:00:00:02:30:
AB:01:
AC:01:
AD:04:C9:
AE:01:
AF:01:D8:00:0A:
B1:14:
B2:31:32:33:34:35:36:00:00:00:00:
B3:C0:BB:9A:1B:AB:A3:20:55:73:
B5:32:31:31:30:
B6:00:01:F6:00:
B7:48:38:2E:58:5F:5F:53:38:2E:31:34:55:5F:5F:5F:
B8:00:
B9:00:00:02:00:
BA:42:54:33:30:37:32:30:32:30:31:32:30:30:30:30:32:30:30:35:32:31:30:30:31:
C0:01:
00:00:00:00:68:00:00:4B:68
```

syssi, 2022-01-22:
Please don't care about the CRC error. Some frames/responses are dropped because of CRC errors. The next reading will be fine again and will update the sensor measurements properly.

trippfam07, 2022-01-22:
Ok so on my bluetooth app There is a value for remain battery and it equals 81%
and there is a value for remain capacity and it equals 457.3 ah
Is there a way to make the remain capacity show up?

syssi, 2022-01-22:
Could you tell me the value of the `capacity_remaining` sensor? May be please provide the full log of sensor measurements.

syssi, 2022-01-22:
https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms/jk_bms.cpp#L127-L128

```
85:4F:  # Register 0x85, value 0x4F = 79%
```

trippfam07, 2022-01-22:
[14:15:27][I][jk_bms:061]: Status frame received
[14:15:27][D][sensor:125]: '${name} cell voltage 1': Sending state 3.37400 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} cell voltage 2': Sending state 3.36500 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} cell voltage 3': Sending state 3.38000 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} cell voltage 4': Sending state 3.36900 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} cell voltage 5': Sending state 3.36400 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} cell voltage 6': Sending state 3.36800 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} cell voltage 7': Sending state 3.36800 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} cell voltage 8': Sending state 3.36300 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} cell voltage 9': Sending state 3.37200 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} cell voltage 10': Sending state 3.36300 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} cell voltage 11': Sending state 3.37900 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} cell voltage 12': Sending state 3.36100 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} cell voltage 13': Sending state 3.38000 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} cell voltage 14': Sending state 3.36400 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} cell voltage 15': Sending state 3.37800 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} cell voltage 16': Sending state 3.35900 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} min cell voltage': Sending state 3.35900 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} max cell voltage': Sending state 3.38000 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} delta cell voltage': Sending state 0.02100 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} power tube temperature': Sending state 20.00000 °C with 0 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} temperature sensor 1': Sending state 17.00000 °C with 0 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} temperature sensor 2': Sending state 16.00000 °C with 0 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} total voltage': Sending state 53.89000 V with 2 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} current': Sending state -45.81000 A with 2 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} capacity remaining': Sending state 82.00000 % with 0 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} temperature sensors': Sending state 2.00000  with 0 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} total charging cycle capacity': Sending state 0.00000  with 0 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} battery strings': Sending state 16.00000  with 0 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} errors bitmask': Sending state 0.00000  with 0 decimals of accuracy
[14:15:27][D][text_sensor:067]: '${name} errors': Sending state ''
[14:15:27][D][sensor:125]: '${name} operation mode bitmask': Sending state 3.00000  with 0 decimals of accuracy
[14:15:27][D][text_sensor:067]: '${name} operation mode': Sending state 'Charging enabled;Discharging enabled'
[14:15:27][D][sensor:125]: '${name} total voltage overvoltage protection': Sending state 58.40000 V with 2 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} total voltage undervoltage protection': Sending state 41.60000 V with 2 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} cell voltage overvoltage protection': Sending state 3.65000 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} cell voltage overvoltage recovery': Sending state 3.55000 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} cell voltage overvoltage delay': Sending state 5.00000 s with 0 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} cell voltage undervoltage protection': Sending state 2.60000 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} cell voltage undervoltage recovery': Sending state 3.00000 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} cell voltage undervoltage delay': Sending state 5.00000 s with 0 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} cell pressure difference protection': Sending state 0.30000 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} discharging overcurrent protection': Sending state 150.00000 A with 0 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} discharging overcurrent delay': Sending state 300.00000 s with 0 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} charging overcurrent protection': Sending state 100.00000 A with 0 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} charging overcurrent delay': Sending state 30.00000 s with 0 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} balance starting voltage': Sending state 3.40000 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} balance opening pressure difference': Sending state 0.00700 V with 3 decimals of accuracy
[14:15:27][D][sensor:125]: '${name} power tube temperature protection': Sending state 90.00000 °C with 0 decimals of accuracy
[... cut ...]
[14:15:27][D][text_sensor:067]: '${name} password': Sending state '123456'
[14:15:27][D][text_sensor:067]: '${name} device type': Sending state 'Input Us'
[14:15:27][D][sensor:125]: '${name} total runtime': Sending state 2141.86670 h with 0 decimals of accuracy
[14:15:27][D][text_sensor:067]: '${name} software version': Sending state 'H8.X__S8.14U___'
[14:15:27][D][sensor:125]: '${name} actual battery capacity': Sending state 512.00000 Ah with 0 decimals of accuracy
[14:15:27][D][text_sensor:067]: '${name} manufacturer': Sending state 'BT3072020120000200521001'

syssi, 2022-01-22:
Another minor thing: Esphome wasn't able to replace the variable `${name}`. I assume it wasn't defined. Please add this to the top of your configuration:

```
substitutions:
  name: jk-bms
```

syssi, 2022-01-22:
May be I didn't understand what you are looking for. This value is available

```
[14:15:27][D][sensor:125]: '${name} capacity remaining': Sending state 82.00000 % with 0 decimals of accuracy
```

and looks good to me. Are you looking for the remaining capacity in Ah? Could you tell me the total capacity of your battery?

trippfam07, 2022-01-22:
Yes but on my bluetooth app there is 2 sections for the battery
1. Remain Battery = 82%
2. Remain Capacity = 463.4 ah
My total capacity is 560 ah

trippfam07, 2022-01-22:
[image]

syssi, 2022-01-22:
The `total_capacity` sensor isn't supported yet:

https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms/jk_bms.cpp#L280-L281

But it looks like your BMS returns the proper value here:

```
AA:00:00:02:30: # Register 0xAA, value 0x0230 = 560 decimal
```

It looks like the `actual_battery_capacity` sensor is useless. I assume the value will drop at some point from 512 to 256 or something. Could you monitor the `actual_battery_capacity` value for some days and provide some updates? I assume the resolution is super bad here.

But we could introduce a new calculated sensor by doing something like this: `560 * 82% = 459,2 Ah`

What do you think?

trippfam07, 2022-01-22:
So how would i program that? for the 560 decimal?

And yes the other would be good to

syssi, 2022-01-22:
I will care about it. It will take a few days.

trippfam07, 2022-01-22:
when I add the substitutions name to the config, do I have to flash again? And if yes can I do it while its connected to the bms?

syssi, 2022-01-22:
Yes. You have to flash the new version but the ESP doesn't need to be attached to the USB port anymore. You can use OTA now.

trippfam07, 2022-01-22:
Ok thank you very much for you help!!!!!
I look forward to hearing from you when you are done.

syssi, 2022-01-23:
I've improved the component. Please extend your configuration by the two new sensors:

```
    capacity_remaining_derived:
      name: "${name} capacity remaining derived"
    total_battery_capacity_setting:
      name: "${name} total battery capacity setting"
```

And try/flash again.

syssi, 2022-01-23:
Feel free to create a new issue if you are missing something else.

closed 2022-01-23 by syssi