syssi/esphome-jk-bms #34: CRC Errors

evil-dog, 2022-01-29:
I'm getting a ton of CRC errors on my esphome log. Intermittently it seems like it does receive a valid packet and shows some data, but the vast majority of the time I'm getting CRC errors.

BMS info:  Model: JK-B2A24S20P HW: 8.X SW: 8.20G Ver: 3.6.3
I'm currently using a HiLetgo ESP-32S module directly connected to the serial port on the BMS.

esphome logs Snippet:
```
[17:35:41][W][jk_modbus:073]: JkModbus CRC Check failed! 4F09!=4F49
[17:35:46][W][jk_modbus:073]: JkModbus CRC Check failed! 4ECA!=4F47
[17:35:51][W][jk_modbus:073]: JkModbus CRC Check failed! 4F50!=4F48
[17:35:56][W][jk_modbus:073]: JkModbus CRC Check failed! 4FE3!=4F48
[17:36:01][W][jk_modbus:073]: JkModbus CRC Check failed! 4E94!=4F45
[17:36:06][W][jk_modbus:073]: JkModbus CRC Check failed! 4F01!=4F49
[17:36:16][W][jk_modbus:073]: JkModbus CRC Check failed! 4FA7!=4F49
[17:36:21][W][jk_modbus:073]: JkModbus CRC Check failed! 73D!=E408
[17:36:26][W][jk_modbus:073]: JkModbus CRC Check failed! 4F50!=4F48
[17:36:31][I][jk_bms:061]: Status frame received
[17:36:31][D][sensor:125]: 'cell voltage 1': Sending state 3.30000 V with 3 decimals of accuracy
[17:36:31][D][sensor:125]: 'cell voltage 2': Sending state 3.30000 V with 3 decimals of accuracy
[17:36:31][D][sensor:125]: 'cell voltage 3': Sending state 3.30000 V with 3 decimals of accuracy
[17:36:31][D][sensor:125]: 'cell voltage 4': Sending state 3.30000 V with 3 decimals of accuracy
[17:36:31][D][sensor:125]: 'cell voltage 5': Sending state 3.30000 V with 3 decimals of accuracy
[17:36:31][D][sensor:125]: 'cell voltage 6': Sending state 3.29900 V with 3 decimals of accuracy
[17:36:31][D][sensor:125]: 'cell voltage 7': Sending state 3.30000 V with 3 decimals of accuracy
[17:36:31][D][sensor:125]: 'cell voltage 8': Sending state 3.30000 V with 3 decimals of accuracy
[17:36:31][D][sensor:125]: 'cell voltage 9': Sending state 3.30000 V with 3 decimals of accuracy
[17:36:31][D][sensor:125]: 'cell voltage 10': Sending state 3.30000 V with 3 decimals of accuracy
[17:36:31][D][sensor:125]: 'cell voltage 11': Sending state 3.30000 V with 3 decimals of accuracy
[17:36:31][D][sensor:125]: 'cell voltage 12': Sending state 3.30000 V with 3 decimals of accuracy
[17:36:31][D][sensor:125]: 'cell voltage 13': Sending state 3.30000 V with 3 decimals of accuracy
[17:36:31][D][sensor:125]: 'cell voltage 14': Sending state 3.30000 V with 3 decimals of accuracy
[17:36:31][D][sensor:125]: 'cell voltage 15': Sending state 3.29800 V with 3 decimals of accuracy
[17:36:31][D][sensor:125]: 'cell voltage 16': Sending state 3.30000 V with 3 decimals of accuracy
[17:36:31][D][sensor:125]: 'min cell voltage': Sending state 3.29800 V with 3 decimals of accuracy
[17:36:31][D][sensor:125]: 'max cell voltage': Sending state 3.30000 V with 3 decimals of accuracy
[17:36:31][D][sensor:125]: 'delta cell voltage': Sending state 0.00200 V with 3 decimals of accuracy
[17:36:31][D][sensor:125]: 'power tube temperature': Sending state 18.00000 °C with 0 decimals of accuracy
[17:36:31][D][sensor:125]: 'temperature 1': Sending state 16.00000 °C with 0 decimals of accuracy
[17:36:31][D][sensor:125]: 'temperature 2': Sending state 17.00000 °C with 0 decimals of accuracy
[17:36:31][D][sensor:125]: 'total voltage': Sending state 52.80000 V with 2 decimals of accuracy
[17:36:31][D][sensor:125]: 'current': Sending state 0.00000 A with 2 decimals of accuracy
[17:36:31][D][sensor:125]: 'capacity remaining': Sending state 30.00000 % with 0 decimals of accuracy
[17:36:31][D][sensor:125]: 'charging cycles': Sending state 0.00000  with 0 decimals of accuracy
[17:36:31][D][sensor:125]: 'total charging cycle capacity': Sending state 0.00000  with 0 decimals of accuracy
[17:36:31][D][text_sensor:067]: 'errors': Sending state ''
[17:36:31][D][text_sensor:067]: 'operation mode': Sending state ''
[17:36:31][D][text_sensor:067]: 'password': Sending state '181207'
[17:36:31][D][sensor:125]: 'total runtime': Sending state 0.00000 h with 0 decimals of accuracy
[17:36:31][D][sensor:125]: 'actual battery capacity': Sending state 256.00000 Ah with 0 decimals of accuracy
[17:36:36][W][jk_modbus:073]: JkModbus CRC Check failed! 4EB6!=4F48
[17:36:41][W][jk_modbus:073]: JkModbus CRC Check failed! 4F48!=4F49
[17:36:46][W][jk_modbus:073]: JkModbus CRC Check failed! 4EF1!=4F49
[17:36:51][W][jk_modbus:073]: JkModbus CRC Check failed! 4ED9!=4F48
```

Any ideas for troubleshooting or if this is maybe related to the other open issue with this BMS model?

syssi, 2022-01-30:
I'm not aware of related issues. How long is your wire between BMS and your ESP32?

evil-dog, 2022-01-30:
The cable I made is about 8in long.

And the issue I was talking about was issue #13.

Is there any additional logging I can enable to provide more info to debug?

syssi, 2022-01-31:
Please enable the debug mode of the uart:

```
uart:
  id: uart0
  baud_rate: 115200
  rx_buffer_size: 384
  tx_pin: GPIO14
  rx_pin: GPIO4
  debug:
    direction: BOTH
```

The log should look like this:

```
[13:43:12][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
[13:43:12][D][uart_debug:114]: <<< 4E:57:01:21:00:00:00:00:06:00:01:79:30:01:0D:38:02:0D:2F:03:0D:34:04:0D:31:05:0D:2F:06:0D:2C:07:0D:3C:08:0D:2C:09:0D:33:0A:0D:2A:0B:0D:3C:0C:0D:2A:0D:0D:3E:0E:0D:2D:58:58:33:AC:18:19:00:15:81:00:11:82:00:08:83:15:1A:84:85:A5:85:4F:86:02:87:00:00:89:00:00:00:00:8A:00:10:8B:00:00:8C:00:03:8E:16:D0:8F:10:40:90:0E:42:91:0D:DE:92:00:05:93:0A:28:94:0B:B8:95:00:05:96:01:2C:97:00:96:98:01:2C:99:00:64:9A:00:1E:9B:0D:48:9C:00:07:9D:01:9E:00:5A:9F:00:46:A0
[13:43:12][D][uart_debug:114]: <<< 00:64:A1:00:64:A2:00:14:A3:00:46:A4:00:46:A5:FF:EC:A6:FF:F6:A7:FF:EC:A8:FF:F6:A9:10:AA:00:00:02:30:AB:01:AC:01:AD:04:C9:AE:01:AF:01:D8:00:0A:B1:14:B2:31:32:33:34:35:36:00:00:00:00:B3:C0:BB:9A:1B:AB:A3:20:55:73:B5:32:31:31:30:B6:00:01:F6:00:B7:48:38:2E:58:5F:5F:53:38:2E:31:34:55:5F:5F:5F:B8:00:B9:00:00:02:00:BA:42:54:33:30:37:32:30:32:30:31:32:30:30:30:30:32:30:30:35:32:31:30:30:31:C0:01:00:00:00:00:68:00:00:4B:68
```

`>>>` means TX. `<<<` means RX.

evil-dog, 2022-02-01:
Well, I guess I had a loose connection or something. When I enabled the uart debug and connected the esp back to the BMS it is now working fine. I've gone back and disabled the uart debug and it's still working.

Another thing that could have affected this is that I also disabled the esphome serial logging. I did this as I was going to use those pins, but for some reason I couldn't seem to get those pins to work, so I moved back to the previous pins that I was using when getting the CRC errors and it started working perfectly.

Thanks for the quick replies and I guess you can close this out.

closed 2022-02-01 by syssi