syssi/esphome-jk-bms #35: Is software v9 not supported?

jonadis, 2022-02-10:
I'm trying out this code on an ESP32 with a JK-BD6A17S8P
Only a few of the parameters are coming through with the rest either showing non-sensical values or non-printing characters.

Details from the App over Bluetooth:
Hardware Ver: 9.X
Software Ver: 9.01G

Example conversation from ESPHome log:

```
[I][jk_bms:061]: Status frame received
[D][sensor:125]: 'jk-bms min cell voltage': Sending state 100.00000 V with 3 decimals of accuracy
[D][sensor:125]: 'jk-bms max cell voltage': Sending state -100.00000 V with 3 decimals of accuracy
[D][sensor:125]: 'jk-bms delta cell voltage': Sending state -200.00000 V with 3 decimals of accuracy
[D][text_sensor:067]: 'jk-bms errors': Sending state ''
[D][text_sensor:067]: 'jk-bms operation mode': Sending state 'Discharging enabled;Balancer enabled;Battery dropped'
[D][text_sensor:067]: 'jk-bms operation mode': Sending state 'Discharging enabled;Balancer enabled;Battery dropped'
[D][text_sensor:067]: 'jk-bms battery type': Sending state 'Unknown'
[D][text_sensor:067]: 'jk-bms password': Sending state ''
[D][text_sensor:067]: 'jk-bms device type': Sending state ''
[D][text_sensor:067]: 'jk-bms software version': Sending state '�?xV���S�?'
[D][sensor:125]: 'jk-bms actual battery capacity': Sending state 744619328.00000 Ah with 0 decimals of accuracy
[D][text_sensor:067]: 'jk-bms manufacturer': Sending state ''
[D][text_sensor:067]: 'jk-bms manufacturer': Sending state ''
```

syssi, 2022-02-10:
It looks like the response payload has changed. Does the BMS return this garbage periodically depending on the update/request interval?

jonadis, 2022-02-10:
Actually I just moved the pins on the ESP from 14 & 4 to 
```
  tx_pin: GPIO16
  rx_pin: GPIO17
```
as it is my understanding that that makes use of the hardware UART: [https://esphome.io/components/logger.html#hardware-uarts](https://esphome.io/components/logger.html#hardware-uarts)
and after that change I'm in good shape.

Perhaps I just had a loose wire and it was just the re-connection that solved it, or perhaps there's something to that hardware UART business.  In any case thanks for putting together this awesome project.