syssi/esphome-jk-bms #37: JK-B2A24S20P hardware 10.X-W, software 10.02

SeByDocKy, 2022-02-15:
Hi,

After being able to integrate your amazing pipsolar package for my axpert max I aka PIP8048, now I am trying your JKBMS Integration.
Actually it's plugged on the hardware UART0 GPIO01/GPIO03.... Unfortunatly absoluptly no oommunication is logged :(
Here is my little yaml
````
esphome:
  name: test_clapet
  platform: ESP32
  board: esp32dev

  

external_components:
  - source: github://syssi/esphome-jk-bms@main

# Enable logging
logger:
  #baud_rate: 0
  #baud_rate: 0
#  level: VERY_VERBOSE
  # logs:
    # uart: VERY_VERBOSE 
    # api.service: WARN
    # ota: WARN
    # sensor: DEBUG

# Enable Home Assistant API
api:

ota:
 
wifi:
  ssid: !secret esphome_ssid
  password: !secret esphome_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Geothermie Fallback Hotspot"
    password: "sebydocky"

captive_portal:

time:
  - platform: homeassistant
    id: homeassistant_time  
 
uart:
  
 ######### UART for the JKBMS ###### 
  
  - id: uart1
    tx_pin: GPIO03 #GPIO18
    rx_pin: GPIO01 #GPIO19
    baud_rate: 115200
    rx_buffer_size: 384 
 
jk_modbus:
   id: modbus1
   uart_id: uart1

jk_bms:
  id: bms0
  jk_modbus_id: modbus1 
 
 
sensor:  
  - platform: jk_bms
    min_cell_voltage:
      name: "test min cell voltage"
    max_cell_voltage:
      name: "test max cell voltage"
    delta_cell_voltage:
      name: "test delta cell voltage"
    cell_voltage_1:
      name: "test cell voltage 1"
    cell_voltage_2:
      name: "test cell voltage 2"
    cell_voltage_3:
      name: "test cell voltage 3"
    cell_voltage_4:
      name: "test cell voltage 4"
    cell_voltage_5:
      name: "test cell voltage 5"
    cell_voltage_6:
      name: "test cell voltage 6"
    cell_voltage_7:
      name: "test cell voltage 7"
    cell_voltage_8:
      name: "test cell voltage 8"
    cell_voltage_9:
      name: "test cell voltage 9"
    cell_voltage_10:
      name: "test cell voltage 10"
    cell_voltage_11:
      name: "test cell voltage 11"
    cell_voltage_12:
      name: "test cell voltage 12"
    cell_voltage_13:
      name: "test cell voltage 13"
    cell_voltage_14:
      name: "test cell voltage 14"
    cell_voltage_15:
      name: "test cell voltage 15"
    cell_voltage_16:
      name: "test cell voltage 16"
    power_tube_temperature:
      name: "test power tube temperature"
    temperature_sensor_1:
      name: "test temperature sensor 1"
    temperature_sensor_2:
      name: "test temperature sensor 2"  
    total_voltage:
      name: "test total voltage"
    current:
      name: "test current"  
    capacity_remaining:
      name: "test capacity remaining"  
````

Of course I trried my swapping TX/RX GPIO definition without more success....

syssi, 2022-02-15:
Where did you buy the BMS? Could you ask the seller about some specifications?

syssi, 2022-02-15:
Please avoid the first UART it's used by the `logger`. If you disable the serial logging by setting the baud rate to `0` the ESP still writes some messages to the UART on boot-up (bootloader etc.). Just use any other GPIO. "All" GPIOs of an ESP32 can be used as hardware UART.

SeByDocKy, 2022-02-15:
> Please avoid the first UART it's used by the `logger`. If you disable the serial logging by setting the baud rate to `0` the ESP still writes some messages to the UART on boot-up (bootloader etc.). Just use any other GPIO. "All" GPIOs of an ESP32 can be used as hardware UART.

Yes I know I should avoir the UART and I disable the baud_rate: 0.... but the problem is I am using the two other UARTs of the ESP32.... and in order to let work, one of the three must be an hardware as far I understood.... Now I am doing some test with a standalone ESP32... I will check via software UART. What GPIO are you advised ?

syssi, 2022-02-15:
There is no software UART implementation (in Esphome) if you use a ESP32. The ESP32 supports 3 hardware UARTs (auto-assigned by some internal matrix).

Software serial is a thing if you use a ESP8266. The ESP8266 doesn't have a "matrix" to wire some features (UARTs) internally. The hardware UARTs are assigned to some selected GPIOs. If you use another GPIOs (which doesn't support hardware UART) the software serial implementation is used (automatically).

syssi, 2022-02-15:
I prefer this setup for poking around on a D1 Mini ESP32:

```
uart:
  id: uart0
  baud_rate: 115200
  rx_buffer_size: 384
  tx_pin: GPIO14
  rx_pin: GPIO4
  debug:
    direction: BOTH
```

SeByDocKy, 2022-02-15:
> In other words: If you use vanilla ESPHome + ESP32 the maximal number of UARTs is limited to 3.

Ok :) ...Actually I am testing on a vanilla ESP32 with one unique softUART (04/14) as in your yaml. 
and it's working ....  Even the choice of GPIO04 is not perfect since the ESP32 don't boot correctly. I need first to disconnect the UART connexion :)

SeByDocKy, 2022-02-15:
Hummm seems I wrote too fast... I had two data sequence ... then anything more :(

syssi, 2022-02-15:
Do you see anything at the logs? If the ESP32 reboots/crashs please try to reduce the number of sensors.

SeByDocKy, 2022-02-15:
Update ... 
I am receiving some data but completly dalse ...

```[14:20:34][I][jk_bms:061]: Status frame received
[14:20:34][D][sensor:125]: 'test min cell voltage': Sending state 100.00000 V with 3 decimals of accuracy
[14:20:34][D][sensor:125]: 'test max cell voltage': Sending state -100.00000 V with 3 decimals of accuracy
[14:20:34][D][sensor:125]: 'test delta cell voltage': Sending state -200.00000 V with 3 decimals of accuracy
[14:20:34][D][sensor:125]: 'test power tube temperature': Sending state 0.00000 °C with 0 decimals of accuracy
[14:20:34][D][sensor:125]: 'test temperature sensor 1': Sending state -22089.00000 °C with 0 decimals of accuracy
[14:20:34][D][sensor:125]: 'test temperature sensor 2': Sending state -22089.00000 °C with 0 decimals of accuracy
[14:20:34][D][sensor:125]: 'test total voltage': Sending state 465.76999 V with 2 decimals of accuracy
[14:20:34][D][sensor:125]: 'test current': Sending state 0.00000 A with 2 decimals of accuracy
[14:20:34][D][sensor:125]: 'test capacity remaining': Sending state 252.00000 % with 0 decimals of accuracy
[14:20:39][I][jk_bms:061]: Status frame received
[14:20:39][D][sensor:125]: 'test min cell voltage': Sending state 100.00000 V with 3 decimals of accuracy
[14:20:39][D][sensor:125]: 'test max cell voltage': Sending state -100.00000 V with 3 decimals of accuracy
[14:20:39][D][sensor:125]: 'test delta cell voltage': Sending state -200.00000 V with 3 decimals of accuracy
[14:20:39][D][sensor:125]: 'test power tube temperature': Sending state 0.00000 °C with 0 decimals of accuracy
[14:20:39][D][sensor:125]: 'test temperature sensor 1': Sending state -22089.00000 °C with 0 decimals of accuracy
[14:20:39][D][sensor:125]: 'test temperature sensor 2': Sending state -22089.00000 °C with 0 decimals of accuracy
[14:20:39][D][sensor:125]: 'test total voltage': Sending state 465.76999 V with 2 decimals of accuracy
[14:20:39][D][sensor:125]: 'test current': Sending state 0.00000 A with 2 decimals of accuracy
[14:20:39][D][sensor:125]: 'test capacity remaining': Sending state 252.00000 % with 0 decimals of accuracy
```

SeByDocKy, 2022-02-15:
I guess I need to recompile in very_verbose for the logger :)

syssi, 2022-02-15:
Please check your wiring and try to swap the wires again.

Another user had success using `GPIO16` and `GPIO17`: https://github.com/syssi/esphome-jk-bms/issues/35

syssi, 2022-02-15:
And please make sure `GND` is connected properly.

syssi, 2022-02-15:
Another user having trouble to interface this BMS model: https://github.com/syssi/esphome-jk-bms/discussions/12

SeByDocKy, 2022-02-15:
Ok first very_verbose in GPIO04/GPIO14

http://pastebin.fr/102039

syssi, 2022-02-15:
Could you provide a photo of your setup?

SeByDocKy, 2022-02-15:
Going to wire with GPIO 16/17

> Could you provide a photo of your setup?

Ok I will try...... coz the BMS is in woodbox.

syssi, 2022-02-15:
Please try to make your log less noisy by:

```
logger:
  level: VERY_VERBOSE
  logs:
    api.service: WARN
    ota: WARN
    sensor: WARN

uart:
  id: uart0
  baud_rate: 115200
  rx_buffer_size: 384
  tx_pin: GPIO14
  rx_pin: GPIO4
  debug:
    direction: BOTH
```

syssi, 2022-02-15:
UART cannot work if you use 2 wires only. You have to attach RX, TX and GND!

SeByDocKy, 2022-02-15:
> api.service: WARN
>     ota: WARN
>     sensor: WARN

Of course there are 3 :)

SeByDocKy, 2022-02-15:
Ok here are the log as you asked : 
````
INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from test_clapet.local using esphome API
INFO Successfully connected to test_clapet.local
[14:49:07][I][app:102]: ESPHome version 2022.3.0-dev compiled on Feb 15 2022, 14:47:50
[14:49:07][C][wifi:491]: WiFi:
[14:49:07][C][wifi:353]:   Local MAC: 3C:71:BF:A1:F0:80
[14:49:07][C][wifi:354]:   SSID: [redacted]
[14:49:07][C][wifi:355]:   IP Address: 192.168.1.87
[14:49:07][C][wifi:357]:   BSSID: [redacted]
[14:49:07][C][wifi:358]:   Hostname: 'test_clapet'
[14:49:07][C][wifi:360]:   Signal strength: -63 dB ▂▄▆█
[14:49:07][V][wifi:362]:   Priority: 0.0
[14:49:07][C][wifi:364]:   Channel: 12
[14:49:07][C][wifi:365]:   Subnet: 255.255.255.0
[14:49:07][C][wifi:366]:   Gateway: 192.168.1.1
[14:49:07][C][wifi:367]:   DNS1: 192.168.1.1
[14:49:07][C][wifi:368]:   DNS2: 192.168.1.1
[14:49:07][C][logger:233]: Logger:
[14:49:07][C][logger:234]:   Level: VERY_VERBOSE
[14:49:07][C][logger:235]:   Log Baud Rate: 115200
[14:49:07][C][logger:236]:   Hardware UART: UART0
[14:49:07][C][logger:238]:   Level for 'api.service': WARN
[14:49:07][C][logger:238]:   Level for 'ota': WARN
[14:49:07][C][logger:238]:   Level for 'sensor': DEBUG
[14:49:07][C][uart.arduino_esp32:107]: UART Bus:
[14:49:07][C][uart.arduino_esp32:108]:   TX Pin: GPIO19
[14:49:07][C][uart.arduino_esp32:109]:   RX Pin: GPIO18
[14:49:07][C][uart.arduino_esp32:111]:   RX Buffer Size: 384
[14:49:07][C][uart.arduino_esp32:113]:   Baud Rate: 115200 baud
[14:49:07][C][uart.arduino_esp32:114]:   Data Bits: 8
[14:49:07][C][uart.arduino_esp32:115]:   Parity: NONE
[14:49:07][C][uart.arduino_esp32:116]:   Stop bits: 1
[14:49:07][C][jk_modbus:095]: JkModbus:
[14:49:07][C][homeassistant.time:010]: Home Assistant Time:
[14:49:07][C][homeassistant.time:011]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[14:49:07][C][jk_bms:462]: JkBms:
[14:49:07][C][jk_bms:463]:   Address: 0x4E
[14:49:07][C][jk_bms:465]: Cell Voltage 1 'test cell voltage 1'
[14:49:07][C][jk_bms:465]:   Device Class: 'voltage'
[14:49:07][C][jk_bms:465]:   State Class: 'measurement'
[14:49:07][C][jk_bms:465]:   Unit of Measurement: 'V'
[14:49:07][C][jk_bms:465]:   Accuracy Decimals: 3
[14:49:07][C][jk_bms:466]: Cell Voltage 2 'test cell voltage 2'
[14:49:07][C][jk_bms:466]:   Device Class: 'voltage'
[14:49:07][C][jk_bms:466]:   State Class: 'measurement'
[14:49:07][C][jk_bms:466]:   Unit of Measurement: 'V'
[14:49:07][C][jk_bms:466]:   Accuracy Decimals: 3
[14:49:07][C][jk_bms:467]: Cell Voltage 3 'test cell voltage 3'
[14:49:07][C][jk_bms:467]:   Device Class: 'voltage'
[14:49:07][C][jk_bms:467]:   State Class: 'measurement'
[14:49:07][C][jk_bms:467]:   Unit of Measurement: 'V'
[14:49:07][C][jk_bms:467]:   Accuracy Decimals: 3
[14:49:07][C][jk_bms:468]: Cell Voltage 4 'test cell voltage 4'
[14:49:07][C][jk_bms:468]:   Device Class: 'voltage'
[14:49:07][C][jk_bms:468]:   State Class: 'measurement'
[14:49:07][C][jk_bms:468]:   Unit of Measurement: 'V'
[14:49:08][C][jk_bms:468]:   Accuracy Decimals: 3
[14:49:08][C][jk_bms:469]: Cell Voltage 5 'test cell voltage 5'
[14:49:08][C][jk_bms:469]:   Device Class: 'voltage'
[14:49:08][C][jk_bms:469]:   State Class: 'measurement'
[14:49:08][C][jk_bms:469]:   Unit of Measurement: 'V'
[14:49:08][C][jk_bms:469]:   Accuracy Decimals: 3
[14:49:08][C][jk_bms:470]: Cell Voltage 6 'test cell voltage 6'
[14:49:08][C][jk_bms:470]:   Device Class: 'voltage'
[14:49:08][C][jk_bms:470]:   State Class: 'measurement'
[14:49:08][C][jk_bms:470]:   Unit of Measurement: 'V'
[... 201 lines cut ...]
[14:49:40][D][sensor:125]: 'test temperature sensor 1': Sending state -22089.00000 °C with 0 decimals of accuracy
[14:49:40][D][sensor:125]: 'test temperature sensor 2': Sending state -22089.00000 °C with 0 decimals of accuracy
[14:49:40][D][sensor:125]: 'test total voltage': Sending state 430.37000 V with 2 decimals of accuracy
[14:49:40][D][sensor:125]: 'test current': Sending state 0.00000 A with 2 decimals of accuracy
[14:49:40][D][sensor:125]: 'test capacity remaining': Sending state 186.00000 % with 0 decimals of accuracy
[14:49:40][V][component:199]: Component jk_modbus took a long time for an operation (0.15 s).
[14:49:40][V][component:200]: Components should block for at most 20-30ms.
[14:49:40][D][uart_debug:114]: <<< 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
[14:49:45][VV][scheduler:186]: Running interval 'update' with interval=5000 last_execution=39997 (now=44997)
[14:49:45][VV][uart.arduino_esp32:150]:     Flushing...
[14:49:45][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
[14:49:45][I][jk_bms:061]: Status frame received
[14:49:45][D][sensor:125]: 'test min cell voltage': Sending state 100.00000 V with 3 decimals of accuracy
[14:49:45][D][sensor:125]: 'test max cell voltage': Sending state -100.00000 V with 3 decimals of accuracy
[14:49:45][D][sensor:125]: 'test delta cell voltage': Sending state -200.00000 V with 3 decimals of accuracy
[14:49:45][D][sensor:125]: 'test power tube temperature': Sending state 0.00000 °C with 0 decimals of accuracy
[14:49:45][D][sensor:125]: 'test temperature sensor 1': Sending state -22089.00000 °C with 0 decimals of accuracy
[14:49:45][D][sensor:125]: 'test temperature sensor 2': Sending state -22089.00000 °C with 0 decimals of accuracy
[14:49:45][D][sensor:125]: 'test total voltage': Sending state 430.37000 V with 2 decimals of accuracy
[14:49:45][D][sensor:125]: 'test current': Sending state 0.00000 A with 2 decimals of accuracy
[14:49:45][D][sensor:125]: 'test capacity remaining': Sending state 186.00000 % with 0 decimals of accuracy
[14:49:45][V][component:199]: Component jk_modbus took a long time for an operation (0.16 s).
[14:49:45][V][component:200]: Components should block for at most 20-30ms.
[14:49:45][D][uart_debug:114]: <<< 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
````

syssi, 2022-02-15:
This is the important part of the capture:

```
# Request
[14:49:45][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29

# Response
[14:49:45][D][uart_debug:114]: <<< 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
```

It looks like the `response` isn't a valid response from the BMS. It could be some echo / crosstalk / short circuit / missing GND issue because there is no difference in the `Request` (TX) and the `Response` (RX).

SeByDocKy, 2022-02-15:
Ok ... I will check that point.... I just soldered a connector (JST XH 3p) from the provided cable. I will by able the check not before 2h. Anyway thanks for your Huge help ...

SeByDocKy, 2022-02-15:
Here are some picts

[https://ibb.co/nb2XqgH](url)

So I wired a cable JST XH 3P, 15cm 
RS485 <=> JST XH 3P
black  <=> black
yellow <=> yellow
white <=> red

Then from this femmal cable into the ESP

[https://ibb.co/hCwwXy2](url)

[https://ibb.co/Ssj6D2Q](url)

[https://ibb.co/VLWf2HF](url)
=

jonadis, 2022-02-15:
I had a very similar output to yours above but then I switched to:

```
uart:
  id: uart2
  baud_rate: 115200
  rx_buffer_size: 384
  tx_pin: GPIO16
  rx_pin: GPIO17
```
 
I have a slightly different BMS than yours but at least on my GPIO16 goes to the pin closest to ground and GPIO17 goes to the pin closest to V_Batt

It's been working well for days now.

SeByDocKy, 2022-02-15:
> I had a very similar output to yours above but then I switched to:
>
> ```
> uart:
>   id: uart2
>   baud_rate: 115200
>   rx_buffer_size: 384
>   tx_pin: GPIO16
>   rx_pin: GPIO17
> ```
>
> I have a slightly different [...]

Can you confirm to me that triplet color from your RS485 module are (black, white, yellow) and black is the ground ?

jonadis, 2022-02-15:
I don't have the RS-485 module.  I made a custom cable that fits the RS-485 port on the BMS.
You can check which ports are +V_Batt and GND with a multi-meter, and that just leaves two other pins for TX&RX that you can just try one way and then another, but as I said above the pin closest to ground on mine is what goes into GPIO16 on the ESP.

SeByDocKy, 2022-02-15:
> I don't have the RS-485 module. I made a custom cable that fits the RS-485 port on the BMS. You can check which ports are +V_Batt and GND with a multi-meter, and that just leaves two other pins for [...]

You mean you attacked directly the RS-485 port from the JKBMS to the ESP32 without using the module ?

jonadis, 2022-02-15:
Correct,
Per the schematic in the instructions: [https://github.com/syssi/esphome-jk-bms](https://github.com/syssi/esphome-jk-bms)

SeByDocKy, 2022-02-15:
> Correct, Per the schematic in the instructions: https://github.com/syssi/esphome-jk-bms

OMG :) maybe it's the trick ..... Hope I have somewhere some JST picoblade 1.25mm 4P :) ..

jonadis, 2022-02-15:
Maybe ... 
But I do have a different model BMS so I can't say for sure.
Mine at least is Molex PicoBlade 1.25mm 4P

SeByDocKy, 2022-02-15:
Good news :) It's WORKING fine :)

closed 2022-02-15 by syssi

SeByDocKy, 2022-02-15:
And FYI, it's also working fine on my "PVbrain" ..... 3 UART used : JKBMS,PIPsolar and dual PZEM 004T :)

syssi, 2022-02-15:
Do you use the official PZEM component? I'm aware of one bug report telling the component is unstable.

SeByDocKy, 2022-02-15:
> Do you use the official PZEM component? I'm aware of one bug report telling the component is unstable.

Yes ... I got some since 2 years ..... Super stable for me....

syssi, 2022-02-15:
Cool! I will use one of these devices (PZEM-017 DC 0-300V 100A) as digital shunt in future.

https://esphome.io/components/sensor/pzemdc.html

SeByDocKy, 2022-02-15:
> Cool! I will use one of these devices (PZEM-017 DC 0-300V 100A) as digital shunt in future.
> 
> https://esphome.io/components/sensor/pzemdc.html

maybe ... I was not clear, the PZEM I installed are PZEM 004T V3 (modbus based). I don't plan to install the DC version with a shunt. The problem, the sens of current is not reported with PZEM ... it's big problem IMHO for solar application. 
I bought such module.... : https://fr.aliexpress.com/item/4000762882822.html?gatewayAdapt=glo2fra&spm=a2g0o.9042311.0.0.27426c37To8Uh6
There are a UART port .... maybe a good excercice for hacking something.