syssi/esphome-jk-bms #45: Not getting always all answers. Any experience?

ortegafernando, 2022-02-27:
HI, I have connected my esp8266 with RS485/port (direct rx and tx comms), and also I have a 40v to 3.3v switch converter, so I power my esp8266 from the bms itself.
I have problems with comms, may be is it because I have my switch converter in the same "shinrktube" as my rx and tx cables? 
What is your error percentaje in comms?
Thanks.

syssi, 2022-02-27:
Which BMS model do you have (for better comparison) and which GPIOs do you use for RX and TX? If it's not `GPIO1` and `GPIO3` Esphome will use a software UART implementation which leads to more errors on high baud rates.

ortegafernando, 2022-02-27:
> Which BMS model do you have (for better comparison) and which GPIOs do you use for RX and TX? If it's not `GPIO1` and `GPIO3` Esphome will use a software UART implementation which leads to more errors on high baud rates.

Thanks, I am using the 8S 2A. I use hardware UART, and "sometimes" I get all answer, buf after a while, very few, then again some, then again few or even anyone, ... 

Probably is not a connection problem. must be some kind of parasic interference?

SeByDocKy, 2022-02-27:
I have no problem with a 1,5m long cable. & ESP32. In order to test your hypethesis, you should power your ESP from a different & independent power source ...

ortegafernando, 2022-02-27:
> I have no problem with a 1,5m long cable. & ESP32. In order to test your hypethesis, you should power your ESP from a different & independent power source ...

Where are you powering from?

ortegafernando, 2022-03-02:
Hi, it was finally that I have very very near the rx and tx cables of the switch converter from battery voltage to 3.3v. After moving them 20 milimeters from switch converter .... everything is ok.

closed 2022-03-02 by ortegafernando