syssi/esphome-jk-bms #49: UART traffic capture: BK3432 BLE module <-> HK32F103C8T6 MCU

syssi, 2022-02-28:
Hardware: JK-BD6A17S6P, hardware version 7.2, software version 7.1.0H

[jk-bms-sigrok.zip](https://github.com/syssi/esphome-jk-bms/files/8154760/jk-bms-sigrok.zip)

I assume it's 19200 baud.

syssi, 2022-05-09:
https://github.com/jblance/mpp-solar/issues/224

Learnings:

1. My traffic capture above is probably garbage because the selected sampling rate was too low
2. The baud rate is `115200` baud
3. Requests & responds are using the `JK02` protocol/frames
4. `P11` must be pulled high (VCC). If P11 isn't pulled the BMS doesn't respond to requests.

TODO: Check the logic level (3.3V vs. 5V?).

Smoeller0, 2022-05-09:
When u start the Communication the Bms sends all Data in interval around 1 second. 
U Need only start once until Power off the Bms.

Start via app is possible.

Smoeller0, 2022-05-09:
When u Recive 300 Byte ur settins are okay.

syssi, 2023-02-23:
This issue is super seeded by #108.

closed 2023-02-23 by syssi

bilogic, 2024-11-16:
Were you guys able to flash the BK3432 with ESPHome?

syssi, 2024-11-16:
No. This wasn't the goal of the issue.

attila-lendvai, 2024-12-10:
> Were you guys able to flash the BK3432 with ESPHome?

@bilogic did you find anything about that^?

my immediate goal is to decide whether flashing my Rojeco PTM-102 cat feeder with ESPhome is feasible or not. (it has a USB connected motorized camera)

and my ultimate goal is to use the feeder without any goddamn online registration!

(sorry for the off topic, a websearch brought me here)

bilogic, 2024-12-11:
@attila-lendvai we share similar goals 

I managed to overcome all software issues, but got stuck with powering the ESP32 on the battery powered device.
If I connect just EN, 3v3 and GND of the ESP32, it doesn't transmit any BLE.

I have no idea how to go about finding the problem, so project currently on hold til I have some ideas to try.