syssi/esphome-jk-bms #56: JKBMS are no more compiling with last dev

SeByDocKy, 2022-03-11:
According to jessierocks, need some update :  https://github.com/esphome/esphome/pull/3216

syssi, 2022-03-12:
I would like to care about this issue as soon the new ESPHome version is released. The CI of this project does a daily build against the current release to make sure everything is in shape: https://github.com/syssi/esphome-jk-bms/actions

Another idea is to introduce a dev branch to work on the ESPhome dev compatibility already. So `dev` branch user could select `@dev` at the external components section.

syssi, 2022-03-12:
Fixed at the main branc by 3eec9f25b65ea8ba126b24c6d1a3ca5f64d29719. Just try again! :-)

closed 2022-03-12 by syssi

SeByDocKy, 2022-03-12:
Super I will try .... :) the same problem fir pipsolar :(

syssi, 2022-03-12:
Let's tackle this at my pipslar repo!

SeByDocKy, 2022-03-12:
Ok the JKBMS part is now compiling :)