syssi/esphome-jk-bms #57: Support for hardware 6.X

miguel300477, 2022-03-24:
Hi,

Im trying to get the BMS to comunicate with the ESP8266 WEMOS D1 mini, but i get alot off rubish data.

`INFO Reading configuration /config/esphome/jkbms.yaml...
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
INFO Starting log output from jk-bms/debug
INFO Connected to MQTT broker!
[10:40:20][D][uart_debug:114]: <<< 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
[10:40:23][VV][scheduler:185]: Running interval 'update' with interval=5000 last_execution=269692 (now=274692)
[10:40:23][VV][uart.arduino_esp8266:180]:     Flushing...
[10:40:23][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
[10:40:23][I][jk_bms:062]: Status frame received
[10:40:23][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_min_cell_voltage/state' payload='100.000' retain=1)
[10:40:23][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_max_cell_voltage/state' payload='-100.000' retain=1)
[10:40:23][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_delta_cell_voltage/state' payload='-200.000' retain=1)
[10:40:23][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_power_tube_temperature/state' payload='0' retain=1)
[10:40:23][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_temperature_sensor_1/state' payload='0' retain=1)
[10:40:23][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_temperature_sensor_2/state' payload='-3739' retain=1)
[10:40:23][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_total_voltage/state' payload='514.67' retain=1)
[10:40:23][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_current/state' payload='0.00' retain=1)
[10:40:23][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_capacity_remaining/state' payload='255' retain=1)
[10:40:23][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_temperature_sensors/state' payload='204' retain=1)
[10:40:23][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_charging_cycles/state' payload='65343' retain=1)
[10:40:23][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_total_charging_cycle_capacity/state' payload='1193427072' retain=1)
[10:40:23][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_battery_strings/state' payload='8256' retain=1)
[10:40:23][V][text_sensor:016]: 'jk-bms errors': Received new state Low capacity;Charging overvoltage;Charging overcurrent;Discharging overcurrent;Overtemperature alarm in the battery box;Battery low temperature;Cell overvoltage;309_A protection
[10:40:23][D][text_sensor:067]: 'jk-bms errors': Sending state 'Low capacity;Charging overvoltage;Charging overcurrent;Discharging overcurrent;Overtemperature alarm in the battery box;Battery low temperature;Cell overvoltage;309_A protection'
[10:40:23][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_errors/state' payload='Low capacity;Charging overvoltage;Charging overcurrent;Discharging overcurrent;Overtemperature alarm in the battery box;Battery low temperature;Cell overvoltage;309_A protection' retain=1)
[10:40:23][V][text_sensor:016]: 'jk-bms operation mode': Received new state Charging enabled;Discharging enabled;Battery dropped
[10:40:23][D][text_sensor:067]: 'jk-bms operation mode': Sending state 'Charging enabled;Discharging enabled;Battery dropped'
[10:40:23][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_operation_mode/state' payload='Charging enabled;Discharging enabled;Battery dropped' retain=1)
[10:40:23][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_manufacturer/state' payload='' retain=1)
[10:40:23][V][component:199]: Component jk_modbus took a long time for an operation (0.05 s).
[10:40:23][V][component:200]: Components should block for at most 20-30ms.
[10:40:23][D][uart_debug:114]: <<< 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
[10:40:24][VV][scheduler:185]: Running timeout 'publish' with interval=1000 last_execution=274745 (now=275745)
[10:40:28][VV][scheduler:185]: Running interval 'update' with interval=5000 last_execution=274692 (now=279692)
[10:40:28][VV][uart.arduino_esp8266:180]:     Flushing...
[10:40:28][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
[10:40:28][I][jk_bms:062]: Status frame received
[10:40:28][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_min_cell_voltage/state' payload='100.000' retain=1)
[10:40:28][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_max_cell_voltage/state' payload='-100.000' retain=1)
[10:40:28][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_delta_cell_voltage/state' payload='-200.000' retain=1)
[10:40:28][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_power_tube_temperature/state' payload='0' retain=1)
[10:40:28][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_temperature_sensor_1/state' payload='0' retain=1)
[10:40:28][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_temperature_sensor_2/state' payload='-3739' retain=1)
[10:40:28][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_total_voltage/state' payload='332.92' retain=1)
[10:40:28][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_current/state' payload='0.00' retain=1)
[10:40:28][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_capacity_remaining/state' payload='255' retain=1)
[10:40:28][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_temperature_sensors/state' payload='40' retain=1)
[10:40:28][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_charging_cycles/state' payload='65343' retain=1)
[10:40:28][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_total_charging_cycle_capacity/state' payload='1193427072' retain=1)
[10:40:28][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_battery_strings/state' payload='8256' retain=1)
[10:40:28][V][text_sensor:016]: 'jk-bms errors': Received new state Power tube overtemperature;Charging overvoltage;Discharging undervoltage;Battery over temperature;Charging overcurrent;Discharging overcurrent;Cell pressure difference;Cell undervoltage;309_A protection
[10:40:28][D][text_sensor:067]: 'jk-bms errors': Sending state 'Power tube overtemperature;Charging overvoltage;Discharging undervoltage;Battery over temperature;Charging overcurrent;Discharging overcurrent;Cell pressure difference;Cell undervoltage;309_A protection'
[10:40:28][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_errors/state' payload='Power tube overtemperature;Charging overvoltage;Discharging undervoltage;Battery over temperature;Charging overcurrent;Discharging overcurrent;Cell pressure difference;Cell undervoltage;309_A protection' retain=1)
[10:40:28][V][text_sensor:016]: 'jk-bms operation mode': Received new state Balancer enabled;Battery dropped
[10:40:28][D][text_sensor:067]: 'jk-bms operation mode': Sending state 'Balancer enabled;Battery dropped'
[10:40:28][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_operation_mode/state' payload='Balancer enabled;Battery dropped' retain=1)
[10:40:28][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_capacity_remaining_derived/state' payload='3043238912' retain=1)
[10:40:28][V][text_sensor:016]: 'jk-bms battery type': Received new state Unknown
[10:40:28][D][text_sensor:067]: 'jk-bms battery type': Sending state 'Unknown'
[10:40:28][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_battery_type/state' payload='Unknown' retain=1)
[10:40:28][V][text_sensor:016]: 'jk-bms password': Received new state \xff?\xb8G"@d\xc4 @
[10:40:28][D][text_sensor:067]: 'jk-bms password': Sending state '\xff?\xb8G"@d\xc4 @'
[10:40:28][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_password/state' payload='\xff?\xb8G"@d\xc4 @' retain=1)
[10:40:28][V][text_sensor:016]: 'jk-bms device type': Received new state  \x94\x8eD \xff
[10:40:28][D][text_sensor:067]: 'jk-bms device type': Sending state ' \x94\x8eD \xff'
[10:40:28][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_device_type/state' payload=' \x94\x8eD \xff' retain=1)
[10:40:28][V][text_sensor:016]: 'jk-bms total runtime formatted': Received new state -44y -245d -19h
[10:40:28][D][text_sensor:067]: 'jk-bms total runtime formatted': Sending state '-44y -245d -19h'
[10:40:28][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_total_runtime_formatted/state' payload='-44y -245d -19h' retain=1)
[10:40:28][V][text_sensor:016]: 'jk-bms software version': Received new state  @$
[10:40:28][D][text_sensor:067]: 'jk-bms software version': Sending state ' @$'
[10:40:28][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_software_version/state' payload=' @$' retain=1)
[... cut ...]
[10:40:28][D][text_sensor:067]: 'jk-bms manufacturer': Sending state '\xc4 @\xd8\xd8\xfe?\x9a\x94\x94$\xff?h\xa7\xfe?\xb8G"@d'
[10:40:28][V][mqtt:405]: Publish(topic='jk-bms/sensor/jk-bms_manufacturer/state' payload='\xc4 @\xd8\xd8\xfe?\x9a\x94\x94$\xff?h\xa7\xfe?\xb8G"@d' retain=1)
[10:40:28][V][component:199]: Component jk_modbus took a long time for an operation (0.11 s).
[10:40:28][V][component:200]: Components should block for at most 20-30ms.
[10:40:28][D][uart_debug:114]: <<< 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
`

Is this compatible with my hardware?

syssi, 2022-03-24:
Could you tell me the model number of your BMS and the software + hardware version? How did you connect the device? A photo would be nice.

Bad news: It looks like you are receiving your own transmission at the moment. It's just an echo.

```
TX >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
RX <<< 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
```

miguel300477, 2022-03-24:
Hi, i have conected it into TX and RX of the WEMOS d1 Mini. i  Use the JKBMS RS485 converter conected to the BMS and to the WEMOS. ( I changed GPIO1 to GPIO3 and i get no transmition like that, so i im using the default config for the ESP8266.

Note im powering the WEMOS by USB, not by the 3.3volt+ground pins. 

When i get home i will acess the BMS to give you the hardware and software version. The model is JK-B2A24S15P

syssi, 2022-03-24:
I photo of your setup especially of the RS485 converter would be nice. This is the background: My BMS is attached directly to the ESP because the socket labeled as "RS485" doesn't provide RS485 signals (it's UART TTL).

To make my BMS RS485 compatible I've to use the JKBMS RS485 converter to convert the UART TTL signal to RS485. If I use this converter I'm not allowed to connect the ESP directly to the A/B pins of the converter. An additional converter is required to convert the RS485 signal back to TTL.

miguel300477, 2022-03-24:
Humm, so i have to connect the ESP direct to the BMS RS485 port?

syssi, 2022-03-24:
Let's check your hardware version first and your JKBMS RS485 converter. If you've connected A/B (-7 to +12V signal level) with RX/TX (0V...3.3V signal level) this cannot work.

miguel300477, 2022-03-24:
[image]
[image]

miguel300477, 2022-03-24:
[image]

syssi, 2022-03-24:
At the top of the RS485 converter is the BMS connector. Right? If this is the case please connect the ESP directly to the BMS and give it another try. Your BMS firmware version should be supported. May be we have to play around with the baud rate a bit.

miguel300477, 2022-03-24:
Its working fine now.

Add some error because i had enabled all sensors. Removed a few and it reading fine the values.

syssi, 2022-03-24:
Good job! I will add you hardware & software version as supported to the docs! Feel free to close the issue.

closed 2022-03-24 by miguel300477