syssi/esphome-jk-bms #63: ESP32-DevKitC Boot Loop

spoonwzd, 2022-04-08:
Similar to [this issue](https://github.com/syssi/esphome-jk-bms/issues/19) my ESP32 seems to be stuck in a boot loop after flashing, but it's not the same error.

I've not connected to the BMS and have only changed the board to nodemcu-32s in esp32-example.yaml.

```
[18:05:22]ets Jun  8 2016 00:22:57
[18:05:22]
[18:05:22]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[18:05:22]configsip: 0, SPIWP:0xee
[18:05:22]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[18:05:22]mode:DIO, clock div:2
[18:05:22]load:0x3fff0018,len:4
[18:05:22]load:0x3fff001c,len:1044
[18:05:22]load:0x40078000,len:10124
[18:05:22]load:0x40080400,len:5828
[18:05:22]entry 0x400806a8
[18:05:23]***ERROR*** A stack overflow in task loopTask has been detected.
[18:05:23]abort() was called at PC 0x40088b98 on core 1
[18:05:23]
[18:05:23]ELF file SHA256: 0000000000000000
[18:05:23]
[18:05:23]Backtrace: 0x40088904:0x3ffafb60 0x40088b81:0x3ffafb80 0x40088b98:0x3ffafba0 0x4008a77b:0x3ffafbc0 0x4008c344:0x3ffafbe0 0x4008c2fa:0x3ffbbcd8
←[33mWARNING Found stack trace! Trying to decode it←[0m
←[33mWARNING Decoded 0x40088904: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715←[0m
←[33mWARNING Decoded 0x40088b81: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715←[0m
←[33mWARNING Decoded 0x40088b98: vApplicationStackOverflowHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715←[0m
←[33mWARNING Decoded 0x4008a77b: vTaskSwitchContext at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507←[0m
←[33mWARNING Decoded 0x4008c344: _frxt_dispatch at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portasm.S:406←[0m
←[33mWARNING Decoded 0x4008c2fa: _frxt_int_exit at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portasm.S:206←[0m
[18:05:24]
[18:05:24]Rebooting...
```

jonadis, 2022-04-08:
Try commenting out some of the sensors.  I think it overloads the stack with all of the sensors in the example.

syssi, 2022-04-08:
Another solution is to increase the stack size:

https://github.com/esphome/issues/issues/855#issuecomment-1069335461

```
esphome:
  name: jk-bms
  platformio_options:
    build_flags: 
      - -DCONFIG_ARDUINO_LOOP_STACK_SIZE=32768

esp32:
  board: nodemcu-32s
  variant: esp32
  framework:
    type: arduino
    version: 2.0.2
    source: https://github.com/espressif/arduino-esp32.git#2.0.2
    platform_version: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
```

spoonwzd, 2022-04-08:
Increasing the stack size as above sorted the issue - thanks!!

closed 2022-04-08 by syssi