syssi/esphome-jk-bms #64: Cannot re-run script

spoonwzd, 2022-04-08:
Runs fine the first time, but any subsequent runs report:

```
←[32mINFO Reading configuration esp32-example.yaml...←[0m
←[32mINFO Updating https://github.com/syssi/esphome-jk-bms.git@main←[0m
Failed config

external_components: [source esp32-example.yaml:13]
  - [source esp32-example.yaml:13]

    usage: git stash list [<options>]
       or: git stash show [<stash>]
       or: git stash drop [-q|--quiet] [<stash>]
       or: git stash ( pop | apply ) [--index] [-q|--quiet] [<stash>]
       or: git stash branch <branchname> [<stash>]
       or: git stash [save [--patch] [-k|--[no-]keep-index] [-q|--quiet]
                       [-u|--include-untracked] [-a|--all] [<message>]]
       or: git stash clear.
    source: github://syssi/esphome-jk-bms@main [source esp32-example.yaml:13]
    refresh: 0s
```

syssi, 2022-04-08:
Do you try to compile the untouched `esp32-example.yaml`? I would like try to reproduce the issue. Which ESPHome version do you use? Which operation system & release do you use? It looks like a general ESPHome issue independent from this custom component.

spoonwzd, 2022-04-08:
Using esphome 2022.3.2

Running on Windows 10

I get the same "failed config" error on an untouched esx32-example.yaml

syssi, 2022-04-08:
I cannot reproduce the issue locally (on linux):

```
$ cp ~/esphome/secrets.yaml .
$ wget https://raw.githubusercontent.com/syssi/esphome-jk-bms/main/esp32-example.yaml 
--2022-04-08 20:17:01--  https://raw.githubusercontent.com/syssi/esphome-jk-bms/main/esp32-example.yaml
Auflösen des Hostnamens raw.githubusercontent.com (raw.githubusercontent.com)… 185.199.111.133, 185.199.110.133, 185.199.108.133, ...
Verbindungsaufbau zu raw.githubusercontent.com (raw.githubusercontent.com)|185.199.111.133|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 7362 (7,2K) [text/plain]
Wird in »esp32-example.yaml« gespeichert.

esp32-example.yaml                        100%[====================================================================================>]   7,19K  --.-KB/s    in 0,001s  

2022-04-08 20:17:01 (14,0 MB/s) - »esp32-example.yaml« gespeichert [7362/7362]

$ esphome compile esp32-example.yaml 
INFO Reading configuration esp32-example.yaml...
INFO Cloning https://github.com/syssi/esphome-jk-bms.git@main
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing jk-bms (board: esp-wrover-kit; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2

Unpacking  [------------------------------------]    0%
Unpacking  [##----------------------------------]    6%
Unpacking  [####--------------------------------]   12%
Unpacking  [######------------------------------]   18%
[...]
Archiving .pioenvs/jk-bms/libFrameworkArduino.a
Indexing .pioenvs/jk-bms/libFrameworkArduino.a
Linking .pioenvs/jk-bms/firmware.elf
RAM:   [=         ]  12.9% (used 42112 bytes from 327680 bytes)
Flash: [=====     ]  51.7% (used 947850 bytes from 1835008 bytes)
Building .pioenvs/jk-bms/firmware.bin
esp32_create_combined_bin([".pioenvs/jk-bms/firmware.bin"], [".pioenvs/jk-bms/firmware.elf"])
Wrote 0xf76f0 bytes to file /tmp/test/.esphome/build/jk-bms/.pioenvs/jk-bms/firmware-factory.bin, ready to flash to offset 0x0
==================================================================== [SUCCESS] Took 30.75 seconds ====================================================================
INFO Successfully compiled program.
$
$
$ esphome compile esp32-example.yaml 
INFO Reading configuration esp32-example.yaml...
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
INFO Generating C++ source...
INFO Compiling app...
Processing jk-bms (board: esp-wrover-kit; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|-- <Update> 1.0
|-- <AsyncMqttClient-esphome> 0.8.6
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <ArduinoJson> 6.18.5
RAM:   [=         ]  12.9% (used 42112 bytes from 327680 bytes)
Flash: [=====     ]  51.7% (used 947850 bytes from 1835008 bytes)
===================================================================== [SUCCESS] Took 1.94 seconds =====================================================================
INFO Successfully compiled program.
$
$
$ esphome compile esp32-example.yaml 
INFO Reading configuration esp32-example.yaml...
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
INFO Generating C++ source...
INFO Compiling app...
Processing jk-bms (board: esp-wrover-kit; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|-- <Update> 1.0
|-- <AsyncMqttClient-esphome> 0.8.6
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <ArduinoJson> 6.18.5
RAM:   [=         ]  12.9% (used 42112 bytes from 327680 bytes)
Flash: [=====     ]  51.7% (used 947850 bytes from 1835008 bytes)
===================================================================== [SUCCESS] Took 1.96 seconds =====================================================================
INFO Successfully compiled program.
$
```

syssi, 2022-04-08:
It looks like this issue happens to other people using ESPHome + custom components every now and then: https://github.com/esphome/feature-requests/issues/611#issuecomment-991800239

Could you try this workaround:

```
esphome clean esp32-example.yaml 
esphome run esp32-example.yaml 
```

spoonwzd, 2022-04-08:
Alas 'clean' just returns the same error

syssi, 2022-04-08:
What a shame. Could you create an official esphome issue?

closed 2022-04-09 by syssi

spoonwzd, 2022-04-09:
Opened https://github.com/esphome/issues/issues/3208