syssi/esphome-jk-bms #67: Expected behaviour of successful flash

spoonwzd, 2022-04-09:
Can you verify the expected behaviour of a successful flash?

Looking at the log output after reboot it seems to be establishing a configuration then saving it to flash.

[image]

Is it normal to take ~3 minutes after establish the sensor config to save preferences to flash?

Once the saving preferences to flash message appears, is it safe to power off the device at this point?

Does the ESP32 device need to be connected to the BMS when it's adding sensors on first boot?

Apologies for the noob questions, I'm still waiting for it to click for me :)

syssi, 2022-04-09:
> Is it normal to take ~3 minutes after establishing the sensor config to save preferences to flash?

This is normal. If the startup doesn't succeed multiple times (= the boot loop counter exceeds `10`: https://esphome.io/components/ota.html) the ESPHome node goes into a recovery mode. So you are able to reflash the device via OTA.

> Once the saving preferences to flash message appears, is it safe to power off the device at this point?

It's always safe to reboot the ESP. If you reset/reboot the ESP manually 10 times in a row the recovery mode kicks in (see above).

> Does the ESP32 device need to be connected to the BMS when it's adding sensors on first boot?

No it doesn't need to be connected. The custom component polls the BMS (with a single request) periodically. The BMS responds with a full status report. You can decide by adding sensors how much data of the status report is published as sensor data.

syssi, 2022-04-09:
Could you provide the BMS type + hardware & software version you are using? I would like to extend the list of supported devices if your device version is missing.

spoonwzd, 2022-04-09:
Model: JK-BD6A24S10P
Hardware: 8.X
Software: 8.0.6G
Version: 3.7.1

This doesn't seem to be on your supported list and may explain why I'm not seeing any data. :/

syssi, 2022-04-09:
It should be supported because the device pretty new. Could you provide a photo of your wiring?

spoonwzd, 2022-04-09:
[image]

syssi, 2022-04-09:
Are you sure the green cable is connected to GND? It looks like `GPIO20`.

spoonwzd, 2022-04-09:
[image]

spoonwzd, 2022-04-09:
[image]

syssi, 2022-04-09:
Alright! Did you try to swap RX and TX just for testing? Your setup looks fine. The BMS is up and running (blinks)?

spoonwzd, 2022-04-09:
Not tried swapping, but I can if you think it's worth a try.

BMS defo up and running.

syssi, 2022-04-09:
Where did you buy the BMS?

syssi, 2022-04-09:
It's worth a try!

spoonwzd, 2022-04-09:
> Where did you buy the BMS?

https://www.aliexpress.com/store/911303103

syssi, 2022-04-09:
No luck?

spoonwzd, 2022-04-09:
Right, so swapping the TX/RX GPIO around didn't help.

Swapped them back to their original states and ran the debug.

[image]

I'm clearly getting data from the BMS but I'm not seeing anything at the MQTT listener.

I'm not running HomeAssistant - I'm using the MQTT listener in node-red which I've used before successfully.

Should I be seeing MQTT related messages in the debug output? Because I'm not!

syssi, 2022-04-09:
Perfect! You are able to retrieve the data from the BMS. The next step is to debug the MQTT connection. I don't know node-red in details. Is there a mosquitto broker or something like that? Could you take a look at the mosquitto.log? Is the ESPHome able to connect? Gets the connection dropped/disconnected because of failed authorization or something?

spoonwzd, 2022-04-09:
Using MQTT explorer I can connect to my MQTT broker and see all the data flowing in, so this must be something throwing me off in node-red.

[image]

I'll suss it, but pretty sure your bit is all working great now.

Thanks for all your help.

closed 2022-04-09 by syssi

spoonwzd, 2022-04-09:
Ok I was subscribing to:

`batt3/sensor/batt3_cell_voltage_1`

instead of

`batt3/sensor/batt3_cell_voltage_1/state`

#facepalm