syssi/esphome-jk-bms #80: Setting up the Battery section of Home Assistant Energy

JohnGalt1717, 2022-04-20:
I can't figure out (because no entries show up) how to set this up under the battery section of Home Assistant. Is there something I'm missing to make this work and get a kWh entry?

syssi, 2022-04-20:
The `total_daily_energy` component is your friend. You've to commulate the instantaneous power (`W`) to `Wh`. I would add the component twice. Once for the `charging_power` and once for the `discharging_power`.

JohnGalt1717, 2022-04-20:
Can you give me a hint?  This gives me an error and the docs are really really really not clear.

```
  - platform: total_daily_energy
    name: "BMS 1 Total Charging Energy Today"
    power_id: bms_1_charging_power
```

power_id is the id of the charging power sensor in the jk_bms_ble platform.  But it appears that you have to do something else that I can't determine from the docs?

syssi, 2022-04-20:
I haven't used the component with the BMS yet but I can provide a similar example of another device (PV inverter):

```
sensor:
  - platform: soyosource_inverter
    soyosource_inverter_id: inverter0
    battery_power:
      name: "${inverter0} battery power"
      id: inverter0_power

  - platform: total_daily_energy
    name: "${inverter0} energy today"
    icon: mdi:counter
    power_id: inverter0_power
    accuracy_decimals: 3
    unit_of_measurement: kWh
    min_save_interval: 2h
    restore: true
    filters:
      - multiply: 0.001

time:
  - platform: sntp
    id: my_time
```

syssi, 2022-04-20:
```
sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    charging_power:
      name: "${name} charging power"
      id: bms0_charging_power
    discharging_power:
      name: "${name} discharging power"
      id: bms0_discharging_power
      
  - platform: total_daily_energy
    name: "bms0 charging energy today"
    icon: mdi:counter
    power_id: bms0_charging_power
    accuracy_decimals: 3
    unit_of_measurement: kWh
    min_save_interval: 2h
    restore: true
    filters:
      - multiply: 0.001
      
  - platform: total_daily_energy
    name: "bms0 discharging energy today"
    icon: mdi:counter
    power_id: bms0_discharging_power
    accuracy_decimals: 3
    unit_of_measurement: kWh
    min_save_interval: 2h
    restore: true
    filters:
      - multiply: 0.001
      
time:
  - platform: sntp
    id: my_time
```

JohnGalt1717, 2022-04-20:
Ok, got it working from there. Tip for anyone else: You have to clean your sources before building or it will error.

closed 2022-04-20 by JohnGalt1717