syssi/esphome-jk-bms #105: Verify JK-B2A8S20P support

syssi, 2022-05-09:
https://github.com/jblance/mpp-solar/issues/224

```
Cell info response (290 bytes)

0xaa, 0xeb, 0x90, 0x02, 0x97, 0xeb, 0x0c, 0xe9, 0x0c, 0xe9, 0x0c, 0xe9, 0x0c, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x0f, 0x00, 0x00, 0x00, 0xeb, 0x0c, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xaa, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xc2, 0x00, 0xba, 0x00, 0xce, 0x00, 0x04, 0x00, 0x00, 0x00, 0x00, 0x00, 0x03, 0x02, 0x00, 0xc0, 0x04, 0x00, 0x00, 0x00, 0x00, 0x00, 0xeb, 0xe0, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x15, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x07, 0x00, 0x00, 0x00, 0x00, 0x00, 0x05, 0x00, 0x00, 0x00, 0x00, 0xbb, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x05, 0x00, 0x00, 0xd7, 0x99, 0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x1e
```

syssi, 2022-05-09:
```
      const uint8_t frame[300] = {
          0x55, 0xaa, 0xeb, 0x90, 0x02, 0x95, 0xed, 0x0c, 0xea, 0x0c, 0xe9, 0x0c, 0xe8, 0x0c, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
          0x0f, 0x00, 0x00, 0x00, 0xe9, 0x0c, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xa5, 0x33, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0xcc, 0x00, 0xc8, 0x00, 0xda, 0x00, 0x04, 0x00, 0x00, 0x00, 0x00, 0x00, 0x03, 0x6c,
          0x02, 0x00, 0xc0, 0x45, 0x04, 0x00, 0x00, 0x00, 0x00, 0x00, 0xeb, 0xe0, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00,
          0xc3, 0x2f, 0x23, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x07, 0x00, 0x00, 0x00, 0x00, 0x00, 0x23, 0x05, 0x00, 0x00, 0x00, 0x00, 0xbb, 0x64, 0x3b, 0x40,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x05, 0x00, 0x00, 0xa3, 0xc4,
          0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xc9};
      this->assemble_(frame, 300);

      const uint8_t frame2[320] = {
          0x55, 0xaa, 0xeb, 0x90, 0x01, 0x9d, 0x58, 0x02, 0x00, 0x00, 0x28, 0x0a, 0x00, 0x00, 0xb8, 0x0b, 0x00, 0x00,
          0x10, 0x0e, 0x00, 0x00, 0x48, 0x0d, 0x00, 0x00, 0x05, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xc4, 0x09, 0x00, 0x00, 0xa8, 0x61, 0x00, 0x00,
          0x1e, 0x00, 0x00, 0x00, 0x3c, 0x00, 0x00, 0x00, 0x40, 0x0d, 0x03, 0x00, 0x2c, 0x01, 0x00, 0x00, 0x3c, 0x00,
          0x00, 0x00, 0x3c, 0x00, 0x00, 0x00, 0xd0, 0x07, 0x00, 0x00, 0xbc, 0x02, 0x00, 0x00, 0x58, 0x02, 0x00, 0x00,
          0xbc, 0x02, 0x00, 0x00, 0x58, 0x02, 0x00, 0x00, 0x38, 0xff, 0xff, 0xff, 0x9c, 0xff, 0xff, 0xff, 0x84, 0x03,
          0x00, 0x00, 0xbc, 0x02, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0xc0, 0x45, 0x04, 0x00, 0xdc, 0x05, 0x00, 0x00, 0xb8, 0x0b, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xd5, 0xaa, 0x55, 0x90, 0xeb, 0xc8, 0x01,
          0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x44};
      this->assemble_(frame2, 320);

      const uint8_t frame3[320] = {
          0x55, 0xaa, 0xeb, 0x90, 0x03, 0xb8, 0x4a, 0x4b, 0x2d, 0x42, 0x32, 0x41, 0x38, 0x53, 0x32, 0x30, 0x50, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x39, 0x2e, 0x58, 0x00, 0x00, 0x00, 0x00, 0x00, 0x39, 0x2e, 0x30, 0x31, 0x4d, 0x33,
          0x00, 0x00, 0x08, 0x2f, 0x23, 0x00, 0x1d, 0x00, 0x00, 0x00, 0x4a, 0x4b, 0x2d, 0x42, 0x32, 0x41, 0x38, 0x53,
          0x32, 0x30, 0x50, 0x00, 0x00, 0x00, 0x00, 0x00, 0x31, 0x32, 0x33, 0x34, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x32, 0x31, 0x31, 0x31, 0x30, 0x39, 0x00, 0x00, 0x31, 0x30, 0x39, 0x31,
          0x35, 0x31, 0x35, 0x31, 0x35, 0x33, 0x00, 0x30, 0x30, 0x30, 0x30, 0x00, 0x30, 0x31, 0x32, 0x33, 0x34, 0x35,
          0x36, 0x37, 0x38, 0x39, 0x30, 0x31, 0x32, 0x61, 0x00, 0x00, 0x31, 0x32, 0x33, 0x34, 0x35, 0x36, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
          0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
[... cut ...]
[13:47:05][I][jk_bms_ble:1043]:   Uptime: 2305800 s
[13:47:05][I][jk_bms_ble:1044]:   Power on count: 29
[13:47:05][I][jk_bms_ble:1045]:   Device name: JK-B2A8S20P
[13:47:05][I][jk_bms_ble:1046]:   Device passcode: 1234
[13:47:05][I][jk_bms_ble:1047]:   Manufacturing date: 211109
[13:47:05][I][jk_bms_ble:1048]:   Serial number: 1091515153
[13:47:05][I][jk_bms_ble:1049]:   Passcode: 0000
[13:47:05][I][jk_bms_ble:1050]:   User data: 0123456789012a
[13:47:05][I][jk_bms_ble:1051]:   Setup passcode: 123456
```

syssi, 2022-05-09:
@EasilyBoredEngineer FYI. Most values are fine. Some are a bit off.

Invalid or strange values:

1. `errors bitmask`: 0b111011100110 (Charge Undertemperature;Error 0x01 0x00;Cell count is not equal to settings;Cell Overvoltage;Error 0x80 0x00')
2. State of charge 0%
3. Charge UTP: 429496704.000000 °C (signed int?)
4. Charge UTP recovery: 429496736.000000 °C (signed int?)
5. Cell count: 8.000000 (setting incorrect?)

EasilyBoredEngineer, 2022-05-14:
Hi @syssi, as requested please find attached app screenshots and updated log file. Note:-

1. I've updated to 4S in the app.
2. SOC 0% makes no sense, actually 63%.
3. Resistance of 0 makes no sense - see app.
4. UVP value makes no sense - actually -20C.

[image]
[image]
[image]
[image]
[image]
[logs_testesp_logs (2).txt](https://github.com/syssi/esphome-jk-bms/files/8692460/logs_testesp_logs.2.txt)

EasilyBoredEngineer, 2022-05-14:
for the sake of having data in one spot, here's a repload of my hex file output from yesterday - note this will still have the 4s / 8s bitmap error.

This is the response to the command:-

````AA5590EB96000000000000000000000000000010````
[getCellDataChallengeAndResponse.txt](https://github.com/syssi/esphome-jk-bms/files/8692475/getCellDataChallengeAndResponse.txt)

syssi, 2022-05-14:
```
Previous (57090d = 1101111100000010b):

- Charge Undertemperature
- Error 0x01 0x00
- Error 0x02 0x00
- Cell count is not equal to settings
- Error 0x08 0x00
- Cell Overvoltage
- Error 0x40 0x00
- Error 0x80 0x00

After (43605d = 1010101001010101b): 

- Charge Overtemperature
- Error 0x00 0x04
- Error 0x00 0x10
- Error 0x00 0x40
- Error 0x02 0x00
- Error 0x08 0x00
- Error 0x20 0x00
- Error 0x80 0x00
```

syssi, 2022-05-14:
If you increase the log level by:

```
logger:
  level: VERY_VERBOSE
  logs:
    scheduler: DEBUG
    component: DEBUG
    esp32_ble_tracker: DEBUG
    sensor: DEBUG
```

you can see the raw traffic if you like.

EasilyBoredEngineer, 2022-05-14:
Thanks @syssi  -

(deleted hex log above as it was mangled - wasn't obtained properly so was showing the hex in the ascii section)

Very verbose logs attached, captured right now after the 4s change.
[logs_testesp_logs (3).txt](https://github.com/syssi/esphome-jk-bms/files/8692501/logs_testesp_logs.3.txt)

EasilyBoredEngineer, 2022-05-14:
reupload of hex dump response, fyi. 

This is how the BMS responds to the following command:-

````AA5590EB96000000000000000000000000000010````

[commandResponse.txt](https://github.com/syssi/esphome-jk-bms/files/8692509/commandResponse.txt)

syssi, 2022-05-14:
I've fixed the decoding of the "under temperature protection" and "under temperature protection recovery". It's a signed integer now.

EasilyBoredEngineer, 2022-05-14:
@syssi 

> I've fixed the decoding of the "under temperature protection" and "under temperature protection recovery". It's a signed integer now.

Thankyou! Now I just have to remember how to synchronise git..... :)

syssi, 2022-05-14:
Let's remove this parts from your config yaml:

```
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
[...]
    cell_voltage_24:
      name: "${name} cell voltage 24"
```

and

```
    cell_resistance_5:
      name: "${name} cell resistance 5"
    cell_resistance_6:
      name: "${name} cell resistance 6"
    cell_resistance_7:
      name: "${name} cell resistance 7"
[...]
    cell_resistance_24:
      name: "${name} cell resistance 24"
```

syssi, 2022-05-14:
@EasilyBoredEngineer If you use this project as `external_component` you don't need a clone of this repository. Just use/keep/edit the `esp32-ble-example.yaml`. This section:

```
substitutions:
  name: jk-bms
  external_components_source: github://syssi/esphome-jk-bms@main

external_components:
  - source: ${external_components_source}
    refresh: 0s
```

clones the main branch of this repository every time (`refresh: 0s`). You have just to recompile & flash again.

EasilyBoredEngineer, 2022-05-14:
great! Thankyou. Yaml changes made.

Attached is new log and a screenshot.

Note the circled items do nothing when clicked.

[image]
[logs_testesp_logs (4).txt](https://github.com/syssi/esphome-jk-bms/files/8692527/logs_testesp_logs.4.txt)

EasilyBoredEngineer, 2022-05-14:
additional screenshots attached - looking good!
[image]
[image]

syssi, 2022-05-14:
Are you sure the buttons doesn't work? Because it's not documented it try to explain what's happing:

If you press the "retrieve settings button" the `0x96` request is sent to the BMS. The BMS should beep (acknowledge the command) and respond with a settings frame (`frame_type: 0x01`). The settings are written with log_level `INFO` to the log and should look something like this:

```
[13:47:04][I][jk_bms_ble:748]: Settings frame (320 bytes):
[13:47:04][I][jk_bms_ble:774]:   Unknown6: 0.600000
[13:47:04][I][jk_bms_ble:776]:   Cell UVP: 2.600000 V
[13:47:04][I][jk_bms_ble:778]:   Cell UVPR: 3.000000 V
[13:47:04][I][jk_bms_ble:780]:   Cell OVP: 3.600000 V
[13:47:04][I][jk_bms_ble:782]:   Cell OVPR: 3.400000 V
[13:47:04][I][jk_bms_ble:784]:   Balance trigger voltage: 0.005000 V
[13:47:04][I][jk_bms_ble:790]:   Power off voltage: 2.500000 V
[13:47:04][I][jk_bms_ble:792]:   Max. charge current: 25.000002 A
[13:47:04][I][jk_bms_ble:794]:   Charge OCP delay: 30.000000 s
[13:47:04][I][jk_bms_ble:796]:   Charge OCP recovery delay: 60.000000 s
[13:47:04][I][jk_bms_ble:798]:   Max. discharge current: 200.000015 A
[13:47:04][I][jk_bms_ble:800]:   Discharge OCP recovery delay: 300.000000 s
```

If you pree the "retrieve device info" button the BMS should beep and the component should write this to the log:

```
[13:47:05][I][jk_bms_ble:988]: Device info frame (320 bytes):
[13:47:05][I][jk_bms_ble:1040]:   Vendor ID: JK-B2A8S20P
[13:47:05][I][jk_bms_ble:1041]:   Hardware version: 9.X
[13:47:05][I][jk_bms_ble:1042]:   Software version: 9.01M3
[13:47:05][I][jk_bms_ble:1043]:   Uptime: 2305800 s
[13:47:05][I][jk_bms_ble:1044]:   Power on count: 29
[13:47:05][I][jk_bms_ble:1045]:   Device name: JK-B2A8S20P
[13:47:05][I][jk_bms_ble:1046]:   Device passcode: 1234
[13:47:05][I][jk_bms_ble:1047]:   Manufacturing date: 211109
[13:47:05][I][jk_bms_ble:1048]:   Serial number: 1091515153
[13:47:05][I][jk_bms_ble:1049]:   Passcode: 0000
[13:47:05][I][jk_bms_ble:1050]:   User data: 0123456789012a
[13:47:05][I][jk_bms_ble:1051]:   Setup passcode: 123456
```

Could you test again? Does it work or doesn't beep the BMS?

EasilyBoredEngineer, 2022-05-14:
@syssi - question for you - if I wished to adapt your code to send the bms data out as a struct via serial IO, as well as sending to home assistant, which part of the code would be best to work on / fork?

EasilyBoredEngineer, 2022-05-14:
> If you press the "retrieve settings button" the `0x96` request is sent to the BMS. The BMS should beep (acknowledge the command) and respond with a settings frame (`frame_type: 0x01`). The settings are written with log_level `INFO` to the log and should look something like this:

No such section noted in log at very verbose level when / after pressing button. Accessing remotely at the moment so cannot confirm whether the BMS beeps or not, apologies, will revisit when next to the ESP.

syssi, 2022-05-14:
This is the most important file about the JK02 (and JK02 protocol): https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/jk_bms_ble.cpp

Take a look at the `decode_` method (to understand the different known frame types). Check out the `decode_jk02_cell_info_` to monitor the most important values periodically.

If the Android app connects to the BMS the device info frame is requested first: https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/jk_bms_ble.cpp#L159

The second step is to request the settings + enable the periodic sendout of the cell info frame: https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/jk_bms_ble.cpp#L275

Please make sure between the first request (0x97) and the second (0x96) must be a gap of ~0.5sec.

EasilyBoredEngineer, 2022-05-14:
Thankyou very much @syssi - really appreciated :)

FYI - just changed log level to INFO and CONFIRM - both buttons work.

attached log shows the relevant responses.
[logs_testesp_upload.txt](https://github.com/syssi/esphome-jk-bms/files/8692574/logs_testesp_upload.txt)

syssi, 2022-05-14:
Lucky guy! I was already afraid that you might have damaged the RX pin of your BMS or Jikong has introduced some encryption / verification bits at the random part of the request to avoid unsecured/unauthorized traffic.

EasilyBoredEngineer, 2022-05-14:
> Lucky guy!

Ha! So was I!!

Nope, looks like all is working well now with your package.. 

Now that's not to say that I haven't possibly killed something else on the bms during my 'investigations' (I did make a big spark with my multimeter probe down towards the mosfet end at one stage - lol) - but I'll know that soon enough :)

Thanks again for all your help - close as successful if you wish?

syssi, 2022-05-14:
One piece of the puzzle is missing unfortunately: The meaning of each bit of the error bitmask of your device is unknown. Your android app doesn't show a red error message anymore but some bits of the error bitmask are still `1`. On other BMS models all bits are `0` if everything is fine / no warning shown.

EasilyBoredEngineer, 2022-05-15:
Ok - understood. I've just gone down to the office and tried to find a way to factory reset the bms, but alas on my app there is no such option.. another potential cause may be that I have strange private user data (0123456789012), and can't change it using the android app - wondering if that might be bleeding over into the next byte... 

I'll try to do some investigations over the coming days:-

EBE 'to do':-

1. Disconnect my friend Dave's 'lytefyba' [cell management  units](https://github.com/dkeenan7/LyteFyba) from my 8s bank and fully charge using the jk in 8s configuration instead of the present workaround 4s.
2. Try to set 'strange' user Private Data back to 0000 using @bullar [suggested command structure](https://github.com/jblance/mpp-solar/issues/170#issuecomment-1050503970)
3. Fork syssi repository and implement rcvd structure via ble>>uart or i2c to allow processing by my existing bms.
4. reconnect via piggy back method and try and get further information for syssi re _RST pin operation and its relevance to #107

syssi, 2022-05-15:
I've seen another user/log yesterday with the same strange error bitmask. I assume everything is fine with your BMS and something has changed here.

```
[21:51:34][V][text_sensor:016]: 'e-kart errors': Received new state Charge Overtemperature;Error 0x00 0x04;Error 0x00 0x10;Error 0x00 0x40;Error 0x02 0x00;Error 0x08 0x00;Error 0x20 0x00;Error 0x80 0x00
[21:51:34][D][text_sensor:067]: 'e-kart errors': Sending state 'Charge Overtemperature;Error 0x00 0x04;Error 0x00 0x10;Error 0x00 0x40;Error 0x02 0x00;Error 0x08 0x00;Error 0x20 0x00;Error 0x80 0x00'
[21:51:34][VV][api.service:174]: send_text_sensor_state_response: TextSensorStateResponse {
  key: 1889080433
  state: 'Charge Overtemperature;Error 0x00 0x04;Error 0x00 0x10;Error 0x00 0x40;Error 0x02 0x00;Error 0x08 0x00;Error 0x20 0x00;Error 0x80 0x00'
  missing_state: NO
}
```

How to identify/reverse engineer the new error bitmask:

1. Declare the pattern above as the "No errors" state
2. Try to trigger every error manually. F.e. cool down one temperature sensor to the UTP. You can increase the setting beforehand so you can reach the limit easily.
3. Next bit is the over temperature protect. Change the setting slightly above room temperature and warm up the temperature sensor with your body temperature. Another bit at the error bitmask should flip from 0 to 1 this time. Note down the position.
4. Change the cell overvoltage setting to something below the current cell voltage of your pack. Another bit should flip. Note down the position.
5. Change the undervoltage setting
6. Change the battery strings settings to a wrong setting (5S). You should see the error message about a wrong battery setting at your app. A bit of the error bitmask should have flipped again.

syssi, 2022-05-15:
New finding: The error bitmask of the JK02 protocol has always been broken / isn't implemented properly. I can reproduce the issue with my BMS model too.

syssi, 2022-05-15:
I've fixed the error bitmask. Just some error bits are well known but may be we can find more in future:

https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/jk_bms_ble.cpp#L455-L479

closed 2022-05-15 by syssi