syssi/esphome-jk-bms #108: Add UART support of the JK02/JK04 protocol

syssi, 2022-05-11:
It's possible to talk the processor of the BMS / control the BMS using the RX/TX lines attached to the bluetooth module. The frames (requests & responses) are equal to the BLE frames. Just the transport layer is different.

TODOs:

- Provide transport layer independent methods to encode/decode frames
- Add UART communication
- Let the user / config decide about the transport layer: UART vs. Bluetooth
- Add instructions to disable the bluetooth module by pulling down the RST pin 
- Add instructions to control/wire P11 and P12 of the bluetooth module properly. It's important to replicate the bluetooth module behavior because the BMS doesn't respond otherwise

Open question: Are optocouplers required / recommended?

syssi, 2022-05-11:
This is a follow-up of https://github.com/syssi/esphome-jk-bms/issues/49.

JohnGalt1717, 2023-03-24:
Any follow up on this? I'd love to be able to set these.

syssi, 2023-03-24:
No news yet but it's on the roadmap.

charbeltabet, 2025-12-14:
> No news yet but it's on the roadmap.

I can work on it because I need it, does anyone have the PDF specification?

syssi, 2025-12-14:
There is no specification but you can use the decoder/code from here: https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/jk_bms_ble.cpp

Just replace the BLE communication with UART.