syssi/esphome-jk-bms #109: Add NEEY/Heltec 4A balancer support

syssi, 2022-05-11:
See  https://github.com/syssi/esphome-jk-bms/discussions/106.

The android apps aren't obfuscated. It should be possible to extract details about the protocol / frame structure and content.

syssi, 2022-05-11:
Do you own a NEEY or Heltec 4A balancer and a Android device with the app installed? Please provide a btsnoop capture which can be easily created if you follow these instructions:

https://medium.com/@charlie.d.anderson/how-to-get-the-bluetooth-host-controller-interface-logs-from-a-modern-android-phone-d23bde00b9fa

The capture is helpful to replicate the behavior / BLE frames of the Android app to talk to the BMS using this component.

@cristi2005

syssi, 2022-05-11:
```
# Frame:

2 Bytes: Start of frame (0x55 0xAA) 
1 Bytes: Address (0x11)
1 Bytes: Read or write
2 Bytes: Command
2 Bytes: Data length
n Bytes: Data
1 Bytes: End of frame (0xFF)

# Commands / Data structure

0x01: Request system data
  - 16 Bytes: Module No
  -  8 Bytes: Hardware version
  -  8 Bytes: Software version
  -  8 Bytes: Protocol version
  -  8 Bytes: Cycle
  -  Int: Reboots counter
  -  Int: Total uptime

0x02: Request battery data
  - 1 Byte: Serial number
  - Cell data
    - Float: Cell voltage 1
    - Float: Cell voltage 2
    - Float: Cell voltage 3
    ...
    - Float: Cell voltage 24
  - Resistance
    - Float: Cell resistance 1
    - Float: Cell resistance 2
    - Float: Cell resistance 3
    ....
    - Float: Cell resistance 24
  - Float: Total voltage
  - Float: SingleVoltage?!
  - Float: MaxMPa
  - 1 Byte: Max Voltage Cell Number
  - 1 Byte: Min Voltage Cell Number
  - 1 Byte: Single Num
  - 1 Byte: Balancing
  - Float: Balancing current
  - 1 Float: Tempature 1
  - 1 Float: Tempature 2
  - 1 Float: Tempature 3
  - 1 Byte: Error bitmask?
  - 1 Byte: NTC warnings?
  - Int: TempStamp?
  
  
0x03: Factory reset
  - 1 Float: Standard voltage 1
  - 1 Float: Standard voltage 2
  - 1 Float: Battery voltage 1
  - 1 Float: Battery voltage 2
  - 1 Float: Standard current 1
  - 1 Float: Standard current 2
  - 1 Float: Super Bat 1
  - 1 Float: Super Bat 2
  - 1 Float: Resistor 1
  - 1 Float: Bat Status
  - 1 Float: Max Voltage
  - 1 Float: Min Voltage
  - 1 Float: NTC Max Temperature
  - 1 Float: NTC Min Temperature
  - Int: Reboot counter
  - 8 Bytes String: Cycle

0x04: Request settings
  - 1 Byte: Num (Battery strings?)
  - 1 Float: MPA
  - 1 Float: Max Current
  - 1 Float: Turn off voltage
  - 1 Byte: Balancer
  - 1 Byte: Alarm mode
  - 1 Byte: Bat mode
  - 1 Int: Bat Volume
  - 1 Float: Balancing start voltage
```

cristi2005, 2022-05-13:
Hi,
In this moment I don't have a balancer, I order one NEEY today, when he arrives I'll let you know here and we'll do tests.
Cristi

syssi, 2022-05-13:
I know. I just tried to inform you if you own the device some day we could give it a try to add device support.

cristi2005, 2022-05-13:
Thank you for the great news and involvement.

cristi2005, 2022-07-29:
Hi I have a Neey balancer, I prepare a phone settings for developer otions....... I downloaded adb files, I conected the phone on laptop, run powershell:

#!/bin/bash
OUTPUT_DIR=$(pwd)
pushd /tmp
echo "Getting Bug Report..."
adb bugreport bug
echo "Unzipping Bluetooth HCI Log..."
unzip bug.zip FS/data/log/bt/btsnoop_hci.log
echo "Copying btsnoop_hci.log to '$OUTPUT_DIR' ..."
cp FS/data/log/bt/btsnoop_hci.log $OUTPUT_DIR/$(date +"%Y%m%d_%H%M%S")_btsnoop_hci.log
echo "Cleaning up..."
rm FS/data/log/bt/btsnoop_hci.log
rm bug.zip
popd

And have a lot of errors and not create btsnoop capture.
If you have some time to help my with the capture please leave message here.

Thank you,
Cristi

syssi, 2022-07-29:
Are you sure the `bug.zip` did not contain a "*btsnoot*log"?

```
unzip bug.zip
find . -name btsnoop_hci.log
```

cristi2005, 2022-07-29:
Only this 4 file:

bugreport-2022-07-29-21-20-00.txt
dumpstate_log.txt
main_entry.txt
version.txt

cristi2005, 2022-07-29:
If you want can allow a remote desktop with AnyDesk ?

syssi, 2022-07-29:
I will provide some additional instructions tomorrow. I assume some important steps of the capturing process are missing. I've to verify some steps locally to provide better instructions. Probably just a "Bluetooth off + Bluetooth on" is missing to trigger the traffic capture. Please start the NEEY app as next step and connect to the BMS. The to retrieve the bug report again. This time the file should be part of the dump. If not we could try to retrieve the dump manually from your device using adb. Where are you from / at which timezone do you life? I've some free time for a anydesk session tomorrow.

cristi2005, 2022-07-29:
I am from Romania, it's ok for me tomorrow, I'm at home all today and can tested all necesary.
Give me a sign tomorrow when you have time and we'll do the tests together.

syssi, 2022-07-29:
Alright. I will ping you tomorrow!

cristi2005, 2022-07-29:
Ok, good nyght.

syssi, 2022-07-30:
Could you tell me the Android version of your device?

cristi2005, 2022-07-30:
8.0.0

syssi, 2022-07-30:
I tried the following with my Android 9 phone:

1. Enable ADB
2. Enable Bluetooth HCI snoop log
3. Turn off bluetooth + turn on bluetooth
4. Start the BMS app and play around
5. Generate a bug report containing the capture `adb bugreport bug`
6. Unzip the bug.zip and copy the `FS/data/misc/bluetooth/logs/btsnoop_hci.log`:

```
$ unzip ../bug.zip
Archive:  ../bug.zip
  inflating: version.txt             
  inflating: proto/SurfaceFlinger_CRITICAL.proto  
  inflating: proto/activity_CRITICAL.proto  
  inflating: proto/window_CRITICAL.proto  
  inflating: FS/data/misc/recovery/ro.build.fingerprint  
  inflating: FS/data/misc/recovery/proc/version.1  
  inflating: FS/data/misc/recovery/proc/version  
  inflating: FS/data/misc/recovery/ro.build.fingerprint.1
[...]
  inflating: lshal-debug/vendor.qti.hardware.radio.qcrilhook@1.0::IQtiOemHook_oemhook1.txt  
  inflating: lshal-debug/vendor.qti.hardware.radio.qtiradio@1.0::IQtiRadio_slot1.txt  
  inflating: lshal-debug/vendor.qti.hardware.radio.qtiradio@1.0::IQtiRadio_slot2.txt  
  inflating: lshal-debug/vendor.qti.hardware.radio.qtiradio@2.0::IQtiRadio_slot1.txt  
  inflating: lshal-debug/vendor.qti.hardware.radio.qtiradio@2.0::IQtiRadio_slot2.txt  
  inflating: lshal-debug/vendor.qti.hardware.radio.uim@1.0::IUim_Uim0.txt  
  inflating: lshal-debug/vendor.qti.hardware.radio.uim@1.0::IUim_Uim1.txt  
  inflating: lshal-debug/vendor.qti.hardware.radio.uim@1.1::IUim_Uim0.txt  
  inflating: lshal-debug/vendor.qti.hardware.radio.uim@1.1::IUim_Uim1.txt  
  inflating: lshal-debug/vendor.qti.hardware.radio.uim_remote_client@1.0::IUimRemoteServiceClient_uimRemoteClient0.txt  
  inflating: lshal-debug/vendor.qti.hardware.radio.uim_remote_client@1.0::IUimRemoteServiceClient_uimRemoteClient1.txt  
  inflating: lshal-debug/vendor.qti.hardware.radio.uim_remote_server@1.0::IUimRemoteServiceServer_uimRemoteServer0.txt  
  inflating: lshal-debug/vendor.qti.hardware.radio.uim_remote_server@1.0::IUimRemoteServiceServer_uimRemoteServer1.txt  
  inflating: lshal-debug/vendor.qti.hardware.tui_comm@1.0::ITuiComm_default.txt  
  inflating: lshal-debug/vendor.qti.ims.callinfo@1.0::IService_default.txt  
  inflating: lshal-debug/vendor.qti.imsrtpservice@1.0::IRTPService_imsrtpservice.txt  
  inflating: FS/data/misc/bluetooth/logs/btsnoop_hci.log  
  inflating: FS/data/misc/bluetooth/logs/firmware_events.log  
  inflating: FS/data/misc/bluetooth/logs/firmware_events.log.last  
  inflating: FS/data/anr/anr_2022-07-29-10-04-23-453  
  inflating: FS/data/tombstones/tombstone_24  
  inflating: FS/data/tombstones/tombstone_23  
  inflating: FS/data/tombstones/tombstone_22  
  inflating: FS/data/tombstones/tombstone_21  
  inflating: FS/data/tombstones/tombstone_20  
  inflating: FS/data/tombstones/tombstone_19  
  inflating: FS/data/tombstones/tombstone_18  
  inflating: FS/data/tombstones/tombstone_17  
  inflating: FS/data/tombstones/tombstone_16  
  inflating: FS/data/tombstones/tombstone_15  
  inflating: proto/activity.proto    
  inflating: bugreport-jasmine-PQ3A.190801.002-2022-07-30-10-22-53.txt  
  inflating: main_entry.txt          
  inflating: dumpstate_log.txt   
```

My `bug.zip` contains hundreds of files.

syssi, 2022-07-30:
Is your phone rooted? If yes we could try this approach: https://stackoverflow.com/a/58384046

cristi2005, 2022-07-30:
Can you asist me on AnyDesk ?
If is yes where send AnyDesk ID ?

syssi, 2022-07-30:
Let's meet at 13.00 o'clock CEST. You could send the anydesk id to basti@linkt.de.

cristi2005, 2022-07-30:
Finaly I have BT snoop log, I send an email to see if ok ?

cristi2005, 2022-07-30:
I send on email.

syssi, 2022-07-30:
Awesome! It looks like your capture contains everything I'm looking for. I will try to extract some requests & responses.

syssi, 2022-07-30:
```
1. Command

<- Write request (0x12)
   Service UUID: 0xffe0
   Char: 0xffe1
   Char Configuration: 0x2902
   (Handle 0x000b, Subscribe to notifications)

-> Write response (0x13)

2. Write command (0x52)
   Service UUID: 0xffe0
   Char: 0xffe1
   Value: aa5511010100140000000000000000000000faff

More commands:

aa5511010400140000000000000000000000ffff
aa5511010200001400000000000000000000f9ff

aa551100050d140000000000000000000000f2ff
aa551100050d140000000000000000000000f2ff

aa551100050d140001000000000000000000f3ff
aa551100050d140001000000000000000000f3ff

Notifications (32 bytes per chunk, 480 bytes per frame):

01: 55aa110102002c0138e7fa5040b6045140850e51
02: 40f0055140b6045140751e51407f4f5140430251
03: 401c3d5140786a5140fe825140167e5140bc7651
04: 40167e51408b805140ca66514000000000000000
05: 0000000000000000000000000000000000000000
06: 00000000003593243e6894263e3d251b3e908e1b
07: 3eb3f3233e2e91253ec61b1a3e4a7c1c3e6f1b1a
08: 3ec2431b3e851e183e4b27193e5edf183ed0eb1a
09: 3ee6d4183e0cfe183e0000000000000000000000
10: 0000000000000000000000000000000000000000
11: 00de405142de4051400017083c0a000f0519a182
12: c0c3f54842c3f548420000000000000000000000
13: 0000000000000000000000000000762e09000000
14: 0000000000000000000000000000000000000000
15: 000000000000000000000000000000000000b6ff

01: 55aa110102002c0139e7fa5040b6045140850e51
02: 40f0055140b6045140751e51407f4f5140430251
03: 401c3d5140786a5140fe825140167e5140bc7651
04: 40167e51408b805140ca66514000000000000000
05: 0000000000000000000000000000000000000000
06: 00000000003593243e6894263e3d251b3e908e1b
07: 3eb3f3233e2e91253ec61b1a3e4a7c1c3e6f1b1a
08: 3ec2431b3e851e183e4b27193e5edf183ed0eb1a
09: 3ee6d4183e0cfe183e0000000000000000000000
10: 0000000000000000000000000000000000000000
11: 00de405142de4051400017083c0a000f0519a182
12: c0c3f54842c3f548420000000000000000000000
13: 0000000000000000000000000000772e09000000
14: 0000000000000000000000000000000000000000
15: 000000000000000000000000000000000000b8ff

01: 55aa110102002c013ae7fa5040b6045140850e51
02: 40f0055140b6045140751e51407f4f5140430251
03: 401c3d5140786a5140fe825140167e5140bc7651
04: 40167e51408b805140ca66514000000000000000
05: 0000000000000000000000000000000000000000
06: 00000000003593243e6894263e3d251b3e908e1b
07: 3eb3f3233e2e91253ec61b1a3e4a7c1c3e6f1b1a
08: 3ec2431b3e851e183e4b27193e5edf183ed0eb1a
09: 3ee6d4183e0cfe183e0000000000000000000000
[... 174 lines cut ...]
12: 0052b84a4252b84a420000000000000000000000
13: 000000000000000000000000000057120a000000
14: 0000000000000000000000000000000000000000
15: 000000000000000000000000000000000000d2ff

-> Command: aa551100050d140001000000000000000000f3ff

Command response: 55aa1100050d14000000000000000000000036ff

01: 55aa110102002c019672f94f40ebba4f405fbd4f
02: 40bac44f4046c24f40cab44f4013a64f40c3cf4f
03: 401d155340c20d53400dff524035ea5240dc0853
04: 402e055340470053403610534000000000000000
05: 0000000000000000000000000000000000000000
06: 00000000003593243e6894263e3d251b3e908e1b
07: 3eb3f3233e2e91253ec61b1a3e4a7c1c3e6f1b1a
08: 3ec2431b3e851e183e4b27193e5edf183ed0eb1a
09: 3ee6d4183e0cfe183e0000000000000000000000
10: 0000000000000000000000000000000000000000
11: 00e0645142e064514080c25b3d08060f04000000
12: 000bd749420bd749420000000000000000000000
13: 00000000000000000000000000005a120a000000
14: 0000000000000000000000000000000000000000
15: 0000000000000000000000000000000000007cff
```

syssi, 2022-07-30:
Could you provide some screenshots of the android app? This will help me a bit to understand what's available/exposed and whats internal + what can be controlled and what's read-only.

cristi2005, 2022-07-30:
I attached a pic, If you need somthing ... please tell me.
Thank you     Pe sâmbătă,

syssi, 2022-07-30:
> I attached a pic, If you need somthing ... please tell me.

Thanks! Could you explain the purpose of the `Run:` switch which is turned on at the moment?

cristi2005, 2022-07-30:
Turn on or off balancer, if is on, on balancer sounds a bip buzer and balancer turn on and cell begin balanced, if switc turn off, on balancer sound a bip buzer and balancer turn off and stop cell balancing. I hope you understand I write here…

Trimis din Yahoo Mail pentru iPhone

syssi, 2022-07-30:
I would be happy if you could create another traffic capture while turning balancing on and off multiple times. I would like to capture the balancing on/off commands.

cristi2005, 2022-07-30:
I attached your request.
    Pe sâmbătă,

syssi, 2022-07-30:
Please flash this first draft to an ESP32: https://github.com/syssi/esphome-jk-bms/blob/fe64af4e8b7dfa85d6ec36985f7c417d5444291c/esp32-heltec-balancer-ble-example.yaml

You have to update the MAC address at the top of the configuration to one of your devices. Please give it a try and provide some logs. The component should be able to connect to your balancer and receive some raw data.

cristi2005, 2022-07-30:
Ehen I try to install on ESP board I have this error:

INFO Reading configuration /config/esphome/neey-bal1.yaml...
INFO Updating https://github.com/syssi/esphome-jk-bms.git@add-heltec-balancer-support
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing heltec-balancer (board: wemos_d1_mini32; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
Tool Manager: Installing platformio/toolchain-esp32ulp @ ~1.22851.0
_**Error: Could not find the package with 'platformio/toolchain-esp32ulp @ ~1.22851.0' requirements for your system 'linux_aarch64'**_

syssi, 2022-07-30:
Please use a linux or windows machine to compile the configuration. The esphome / platformio build stack isn't Raspberry Pi compatible at the moment.

cristi2005, 2022-07-30:
Sorry, I don't have Linux and on Windows I don't know how ?

syssi, 2022-07-30:
https://esphome.io/guides/installing_esphome.html

1. Install python 3.10
2. Make sure to "Add Python to PATH".
3. Open a terminal window and try to use the python package manager:

```
pip3 install wheel
pip3 install esphome
```

4. Copy the `neey-bal1.yaml` to your windows machine
5. Go to the directory and execute `esphome run neey-bal1.yaml`:

```
cd Desktop
esphome run neey-bal1.yaml
```

cristi2005, 2022-07-30:
Finaly, I have this:

[19:28:03][I][heltec_balancer_ble:188]: Request status notification
[19:28:03][D][heltec_balancer_ble:338]: Write register: AA.55.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.26.FF (20)
[19:28:03][I][heltec_balancer_ble:148]: Notification: 55.AA.11.01.01.00.64.00.47.57.2D.32.34.53.34.45.42.00.00.00.00.00.00.00.48.57.2D.32.2E.38.2E.30.5A.48.2D.31.2E.32.2E.33.56.31.2E.30.2E.30.00.00.32.30.32.32.30.35.33.31.05.00.00.00.01.91.0A.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.AB.FF (100)
[19:28:08][I][heltec_balancer_ble:188]: Request status notification
[19:28:08][D][heltec_balancer_ble:338]: Write register: AA.55.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.26.FF (20)
[19:28:08][I][heltec_balancer_ble:148]: Notification: 55.AA.11.01.01.00.64.00.47.57.2D.32.34.53.34.45.42.00.00.00.00.00.00.00.48.57.2D.32.2E.38.2E.30.5A.48.2D.31.2E.32.2E.33.56.31.2E.30.2E.30.00.00.32.30.32.32.30.35.33.31.05.00.00.00.01.91.0A.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.AB.FF (100)
[19:28:13][I][heltec_balancer_ble:188]: Request status notification
[19:28:13][D][heltec_balancer_ble:338]: Write register: AA.55.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.26.FF (20)
[19:28:13][I][heltec_balancer_ble:148]: Notification: 55.AA.11.01.01.00.64.00.47.57.2D.32.34.53.34.45.42.00.00.00.00.00.00.00.48.57.2D.32.2E.38.2E.30.5A.48.2D.31.2E.32.2E.33.56.31.2E.30.2E.30.00.00.32.30.32.32.30.35.33.31.05.00.00.00.01.91.0A.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.AB.FF (100)
[19:28:18][I][heltec_balancer_ble:188]: Request status notification
[19:28:18][D][heltec_balancer_ble:338]: Write register: AA.55.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.26.FF (20)
[19:28:18][I][heltec_balancer_ble:148]: Notification: 55.AA.11.01.01.00.64.00.47.57.2D.32.34.53.34.45.42.00.00.00.00.00.00.00.48.57.2D.32.2E.38.2E.30.5A.48.2D.31.2E.32.2E.33.56.31.2E.30.2E.30.00.00.32.30.32.32.30.35.33.31.05.00.00.00.01.91.0A.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.AB.FF (100)
[19:28:23][I][heltec_balancer_ble:188]: Request status notification
[19:28:23][D][heltec_balancer_ble:338]: Write register: AA.55.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.26.FF (20)
[19:28:23][I][heltec_balancer_ble:148]: Notification: 55.AA.11.01.01.00.64.00.47.57.2D.32.34.53.34.45.42.00.00.00.00.00.00.00.48.57.2D.32.2E.38.2E.30.5A.48.2D.31.2E.32.2E.33.56.31.2E.30.2E.30.00.00.32.30.32.32.30.35.33.31.05.00.00.00.01.91.0A.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.AB.FF (100)
[19:28:28][I][heltec_balancer_ble:188]: Request status notification
[19:28:28][D][heltec_balancer_ble:338]: Write register: AA.55.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.26.FF (20)
[19:28:28][I][heltec_balancer_ble:148]: Notification: 55.AA.11.01.01.00.64.00.47.57.2D.32.34.53.34.45.42.00.00.00.00.00.00.00.48.57.2D.32.2E.38.2E.30.5A.48.2D.31.2E.32.2E.33.56.31.2E.30.2E.30.00.00.32.30.32.32.30.35.33.31.05.00.00.00.01.91.0A.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.AB.FF (100)
[19:28:33][I][heltec_balancer_ble:188]: Request status notification
[19:28:33][D][heltec_balancer_ble:338]: Write register: AA.55.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.26.FF (20)
[19:28:33][I][heltec_balancer_ble:148]: Notification: 55.AA.11.01.01.00.64.00.47.57.2D.32.34.53.34.45.42.00.00.00.00.00.00.00.48.57.2D.32.2E.38.2E.30.5A.48.2D.31.2E.32.2E.33.56.31.2E.30.2E.30.00.00.32.30.32.32.30.35.33.31.05.00.00.00.01.91.0A.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.AB.FF (100)
[19:28:37][D][esp32_ble_tracker:726]: Found device 6A:D4:01:69:F3:5B RSSI=-82
[19:28:37][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[19:28:38][I][heltec_balancer_ble:188]: Request status notification
[19:28:38][D][heltec_balancer_ble:338]: Write register: AA.55.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.26.FF (20)
[19:28:38][I][heltec_balancer_ble:148]: Notification: 55.AA.11.01.01.00.64.00.47.57.2D.32.34.53.34.45.42.00.00.00.00.00.00.00.48.57.2D.32.2E.38.2E.30.5A.48.2D.31.2E.32.2E.33.56.31.2E.30.2E.30.00.00.32.30.32.32.30.35.33.31.05.00.00.00.01.91.0A.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.AB.FF (100)
[19:28:42][D][api:102]: Accepted ::FFFF:192.168.1.39
[19:28:42][D][api:102]: Accepted ::FFFF:192.168.1.39
[19:28:42][W][api.connection:080]: Home Assistant 2022.7.7 (::FFFF:192.168.1.39): Connection closed
[19:28:42][W][api.connection:080]: Home Assistant 2022.7.7 (::FFFF:192.168.1.39): Connection closed
[19:28:42][D][esp32_ble_tracker:726]: Found device 53:33:B3:8F:20:DC RSSI=-91
[19:28:42][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[19:28:42][D][esp32_ble_tracker:751]:   TX Power: 2
[19:28:43][I][heltec_balancer_ble:188]: Request status notification
[19:28:43][D][heltec_balancer_ble:338]: Write register: AA.55.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.26.FF (20)
[19:28:43][I][heltec_balancer_ble:148]: Notification: 55.AA.11.01.01.00.64.00.47.57.2D.32.34.53.34.45.42.00.00.00.00.00.00.00.48.57.2D.32.2E.38.2E.30.5A.48.2D.31.2E.32.2E.33.56.31.2E.30.2E.30.00.00.32.30.32.32.30.35.33.31.05.00.00.00.01.91.0A.00.00.00.00.00.00. [... cut ...]
[19:28:53][I][heltec_balancer_ble:188]: Request status notification
[19:28:53][D][heltec_balancer_ble:338]: Write register: AA.55.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.26.FF (20)
[19:28:53][I][heltec_balancer_ble:148]: Notification: 55.AA.11.01.01.00.64.00.47.57.2D.32.34.53.34.45.42.00.00.00.00.00.00.00.48.57.2D.32.2E.38.2E.30.5A.48.2D.31.2E.32.2E.33.56.31.2E.30.2E.30.00.00.32.30.32.32.30.35.33.31.05.00.00.00.01.91.0A.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.AB.FF (100)

syssi, 2022-07-30:
Good job! We are able to talk to the Balancer in general. I will try to enhance to communication now. There won't be any progress today anymore. I will ping you tomorrow as soon as I made some progress. You only need to execute `esphome run config.yaml` everytime and provide a new log. It's totally fine if it takes a few days to get everything up and running.

syssi, 2022-07-30:
> I attached your request.

I didn't receive any attachment yet. Could your provide another capturing containing the "balancer on/off" command? Thanks!

In near future I'm also interested in a traffic capture of some the "change settings" commands. I would like to be able to change the "start balancing voltage" etc.

cristi2005, 2022-07-30:
Its Ok for me, when you need something to tested please ping me.

cristi2005, 2022-07-30:
I

> I didn't receive any attachment yet. Could your provide another capturing containing the "balancer on/off" command? Thanks!
>
> In near future I'm also interested in a traffic capture of some the [...]

I sended again on mail.

cristi2005, 2022-07-30:
Please confirm if you received and if is ok.

syssi, 2022-07-30:
```
*READ*
        
System request:      0xAA 0x55 0x11 0x01 0x01 0x00 0x14 0x00 ... 0x00 0x00 CRC 0xFF
Battery request:     0xAA 0x55 0x11 0x01 0x02 0x00 0x00 0x14 0x00 ... 0x00 CRC 0xFF
Factory request:     0xAA 0x55 0x11 0x01 0x03 0x00 0x14 0x00 ... 0x00 0x00 CRC 0xFF
Settings request:    0xAA 0x55 0x11 0x01 0x04 0x00 0x14 0x00 ... 0x00 0x00 CRC 0xFF

*WRITE*

int_x, int_y, int_z: 0xAA 0x55 0x11 0x00  x    y   0x14 0x00  z  0x00 ... 0x00 CRC 0xFF

int_x, int_y, string s, int_z:
                     0xAA 0x55 0x11 0x00  x    y   0x14 0x00  substr(s, z) ... CRC 0xFF

int_x, int_y, float_f:
                     0xAA 0x55 0x11 0x00  x    y   0x14 0x00  f  0x00 ... 0x00 CRC 0xFF

int_x, int_y, byte_b:
                     0xAA 0x55 0x11 0x00  x    y   0x14 0x00  b  0x00 ... 0x00 CRC 0xFF

```

syssi, 2022-07-30:
Balacing on/off:

```
aa551100050d140000000000000000000000f2ff
aa551100050d140001000000000000000000f3ff
```

syssi, 2022-07-31:
@cristi2005 Please compile & flash again. The component should be able to decode the `device info frame` now and in best case the `cell info frame` will be requested. Please provide a new log containing some responses.

cristi2005, 2022-07-31:
[13:56:29][I][heltec_balancer_ble:198]: Request status notification
[13:56:29][D][heltec_balancer_ble:386]: Write register: 55.AA.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[13:56:32][D][esp32_ble_tracker:726]: Found device 55:AA:92:51:E3:8D RSSI=-88
[13:56:32][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:56:32][D][esp32_ble_tracker:751]:   TX Power: 2
[13:56:34][I][heltec_balancer_ble:198]: Request status notification
[13:56:34][D][heltec_balancer_ble:386]: Write register: 55.AA.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[13:56:36][D][esp32_ble_tracker:726]: Found device C4:1A:2E:E6:F1:8E RSSI=-85
[13:56:36][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:56:37][D][esp32_ble_tracker:726]: Found device 4E:79:B7:85:B5:DB RSSI=-86
[13:56:37][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:56:37][D][esp32_ble_tracker:751]:   TX Power: 2
[13:56:39][I][heltec_balancer_ble:198]: Request status notification
[13:56:39][D][heltec_balancer_ble:386]: Write register: 55.AA.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[13:56:40][D][esp32_ble_tracker:726]: Found device 5D:A2:39:D2:B7:93 RSSI=-88
[13:56:40][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[13:56:40][D][esp32_ble_tracker:751]:   TX Power: 2
[13:56:40][D][api:102]: Accepted ::FFFF:192.168.1.39
[13:56:40][D][api:102]: Accepted ::FFFF:192.168.1.39
[13:56:40][W][api.connection:080]: Home Assistant 2022.7.7 (::FFFF:192.168.1.39): Connection closed
[13:56:40][W][api.connection:080]: Home Assistant 2022.7.7 (::FFFF:192.168.1.39): Connection closed
[13:56:44][I][heltec_balancer_ble:198]: Request status notification
[13:56:44][D][heltec_balancer_ble:386]: Write register: 55.AA.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[13:56:49][I][heltec_balancer_ble:198]: Request status notification
[13:56:49][D][heltec_balancer_ble:386]: Write register: 55.AA.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[13:56:54][I][heltec_balancer_ble:198]: Request status notification
[13:56:54][D][heltec_balancer_ble:386]: Write register: 55.AA.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[13:56:59][I][heltec_balancer_ble:198]: Request status notification
[13:56:59][D][heltec_balancer_ble:386]: Write register: 55.AA.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[13:57:04][I][heltec_balancer_ble:198]: Request status notification
[13:57:04][D][heltec_balancer_ble:386]: Write register: 55.AA.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[13:57:09][I][heltec_balancer_ble:198]: Request status notification
[13:57:09][D][heltec_balancer_ble:386]: Write register: 55.AA.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[13:57:14][I][heltec_balancer_ble:198]: Request status notification
[13:57:14][D][heltec_balancer_ble:386]: Write register: 55.AA.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[13:57:19][I][heltec_balancer_ble:198]: Request status notification
[13:57:19][D][heltec_balancer_ble:386]: Write register: 55.AA.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[13:57:24][I][heltec_balancer_ble:198]: Request status notification
[13:57:24][D][heltec_balancer_ble:386]: Write register: 55.AA.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[13:57:29][I][heltec_balancer_ble:198]: Request status notification
[13:57:29][D][heltec_balancer_ble:386]: Write register: 55.AA.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[13:57:34][I][heltec_balancer_ble:198]: Request status notification
[13:57:34][D][heltec_balancer_ble:386]: Write register: 55.AA.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)

syssi, 2022-07-31:
This doesn't look good. The balancer didn't respond anymore. I've made a small change. Please try again!

cristi2005, 2022-07-31:
When try to compile I  have a lot fo error:

C:\tools>esphome run neey-bal1.yaml
←[32mINFO Reading configuration neey-bal1.yaml...←[0m
←[32mINFO Cloning https://github.com/syssi/esphome-jk-bms.git@add-heltec-balancer-support←[0m
←[32mINFO Generating C++ source...←[0m
←[32mINFO Core config or version changed, cleaning build files...←[0m
←[32mINFO Compiling app...←[0m
Processing neey-bal1 (board: wemos_d1_mini32; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/wemos_d1_mini32.html
PLATFORM: Espressif 32 (3.5.0) > WeMos D1 MINI ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40302.0 (4.3.2)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-riscv32-esp 8.4.0+2021r2-patch2
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch2
 - toolchain-xtensa-esp32s2 8.4.0+2021r2-patch2
Reading CMake configuration...
Generating assembly for certificate bundle...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pioenvs\neey-bal1\src\esphome\components\api\api_connection.o
Compiling .pioenvs\neey-bal1\src\esphome\components\api\api_frame_helper.o
Compiling .pioenvs\neey-bal1\src\esphome\components\api\api_pb2.o
Compiling .pioenvs\neey-bal1\src\esphome\components\api\api_pb2_service.o
Compiling .pioenvs\neey-bal1\src\esphome\components\api\api_server.o
Compiling .pioenvs\neey-bal1\src\esphome\components\api\list_entities.o
Compiling .pioenvs\neey-bal1\src\esphome\components\api\proto.o
Compiling .pioenvs\neey-bal1\src\esphome\components\api\subscribe_state.o
Compiling .pioenvs\neey-bal1\src\esphome\components\api\user_services.o
Compiling .pioenvs\neey-bal1\src\esphome\components\binary_sensor\automation.o
Compiling .pioenvs\neey-bal1\src\esphome\components\binary_sensor\binary_sensor.o
Compiling .pioenvs\neey-bal1\src\esphome\components\binary_sensor\filter.o
Compiling .pioenvs\neey-bal1\src\esphome\components\ble_client\ble_client.o
Compiling .pioenvs\neey-bal1\src\esphome\components\button\button.o
Compiling .pioenvs\neey-bal1\src\esphome\components\esp32\core.o
Compiling .pioenvs\neey-bal1\src\esphome\components\esp32\gpio_arduino.o
Compiling .pioenvs\neey-bal1\src\esphome\components\esp32\gpio_idf.o
Compiling .pioenvs\neey-bal1\src\esphome\components\esp32\preferences.o
Compiling .pioenvs\neey-bal1\src\esphome\components\esp32_ble_tracker\esp32_ble_tracker.o
src/esphome/components/ble_client/ble_client.cpp: In member function 'float esphome::ble_client::BLEClient::parse_char_value(uint8_t*, uint16_t)':
src/esphome/components/ble_client/ble_client.cpp:257:7: warning: this statement may fall through [-Wimplicit-fallthrough=]
       if (length > 2) {
       ^~
src/esphome/components/ble_client/ble_client.cpp:260:5: note: here
     case 0x7:  // uint24.
     ^~~~
src/esphome/components/ble_client/ble_client.cpp:261:7: warning: this statement may fall through [-Wimplicit-fallthrough=]
       if (length > 3) {
       ^~
src/esphome/components/ble_client/ble_client.cpp:264:5: note: here
     case 0x8:  // uint32.
     ^~~~
src/esphome/components/ble_client/ble_client.cpp:265:7: warning: this statement may fall through [-Wimplicit-fallthrough=]
       if (length > 4) {
       ^~
src/esphome/components/ble_client/ble_client.cpp:269:5: note: here
[... 9 lines cut ...]
       if (length > 3) {
       ^~
src/esphome/components/ble_client/ble_client.cpp:280:5: note: here
     case 0x10:  // int32.
     ^~~~
Compiling .pioenvs\neey-bal1\src\esphome\components\heltec_balancer_ble\heltec_balancer_ble.o
Compiling .pioenvs\neey-bal1\src\esphome\components\logger\logger.o
Compiling .pioenvs\neey-bal1\src\esphome\components\md5\md5.o
In file included from src/esphome/components/sensor/sensor.h:3,
                 from src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.h:9,
                 from src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.cpp:1:
src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.cpp: In member function 'void esphome::heltec_balancer_ble::HeltecBalancerBle::decode_settings_(const std::vector<unsigned char>&)':
src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.cpp:354:34: error: 'heltec_get_16bit' was not declared in this scope
   ESP_LOGD(TAG, "  Command: %d", heltec_get_16bit(4));
                                  ^~~~~~~~~~~~~~~~
src/esphome/core/log.h:92:89: note: in definition of macro 'esph_log_d'
   esp_log_printf_(ESPHOME_LOG_LEVEL_DEBUG, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                         ^~~~~~~~~~~
src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.cpp:354:3: note: in expansion of macro 'ESP_LOGD'
   ESP_LOGD(TAG, "  Command: %d", heltec_get_16bit(4));
   ^~~~~~~~
*** [.pioenvs\neey-bal1\src\esphome\components\heltec_balancer_ble\heltec_balancer_ble.o] Error 1
============================================= [FAILED] Took 65.42 seconds =============================================

C:\tools>

syssi, 2022-07-31:
My fault. Wait a sec.

syssi, 2022-07-31:
It should build fine again.

cristi2005, 2022-07-31:
[14:48:57][D][ble_client:047]: Found device at MAC address [3C:A5:51:89:40:FF]
[14:48:57][I][ble_client:085]: Attempting BLE connection to 3c:a5:51:89:40:ff
[14:48:58][I][ble_client:170]: Service UUID: 0x1800
[14:48:58][I][ble_client:171]:   start_handle: 0x1  end_handle: 0x7
[14:48:58][I][ble_client:383]:  characteristic 0x2A00, handle 0x3, properties 0x2
[14:48:58][I][ble_client:383]:  characteristic 0x2A01, handle 0x5, properties 0x2
[14:48:58][I][ble_client:383]:  characteristic 0x2A04, handle 0x7, properties 0x2
[14:48:58][I][ble_client:170]: Service UUID: 0xFFE0
[14:48:58][I][ble_client:171]:   start_handle: 0x8  end_handle: 0xffff
[14:48:58][I][ble_client:383]:  characteristic 0xFFE1, handle 0xa, properties 0x1c
[14:48:58][I][ble_client:383]:  characteristic 0xFFE2, handle 0xd, properties 0x1c
[14:48:58][I][ble_client:383]:  characteristic 0xFFE3, handle 0x10, properties 0xc
[14:48:58][D][esp32_ble_tracker:217]: Starting scan...
[14:48:58][I][heltec_balancer_ble:159]: Request device info
[14:48:58][D][heltec_balancer_ble:470]: Write register: 55.AA.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.FA.FF (20)
[14:48:59][I][heltec_balancer_ble:250]: Request status notification
[14:48:59][D][heltec_balancer_ble:470]: Write register: 55.AA.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.FA.FF (20)
[14:48:59][D][esp32_ble_tracker:726]: Found device 3C:A5:51:85:E8:F6 RSSI=-90
[14:48:59][D][esp32_ble_tracker:747]:   Address Type: PUBLIC
[14:48:59][D][esp32_ble_tracker:749]:   Name: 'Bank1     '
[14:49:03][D][esp32_ble_tracker:726]: Found device 69:E8:85:7D:E0:53 RSSI=-38
[14:49:03][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[14:49:04][I][heltec_balancer_ble:250]: Request status notification
[14:49:04][D][heltec_balancer_ble:470]: Write register: 55.AA.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.FA.FF (20)
[14:49:06][D][esp32_ble_tracker:726]: Found device 40:FE:1D:65:95:E9 RSSI=-83
[14:49:06][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[14:49:06][D][esp32_ble_tracker:751]:   TX Power: 2
[14:49:08][D][esp32_ble_tracker:726]: Found device EB:91:28:14:6D:AD RSSI=-81
[14:49:08][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[14:49:09][I][heltec_balancer_ble:250]: Request status notification
[14:49:09][D][heltec_balancer_ble:470]: Write register: 55.AA.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.FA.FF (20)
[14:49:14][I][heltec_balancer_ble:250]: Request status notification
[14:49:14][D][heltec_balancer_ble:470]: Write register: 55.AA.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.FA.FF (20)
[14:49:19][I][heltec_balancer_ble:250]: Request status notification
[14:49:19][D][heltec_balancer_ble:470]: Write register: 55.AA.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.FA.FF (20)
[14:49:24][I][heltec_balancer_ble:250]: Request status notification
[14:49:24][D][heltec_balancer_ble:470]: Write register: 55.AA.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.FA.FF (20)
[14:49:29][I][heltec_balancer_ble:250]: Request status notification
[14:49:29][D][heltec_balancer_ble:470]: Write register: 55.AA.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.FA.FF (20)
[14:49:29][D][esp32_ble_tracker:726]: Found device 64:32:28:B1:1F:0D RSSI=-89
[14:49:29][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[14:49:29][D][esp32_ble_tracker:751]:   TX Power: 2
[14:49:34][I][heltec_balancer_ble:250]: Request status notification
[14:49:34][D][heltec_balancer_ble:470]: Write register: 55.AA.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.FA.FF (20)
[14:49:39][I][heltec_balancer_ble:250]: Request status notification
[14:49:39][D][heltec_balancer_ble:470]: Write register: 55.AA.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.FA.FF (20)
[14:49:44][I][heltec_balancer_ble:250]: Request status notification
[14:49:44][D][heltec_balancer_ble:470]: Write register: 55.AA.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.FA.FF (20)
[14:49:49][I][heltec_balancer_ble:250]: Request status notification
[14:49:49][D][heltec_balancer_ble:470]: Write register: 55.AA.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.FA.FF (20)
[14:49:54][I][heltec_balancer_ble:250]: Request status notification
[14:49:54][D][heltec_balancer_ble:470]: Write register: 55.AA.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.FA.FF (20)
[14:49:57][D][api:102]: Accepted ::FFFF:192.168.1.39
[14:49:57][D][api:102]: Accepted ::FFFF:192.168.1.39
[14:49:57][W][api.connection:080]: Home Assistant 2022.7.7 (::FFFF:192.168.1.39): Connection closed
[14:49:57][W][api.connection:080]: Home Assistant 2022.7.7 (::FFFF:192.168.1.39): Connection closed
[14:49:59][I][heltec_balancer_ble:250]: Request status notification
[14:49:59][D][heltec_balancer_ble:470]: Write register: 55.AA.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.FA.FF (20)
[14:50:04][I][heltec_balancer_ble:250]: Request status notification
[14:50:04][D][heltec_balancer_ble:470]: Write register: 55.AA.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.FA.FF (20)
[14:50:09][I][heltec_balancer_ble:250]: Request status notification
[14:50:09][D][heltec_balancer_ble:470]: Write register: 55.AA.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.FA.FF (20)
[14:50:14][I][heltec_balancer_ble:250]: Request status notification
[14:50:14][D][heltec_balancer_ble:470]: Write register: 55.AA.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.FA.FF (20)
[14:50:19][I][heltec_balancer_ble:250]: Request status notification
[14:50:19][D][heltec_balancer_ble:470]: Write register: 55.AA.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.FA.FF (20)
[14:50:24][I][heltec_balancer_ble:250]: Request status notification
[14:50:24][D][heltec_balancer_ble:470]: Write register: 55.AA.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.FA.FF (20)
[14:50:29][I][heltec_balancer_ble:250]: Request status notification
[14:50:29][D][heltec_balancer_ble:470]: Write register: 55.AA.11.01.01.00.14.00.00.00.00.00.00.00.00.00.00.00.FA.FF (20)

syssi, 2022-07-31:
I've pushed another fix. The balancer should respond again this time.

cristi2005, 2022-07-31:
[15:34:57][I][heltec_balancer_ble:252]: Request status notification
[15:34:57][D][heltec_balancer_ble:480]: Write register: AA.55.11.01.02.00.00.14.00.00.00.00.00.00.00.00.00.00.F9.FF (20)
[15:34:57][D][esp32_ble_tracker:726]: Found device 5F:48:B5:AB:4A:3C RSSI=-36
[15:34:57][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[15:34:57][I][heltec_balancer_ble:171]: Notification: 55.AA.11.01.02.00.2C.01.EF.DB.46.56.40.B0.0F.56.40.FA.00.56.40.0A.F1.55.40.6E.03.56.40.7F.19.56.40.66.1E.56.40.F0.CF.55.40.68.44.56.40.BB.40.56.40.F4.41.56.40.D3.3B.56.40.68.44.56.40.A2.45.56.40.81.3F.56.40.68.44.56.40.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.35.93.24.3E.68.94.26.3E.3D.25.1B.3E.90.8E.1B.3E.B3.F3.23.3E.2E.91.25.3E.C6.1B.1A.3E.4A.7C.1C.3E.6F.1B.1A.3E.C2.43.1B.3E.85.1E.18.3E.4B.27.19.3E.5
[15:34:57][I][heltec_balancer_ble:171]: Notification: 00.00.00.00.00.00.00.00.00.00.00.00.00.39.99.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.DF.FF (59)
[15:34:58][I][heltec_balancer_ble:171]: Notification: 55.AA.11.01.02.00.2C.01.F0.DB.46.56.40.B0.0F.56.40.FA.00.56.40.0A.F1.55.40.6E.03.56.40.7F.19.56.40.66.1E.56.40.F0.CF.55.40.68.44.56.40.BB.40.56.40.F4.41.56.40.D3.3B.56.40.68.44.56.40.A2.45.56.40.81.3F.56.40.68.44.56.40.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.35.93.24.3E.68.94.26.3E.3D.25.1B.3E.90.8E.1B.3E.B3.F3.23.3E.2E.91.25.3E.C6.1B.1A.3E.4A.7C.1C.3E.6F.1B.1A.3E.C2.43.1B.3E.85.1E.18.3E.4B.27.19.3E.5
[15:34:58][I][heltec_balancer_ble:171]: Notification: 00.00.00.00.00.00.00.00.00.00.00.00.00.3A.99.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.18.FF (59)
[15:34:58][I][heltec_balancer_ble:171]: Notification: 55.AA.11.01.02.00.2C.01.F1.DB.46.56.40.B0.0F.56.40.FA.00.56.40.0A.F1.55.40.6E.03.56.40.7F.19.56.40.66.1E.56.40.F0.CF.55.40.68.44.56.40.BB.40.56.40.F4.41.56.40.D3.3B.56.40.68.44.56.40.A2.45.56.40.81.3F.56.40.68.44.56.40.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.35.93.24.3E.68.94.26.3E.3D.25.1B.3E.90.8E.1B.3E.B3.F3.23.3E.2E.91.25.3E.C6.1B.1A.3E.4A.7C.1C.3E.6F.1B.1A.3E.C2.43.1B.3E.85.1E.18.3E.4B.27.19.3E.5
[15:34:58][I][heltec_balancer_ble:171]: Notification: 00.00.00.00.00.00.00.00.00.00.00.00.00.3A.99.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.92.FF (59)
[15:34:59][I][heltec_balancer_ble:171]: Notification: 55.AA.11.01.02.00.2C.01.F2.DB.46.56.40.B0.0F.56.40.FA.00.56.40.0A.F1.55.40.6E.03.56.40.7F.19.56.40.66.1E.56.40.F0.CF.55.40.68.44.56.40.BB.40.56.40.F4.41.56.40.D3.3B.56.40.68.44.56.40.A2.45.56.40.81.3F.56.40.68.44.56.40.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.35.93.24.3E.68.94.26.3E.3D.25.1B.3E.90.8E.1B.3E.B3.F3.23.3E.2E.91.25.3E.C6.1B.1A.3E.4A.7C.1C.3E.6F.1B.1A.3E.C2.43.1B.3E.85.1E.18.3E.4B.27.19.3E.5
[15:34:59][I][heltec_balancer_ble:171]: Notification: 00.00.00.00.00.00.00.00.00.00.00.00.00.3B.99.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.94.FF (59)
[15:34:59][I][heltec_balancer_ble:171]: Notification: 55.AA.11.01.02.00.2C.01.F3.DB.46.56.40.24.12.56.40.4D.FD.55.40.0A.F1.55.40.E2.05.56.40.46.18.56.40.66.1E.56.40.2A.D1.55.40.DB.46.56.40.F4.41.56.40.F4.41.56.40.D3.3B.56.40.2E.43.56.40.A2.45.56.40.F4.41.56.40.DB.46.56.40.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.35.93.24.3E.68.94.26.3E.3D.25.1B.3E.90.8E.1B.3E.B3.F3.23.3E.2E.91.25.3E.C6.1B.1A.3E.4A.7C.1C.3E.6F.1B.1A.3E.C2.43.1B.3E.85.1E.18.3E.4B.27.19.3E.5
[15:34:59][I][heltec_balancer_ble:171]: Notification: 00.00.00.00.00.00.00.00.00.00.00.00.00.3B.99.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.DE.FF (59)
[15:35:00][I][heltec_balancer_ble:171]: Notification: 55.AA.11.01.02.00.2C.01.F4.DB.46.56.40.24.12.56.40.4D.FD.55.40.0A.F1.55.40.E2.05.56.40.46.18.56.40.66.1E.56.40.2A.D1.55.40.DB.46.56.40.F4.41.56.40.F4.41.56.40.D3.3B.56.40.2E.43.56.40.A2.45.56.40.F4.41.56.40.DB.46.56.40.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.35.93.24.3E.68.94.26.3E.3D.25.1B.3E.90.8E.1B.3E.B3.F3.23.3E.2E.91.25.3E.C6.1B.1A.3E.4A.7C.1C.3E.6F.1B.1A.3E.C2.43.1B.3E.85.1E.18.3E.4B.27.19.3E.5
[15:35:00][I][heltec_balancer_ble:171]: Notification: 00.00.00.00.00.00.00.00.00.00.00.00.00.3C.99.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.E0.FF (59)
[15:35:00][I][heltec_balancer_ble:171]: Notification: 55.AA.11.01.02.00.2C.01.F5.DB.46.56.40.24.12.56.40.4D.FD.55.40.0A.F1.55.40.E2.05.56.40.46.18.56.40.66.1E.56.40.2A.D1.55.40.DB.46.56.40.F4.41.56.40.F4.41.56.40.D3.3B.56.40.2E.43.56.40.A2.45.56.40.F4.41.56.40.DB.46.56.40.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.35.93.24.3E.68.94.26.3E.3D.25.1B.3E.90.8E.1B.3E.B3.F3.23.3E.2E.91.25.3E.C6.1B.1A.3E.4A.7C.1C.3E.6F.1B.1A.3E.C2.43.1B.3E.85.1E.18.3E.4B.27.19.3E.5
[... cut ...] C.3D.56.40.BB.40.56.40.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.35.93.24.3E.68.94.26.3E.3D.25.1B.3E.90.8E.1B.3E.B3.F3.23.3E.2E.91.25.3E.C6.1B.1A.3E.4A.7C.1C.3E.6F.1B.1A.3E.C2.43.1B.3E.85.1E.18.3E.4B.27.19.3E.5
[15:35:45][I][heltec_balancer_ble:171]: Notification: 00.00.00.00.00.00.00.00.00.00.00.00.00.68.99.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.16.FF (59)

syssi, 2022-07-31:
Perfect! If you press the reset button of your ESP the system should reboot. If the bluetooth connection to the balancer is established you should see the `device info frame` once at the logs. The content is pretty similar to the about page of the android app.

cristi2005, 2022-07-31:
[15:49:29][I][heltec_balancer_ble:171]: Notification: 00.00.00.00.00.00.00.00.00.00.00.00.00.9F.9C.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10.FF (59)
[15:49:30][I][heltec_balancer_ble:171]: Notification: 55.AA.11.01.02.00.2C.01.EF.14.22.56.40.F1.F5.55.40.3B.E7.55.40.D8.D4.55.40.22.EC.55.40.6E.03.56.40.1C.07.56.40.F6.B4.55.40.14.22.56.40.DA.20.56.40.14.22.56.40.F3.1B.56.40.14.22.56.40.DA.20.56.40.66.1E.56.40.14.22.56.40.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.35.93.24.3E.68.94.26.3E.3D.25.1B.3E.90.8E.1B.3E.B3.F3.23.3E.2E.91.25.3E.C6.1B.1A.3E.4A.7C.1C.3E.6F.1B.1A.3E.C2.43.1B.3E.85.1E.18.3E.4B.27.19.3E.5
[15:49:30][I][heltec_balancer_ble:171]: Notification: 00.00.00.00.00.00.00.00.00.00.00.00.00.9F.9C.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.E9.FF (59)
[15:49:30][I][heltec_balancer_ble:171]: Notification: 55.AA.11.01.02.00.2C.01.F0.14.22.56.40.F1.F5.55.40.3B.E7.55.40.D8.D4.55.40.22.EC.55.40.6E.03.56.40.1C.07.56.40.F6.B4.55.40.14.22.56.40.DA.20.56.40.14.22.56.40.F3.1B.56.40.14.22.56.40.DA.20.56.40.66.1E.56.40.14.22.56.40.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.35.93.24.3E.68.94.26.3E.3D.25.1B.3E.90.8E.1B.3E.B3.F3.23.3E.2E.91.25.3E.C6.1B.1A.3E.4A.7C.1C.3E.6F.1B.1A.3E.C2.43.1B.3E.85.1E.18.3E.4B.27.19.3E.5
[15:49:30][I][heltec_balancer_ble:171]: Notification: 00.00.00.00.00.00.00.00.00.00.00.00.00.A0.9C.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.42.FF (59)
[15:49:31][I][heltec_balancer_ble:171]: Notification: 55.AA.11.01.02.00.2C.01.F1.14.22.56.40.F1.F5.55.40.3B.E7.55.40.D8.D4.55.40.22.EC.55.40.6E.03.56.40.1C.07.56.40.F6.B4.55.40.14.22.56.40.DA.20.56.40.14.22.56.40.F3.1B.56.40.14.22.56.40.DA.20.56.40.66.1E.56.40.14.22.56.40.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.35.93.24.3E.68.94.26.3E.3D.25.1B.3E.90.8E.1B.3E.B3.F3.23.3E.2E.91.25.3E.C6.1B.1A.3E.4A.7C.1C.3E.6F.1B.1A.3E.C2.43.1B.3E.85.1E.18.3E.4B.27.19.3E.5
[15:49:31][I][heltec_balancer_ble:171]: Notification: 00.00.00.00.00.00.00.00.00.00.00.00.00.A0.9C.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.B0.FF (59)
[15:49:31][I][heltec_balancer_ble:171]: Notification: 55.AA.11.01.02.00.2C.01.F2.14.22.56.40.F1.F5.55.40.3B.E7.55.40.D8.D4.55.40.22.EC.55.40.6E.03.56.40.1C.07.56.40.F6.B4.55.40.14.22.56.40.DA.20.56.40.14.22.56.40.F3.1B.56.40.14.22.56.40.DA.20.56.40.66.1E.56.40.14.22.56.40.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.35.93.24.3E.68.94.26.3E.3D.25.1B.3E.90.8E.1B.3E.B3.F3.23.3E.2E.91.25.3E.C6.1B.1A.3E.4A.7C.1C.3E.6F.1B.1A.3E.C2.43.1B.3E.85.1E.18.3E.4B.27.19.3E.5
[15:49:31][I][heltec_balancer_ble:171]: Notification: 00.00.00.00.00.00.00.00.00.00.00.00.00.A1.9C.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.4A.FF (59)
[15:49:32][I][heltec_balancer_ble:171]: Notification: 55.AA.11.01.02.00.2C.01.F3.14.22.56.40.F1.F5.55.40.3B.E7.55.40.D8.D4.55.40.22.EC.55.40.6E.03.56.40.1C.07.56.40.F6.B4.55.40.14.22.56.40.DA.20.56.40.14.22.56.40.F3.1B.56.40.14.22.56.40.DA.20.56.40.66.1E.56.40.14.22.56.40.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.35.93.24.3E.68.94.26.3E.3D.25.1B.3E.90.8E.1B.3E.B3.F3.23.3E.2E.91.25.3E.C6.1B.1A.3E.4A.7C.1C.3E.6F.1B.1A.3E.C2.43.1B.3E.85.1E.18.3E.4B.27.19.3E.5
[15:49:32][I][heltec_balancer_ble:171]: Notification: 00.00.00.00.00.00.00.00.00.00.00.00.00.A1.9C.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.BD.FF (59)
INFO neey-bal1.local: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for neey-bal1.local
WARNING Disconnected from API
INFO neey-bal1.local: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Successfully connected to neey-bal1.local
[15:49:46][D][api:102]: Accepted ::FFFF:192.168.1.39
[15:49:46][D][api:102]: Accepted ::FFFF:192.168.1.39
[15:49:46][W][api.connection:080]: Home Assistant 2022.7.7 (::FFFF:192.168.1.39): Connection closed
[15:49:46][W][api.connection:080]: Home Assistant 2022.7.7 (::FFFF:192.168.1.39): Connection closed
[15:49:48][W][heltec_balancer_ble:247]: [3c:a5:51:89:40:ff] Not connected
INFO neey-bal1.local: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for neey-bal1.local
WARNING Disconnected from API
INFO neey-bal1.local: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Successfully connected to neey-bal1.local
[15:50:02][I][heltec_balancer_ble:171]: Notification: 55.AA.11.01.02.00.2C.01.00.14.22.56.40.7E.F3.55.40.C8.E4.55.40.9E.D3.55.40.75.E8.55.40.13.FC.55.40.6E.03.56.40.6B.B7.55.40.14.22.56.40.2D.1D.56.40.DA.20.56.40.7F.19.56.40.A0.1F.56.40.DA.20.56.40.66.1E.56.40.66.1E.56.40.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.35.93.24.3E.68.94.26.3E.3D.25.1B.3E.90.8E.1B.3E.B3.F [... cut ...] D.1D.56.40.DA.20.56.40.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.35.93.24.3E.68.94.26.3E.3D.25.1B.3E.90.8E.1B.3E.B3.F3.23.3E.2E.91.25.3E.C6.1B.1A.3E.4A.7C.1C.3E.6F.1B.1A.3E.C2.43.1B.3E.85.1E.18.3E.4B.27.19.3E.5
[15:50:07][I][heltec_balancer_ble:171]: Notification: 00.00.00.00.00.00.00.00.00.00.00.00.00.C4.9C.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.C3.FF (59)

cristi2005, 2022-07-31:
Until now I didn't understand any of these codes, but if you say it's ok... I believe you :)))

syssi, 2022-07-31:
Please compile & flash again. :-)

cristi2005, 2022-07-31:
[16:11:13][I][heltec_balancer_ble:323]: Cell info frame (300 bytes):
[16:11:13][D][heltec_balancer_ble:324]:   55.AA.11.01.02.00.2C.01.78.C8.E4.55.40.00.C0.55.40.49.B1.55.40.71.9C.55.40.BD.B3.55.40.43.CC.55.40.D8.D4.55.40.CB.7D.55.40.3B.E7.55.40.C8.E4.55.40.02.E6.55.40.8D.E3.55.40.02.E6.55.40.8D.E3.55.40.8D.E3.55.40.02.E6.55.40.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.35.93.24.3E.68.94.26.3E.3D.25.1B.3E.90.8E.1B.3E.B3.F3.23.3E.2E.91.25.3E.C6.1B.1A.3E.4A.7C.1C.3E.6F.1B.1A.3E.C2.43.1B.3E.85.1E.18.3E.4B.27.19.3E.5E.DF.18.3E.D
[16:11:13][D][heltec_balancer_ble:325]:   3E.E6.D4.18.3E.0C.FE.18.3E.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.DD.CE.55.42.DD.CE.55.40.00.E0.D2.3B.08.07.0F.05.FD.EF.80.40.0B.D7.49.42.0B.D7.49.42.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.B5.A1.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.93.FF (140)
[16:11:18][I][heltec_balancer_ble:323]: Cell info frame (300 bytes):
[16:11:19][D][heltec_balancer_ble:324]:   55.AA.11.01.02.00.2C.01.82.8D.E3.55.40.3A.C1.55.40.0E.B0.55.40.38.9B.55.40.BD.B3.55.40.43.CC.55.40.9E.D3.55.40.79.81.55.40.3B.E7.55.40.8D.E3.55.40.3B.E7.55.40.53.E2.55.40.02.E6.55.40.8D.E3.55.40.53.E2.55.40.02.E6.55.40.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.35.93.24.3E.68.94.26.3E.3D.25.1B.3E.90.8E.1B.3E.B3.F3.23.3E.2E.91.25.3E.C6.1B.1A.3E.4A.7C.1C.3E.6F.1B.1A.3E.C2.43.1B.3E.85.1E.18.3E.4B.27.19.3E.5E.DF.18.3E.D
[16:11:19][D][heltec_balancer_ble:325]:   3E.E6.D4.18.3E.0C.FE.18.3E.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.B6.CE.55.42.B6.CE.55.40.00.84.CB.3B.0A.07.0F.05.33.F8.86.40.F6.28.4B.42.F6.28.4B.42.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.BA.A1.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.3C.FF (140)
[16:11:24][I][heltec_balancer_ble:323]: Cell info frame (300 bytes):
[16:11:24][D][heltec_balancer_ble:324]:   55.AA.11.01.02.00.2C.01.8C.8D.E3.55.40.74.C2.55.40.0E.B0.55.40.FE.99.55.40.F6.B4.55.40.43.CC.55.40.D8.D4.55.40.3F.80.55.40.02.E6.55.40.C8.E4.55.40.C8.E4.55.40.1B.E1.55.40.75.E8.55.40.02.E6.55.40.02.E6.55.40.C8.E4.55.40.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.35.93.24.3E.68.94.26.3E.3D.25.1B.3E.90.8E.1B.3E.B3.F3.23.3E.2E.91.25.3E.C6.1B.1A.3E.4A.7C.1C.3E.6F.1B.1A.3E.C2.43.1B.3E.85.1E.18.3E.4B.27.19.3E.5E.DF.18.3E.D
[16:11:24][D][heltec_balancer_ble:325]:   3E.E6.D4.18.3E.0C.FE.18.3E.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.04.CF.55.42.04.CF.55.40.00.6C.D0.3B.0C.07.0F.05.FD.4F.73.40.3E.0A.4C.42.3E.0A.4C.42.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.BF.A1.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.2C.FF (140)
[16:11:29][I][heltec_balancer_ble:323]: Cell info frame (300 bytes):
[16:11:29][D][heltec_balancer_ble:324]:   55.AA.11.01.02.00.2C.01.96.C8.E4.55.40.00.C0.55.40.49.B1.55.40.FE.99.55.40.83.B2.55.40.09.CB.55.40.9E.D3.55.40.3F.80.55.40.3B.E7.55.40.C8.E4.55.40.02.E6.55.40.A6.DE.55.40.8D.E3.55.40.02.E6.55.40.1B.E1.55.40.8D.E3.55.40.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.35.93.24.3E.68.94.26.3E.3D.25.1B.3E.90.8E.1B.3E.B3.F3.23.3E.2E.91.25.3E.C6.1B.1A.3E.4A.7C.1C.3E.6F.1B.1A.3E.C2.43.1B.3E.85.1E.18.3E.4B.27.19.3E.5E.DF.18.3E.D
[16:11:29][D][heltec_balancer_ble:325]:   3E.E6.D4.18.3E.0C.FE.18.3E.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.06.CE.55.42.06.CE.55.40.00.F8.CD.3B.08.07.0F.05.68.84.85.C0.E2.7A.4C.42.E2.7A.4C.42.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.C4.A1.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.1C.FF (140)
INFO neey-bal1.local: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for neey-bal1.local
WARNING Disconnected from API
INFO neey-bal1.local: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Successfully connected to neey-bal1.local
[16:11:44][D][ble_client:047]: Found device at MAC address [3C:A5:51:89:40:FF]
[16:11:44][I][ble_client:085]: Attempting BLE connection to 3c:a5:51:89:40:ff
[16:11:44][W][heltec_balancer_ble:246]: [3c:a5:51:89:40:ff] Not connected
[16:11:46][I][ble_client:170]: Service UUID: 0x1800
[16:11:46][I][ble_client:171]:   start_handle: 0x1  end_handle: 0x7
[16:11:46][I][ble_client:383]:  characteristic 0x2A00, handle 0x3, properties 0x2
[16:11:46][I][ble_client:383]:  characteristic 0x2A01, handle 0x5, properties 0x2
[16:11:46][I][ble_client:383]:  characteristic 0x2A04, handle 0x7, properties 0x2
[16:11:46][I][ble_client:170]: Service UUID: 0xFFE0
[16:11:46][I][ble_client:171]:   start_handle: 0x8  end_handle: 0xffff
[16:11:46][I][ble_client:383]:  characteristic 0xFFE1, handle 0xa, properties 0x1c
[16:11:46][I][ble_client:383]:  characteristic 0xFFE2, handle 0xd, properties 0x1c
[... cut ...] .18.3E.4B.27.19.3E.5E.DF.18.3E.D
[16:12:31][D][heltec_balancer_ble:325]:   3E.E6.D4.18.3E.0C.FE.18.3E.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.CD.CB.55.42.CD.CB.55.40.00.6C.D0.3B.0C.07.0F.05.37.CF.89.40.F6.28.4B.42.F6.28.4B.42.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.01.A2.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.24.FF (140)

syssi, 2022-07-31:
Please compile & flash again. This time the cell info frame should be decoded and written into the log.

Do you know the meaning of these values:

```
[17:05:06][I][heltec_balancer_ble:426]:   Single num?!: 15
[17:05:07][I][heltec_balancer_ble:428]:   Isostasy: 5
[17:05:07][I][heltec_balancer_ble:430]:   Isostasy current?!: -4.082165
```

And could you verify the

```
[17:05:06][I][heltec_balancer_ble:417]:   Balacing Current: 0.008306 A
```

Balancing current is correct?

cristi2005, 2022-07-31:
[17:05:07][I][heltec_balancer_ble:430]:   Isostasy current?!: -4.082165

This is the balancing curent.

I sent another foto on email.

cristi2005, 2022-07-31:
[18:23:00][D][heltec_balancer_ble:324]:   55.AA.11.01.02.00.2C.01.67.BB.8D.55.40.0B.3E.55.40.43.19.55.40.96.15.55.40.F9.27.55.40.5F.60.55.40.58.7B.55.40.21.13.55.40.62.AC.55.40.28.AB.55.40.62.AC.55.40.B4.A8.55.40.EE.A9.55.40.9C.AD.55.40.B4.A8.55.40.28.AB.55.40.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.35.93.24.3E.68.94.26.3E.3D.25.1B.3E.90.8E.1B.3E.B3.F3.23.3E.2E.91.25.3E.C6.1B.1A.3E.4A.7C.1C.3E.6F.1B.1A.3E.C2.43.1B.3E.85.1E.18.3E.4B (150)
[18:23:00][D][heltec_balancer_ble:325]:   27.19.3E.5E.DF.18.3E.D0.EB.1A.3E.E6.D4.18.3E.0C.FE.18.3E.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.98.76.55.42.98.76.55.40.00.7B.1A.3C.0D.07.0F.05.B7.16.80.40.C3.F5.48.42.C3.F5.48.42.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.96.C0.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.D8.FF (150)
[18:23:00][D][heltec_balancer_ble:349]:   Function: 1
[18:23:00][D][heltec_balancer_ble:351]:   Command: 2
[18:23:00][D][heltec_balancer_ble:353]:   Length: 300
[18:23:00][I][heltec_balancer_ble:355]:   Serial number: 103
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 1: 3.336776 V
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 2: 3.331912 V
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 3: 3.329667 V
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 4: 3.329443 V
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 5: 3.330565 V
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 6: 3.334007 V
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 7: 3.335653 V
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 8: 3.329293 V
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 9: 3.338646 V
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 10: 3.338572 V
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 11: 3.338646 V
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 12: 3.338422 V
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 13: 3.338497 V
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 14: 3.338721 V
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 15: 3.338422 V
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 16: 3.338572 V
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 17: 0.000000 V
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 18: 0.000000 V
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 19: 0.000000 V
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 20: 0.000000 V
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 21: 0.000000 V
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 22: 0.000000 V
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 23: 0.000000 V
[18:23:00][I][heltec_balancer_ble:381]:   Cell voltage 24: 0.000000 V
[18:23:00][I][heltec_balancer_ble:408]:   Con. wire resistance 1: 0.160718 Ohm
[18:23:00][I][heltec_balancer_ble:408]:   Con. wire resistance 2: 0.162675 Ohm
[18:23:00][I][heltec_balancer_ble:408]:   Con. wire resistance 3: 0.151509 Ohm
[18:23:00][I][heltec_balancer_ble:408]:   Con. wire resistance 4: 0.151911 Ohm
[18:23:00][I][heltec_balancer_ble:408]:   Con. wire resistance 5: 0.160109 Ohm
[18:23:00][I][heltec_balancer_ble:408]:   Con. wire resistance 6: 0.161687 Ohm
[18:23:00][I][heltec_balancer_ble:408]:   Con. wire resistance 7: 0.150497 Ohm
[18:23:00][I][heltec_balancer_ble:408]:   Con. wire resistance 8: 0.152818 Ohm
[18:23:00][I][heltec_balancer_ble:408]:   Con. wire resistance 9: 0.150495 Ohm
[18:23:00][I][heltec_balancer_ble:408]:   Con. wire resistance 10: 0.151626 Ohm
[18:23:00][I][heltec_balancer_ble:408]:   Con. wire resistance 11: 0.148554 Ohm
[18:23:00][I][heltec_balancer_ble:408]:   Con. wire resistance 12: 0.149564 Ohm
[18:23:00][I][heltec_balancer_ble:408]:   Con. wire resistance 13: 0.149290 Ohm
[18:23:00][I][heltec_balancer_ble:408]:   Con. wire resistance 14: 0.151290 Ohm
[18:23:00][I][heltec_balancer_ble:408]:   Con. wire resistance 15: 0.149250 Ohm
[18:23:00][I][heltec_balancer_ble:408]:   Con. wire resistance 16: 0.149407 Ohm
[18:23:00][I][heltec_balancer_ble:408]:   Con. wire resistance 17: 0.000000 Ohm
[18:23:00][I][heltec_balancer_ble:408]:   Con. wire resistance 18: 0.000000 Ohm
[18:23:00][I][heltec_balancer_ble:408]:   Con. wire resistance 19: 0.000000 Ohm
[18:23:00][I][heltec_balancer_ble:408]:   Con. wire resistance 20: 0.000000 Ohm
[18:23:00][I][heltec_balancer_ble:408]:   Con. wire resistance 21: 0.000000 Ohm
[18:23:00][I][heltec_balancer_ble:408]:   Con. wire resistance 22: 0.000000 Ohm
[18:23:00][I][heltec_balancer_ble:408]:   Con. wire resistance 23: 0.000000 Ohm
[18:23:01][I][heltec_balancer_ble:408]:   Con. wire resistance 24: 0.000000 Ohm
[18:23:01][I][heltec_balancer_ble:411]:   Total voltage: 53.365814 V
[18:23:01][I][heltec_balancer_ble:414]:   Single voltage: 3.335363 V
[18:23:01][I][heltec_balancer_ble:417]:   Balacing Current: 0.009429 A
[18:23:01][I][heltec_balancer_ble:420]:   Maximum voltage cell: 14
[18:23:01][I][heltec_balancer_ble:423]:   Minimum voltage cell: 8
[18:23:01][I][heltec_balancer_ble:426]:   Single num?!: 15
[18:23:01][I][heltec_balancer_ble:428]:   Isostasy: 5
[18:23:01][I][heltec_balancer_ble:430]:   Isostasy current?!: 4.002773
[18:23:01][I][heltec_balancer_ble:432]:   Temperature 1: 50.240002 °C
[18:23:01][I][heltec_balancer_ble:434]:   Temperature 2: 50.240002 °C
[... cut ...]
[18:23:11][I][heltec_balancer_ble:420]:   Maximum voltage cell: 14
[18:23:11][I][heltec_balancer_ble:423]:   Minimum voltage cell: 4
[18:23:11][I][heltec_balancer_ble:426]:   Single num?!: 15
[18:23:11][I][heltec_balancer_ble:428]:   Isostasy: 5
[18:23:11][I][heltec_balancer_ble:430]:   Isostasy current?!: -4.087837
[18:23:11][I][heltec_balancer_ble:432]:   Temperature 1: 50.240002 °C
[18:23:11][I][heltec_balancer_ble:434]:   Temperature 2: 50.240002 °C
[18:23:15][I][heltec_balancer_ble:323]: Cell info frame (300 bytes):

syssi, 2022-07-31:
I've added some `button` entities. Please extend your configuration yaml by:

```
button:
  - platform: heltec_balancer_ble
    retrieve_settings:
      name: "${name} retrieve settings"
    retrieve_device_info:
      name: "${name} retrieve device info"
    retrieve_factory_defaults:
      name: "${name} retrieve factory defaults"
```

And clean build the project by:

```
esphome clean neey-bal1.yaml
esphome run neey-bal1.yaml
```

Do you use Home Assistant? You should see the button entities here:

[image]

Please pree each button once and provide the output / log messages. Each button requests some data (`device info`, `settings`, `factory defualts`) from the balancer. The balancer should respond with some data. The `device info` and `settings` frame can be decoded already. The response of the `factory defaults` frame isn't known yet.

[66 more replies]

commit 854196d1 (2022-08-14): Add Heltec/NEEY 4A balancer support (Closes: #109) (#145)
closed 2022-08-14 by syssi