syssi/esphome-jk-bms #111: ESP-32 Bluetooth connection JK-B2A24S20P

s1lvi0, 2022-05-14:
Hi, I tried to connect the bms through bluetooth with an esp32 mcu node but the board restarts at some point with error:
 `CORRUPT HEAP: multi_heap.c: 187 detected in 0x3fff4a14`
What could be the problem?

This is my esphome configuration:
```
substitutions:
  name: electric-kart
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: c8:47:8c:e2:95:b2
  # Defaults to "JK02". Please use "JK04" if you have some old JK-BMS version (f.e. JK-B2A16S hw 3.0, sw. 3.3.0)
  protocol_version: JK02

esphome:
  name: ${name}

esp32:
  board: nodemcu-32s
  framework:
    type: arduino

# Enable logging
logger:
  level: VERY_VERBOSE
  logs:
    scheduler: DEBUG
    component: DEBUG
    esp32_ble_tracker: DEBUG
    sensor: DEBUG

# Enable Home Assistant API
api:

external_components:
  - source: ${external_components_source}
    refresh: 0s

ota:
  password: "69ca71c5c1e5959c66b9f4c61caa0f69"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Electric-Kart Fallback Hotspot"
    password: "sfDGMvdU652h"

captive_portal:
    

esp32_ble_tracker:

ble_client:
  - mac_address: ${mac_address}
    id: client0

jk_bms_ble:
  - ble_client_id: client0
    throttle: 5s
    id: bms0
    protocol_version: ${protocol_version}

binary_sensor:
  - platform: jk_bms_ble
    balancing:
      name: "${name} balancing"
    charging:
      name: "${name} charging"
[... 666 lines cut ...]
[VV][esp-idf:000]: Init static rx buffer num: 16
[VV][esp-idf:000]: 

[VV][esp-idf:000]: I (2008) wifi:
[VV][esp-idf:000]: Init dynamic rx buffer num: 32
[VV][esp-idf:000]: 

[VV][esp-idf:000]: I (2032) wifi:
[VV][esp-idf:000]: mode : sta (fc:f5:c4:00:f6:34)
[VV][esp-idf:000]: 

[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
[V][wifi_esp32:460]: Event: WiFi STA start
[VV][esp-idf:000]: I (2055) wifi:
[VV][esp-idf:000]: Set ps type: 1

[VV][esp-idf:000]: 

[D][wifi:373]: Starting scan...
[VV][esp-idf:000]: W (2065) wifi:
[VV][esp-idf:000]: Error! Should use default active scan time parameter for WiFi scan when Bluetooth is enabled!!!!!!

[VV][esp-idf:000]: 
```

syssi, 2022-05-14:
Please don't use the Arduino framework because of this issue. If you stick to ESP-IDF the device won't crash.

s1lvi0, 2022-05-14:
Ok thanks, my mistake. 

The Total Voltage value display around 13V insted of around 80V, is that normal?

syssi, 2022-05-14:
No. What does the android app show? Did you set the cell count setting correctly?

syssi, 2022-05-14:
Could you provide a log with log level `VERY_VERBOSE` enabled? I would like to compare the raw frame with the sensor value.

s1lvi0, 2022-05-14:
On the app the voltage is correct and the cells number is correct (24 cells).

<details>
  <summary>VERBOSE LOG</summary>

```
INFO Starting log output from e-kart.local using esphome API
INFO Successfully connected to e-kart.local
[21:51:27][VV][api.service:373]: on_device_info_request: DeviceInfoRequest {}
[21:51:27][VV][api.service:049]: send_device_info_response: DeviceInfoResponse {
  uses_password: NO
  name: 'e-kart'
  mac_address: '9C:9C:1F:C7:20:60'
  esphome_version: '2022.4.0'
  compilation_time: 'May 14 2022, 21:50:34'
  model: 'nodemcu-32s'
  has_deep_sleep: NO
  project_name: ''
  project_version: ''
  webserver_port: 0
}
[21:51:27][I][app:102]: ESPHome version 2022.4.0 compiled on May 14 2022, 21:50:34
[21:51:27][VV][api.service:382]: on_list_entities_request: ListEntitiesRequest {}
[21:51:27][C][logger:233]: Logger:
[21:51:27][C][logger:234]:   Level: VERY_VERBOSE
[21:51:27][C][logger:235]:   Log Baud Rate: 115200
[21:51:27][C][logger:236]:   Hardware UART: UART0
[21:51:27][VV][api.service:062]: send_list_entities_binary_sensor_response: ListEntitiesBinarySensorResponse {
  object_id: 'e-kart_charging'
  key: 659392089
  name: 'e-kart charging'
  unique_id: 'e-kartbinary_sensore-kart_charging'
  device_class: ''
  is_status_binary_sensor: NO
  disabled_by_default: NO
  icon: 'mdi:battery-charging'
  entity_category: ENTITY_CATEGORY_NONE
}
[21:51:27][VV][api.service:062]: send_list_entities_binary_sensor_response: ListEntitiesBinarySensorResponse {
  object_id: 'e-kart_discharging'
  key: 757500221
  name: 'e-kart discharging'
  unique_id: 'e-kartbinary_sensore-kart_discharging'
  device_class: ''
  is_status_binary_sensor: NO
  disabled_by_default: NO
  icon: 'mdi:power-plug'
  entity_category: ENTITY_CATEGORY_NONE
}
[21:51:27][C][jk_bms_ble:049]: JkBmsBle
[21:51:27][C][jk_bms_ble:050]: Minimum Cell Voltage 'e-kart min cell voltage'
[21:51:27][C][jk_bms_ble:050]:   Device Class: 'voltage'
[21:51:27][C][jk_bms_ble:050]:   State Class: 'measurement'
[21:51:27][C][jk_bms_ble:050]:   Unit of Measurement: 'V'
[21:51:27][C][jk_bms_ble:050]:   Accuracy Decimals: 3
[21:51:27][C][jk_bms_ble:051]: Maximum Cell Voltage 'e-kart max cell voltage'
[21:51:27][C][jk_bms_ble:051]:   Device Class: 'voltage'
[21:51:27][C][jk_bms_ble:051]:   State Class: 'measurement'
[21:51:27][C][jk_bms_ble:051]:   Unit of Measurement: 'V'
[21:51:27][C][jk_bms_ble:051]:   Accuracy Decimals: 3
[21:51:27][C][jk_bms_ble:052]: Minimum Voltage Cell 'e-kart min voltage cell'
[21:51:27][C][jk_bms_ble:052]:   State Class: 'measurement'
[21:51:27][C][jk_bms_ble:052]:   Unit of Measurement: ''
[21:51:27][C][jk_bms_ble:052]:   Accuracy Decimals: 0
[21:51:27][C][jk_bms_ble:052]:   Icon: 'mdi:battery-minus-outline'
[21:51:27][C][jk_bms_ble:053]: Maximum Voltage Cell 'e-kart max voltage cell'
[21:51:27][C][jk_bms_ble:053]:   State Class: 'measurement'
[21:51:27][C][jk_bms_ble:053]:   Unit of Measurement: ''
[21:51:27][C][jk_bms_ble:053]:   Accuracy Decimals: 0
[21:51:27][C][jk_bms_ble:054]: Delta Cell Voltage 'e-kart delta cell voltage'
[... 1859 lines cut ...]
[21:51:39][V][sensor:074]: 'e-kart cell voltage 13': Received new state 3.282000
[21:51:39][D][sensor:125]: 'e-kart cell voltage 13': Sending state 3.28200 V with 3 decimals of accuracy
[21:51:39][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 647864942
  state: 3.282
  missing_state: NO
}
[21:51:39][V][sensor:074]: 'e-kart cell voltage 14': Received new state 3.284000
[21:51:39][D][sensor:125]: 'e-kart cell voltage 14': Sending state 3.28400 V with 3 decimals of accuracy
[21:51:39][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 647864937
  state: 3.284
  missing_state: NO
}
[21:51:39][V][sensor:074]: 'e-kart cell voltage 15': Received new state 3.280000
[21:51:39][D][sensor:125]: 'e-kart cell voltage 15': Sending state 3.28000 V with 3 decimals of accuracy
[21:51:39][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 647864936
  state: 3.28
  missing_state: NO
}
[21:51:39][V][sensor:074]: 'e-kart cell voltage 16': Received new state 3.280000

```
</details>

syssi, 2022-05-15:
Some most important bytes from your log:

```
Device info frame:

55.AA.EB.90.03.FC.4A.4B.2D.42.32.41.32.34.53.32.30.50.00.00.00.00.31.30.2E.58.47.00.00.00.31.30.2E.30.37.00.00.00.2C.1B.04.00.01.00.00.00.4A.4B.2D.42.32.41.32.34.53.32.30.50.00.00.00.00.31.32.33.34.00.00.00.00.00.00.00.00.00.00.00.00.32.32.30.35.31.31.00.00.32.30.34.31.38.30.33.30.32.38.00.30.30.30.30.00.49.6E.70.75.74.20.55.73.65.72.64.61.74.61.00.00.31.32.33.34.35.36.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FC

Settings frame:

55.AA.EB.90.01.FC.58.02.00.00.28.0A.00.00.78.0A.00.00.DE.0D.00.00.48.0D.00.00.05.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.F6.09.00.00.50.C3.00.00.1E.00.00.00.3C.00.00.00.20.BF.02.00.2C.01.00.00.3C.00.00.00.3C.00.00.00.BC.02.00.00.58.02.00.00.F4.01.00.00.58.02.00.00.26.02.00.00.00.00.00.00.0A.00.00.00.84.03.00.00.BC.02.00.00.18.00.00.00.01.00.00.00.00.00.00.00.00.00.00.00.50.C3.00.00.E8.03.00.00.B8.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.74

Cell info frame:

55.AA.EB.90.02.FC.D4.0C.D0.0C.CF.0C.D0.0C.D0.0C.CF.0C.CF.0C.D1.0C.D0.0C.CF.0C.D0.0C.CF.0C.D2.0C.D4.0C.CF.0C.CF.0C.D0.0C.CF.0C.D1.0C.D2.0C.D2.0C.D2.0C.D5.0C.D1.0C.FF.FF.FF.00.D1.0C.05.00.00.03.64.00.6C.00.68.00.63.00.5F.00.5D.00.5B.00.5E.00.7F.00.79.00.6B.00.6F.00.6F.00.68.00.64.00.60.00.5B.00.58.00.58.00.5B.00.60.00.65.00.68.00.61.00.00.00.00.00.00.00.8F.33.01.00.00.00.00.00.00.00.00.00.CF.00.CF.00.EA.00.00.00.00.00.00.55.34.A7.00.00.50.C3.00.00.00.00.00.00.F9.34.00.00.64.00.40.05.1C.1C.04.00.01.00.85.06.00.00.00.00.00.00.00.00.00.00.00.00.07.00.01.00.00.00.FA.03.00.00.00.00.00.D2.3F.40.00.00.00.00.E2.04.00.00.00.00.00.01.00.03.00.00.1A.19.29.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.43
```

syssi, 2022-05-15:
I've fixed the total voltage. Compile & flash again.

closed 2022-05-15 by syssi

s1lvi0, 2022-05-15:
Now works perfect, thank you so much.

commit 12b3cf58 (2022-05-15): Fix total voltage (Fixes: #111)
commit c2b1f2d7 (2022-05-15): Fix total voltage (Fixes: #111)