syssi/esphome-jk-bms #115: Improve error bitmask of the JK02 protocol

syssi, 2022-05-15:
```
        0003d98c 4d 4f 53        ds         "MOS over temperature protection"
        0003d9ac 4d 4f 53        ds         "MOS over-temperature protection is released"

        0003d9d8 41 62 6e        ds         "Abnormal current sensor"
        0003d9f0 41 62 6e        ds         "Abnormal release of current sensor"
        
        0003da13 41 62 6e        ds         "Abnormal coprocessor communication"
        0003da36 41 62 6e        ds         "Abnormal cancellation of coprocessor communic
        
        0003da69 43 65 6c        ds         "Cell overcharge protection"
        0003da84 43 65 6c        ds         "Cell overcharge protection is released"
        
        0003daab 42 61 74        ds         "Battery overcharge protection"
        0003dac9 42 61 74        ds         "Battery overcharge protection is released"
        
        0003daf3 43 68 61        ds         "Charge overcurrent protection"
        0003db11 43 68 61        ds         "Charge overcurrent protection is released"
        
        0003db3b 43 68 61        ds         "Charge short circuit protection"
        0003db5b 43 68 61        ds         "Charge short circuit protection is released"
        
        0003db87 43 68 61        ds         "Charge over temperature protection"
        0003dbaa 43 68 61        ds         "Charge over temperature protection is released"

        0003dbd9 43 68 61        ds         "Charge low temperature protection"
        0003dbfb 43 68 61        ds         "Charge low temperature protection is released"
        
        0003dc29 43 65 6c        ds         "Cell undervoltage protection"
        0003dc46 43 65 6c        ds         "Cell undervoltage protection is released"
        
        0003dc6f 42 61 74        ds         "Battery undervoltage protection"
        0003dc8f 42 61 74        ds         "Battery undervoltage protection is released"
        
        0003dcbb 44 69 73        ds         "Discharge overcurrent protection"
        0003dcdc 44 69 73        ds         "Discharge overcurrent protection is released"
        
        0003dd09 44 69 73        ds         "Discharge short circuit protection"
        0003dd2c 44 69 73        ds         "Discharge short circuit protectionis released"
        
        0003dd5a 44 69 73        ds         "Discharge over temperature protection"
        0003dd80 44 69 73        ds         "Discharge over-temperature protection is released"

        0003ddb2 52 65 73        ds         "Reset Watch-Dog"
        0003ddc2 43 65 6c        ds         "Cell %1 over charge protection"
        0003dde1 43 65 6c        ds         "Cell %1 over discharge protection"
```

EasilyBoredEngineer, 2022-05-16:
Hi @syssi - interested to know where you obtained those bitmasks?

Can confirm that I have installed your mods and on my JK 8S bitmask now reads '0'
[image]

syssi, 2022-05-16:
I've extracted these strings from the android app.

Jalle19, 2023-01-16:
Not really, I just stumbled upon "current sensor anomaly" when connecting a charger to the battery while discharging. Learned that the ESP can be forced to disconnect from Bluetooth to allow the official app to find the BMS 👍

syssi, 2024-07-07:
Does the status bits change? https://github.com/jblance/mpp-solar/issues/460#issuecomment-1973313087