syssi/esphome-jk-bms #118: Decode display port traffic

syssi, 2022-05-20:
[ScreenCapture9600.txt](https://github.com/syssi/esphome-jk-bms/files/8738581/ScreenCapture9600.txt)

TODO: Ideantify SOF and EOF.

syssi, 2022-05-20:
@EasilyBoredEngineer Could you help to identify the start byte and end byte of a display frame by timing? It's hard to guess. :-)

23900, 2024-01-27:
display port protocol:
http://www.jk-bms.com/en/Upload/2023-12-05/1556482191.pdf

syssi, 2024-01-27:
```
# Offset  Content                                 Type     Unit    Remarks
#  0      Battery voltage                        UINT16    10mV
#  2      Battery current                         INT16    0.1A
#  4      Reserved                               UINT16    -
#  6      State of charge                        UINT16    %
#  8      Maximum pressure difference            UINT16    mV
# 10      MOSFET temperature                      INT16    ℃
# 12      Battery temperature                     INT16    ℃
# 14      System warning                         UINT16    -       0: no alarm, 1: alarm
# 16      Cell average voltage                   UINT16    mV
# 18      Balanced switch state                  UINT16    -       0: close, 1: open
# 20      Charging MOS status                    UINT16    -       0: close, 1: open
# 22      Discharging MOS status                 UINT16    -       0: close, 1: open
# 24      Cell voltages[24]                   24*UINT16    mV
# 72      Single unit over-discharge alarm       UINT16    -       0: no alarm, 1: alarm
# 74      Cell overcharge alarm                  UINT16    -       0: no alarm, 1: alarm
# 76      Overcurrent alarm                      UINT16    -       0: no alarm, 1: alarm
# 78      MOS over temperature alarm             UINT16    -       0: no alarm, 1: alarm
# 80      Battery over temperature alarm         UINT16    -       0: no alarm, 1: alarm
# 82      Short circuit alarm                    UINT16    -       0: no alarm, 1: alarm
# 84      Co-handling communication exceptions   UINT16    -       0: no alarm, 1: alarm
# 86      Balance line resistance is too large   UINT16    -       0: no alarm, 1: alarm
# 88      The number of strings does not match   UINT16    -       0: no alarm, 1: alarm
```

syssi, 2024-01-27:
```
# Frame structure

0xA5 0x5A  # Frame header
0x5D       # Data length (fixed / always 93 bytes)
0x82       # Frame type (fixed, it's always the status frame)
0x10 0x00  # Address
... 90 bytes data ...
+ Checksum?
```

syssi, 2024-01-27:
I would be happy about a traffic capture. I could provide some instructions how to sniff the traffic between the BMS and the display using an ESP + RS485 module.

23900, 2024-01-28:
Unfortunately, the display port and CAN port are both one-way, which means it can only receive BMS data but cannot send control instructions.

23900, 2024-01-28:
If you remove the RS485 chip, connect RO to B and DI to A, the RS485 function of the display port becomes a 5V level uart port. A=uart tx B= uart RX.
[image]

blacklopo, 2024-02-23:
Today I tried to capture display data. According manual the protocol parameters are 2400bps, RS485, pins according https://github.com/syssi/esphome-jk-bms/blob/main/docs/display-port.md. It worked perfectly.

Captured data:
[EXPORT3.hex.zip](https://github.com/syssi/esphome-jk-bms/files/14389950/EXPORT3.hex.zip)

Decoded data:
b'\xa5Z' -23206
b']\x82' 23938
b'\x10\x00' 4096
b'\nw' 2679
b'\xff\xf8' -8
b'\x00\x02' 2
b'\x008' 56
b'\x00\x01' 1
b'\x00\x18' 24
b'\x00\x16' 22
b'\x00\x01' 1
b'\r\x15' 3349
b'\x00\x01' 1
b'\x00\x01' 1
b'\x00\x01' 1
b'\r\x15' 3349
b'\r\x15' 3349
b'\r\x15' 3349
b'\r\x15' 3349
b'\r\x15' 3349
b'\r\x15' 3349
b'\r\x15' 3349
b'\r\x16' 3350
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0
b'\x00\x00' 0

Villy-web, 2024-08-11:
> Today I tried to capture display data.

I noticed that there are several data blocks that start with "A5 5A". I counted 27 such blocks.

`A5 5A 03 80 01 00 A5 5A 5D 82 10 00 0A 72 00 00 00 03 00 63 00 01 00 1F 00 1D 00 01 0D 0F 00 01 00 01 00 01 0D 10 0D 0F 0D 0F 0D 0F 0D 0F 0D 0F 0D 10 0D 0F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 A5 5A 05 82 20 03 F8 00 A5 5A 05 82 20 13 07 E0 A5 5A 05 82 20 23 07 E0 A5 5A 05 82 20 33 07 E0 A5 5A 05 82 20 43 00 1F A5 5A 05 82 20 53 07 E0 A5 5A 05 82 20 63 07 E0 A5 5A 05 82 20 73 07 E0 A5 5A 05 82 20 83 00 00 A5 5A 05 82 20 93 00 00 A5 5A 05 82 20 A3 00 00 A5 5A 05 82 20 B3 00 00 A5 5A 05 82 20 C3 00 00 A5 5A 05 82 20 D3 00 00 A5 5A 05 82 20 E3 00 00 A5 5A 05 82 20 F3 00 00 A5 5A 05 82 21 03 00 00 A5 5A 05 82 21 13 00 00 A5 5A 05 82 21 23 00 00 A5 5A 05 82 21 33 00 00 A5 5A 05 82 21 43 00 00 A5 5A 05 82 21 53 00 00 A5 5A 05 82 21 63 00 00 A5 5A 05 82 21 73 00 00 A5 5A 05 82 21 83 00 00 A5 5A 2B 82 13 00 00 05 0A D7 00 00 00 00 00 00 00 00 00 00 0D 10 0D 10 0D 0F 00 08 00 00 00 34 00 1D 00 01 00 1F 00 01 00 01 00 01 00 01 `

The last 27th block partially repeats the data of the second block "...A5 5A 5D 82 10 00..."

syssi, 2024-08-11:
```
A5 5A 03 80 01 00 
A5 5A 5D 82 10 00 0A 72 00 00 00 03 00 63 00 01 00 1F 00 1D 00 01 0D 0F 00 01 00 01 00 01 0D 10 0D 0F 0D 0F 0D 0F 0D 0F 0D 0F 0D 10 0D 0F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
A5 5A 05 82 20 03 F8 00 
A5 5A 05 82 20 13 07 E0 
A5 5A 05 82 20 23 07 E0 
A5 5A 05 82 20 33 07 E0 
A5 5A 05 82 20 43 00 1F 
A5 5A 05 82 20 53 07 E0 
A5 5A 05 82 20 63 07 E0 
A5 5A 05 82 20 73 07 E0 
A5 5A 05 82 20 83 00 00 
A5 5A 05 82 20 93 00 00 
A5 5A 05 82 20 A3 00 00 
A5 5A 05 82 20 B3 00 00 
A5 5A 05 82 20 C3 00 00 
A5 5A 05 82 20 D3 00 00 
A5 5A 05 82 20 E3 00 00 
A5 5A 05 82 20 F3 00 00 
A5 5A 05 82 21 03 00 00 
A5 5A 05 82 21 13 00 00 
A5 5A 05 82 21 23 00 00 
A5 5A 05 82 21 33 00 00 
A5 5A 05 82 21 43 00 00 
A5 5A 05 82 21 53 00 00 
A5 5A 05 82 21 63 00 00 
A5 5A 05 82 21 73 00 00 
A5 5A 05 82 21 83 00 00 
A5 5A 2B 82 13 00 00 05 0A D7 00 00 00 00 00 00 00 00 00 00 0D 10 0D 10 0D 0F 00 08 00 00 00 34 00 1D 00 01 00 1F 00 01 00 01 00 01 00 01 
```

syssi, 2024-08-11:
```
A5 5A 5D 82 10 00 0A 72 00 00 00 03 00 63 00 01 00 1F 00 1D 00 01 0D 0F 00 01 00 01 00 01 0D 10 0D 0F 0D 0F 0D 0F 0D 0F 0D 0F 0D 10 0D 0F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
       |  1  2  3  4  5  6  7  8  9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 40 41 42 43 44 45 46 47 48 49 50 51 52 53 54 55 56 57 58 59 60 61 62 63 64 65 66 67 68 69 70 71 72 73 74 75 76 77 78 79 80 81 82 83 84 85 86 87 88 89 90 91 92 93
       data_len: 93 bytes

A5 5A 05 82 20 33 07 E0 
       |  1  2  3  4  5
       data_len: 5
      
A5 5A 2B 82 13 00 00 05 0A D7 00 00 00 00 00 00 00 00 00 00 0D 10 0D 10 0D 0F 00 08 00 00 00 34 00 1D 00 01 00 1F 00 01 00 01 00 01 00 01 
       |  1  2  3  4  5  6 7  8   9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 40 41 42 43
       data_len: 43 bytes
```

Villy-web, 2024-08-11:
> A5 5A 2B 82 13 00 00 05 0A D7 00 00 00 00 00 00 00 00 00 00 0D 10 0D 10 0D 0F 00 08 00 00 00 34 00 1D 00 01 00 1F 00 01 00 01 00 01 00 01
>        |  1  2  3  4  5  6 7  8   9 10 11 12 13 14 15 16 17 [...]

Is there a decryption of this block? I assume that with an offset of 0x02(byte 6,7 UINT 0.1Ah), Remain Capacity is located here.

syssi, 2024-08-11:
Hmm.. I don't know. Do you have a JK-BMS and a display attached to capture some day and provide the screen content as reference? There is just the description of the address 0x1000 frame at the PDF document above. A description of address 0x1300 is missing.

@blacklopo Do you remember how many cell where attached to your test setup above?

michael9dk, 2024-08-14:
> ```
> A5 5A 5D 82 10 00 0A 72 00 00 00 03 00 63 00 01 00 1F 00 1D 00 01 0D 0F 00 01 00 01 00 01 0D 10 0D 0F 0D 0F 0D 0F 0D 0F 0D 0F 0D 10 0D 0F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [...]

Just wondering if those shorter datasets are a subset of the full 90 bytes, to support other minimalistic displays.
I havent digged into it, but is there 5 sequential usefull bytes at it's corresponding offset.

I have a JK BMS and Display. Can provide a capture of the oneway simplex RS232 signal (inside the Display), when I get it all connected.

syssi, 2024-08-14:
I've implemented a simple decoder here: https://github.com/syssi/esphome-jk-bms/pull/572

Just to get an better idea of the payloads.

syssi, 2024-08-15:
@michael9dk Cool! I would be happy about some more traffic examples including photos what the display is showing.

michael9dk, 2024-11-21:
> ```
> # Offset  Content                                 Type     Unit    Remarks
> #  0      Battery voltage                        UINT16    10mV
> #  2      Battery current [...]

# 80 includes alarm for Charge under-temperature protection. It triggers wether charging or not.
[image]

syssi, 2024-11-21:
Thanks for your feedback! Do you use the decoder and could you provide some raw traffic?

michael9dk, 2024-11-21:
I'm not using esphome/decoder. 
Finally got the battery and bms mounted. Will make a raw capture when I switch from the 2" LCD Display to a custom board.

syssi, 2024-12-29:
Resolved by https://github.com/syssi/esphome-jk-bms/pull/572

closed 2024-12-29 by syssi

Villy-web, 2025-05-22:
JK BMS Inverter V19 LCD Protocol:

A5 5A 03 80 01 40
A5 5A 5D 82 10 00 0A 65 00 00 00 03 00 64 00 05 00 00 00 00 00 01 0C FF 00 01 00 01 00 01 0D 00 0D 01 0D 00 0C FC 0D 01 0C FD 0C FE 0C FE 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
A5 5A 05 82 20 03 07 E0
A5 5A 05 82 20 13 00 1F
A5 5A 05 82 20 23 07 E0
A5 5A 05 82 20 33 F8 00
A5 5A 05 82 20 43 07 E0
A5 5A 05 82 20 53 07 E0
A5 5A 05 82 20 63 07 E0
A5 5A 05 82 20 73 07 E0
A5 5A 05 82 20 83 00 00
A5 5A 05 82 20 93 00 00
A5 5A 05 82 20 A3 00 00
A5 5A 05 82 20 B3 00 00
A5 5A 05 82 20 C3 00 00
A5 5A 05 82 20 D3 00 00
A5 5A 05 82 20 E3 00 00
A5 5A 05 82 20 F3 00 00
A5 5A 05 82 21 03 00 00
A5 5A 05 82 21 13 00 00
A5 5A 05 82 21 23 00 00
A5 5A 05 82 21 33 00 00
A5 5A 05 82 21 43 00 00
A5 5A 05 82 21 53 00 00
A5 5A 05 82 21 63 00 00
A5 5A 05 82 21 73 00 00
A5 5A 05 82 21 83 00 00
A5 5A 05 82 22 03 07 E0
A5 5A 05 82 22 13 07 E0
A5 5A 2B 82 13 00 00 05 01 8F 00 00 00 00 00 00 00 00 00 00 0D 01 0C FC 0C FF 00 48 00 00 00 33 00 1E 00 05 00 17 00 01 00 01 00 01 00 01

Villy-web, 2025-05-24:
**JK BMS Inverter V19**. # 15 - **JK BMS PBxx SERIES LCD Protocol V1.0**:
00 A9 4B DB BE FF EB 2B FF FF FF 37 FF D7 FD DF 7F 38 79 EF D3 FF FD FF FD E5 FD FF FF FF FD FF FD FF FB 4B B5 63 9F BB 57 F9 E5 F7 E5 F3 E5 FD E5 F3 E5 F1 E5 F3 E5 EF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF F9 FF 37 FD DF 7F F3 F1 FF FD FF D3 FF FF FF FD FF FD FF FF 4B B5 7D 9F CB F1 3F A9 4B DF 7E D9 F1 3F A9 4B DF 7E B9 F1 3F A9 4B DF 7E 99 0F 4B B5 7D 9F CB C5 3F A9 4B DF 7E 59 F1 3F A9 4B DF 7E 39 F1 3F A9 4B DF 7E 19 FF C1 4B B5 7D 9F CB FE FF 4B B5 7D 9F CB FE FF 4B B5 7D 9F CB F9 FF 4B B5 7D 9F CB FE FF 4B B5 7D 9F CB FC FF 4B B5 7D 9F CB FC FF 4B B5 7D 9F CB F8 FF 4B B5 7D 9F CB FF 4B B5 7D 9F CB F1 3F A9 4B DF 7A D9 F1 3F A9 4B FF 7F 2D D5 F2 FF BF 
00 A9 4B DF EA FE FF FF 4B B5 7D 9F 4F FF DF 4B B5 7D 9F CF FF DF 4B B5 7D 9F 4F FF DF 4B B5 7D 9F CF DF DF 4B B5 7D 9F 4F DF DF 4B B5 7D 9F CF FF DF 4B B5 7D 9F 4F FF DF 4B B5 7D 9F CF EF DF 4B B5 7D 9F 4F EF DF 4B B5 7D 9F CF EF DF 4B B5 7D 9F 4F EF DF 4B B5 7D 9F CF DE DF 4B B5 7D 9F 4F DE DF 4B B5 7D 9F CF FF DF 4B B5 7D 9F 4F FF DF 4B B5 7A 9F 9F A9 CB FD FF F5 FF FD 4B B5 7A 9F 9F A9 CB FD FD F5 FF FF 4B B5 7A 9F 9F A9 CB FD FB F5 FF FF 4B B5 7A 9F 9F A9 CB FD F9 F5 FF FF 4B B5 7A 9F 9F A9 CB FD F7 F5 FF FF 4B B5 7A 9F 9F A9 CB FD F5 F5 FF FF 4B B5 7A 9F 9F A9 CB FD F3 F5 FF FF 4B B5 7A 9F 9F A9 CB FD F1 F5 FF FF 4B B5 7A 9F 9F A9 CB FD EF F5 FF FF 4B B5 7A 9F 9F A9 CB FD ED F5 FF FF 4B B5 7A 9F 9F A9 CB FD EB F5 FF FF 4B B5 7A 9F 9F A9 CB FD E9 F5 FF FF 4B B5 7A 9F 9F A9 CB FD E7 F5 FF FF 4B B5 7A 9F 9F A9 CB FD E5 F5 FF FF 4B B5 7A 9F 9F A9 CB FD E3 F5 FF FF 4B B5 7A 9F 9F A9 CB FD E1 F5 FF FF 
00 A9 4B DF A2 BF FF FF 4B B5 7D 9F EE FF FF 4B B5 7D 9F AE FF FD 4B B5 7D 9F 6E FF FB 4B B5 7D 9F 2E FF F9 4B B5 7D 9F EE FF F7 4B B5 7D 9F AE FF F5 4B B5 7D 9F 6E FF F3 4B B5 7D 9F 2E FF F1 4B B5 7D 9F EE EF EF 4B B5 7D 9F AE EF ED 4B B5 7D 9F 6E EF EB 4B B5 7D 9F 2E EF E5 4B B5 7D 9F EE FF E3 4B B5 7D 9F AE FF E1 4B B5 7D 9F 6E FF DF 4B B5 7D 9F 2E FF DF 4B B5 7D 9F 2E F7 FF 4B B5 7A 9F 9F A9 CB F7 F1 F5 FF FD 4B B5 7D 9F AE F7 FD 4B B5 7A 9F 9F A9 CB F5 EF F5 FF FF 4B B5 7A 9F 9F A9 CB F7 FF F5 FF FD 4B B5 7A 9F 9F A9 CB F7 FD F5 FF FD 4B B5 7A 9F 9F A9 CB F7 FB F5 FF FD 4B B5 7A 9F 9F A9 CB F7 F9 F5 FF FD 4B B5 7A 9F 9F A9 CB F7 F7 F5 FF FD 4B B5 7A 9F 9F A9 CB F7 F5 F5 FF FD 4B B5 7A 9F 9F A9 CB F7 F3 F5 FF FD 4B B5 7A 9F 9F A9 CB F5 FF F5 FF FD 4B B5 7A 9F 9F A9 CB F5 FD F5 FF FD 4B B5 7A 9F 9F A9 CB F5 FB F5 FF FD 4B B5 7A 9F 9F A9 CB F5 F9 F5 FF FD 4B B5 7A 9F 9F A9 CB F5 F7 F5 FF FD 4B B5 7A 9F 9F A9 CB F5 F5 F5 FF FD 4B B5 7A 9F 9F A9 CB F5 F3 F5 FF FD 4B B5 7A 9F 9F A9 CB F3 FF F5 FF FD 4B B5 7A 9F 9F A9 CB F3 FD F5 FF FD 4B B5 7D 9F 37 FF F9 4B B5 7D 9F 97 FF D7 
00 A9 4B DF A2 BD FF FF 4B B5 7D 9F EE FD FF 4B B5 7D 9F AE FD FD 4B B5 7D 9F 6E FD FB 4B B5 7D 9F 2E FD F9 4B B5 7D 9F EE FD F7 4B B5 7D 9F AE FD F5 4B B5 7D 9F 6E FD F3 4B B5 7D 9F 2E FD F1 4B B5 7D 9F EE EB EF 4B B5 7D 9F AE EB ED 4B B5 7D 9F 6E EB EB 4B B5 7D 9F 2E EB E9 4B B5 7D 9F EE FD DF 4B B5 7D 9F AE FD DF 4B B5 7D 9F 6E FD DF 4B B5 7D 9F 2E F5 FF 4B B5 7A 9F 9F A9 CB F1 F1 F5 FF FD 4B B5 7D 9F AE F5 FD 4B B5 7A 9F 9F A9 CB EF EF F5 FF FF 4B B5 7A 9F 9F A9 CB F1 FF F5 FF FD 4B B5 7A 9F 9F A9 CB F1 FD F5 FF FD 4B B5 7A 9F 9F A9 CB F1 FB F5 FF FD 4B B5 7A 9F 9F A9 CB F1 F9 F5 FF FD 4B B5 7A 9F 9F A9 CB F1 F7 F5 FF FD 4B B5 7A 9F 9F A9 CB F1 F5 F5 FF FD 4B B5 7A 9F 9F A9 CB F1 F3 F5 FF FD 4B B5 7A 9F 9F A9 CB EF FF F5 FF FD 4B B5 7A 9F 9F A9 CB EF FD F5 FF FD 4B B5 7A 9F 9F A9 CB EF FB F5 FF FD 4B B5 7A 9F 9F A9 CB EF F9 F5 FF FD 4B B5 7A 9F 9F A9 CB EF F7 F5 FF FD 4B B5 7A 9F 9F A9 CB EF F5 F5 FF FF 4B B5 7A 9F 9F A9 CB EF F3 F5 FF FF 4B B5 7A 9F 9F A9 CB ED FF F5 FF FD

Villy-web, 2025-05-26:
> **JK BMS Inverter V19**. # 15 - **JK BMS PBxx SERIES LCD Protocol V1.0** (UART, baud rate: 115200):
> 00 A9 4B DB BE FF EB 2B FF FF FF 37 FF D7 FD DF 7F 38 79 EF D3 FF FD FF FD E5 FD FF FF FF FD FF

This doesn't look like a protocol. Maybe the baud rate is wrong?