syssi/esphome-jk-bms #120: JK-B2A24S15P JkModbus CRC Check failed! 47F0!=4882

mariusvaida, 2022-05-27:
Hello, i have a JK bms, model JK-B2A24S15P and i'm trying to make it work with esphome, but i get some errors like invalid hearders or CRC check failed.

```
[12:51:20][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
[12:51:20][D][uart_debug:114]: <<< 4E:57:01:1B:00:00:00:00:06:00:01:79:2A:01:0F:4E:02:0F:50:03:0F:50:04:0F:50:05:0F:50:06:0F:00:07:0F:50:08:0E:50:09:0F:50:0A:0F:50:0B:0F:50:0C:0F:50:0D:0F:50:0E:0F:50:80:00:1A:81:00:18:82:00:18:83:15:6F:84:00:00:85:63:86:02:87:00:11:89:00:00:0D:00:8A:00:0E:8B:00:00:8C:00:03:8E:16:26:8F:0F:DC:90:0F:D2:91:0F:6E:92:00:05:93:0B:54:94:0B:B8:95:00:05:96:01:2C:97:00:96:98:01:2C:99:00:64:9A:00:1E:9B:0C:E4:9C:00:05:9D:01:9E:00:5A:9F:00:46:A0:00:64:A1:00:64
[12:51:21][W][jk_modbus:075]: JkModbus CRC Check failed! 47F0!=4882
```

```
[12:51:21][D][uart_debug:114]: <<< A2:00:14:A3:00:06:A4:00:46:A5:00:01:A6:00:02:A7:FF:EC:A8:FF:F6:A9:0E:AA:00:00:00:C3:AB:01:AC:01:AD:03:E8:AE:01:AF:00:B0:00:0A:B1:14:B2:37:32:38:32:00:00:00:00:00:00:B3:00:B4:49:6E:70:75:74:20:55:73:B5:32:31:31:31:B6:00:03:3E:00:B7:48:38:2E:58:5F:5F:53:38:2E:32:31:57:5F:5F:5F:B8:00:B9:00:00:00:00:BA:42:54:33:30:37:32:30:32:30:31:32:30:30:30:30:32:30:30:35:32:31:30:30:31:C0:01:00:00:00:00:68:00:00:48:82
[12:51:22][D][uart_debug:114]: <<< DF
[12:51:23][D][uart_debug:114]: <<< FF
[12:51:25][D][uart_debug:114]: <<< FF
[12:51:25][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
[12:51:25][W][jk_modbus:050]: Invalid header
```

```
[12:51:26][D][uart_debug:114]: <<< FF:4E:57:01:1B:00:00:00:00:06:00:01:79:2A:01:0F:4D:02:0F:50:03:0E:4E:04:0F:4E:05:0F:4E:06:0F:4E:07:0F:4E:08:0F:50:09:0F:4E:0A:0F:4E:0B:0F:4E:0C:0F:50:0D:0F:4E:0E:0F:50:80:00:1A:81:00:18:82:00:10:83:15:6E:84:00:00:85:63:86:02:87:00:11:89:00:00:0D:00:8A:00:0E:8B:00:00:8C:00:03:8E:16:26:8F:0F:DC:90:0F:D2:91:0F:6E:92:00:05:93:0B:54:94:0B:B8:95:00:05:96:02:2C:97:00:96:98:01:2C:99:00:64:9A:00:1E:9B:0C:E4:9C:00:05:9D:01:9E:00:5A:9F:00:46:A0:00:64:A1:00
[12:51:26][W][jk_modbus:050]: Invalid header
```

[image]

Please help me and tell me if you need more info.
Thank you!
Marius.

syssi, 2022-05-27:
Do you use an ESP8266 or ESP32? Which GPIOs did you use?

mariusvaida, 2022-05-27:
Hi Syssi, i'm using ESP32 with pins 16&17.
Thanks.

syssi, 2022-05-27:
Alright. In this case it's no "software serial is unable to handle 115200 bauds properly" issue. Could you check your wiring and/or provide a photo? Make sure `GND` is connected properly. It looks like there is much noise on the wire. F.e. take a look at this bunch of received bytes:

```
FF:4E:57:01:1B:00:00:00:00:06:00:01:79:2A:01:0F:4D:02:0F:50:03:0E:4E:04:0F:4E:05:0F:4E:06:0F:4E:07:0F:4E:08:0F:50:09:0F:4E:0A:0F:4E:0B:0F:4E:0C:0F:50:0D:0F:4E:0E:0F:50:80:00:1A:81:00:18:82:00:10:83:15:6E:84:00:00:85:63:86:02:87:00:11:89:00:00:0D:00:8A:00:0E:8B:00:00:8C:00:03:8E:16:26:8F:0F:DC:90:0F:D2:91:0F:6E:92:00:05:93:0B:54:94:0B:B8:95:00:05:96:02:2C:97:00:96:98:01:2C:99:00:64:9A:00:1E:9B:0C:E4:9C:00:05:9D:01:9E:00:5A:9F:00:46:A0:00:64:A1:00
```

The start sequence of a frame is `0x4e 0x57`. Your frame starts with `0xff 0x4e 0x57`. Most of the payload/bytes look good:

```
FF   <- garbage
4E:57:  <- start sequence
01:1B:  <- data length
00:00:00:00: <- terminal number
06: <- "read all" response
00: <- source id
01: <- frame type
79: <- 12.1V
2A: <- 42 / 3 = 14 Cells
  01: 0F:4D: 3.917V
  02: 0F:50: 3.920V
  03: 0E:4E:
  04: 0F:4E:
  05: 0F:4E:
  06: 0F:4E:
  07: 0F:4E:
  08: 0F:50:
  09: 0F:4E:
  0A: 0F:4E:
  0B: 0F:4E:
  0C: 0F:50:
  0D: 0F:4E:
  0E:0 F:50:
  80:00:1A:
  81:00:18:
  82:00:10:
  83:15:6E:
  84:00:00:
  85:63:
  86:02:
  87:00:11:
  89:00:00:0D:00:
  8A:00:0E:
  8B:00:00:
  8C:00:03:
  8E:16:26:
  8F:0F:DC:
  90:0F:D2:
  91:0F:6E:
  92:00:05:
  93:0B:54:
  94:0B:B8:
  95:00:05:
  96:02:2C:
  97:00:96:
  98:01:2C:
  99:00:64:
  9A:00:1E:
  9B:0C:E4:
  9C:00:05:
  9D:01:
  9E:00:5A:
  9F:00:46:
  A0:00:64: Temperature protection value in the battery box       100°C
  A1:00 .... some bytes are missing here
```

Missing bytes at the end of the frame: https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms/jk_bms.cpp#L268-L368

mariusvaida, 2022-05-27:
Thank you, i will check the wire again. As a test, the cells voltages are working fine, but as soon i add the total voltage or current sensor, it stops working and throws the invalid header error.

syssi, 2022-05-27:
Could you provide a not compiling config example? Please try a `esphome clean yourconfig.yaml` to do a clean build afterwards.

syssi, 2022-05-27:
If you see the CRC error every now and then this is fine. The communication is super simpel: Depending on the `update_interval` a "read all" request is sent to the BMS. The BMS responds with a big frame including all measurements & config settings. Everytime you see a CRC error the response was incomplete/broken and gets skipped.

mariusvaida, 2022-05-27:
Succes! You were right about the cable. I made a new one and i used a shielded one and now it works fine.
Thank you very very much! You are awesome!

closed 2022-05-27 by syssi

mariusvaida, 2022-06-03:
Sorry to open this again, but i need your opinion. Since i opened this support issue, it worked (almost) fine, there were moments when i got no response from the bms and then suddenly worked again, but for the last 2 days is completely dead, i get no reponse from the bms, so today i had some time to debug.
What i discovered is that the TX pin from the bms is in low state, even when the bms is powered off. I tested it with a multimeter on continuity mode between GND and TX. It could be that the rs232 chip from the bms is dead? It would be strange since all the soldering i've done was with the cable unplugged from the bms.

syssi, 2022-06-03:
The BMS is still working just the UART-TTL connection isn't working anymore?

syssi, 2022-06-03:
Please remove the ESP and the battery cells (+ balancer cables) from the BMS and measure the resistance between `GND<->RX` and `GND<->TX`. In my case its around 3.3 MOhm in both cases.