syssi/esphome-jk-bms #121: Can't compile using esp-idf framework - arduino framework generates boot loop

TheSmartGerman, 2022-06-01:
Hello,
I tried to install the "esp32-ble-example.yaml", but I can't build the example. As far I can track down the issue binutils-esp32ulp is not supportetd on aarch64. 
I'll get an error like this: "Error: Could not find the package with 'platformio/toolchain-esp32ulp @ ~1.22851.0' requirements for your system 'linux_aarch64'"

So I tried to switch to arduino framework, this seems to be ok. 

But it send's the esp32dev board into boot loop:

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

Any advice on this?

Is it only possible to compile on x86 machine? 

thx

syssi, 2022-06-01:
Yes. The esp-idf toolchain isn't compatible to the raspberry pi cpu arch at the moment. Please use a amd64 machine to build the yaml.

syssi, 2022-06-01:
Please don't use the Arduino framework because it crashs as soon a BLE notification is received.

TheSmartGerman, 2022-06-01:
ok, thx. Does this behavior also applies to other ble features? I tried to get a ble_gateway running without success. can this also solve me problems? 

But I think this topic can be closed

syssi, 2022-06-01:
I don't know. You could give it a try!