syssi/esphome-jk-bms #124: Log: "unsuccessful boot attempts"

amagr0, 2022-06-07:
Hello, 

After compiling the ESPhome firmware I'm getting this erros:

`Showing logs:
[18:55:53][I][logger:214]: Log initialized
[18:55:53][C][ota:461]: There have been 2 suspected unsuccessful boot attempts.
[18:55:53][D][esp32.preferences:114]: Saving preferences to flash...
[18:55:53][I][app:029]: Running through setup()...
[18:55:53][C][uart.arduino_esp32:077]: Setting up UART...
[18:55:53][C][wifi:037]: Setting up WiFi...
[18:55:53]Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
[18:55:53]Debug exception reason: Stack canary watchpoint triggered (loopTask) 
[18:55:53]Core 1 register dump:
[18:55:53]PC      : 0x40086f94  PS      : 0x00060836  A0      : 0x3ffb0110  A1      : 0x3ffb0050  
[18:55:53]A2      : 0x00000001  A3      : 0x3ffc7c48  A4      : 0x3ffc7c48  A5      : 0x00000001  
[18:55:53]A6      : 0x00060820  A7      : 0x00000000  A8      : 0x800855c8  A9      : 0x3ffb00f0  
[18:55:53]A10     : 0x3ff000e0  A11     : 0x00000001  A12     : 0x3ffbe8e8  A13     : 0x00000001  
[18:55:54]A14     : 0x00060823  A15     : 0x00000000  SAR     : 0x0000000b  EXCCAUSE: 0x00000001  
[18:55:54]EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
[18:55:54]
[18:55:54]ELF file SHA256: 0000000000000000
[18:55:54]
[18:55:54]Backtrace: 0x40086f94:0x3ffb0050 0x3ffb010d:0x3ffb0130 0x401652cf:0x3ffb0170 0x401653a6:0x3ffb0190 0x400878e2:0x3ffb01b0 0x40087d08:0x3ffb01d0 0x4016cb51:0x3ffb0220 0x4016afaa:0x3ffb0240 0x4016b315:0x3ffb0260 0x4016a33c:0x3ffb02e0 0x4016a8d2:0x3ffb0360 0x40169cd7:0x3ffb03d0 0x40140c0e:0x3ffb0410 0x40140d26:0x3ffb0450 0x40140d4d:0x3ffb0470 0x40100757:0x3ffb0490 0x4010086f:0x3ffb04b0 0x401633d9:0x3ffb04d0 0x400f26d1:0x3ffb04f0 0x400e82e5:0x3ffb0600 0x400e8344:0x3ffb0630 0x400e7215:0x3ffb0660 0x401796ad:0x3ffb0770 0x401797a5:0x3ffb0790 0x400ea015:0x3ffb07b0 0x400f097a:0x3ffb0800 0x400fe6c6:0x3ffb1fb0 0x40089b92:0x3ffb1fd0
[18:55:54]
[18:55:54]Rebooting...
[18:55:54]ets Jun  8 2016 00:22:57
[18:55:54]
[18:55:54]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[18:55:54]configsip: 0, SPIWP:0xee
[18:55:54]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[18:55:54]mode:DIO, clock div:2
[18:55:54]load:0x3fff0018,len:4
[18:55:54]load:0x3fff001c,len:952
[18:55:54]load:0x40078000,len:6084
[18:55:54]load:0x40080000,len:7944
[18:55:54]entry 0x40080310
[18:55:54][I][logger:214]: Log initialized
[18:55:54][C][ota:461]: There have been 3 suspected unsuccessful boot attempts.
[18:55:54][D][esp32.preferences:114]: Saving preferences to flash...
[18:55:54][I][app:029]: Running through setup()...
[18:55:54][C][uart.arduino_esp32:077]: Setting up UART...
[18:55:54][C][wifi:037]: Setting up WiFi...
[18:55:54]Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
[18:55:54]Debug exception reason: Stack canary watchpoint triggered (loopTask) 
[18:55:54]Core 1 register dump:
[18:55:54]PC      : 0x40086f94  PS      : 0x00060836  A0      : 0x3ffb0110  A1      : 0x3ffb0050  
[18:55:54]A2      : 0x00000001  A3      : 0x3ffc7c48  A4      : 0x3ffc7c48  A5      : 0x00000001  
[18:55:54]A6      : 0x00060820  A7      : 0x00000000  A8      : 0x800855c8  A9      : 0x3ffb00f0  
[18:55:54]A10     : 0x3ff000e0  A11     : 0x00000001  A12     : 0x3ffbe8e8  A13     : 0x00000001  
[18:55:54]A14     : 0x00060823  A15     : 0x00000000  SAR     : 0x0000000b  EXCCAUSE: 0x00000001  
[18:55:54]EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
[18:55:54]
[18:55:54]ELF file SHA256: 0000000000000000
[18:55:54]
[18:55:54]Backtrace: 0x40086f94:0x3ffb0050 0x3ffb010d:0x3ffb0130 0x401652cf:0x3ffb0170 0x401653a6:0x3ffb0190 0x400878e2:0x3ffb01b0 0x40087d08:0x3ffb01d0 0x4016cb51:0x3ffb0220 0x4016afaa:0x3ffb0240 0x4016b315:0x3ffb0260 0x4016a33c:0x3ff`

And this is the firmware without the sensors, I have disabled a lot off them.

`esphome:
  name: ${name}

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:

api:

ota:
  password: "fc7210241722b4adab192bed49d9cb36"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "Bms-Data Fallback Hotspot"
    password: "aWEChqqSgAq4"

captive_portal:

substitutions:
  name: jk-bms
  external_components_source: github://syssi/esphome-jk-bms@main
  tx_pin: GPIO16
  rx_pin: GPIO17

external_components:
  - source: ${external_components_source}
    refresh: 0s

mqtt:
  broker: 192.168.1.000
  username: mqtt-user
  password: mqtt-pass
  id: mqtt_client

uart:
  id: uart0
  baud_rate: 115200
  rx_buffer_size: 384
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}

jk_modbus:
  id: modbus0
  uart_id: uart0

jk_bms:
  id: bms0
  jk_modbus_id: modbus0
...
..
.`

What could cause this erros?
I have a ESP32 WROOM-32 board.

Thanks

syssi, 2022-06-07:
As first step please try to use `esp-idf` framework instead of `arduino`:

```
esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: latest
```

If this isn't possible please reduce the number of sensor to a minimum (just for testing). If you use f.e. just one jk_bms sensor to publish the battery voltage does this fix the boot loop?

amagr0, 2022-06-07:
I'm getting warnings just by add your suggestion.

[image]

syssi, 2022-06-07:
Do you need the `captive_portal` for sure? You could remove the line/component if not.

syssi, 2022-06-07:
```
esphome:
  name: ${name}

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:

api:

ota:
  password: "fc7210241722b4adab192bed49d9cb36"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

substitutions:
  name: jk-bms
  external_components_source: github://syssi/esphome-jk-bms@main
  tx_pin: GPIO16
  rx_pin: GPIO17

external_components:
  - source: ${external_components_source}
    refresh: 0s

mqtt:
  broker: 192.168.1.000
  username: mqtt-user
  password: mqtt-pass
  id: mqtt_client

uart:
  id: uart0
  baud_rate: 115200
  rx_buffer_size: 384
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}

jk_modbus:
  id: modbus0
  uart_id: uart0

jk_bms:
  id: bms0
  jk_modbus_id: modbus0

sensor:
  - platform: jk_bms
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
```

amagr0, 2022-06-07:
I removed, but I still have the first warning, and actually a have another one in mqtt settings:

[image]
[image]

amagr0, 2022-06-07:
To use the arduino framework?

[image]

syssi, 2022-06-07:
This was a copy&paste issue (edited now!). ;-)

amagr0, 2022-06-07:
I still have the warnings:

[image]

syssi, 2022-06-07:
Could you check your ESPHome version? It's important you are using the most recent version.

This config compiles fine using `esphome 2022.5.0` here:

```
substitutions:
  name: jk-bms
  external_components_source: github://syssi/esphome-jk-bms@main
  tx_pin: GPIO16
  rx_pin: GPIO17

esphome:
  name: ${name}

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: latest

logger:

api:

ota:
  password: "fc7210241722b4adab192bed49d9cb36"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

external_components:
  - source: ${external_components_source}
    refresh: 0s

mqtt:
  broker: 192.168.1.000
  username: mqtt-user
  password: mqtt-pass
  id: mqtt_client

uart:
  id: uart0
  baud_rate: 115200
  rx_buffer_size: 384
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}

jk_modbus:
  id: modbus0
  uart_id: uart0

jk_bms:
  id: bms0
  jk_modbus_id: modbus0

sensor:
  - platform: jk_bms
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
```

Could you hover over the read crosses? Do you see some tooltip what's wrong here?

amagr0, 2022-06-07:
[image]
[image]

I have version 2022.2.6, but not have any update available.
[image]

syssi, 2022-06-07:
The `mqtt` component of your outdated esphome version isn't `esp-idf` compatible yet. I don't use the embedded esphome install into Home Assistant. If you have a linux machine around you could install esphome locally using:

```
pip3 install esphome
```

Afterwards you can access the esphome CLI command using:

```
~/.local/bin/esphome
```

And compile/flash the yaml using:

```
.local/bin/esphome run test.yaml
```

amagr0, 2022-06-08:
Hello Sebastian,

I finally update my ESPhome to the newest version in HA.

I found this information in Discord ESPhome channel:
[image]

I will try to compile in the afternoon, then I will give you feedback about it.
Tks.

syssi, 2022-06-08:
Good to know! I hope your Home Assistant Host isn't a Raspberry Pi (ARM64) device because the toolchain required by ESP-IDF isn't ARM64 compatible at the moment. In case of ARM64 you will see something like this:

```
Error: Could not find the package with 'platformio/toolchain-esp32ulp @ ~1.22851.0' requirements for your system 'linux_aarch64'
```

If your HA host is a x86 architecture you are on a good track!

amagr0, 2022-06-08:
Ouhh! Yes it's a ARM64 version :/
What is the possibilities that I have now?

syssi, 2022-06-11:
Any news? :-) I would like to close this issue as soon your setup is on air.

syssi, 2022-06-22:
I will close this issue. Feel free to re-open if you make some progress.

syssi, 2022-06-23:
Yes! Pick any text editor. Nothing special is required. The most important part is don't mess up the indents.

amagr0, 2022-06-25:
Hello @syssi,

I think I finally made it, let's see how it goes in the next days.

[image]

I found that ESPhome has a [dashboard](https://esphome.io/guides/getting_started_command_line.html#bonus-esphome-dashboard) for linux, that simplifies the life of newbies like me.

[image]

Btw, the specs of my unit are:
ID: JK-B2A24S20P
Hardware ver: 10.XW
Software ver: 10.07

Thanks for everything!

syssi, 2022-06-26:
Good job! I will mention your BMS model version at the README.

amagr0, 2022-06-28:
Hello @syssi,

Is possible to get the Balance Current data, I did not found that sensor?
Thanks.

syssi, 2022-06-28:
No. This value is only available via bluetooth.