syssi/esphome-jk-bms #130: UART (non BLE) version appears to not be able to control switches

derekyle, 2022-06-27:
I don't see any mention in the readme, and in fact, the readme seems to focus solely on the uart method of connection, but I can't seem to get the switches to show up in Home Assistant. I am using the api method rather than mqtt.

Is this a limitation of the uart protocol? Do I need to use an esp32 and use bluetooth in order to control the charge/dischange/balance switches from within home assistant?

syssi, 2022-06-27:
This is a limitation of the UART protocol if we trust the seller & manufacturer. It's read-only. May be new software versions will allow write operations some day. You have to use an ESP32 + BLE if you want to control the BMS.

This is my idea for a future release of this component:

1. Turn off the BLE module by pulling to RST pin to GND: https://github.com/syssi/esphome-jk-bms/issues/107
2. Add wires to `RX`, `TX`, `GND`, `P11` and `P12` of the bluetooth module: https://github.com/syssi/esphome-jk-bms/issues/108
3. Rewrite the JK02/04 protocol implementation to skip BLE as transport layer and use UART-TTL to talk directly to the BMS (using the JK02/04 protocol)

derekyle, 2022-06-28:
Interesting. Do most jkbms units have a separate Bluetooth module? Or is this something that needs to be accessed on the internal board?

syssi, 2022-06-28:
You have to open the case to access the pins. It's always internal:

https://raw.githubusercontent.com/syssi/esphome-jk-bms/main/images/jk-bd6a17s6p/004.jpg
https://raw.githubusercontent.com/syssi/esphome-jk-bms/main/images/jk-bd6a17s6p/005-labeled.jpg

derekyle, 2022-06-28:
I see. Thank you very much for your work on this project. It has been very helpful to be able to easily connect my BMS into my Home Assistant.

syssi, 2022-06-28:
Thanks for your feedback! If you have no further questions feel free to close the issue. :-)

closed 2022-06-30 by syssi

paulsteigel, 2024-03-15:
Hi @syssi and all, 
I have recently purchased a JKBD6A20S10P and I built an esphome interface using your esp32-example.yaml with esp32lolin. However, the JKBMS does not response to query command via UAR TTL. This happened with my friends' version too (11.28 and 11.36, mine is 11.42). I think JK has changed the protocol. Can you advice on this?

The response code to the "4E:57:00:13:00:00:00:00:06:02:00:00:00:00:00:00:68:00:00:01:28" command is nothing and time out happened
Please help with thanks.
Best!