syssi/esphome-jk-bms #135: ESP32S node MCU Compiling error.

MrYoshii, 2022-07-03:
Hello when I want to flash the code to the ESP32 I get an error message.
It can still be flashed but some sensors update only once at the beginning.

`src/esphome/components/ble_client/ble_client.cpp: In member function 'float esphome::ble_client::BLEClient::parse_char_value(uint8_t*, uint16_t)':
src/esphome/components/ble_client/ble_client.cpp:257:7: warning: this statement may fall through [-Wimplicit-fallthrough=]
       if (length > 2) {
       ^~
src/esphome/components/ble_client/ble_client.cpp:260:5: note: here
     case 0x7:  // uint24.
     ^~~~
src/esphome/components/ble_client/ble_client.cpp:261:7: warning: this statement may fall through [-Wimplicit-fallthrough=]
       if (length > 3) {
       ^~
src/esphome/components/ble_client/ble_client.cpp:264:5: note: here
     case 0x8:  // uint32.
     ^~~~
src/esphome/components/ble_client/ble_client.cpp:265:7: warning: this statement may fall through [-Wimplicit-fallthrough=]
       if (length > 4) {
       ^~
src/esphome/components/ble_client/ble_client.cpp:269:5: note: here
     case 0xC:  // int8.
     ^~~~
src/esphome/components/ble_client/ble_client.cpp:273:7: warning: this statement may fall through [-Wimplicit-fallthrough=]
       if (length > 2) {
       ^~
src/esphome/components/ble_client/ble_client.cpp:276:5: note: here
     case 0xF:  // int24.
     ^~~~
src/esphome/components/ble_client/ble_client.cpp:277:7: warning: this statement may fall through [-Wimplicit-fallthrough=]
       if (length > 3) {
       ^~
src/esphome/components/ble_client/ble_client.cpp:280:5: note: here
     case 0x10:  // int32.
     ^~~~`

syssi, 2022-07-04:
Which ESPHome version do you use? Could you provide a full log and describe the issue in detail? The mentioned warnings are normal and shouldn't affect the operation.

MrYoshii, 2022-07-04:
Hi sorry ok so:
I use ESPHome v2022.6.2

The problem is that the values are not updated.
So not all values. Some are updated and others are not.
For example. The voltage is updated but the temperature is not.
If you need more let me know.

[image]

[image]

Log: 
```
[10:18:50][D][api.connection:861]: Home Assistant 2022.6.7 (::FFFF:192.168.1.110): Connected successfully
[10:18:50][I][jk_bms_ble:746]: Settings frame (300 bytes):
[10:18:50][D][jk_bms_ble:772]:   Unknown6: 0.600000
[10:18:50][I][jk_bms_ble:774]:   Cell UVP: 2.500000 V
[10:18:50][D][number:012]: 'jk-bms cell voltage undervoltage protection': Sending state 2.500000
[10:18:50][I][jk_bms_ble:778]:   Cell UVPR: 2.550000 V
[10:18:50][D][number:012]: 'jk-bms cell voltage undervoltage recovery': Sending state 2.550000
[10:18:50][I][jk_bms_ble:782]:   Cell OVP: 3.600000 V
[10:18:50][D][number:012]: 'jk-bms cell voltage overvoltage protection': Sending state 3.600000
[10:18:50][I][jk_bms_ble:786]:   Cell OVPR: 3.550000 V
[10:18:50][D][number:012]: 'jk-bms cell voltage overvoltage recovery': Sending state 3.550000
[10:18:50][I][jk_bms_ble:790]:   Balance trigger voltage: 0.003000 V
[10:18:51][D][number:012]: 'jk-bms balance trigger voltage': Sending state 0.003000
[10:18:51][I][jk_bms_ble:798]:   Power off voltage: 2.450000 V
[10:18:51][D][number:012]: 'jk-bms power off voltage': Sending state 2.450000
[10:18:51][I][jk_bms_ble:802]:   Max. charge current: 200.000015 A
[10:18:51][D][number:012]: 'jk-bms max charge current': Sending state 200.000015
[10:18:51][I][jk_bms_ble:806]:   Charge OCP delay: 30.000000 s
[10:18:51][I][jk_bms_ble:808]:   Charge OCP recovery delay: 60.000000 s
[10:18:51][I][jk_bms_ble:810]:   Max. discharge current: 200.000015 A
[10:18:51][D][number:012]: 'jk-bms max discharge current': Sending state 200.000015
[10:18:51][I][jk_bms_ble:814]:   Discharge OCP recovery delay: 300.000000 s
[10:18:51][I][jk_bms_ble:816]:   Discharge OCP recovery delay: 60.000000 s
[10:18:51][I][jk_bms_ble:818]:   SCP recovery time: 60.000000 s
[10:18:51][I][jk_bms_ble:820]:   Max. balance current: 2.000000 A
[10:18:51][D][number:012]: 'jk-bms max balance current': Sending state 2.000000
[10:18:51][I][jk_bms_ble:824]:   Charge OTP: 50.000000 °C
[10:18:51][I][jk_bms_ble:826]:   Charge OTP recovery: 45.000000 °C
[10:18:51][I][jk_bms_ble:828]:   Discharge OTP: 50.000000 °C
[10:18:51][I][jk_bms_ble:830]:   Discharge OTP recovery: 45.000000 °C
[10:18:51][I][jk_bms_ble:832]:   Charge UTP: 3.000000 °C
[10:18:51][I][jk_bms_ble:834]:   Charge UTP recovery: 7.000000 °C
[10:18:51][I][jk_bms_ble:836]:   MOS OTP: 90.000000 °C
[10:18:51][I][jk_bms_ble:838]:   MOS OTP recovery: 70.000000 °C
[10:18:51][I][jk_bms_ble:840]:   Cell count: 8.000000
[10:18:51][D][number:012]: 'jk-bms cell count': Sending state 8.000000
[10:18:51][I][jk_bms_ble:844]:   Charge switch: on
[10:18:51][D][switch:037]: 'jk-bms charging': Sending state ON
[10:18:51][I][jk_bms_ble:848]:   Discharge switch: on
[10:18:51][D][switch:037]: 'jk-bms discharging': Sending state ON
[10:18:51][I][jk_bms_ble:852]:   Balancer switch: on
[10:18:51][D][switch:037]: 'jk-bms balancer': Sending state ON
[10:18:51][I][jk_bms_ble:856]:   Nominal battery capacity: 280.000000 Ah
[10:18:51][D][number:012]: 'jk-bms total battery capacity': Sending state 280.000000
[10:18:51][D][jk_bms_ble:860]:   Unknown134: 1.500000
[10:18:51][I][jk_bms_ble:862]:   Start balance voltage: 3.400000 V
[10:18:51][D][number:012]: 'jk-bms balance starting voltage': Sending state 3.400000
[10:18:51][I][jk_bms_ble:894]:   Con. wire resistance 1: 0.000000 Ohm
[10:18:51][I][jk_bms_ble:894]:   Con. wire resistance 2: 0.000000 Ohm
[10:18:51][I][jk_bms_ble:894]:   Con. wire resistance 3: 0.000000 Ohm
[10:18:51][I][jk_bms_ble:894]:   Con. wire resistance 4: 0.000000 Ohm
[10:18:51][I][jk_bms_ble:894]:   Con. wire resistance 5: 0.000000 Ohm
[10:18:51][I][jk_bms_ble:894]:   Con. wire resistance 6: 0.000000 Ohm
[10:18:51][I][jk_bms_ble:894]:   Con. wire resistance 7: 0.000000 Ohm
[10:18:51][I][jk_bms_ble:894]:   Con. wire resistance 8:ce 19: 0.000000 Ohm
[10:18:51][I][jk_bms_ble:894]:   Con. wire resistance 20: 0.000000 Ohm
[... 110 lines cut ...]
[10:19:01][D][sensor:125]: 'jk-bms cell resistance 1': Sending state 0.03600 Ω with 3 decimals of accuracy
[10:19:01][D][sensor:125]: 'jk-bms cell voltage 2': Sending state 3.32300 V with 3 decimals of accuracy
[10:19:02][D][sensor:125]: 'jk-bms cell resistance 2': Sending state 0.03600 Ω with 3 decimals of accuracy
[10:19:02][D][sensor:125]: 'jk-bms cell voltage 3': Sending state 3.32100 V with 3 decimals of accuracy
[10:19:02][D][sensor:125]: 'jk-bms cell resistance 3': Sending state 0.03600 Ω with 3 decimals of accuracy
[10:19:02][D][sensor:125]: 'jk-bms cell voltage 4': Sending state 3.31500 V with 3 decimals of accuracy
[10:19:02][D][sensor:125]: 'jk-bms cell resistance 4': Sending state 0.03600 Ω with 3 decimals of accuracy
[10:19:02][D][sensor:125]: 'jk-bms cell voltage 5': Sending state 3.31200 V with 3 decimals of accuracy
[10:19:02][D][sensor:125]: 'jk-bms cell resistance 5': Sending state 0.03600 Ω with 3 decimals of accuracy
[10:19:02][D][sensor:125]: 'jk-bms cell voltage 6': Sending state 3.31000 V with 3 decimals of accuracy
[10:19:02][D][sensor:125]: 'jk-bms cell resistance 6': Sending state 0.03600 Ω with 3 decimals of accuracy
[10:19:02][D][sensor:125]: 'jk-bms cell voltage 7': Sending state 3.31200 V with 3 decimals of accuracy
[... cut ...]
[10:19:02][D][sensor:125]: 'jk-bms average cell voltage': Sending state 3.31500 V with 3 decimals of accuracy
[10:19:02][D][sensor:125]: 'jk-bms delta cell voltage': Sending state 0.00500 V with 3 decimals of accuracy
[10:19:02][D][sensor:125]: 'jk-bms max voltage cell': Sending state 2.00000  with 0 decimals of accuracy
[10:19:02][D][sensor:125]: 'jk-bms min voltage cell': Sending state 1.00000  with 0 decimals of accuracy
[10:19:02][D][jk_bms_ble:426]: Unknown112: 00 00
[10:19:02][I][jk_bms_ble:428]: Wire resistance warning: 00 00 00 00
```

syssi, 2022-07-04:
> For example. The voltage is updated but the temperature is not.

This is the default behavior of ESPHome: If a measurement / sensor state didn't change the update isn't published. You can change this behavior per sensor by adding `force_update: true`: https://esphome.io/components/sensor/index.html

MrYoshii, 2022-07-04:
oh ok thank you for the help.

closed 2022-07-04 by MrYoshii