syssi/esphome-jk-bms #137: ERROR Unable to load component jk_bms_ble.number:

dremeier, 2022-07-07:
Hello, I'm not new to ESP, Arduino and so on but new to ESPhome. So I don`t understand how to work with it. But I got some simple samples working. But not the esp32-ble-example.yaml.

What I did:
Installed HomeAssistance with ESPhome (have it on ioBroker as well) put the RAW esp32-ble-example.yaml in a new ESPhome node, edit some values to my needs and clicked install then got the error "Unable to load component jk_bms_ble.number" I tried it several times and tested around also with api and mqtt, esp-idf and arduino.
So what I did wrong?

```
# Enable logging
logger:

# Enable Home Assistant API
#api:
#  encryption:
#    key: "xxxxxxxx"

ota:
  password: "ec30....................."

wifi:
  ssid: "xxxxxxx"
  password: "xxxxxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Jk-Bms Fallback Hotspot"
    password: "c38BzekXLFWE"

#captive_portal:
    
substitutions:
  name: jk-bms
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: C8:47:8C:E1:E2:E1
  # Defaults to "JK02". Please use "JK04" if you have some old JK-BMS version (f.e. JK-B2A16S hw 3.0, sw. 3.3.0)
  protocol_version: JK02

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.1.0

esp32:
  board: esp32dev
  framework:
#  The arduino platform crashs
    type: arduino
    version: latest
#    type: esp-idf
#    version: latest

external_components:
  - source: ${external_components_source}
    refresh: 0s

# This feature is only available with frameworks ['arduino'] <= 2022.3.1
# https://github.com/esphome/esphome/pull/2930: Add mqtt for idf >= 2022.4.0
#
mqtt:
  broker: 192.168.107.220
  username: "xxxxxx"
  password: "xxxxxx"
  port: 1893
  id: "mqtt_client"

esp32_ble_tracker:

ble_client:
  - mac_address: ${mac_address}
    id: client0

jk_bms_ble:
  - ble_client_id: client0
    protocol_version: ${protocol_version}
    throttle: 5s
    id: bms0

binary_sensor:
  - platform: jk_bms_ble
    balancing:
      name: "${name} balancing"
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"

button:
  - platform: jk_bms_ble
    retrieve_settings:
      name: "${name} retrieve settings"
    retrieve_device_info:
      name: "${name} retrieve device info"

number:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    balance_trigger_voltage:
      name: "${name} balance trigger voltage"
    cell_count:
      name: "${name} cell count"
    total_battery_capacity:
      name: "${name} total battery capacity"
    cell_voltage_overvoltage_protection:
      name: "${name} cell voltage overvoltage protection"
    cell_voltage_overvoltage_recovery:
      name: "${name} cell voltage overvoltage recovery"
    cell_voltage_undervoltage_protection:
      name: "${name} cell voltage undervoltage protection"
    cell_voltage_undervoltage_recovery:
      name: "${name} cell voltage undervoltage recovery"
    balance_starting_voltage:
      name: "${name} balance starting voltage"
    voltage_calibration:
      name: "${name} voltage calibration"
    current_calibration:
      name: "${name} current calibration"
    power_off_voltage:
      name: "${name} power off voltage"
    max_balance_current:
      name: "${name} max balance current"
    max_charge_current:
      name: "${name} max charge current"
    max_discharge_current:
      name: "${name} max discharge current"

sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    average_cell_voltage:
      name: "${name} average cell voltage"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
    cell_voltage_8:
      name: "${name} cell voltage 8"
    cell_voltage_9:
      name: "${name} cell voltage 9"
    cell_voltage_10:
      name: "${name} cell voltage 10"
    cell_voltage_11:
      name: "${name} cell voltage 11"
    cell_voltage_12:
      name: "${name} cell voltage 12"
    cell_voltage_13:
      name: "${name} cell voltage 13"
    cell_voltage_14:
      name: "${name} cell voltage 14"
    cell_voltage_15:
      name: "${name} cell voltage 15"
    cell_voltage_16:
      name: "${name} cell voltage 16"
    cell_voltage_17:
      name: "${name} cell voltage 17"
    cell_voltage_18:
      name: "${name} cell voltage 18"
    cell_voltage_19:
      name: "${name} cell voltage 19"
    cell_voltage_20:
      name: "${name} cell voltage 20"
    cell_voltage_21:
      name: "${name} cell voltage 21"
    cell_voltage_22:
      name: "${name} cell voltage 22"
    cell_voltage_23:
      name: "${name} cell voltage 23"
    cell_voltage_24:
      name: "${name} cell voltage 24"
    cell_resistance_1:
      name: "${name} cell resistance 1"
    cell_resistance_2:
      name: "${name} cell resistance 2"
    cell_resistance_3:
      name: "${name} cell resistance 3"
    cell_resistance_4:
      name: "${name} cell resistance 4"
    cell_resistance_5:
      name: "${name} cell resistance 5"
    cell_resistance_6:
      name: "${name} cell resistance 6"
    cell_resistance_7:
      name: "${name} cell resistance 7"
    cell_resistance_8:
      name: "${name} cell resistance 8"
    cell_resistance_9:
      name: "${name} cell resistance 9"
    cell_resistance_10:
      name: "${name} cell resistance 10"
    cell_resistance_11:
      name: "${name} cell resistance 11"
    cell_resistance_12:
      name: "${name} cell resistance 12"
    cell_resistance_13:
      name: "${name} cell resistance 13"
    cell_resistance_14:
      name: "${name} cell resistance 14"
    cell_resistance_15:
      name: "${name} cell resistance 15"
    cell_resistance_16:
      name: "${name} cell resistance 16"
    cell_resistance_17:
      name: "${name} cell resistance 17"
    cell_resistance_18:
      name: "${name} cell resistance 18"
    cell_resistance_19:
      name: "${name} cell resistance 19"
    cell_resistance_20:
      name: "${name} cell resistance 20"
    cell_resistance_21:
      name: "${name} cell resistance 21"
    cell_resistance_22:
      name: "${name} cell resistance 22"
    cell_resistance_23:
      name: "${name} cell resistance 23"
    cell_resistance_24:
      name: "${name} cell resistance 24"
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    charging_power:
      name: "${name} charging power"
    discharging_power:
      name: "${name} discharging power"
    temperature_sensor_1:
      name: "${name} temperature sensor 1"
    temperature_sensor_2:
      name: "${name} temperature sensor 2"
    power_tube_temperature:
      name: "${name} power tube temperature"
    state_of_charge:
      name: "${name} state of charge"
    capacity_remaining:
      name: "${name} capacity remaining"
    total_battery_capacity_setting:
      name: "${name} total battery capacity setting"
    charging_cycles:
      name: "${name} charging cycles"
    total_charging_cycle_capacity:
      name: "${name} total charging cycle capacity"
    total_runtime:
      name: "${name} total runtime"
    balancing_current:
      name: "${name} balancing current"
    errors_bitmask:
      name: "${name} errors bitmask"

switch:
  - platform: jk_bms_ble
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"
    balancer:
      name: "${name} balancer"

  - platform: ble_client
    ble_client_id: client0
    name: "${name} enable bluetooth connection"

text_sensor:
  - platform: jk_bms_ble
    errors:
      name: "${name} errors"
    total_runtime_formatted:
      name: "${name} total runtime formatted"
```

syssi, 2022-07-08:
Which ESPHome version do you use?

dremeier, 2022-07-08:
[v2021.11.4]

syssi, 2022-07-08:
Could you try to update ESPHome to 2022.6? I've made a copy of your configuration above. I've changed

```
esp32:
  board: esp32dev
  framework:
#  The arduino platform crashs
    type: arduino
    version: latest
#    type: esp-idf
#    version: latest
```

to

```
esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: latest
```

and executed the compile & flash task:

```
$ esphome run dremeier.yaml 
INFO Reading configuration dremeier.yaml...
INFO Cloning https://github.com/syssi/esphome-jk-bms.git@main
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing jk-bms (board: esp32dev; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf 3.40302.0 (4.3.2) 
 - tool-cmake 3.16.4 
 - tool-ninja 1.7.1 
 - toolchain-riscv32-esp 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32s2 8.4.0+2021r2-patch2
Reading CMake configuration...
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5

Unpacking  [------------------------------------]    0%
Unpacking  [------------------------------------]    1%
Unpacking  [------------------------------------]    2%
[...]

INFO Successfully compiled program.
INFO Resolving IP address of jk-bms.local
```

with trouble.

syssi, 2022-07-08:
I it doesn't work out of the box please clean up a build folder by:

```
esphome clean config.yaml
esphome run config.yaml
```

If your dashboard doesn't provide a "clean build" button you could also rename the node to create a new build folder.

dremeier, 2022-07-08:
can´t update the ESPhome dashboard it is fixed in the sources for the iobroker app. But I´ll try to install a comadline ESPhome in a VM. Do you have a hint how to do?

syssi, 2022-07-08:
On linux this is pretty easy. In best case `python3` and `pip3` (the python package manager) is already installed. In worst-case you have to install both first:

```
apt-get install python3 python3-pip
```

Now you can use `pip3` to install the latest ESPHome version:

```
pip3 install esphome
```

In a few weeks/month you can update by:

```
pip3 install esphome -U
```

The package manage will install the package into your home directory. The executable / command line tool can be found here:

```
.local/bin/esphome
```

Either you write the path always in front of the command:

```
.local/bin/esphome run config.yaml
``` 

Or you have to extend your `$PATH` to include ".local/bin/" f.e. by extending the `.bashrc`:

```
if [ -d "$HOME/.local/bin" ] ; then
    PATH="$HOME/.local/bin:$PATH"
fi
```

If you start a new shell now / open a new terminal window the CLI command `esphome` should be available without passing the path.

dremeier, 2022-07-08:
perfect! will do tomorrow, have to go now :-(

syssi, 2022-07-08:
Same here. Have a nice workday! ;-)

dremeier, 2022-07-10:
So, I installed ESPhome on a fresh ubuntu 20.04 and run the the above .yaml with success!
Then I used the esphome-flasher to flash the binary and got the error: "Unexpected error: The firmware binary is invalid (magic byte=FF, should be E9)"

```
Using 'COM4' as serial port.
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Connecting....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WDQ6 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 40:91:51:AB:8C:B8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
Unexpected error: The firmware binary is invalid (magic byte=FF, should be E9)
```

btw:
-in the .yaml, does I have to change the "mac_address: C8:47:8C:E1:E2:E1" to the mac of the BMS? 
-can I use any OTA pw?
-I like to use only mqtt, the framework can still be esp-idf? what I´ve to do with the "api:"?

syssi, 2022-07-10:
1. If you are able to passthrough a USB device you could use `esphome run config.yaml` to flash the ESP using esphome via ttyUSB0 f.e. 

Flashing the firmware file using a generic flasher should also be possible. In this case it is important to pick the correct file. Which file did you try exactly?

2. You have to change the MAC address to the address of your BMS. If you don't know the MAC address yet just let the ESPHome node boot with a wrong MAC address. The node will report all BLE devices at the log. You can lookup the address of your BMS there and flash again using the correct address.

3. You can use any OTA password. If you set none everybody having access to your local network could re-flash/OTA your node.

4. Yes. MQTT is supported at ESP-IDF since a few ESPHome releases. The `api` is just another communication protocol (to talk to a Home Assistant instance). Home Assistant is able to discover and integrate ESPHome nodes with a single click. If you don't use Home Assistant MQTT is the way to go.

dremeier, 2022-07-10:
Thanks!
I flashed now with a passthrough USB and also with the esphome-flasher successfully. I don't know where was the mistake before.
Wifi and BLE connects fine and mqtt with data comes in. All fine so far.
But, after 8 attempts of reconnecting ble, the connection refused. 

maybe you can read more out of the log file?
[output_2022-07-10_15-21-51.log](https://github.com/syssi/esphome-jk-bms/files/9079146/output_2022-07-10_15-21-51.log)

syssi, 2022-07-10:
The BMS can handle a single connection simultaneously only. A new connection can be established only if the red status LED is blinking. Closing the Android app (for example) doesn't stop the BLE connection. You have to force close the app to disconnect from the BMS.

If the status LED is blinking and you are unable to connect using the ESP please reboot the ESP first and restart the BMS as second step to identify the origin of the issue.

dremeier, 2022-07-10:
but the esp is connecting with the BMS for short time and data are showing up in the mqtt broker, just after 8 reconnection the esp is giving up. The BMS app on the iphone is closed and out of range. There is just a second bms with different mac which is connected with another ESP-ble projekt.

syssi, 2022-07-10:
I'm on vacation and didn't have the chance to inspect your log yet. I will have a look later the week.

dremeier, 2022-07-10:
some news:
the esp is now connected to the uart-rs485 port of the BMS. Thats works perfect and stable to the mqtt broker.
Sending data to the api, the most sensor and switch data are missing some are working and after a while the the connection to the api closed, maybe dueto my old version in the esphome app of ioBroker. 
For me just the mqtt is working and is fine. But I guess some like to use the api. How can we debug that issue?

here my yaml files
[batterie-1_uart_api.yaml.txt](h
[batterie-1_UART_MQTT.yaml.txt](https://github.com/syssi/esphome-jk-bms/files/9079480/batterie-1_UART_MQTT.yaml.txt)
ttps://github.com/syssi/esphome-jk-bms/files/9079478/batterie-1_uart_api.yaml.txt)

dremeier, 2022-07-10:
> I'm on vacation and didn't have the chance to inspect your log yet. I will have a look later the week.

enjoy your vacation! Take your time. I´ll try to investigate further :-)

syssi, 2022-07-10:
The UART connection is another protocol and read-only. Please don't expect the same data/measurements/values & features here compared to BLE. I recommend the UART connect to monitor the BMS. To control the switches or change settings a BLE connection is required.

syssi, 2022-07-11:
I've inspected your `output_2022-07-10_15-21-51.log`. I haven't seen any of the exceptions and assertions yet. I will update my ESPHome install in a few weeks and try to reproduce the issue.

syssi, 2022-07-11:
The issue is may be related to this one:

https://github.com/esphome/issues/issues/3327
https://github.com/esphome/esphome/pull/3594

dremeier, 2022-07-11:
the ESP is now up and running with direkt wires. Just reading the data is fine for me.

syssi, 2022-07-12:
Alright. Let's close this issue because it's already about different aspects. Feel free to create a new issue if something doesn't work as expected!

closed 2022-07-12 by dremeier