syssi/esphome-jk-bms #138: esp32 goes offline

magnetus26, 2022-07-08:
I tried a fe different esp32 modules and all stops working after some random time. I switch esps modules with others that I have around the house and they work but the one containing esphome-jk-bms code will go offline. I do have good Wi-Fi signal in the area esphome mmpsolar and victron shunt are working in the same area but this code will ho offline.

syssi, 2022-07-08:
Is the ESPHome node unavailable after some time or does the BLE connection to the BMS stop?

syssi, 2022-07-08:
I'm only aware of a user where the BLE connection doesn't recover after a certain time and a reboot is required. This user did add the `restart` component to the Yaml to be able to restart the node on demand / via automation: https://github.com/syssi/esphome-jk-bms/discussions/119#discussioncomment-3069605

magnetus26, 2022-07-08:
esphome unabailable after some time, i do have the restart component ans an automation set fot 3am to restart the esp32. like i said its weird due to the fact that i switch esp32 modules a few times around and still have the issue. if i programe another code it works fine( for example i have 3 esp32s reading 10 ble thermostats around the house and another fot my mppsolar inverter it happends with all esps modules if i program the jk-bms code but they work fine work weeks even months if i program some other code

syssi, 2022-07-08:
Could you try to monitor (record the log) of the failing setup via a serial connection? I would like to know the reason for getting stuck. It's a boot loop or is the node unable to connect to the WiFi at some point?

magnetus26, 2022-07-08:
I really cant
1. computer too far from esp32
2. if i move esp32 to computer side will not connect to ble device
3. if I was able somehow to move the computer nest to the esp32, how should I connect to the serial port (what pins and speed in the esp32
4. why the node will be unable to connect to the WIFI. I have 2 other nodes mmpsolar and smartshunt and the are connected to the WIFI all the time I never lost connection on those 2 devices the only difference they dont have a ble component

syssi, 2022-07-08:
1. Do you own a notebook?
2. Move the notebook to ESP32 + BMS. ;-)
3. What kind of ESP32 do you use? If it's a development board having a USB-to-Serial chip it's super easy and you should see some log output there already.
5. Yes! "esphome logs config.yaml" will ask something like: Do you want to connect to the ESP via ttyUSB0 or network? Please select "ttyUSB0".

Some guesses about point 4:

1. It's possible the BLE stack disturbs the WiFi stack and the WiFi stack is unable to connect after some time. The strange thing: The native API component has a `reboot_timeout` which defaults to 15 minutes. If the API component is unable to talk to your Home Assistant instance it will reboot the node for safety's sake.

2. It's possible the ESP enters a boot loop. In this case ESPHome should enter the safe mode (after ~10 reboots). No components are loaded at the safe mode. The node should connect to the WiFi and will wait for some update / OTA. Again: The node is available (pingable) in this case but doesn't provide sensor data.

3. If the node is unstable at all and doesn't enter the safe mode / get stuck I look like to see the last log message(s). For example if you power supply isn't beefy enough we should see some message of the brownout detection.

syssi, 2022-07-08:
Another detail I would like to share: You tried to compare the `jk_bms_ble` (unstable) component with the `atc_mithermometer` (stable) component. Both components are using the BLE stack but the `atc_mithermometer` is a pure passive implementation. There is something like a public BLE beacon which is published periodically. The `atc_mithermometer` component acts like an RX-only radio and decodes the beacons. The `jk_bms_ble` component is more complex. The requests are transmitted here using a `ble_client` implementation. The BMS responds with a periodic BLE notification.

syssi, 2022-07-13:
Any updates / new findings? :-)

magnetus26, 2022-07-13:
sorry no updates. have t have time to play with it. taking care of my mom
I do have a laptop I will need to dl phyton and esphome command line will try that route and will post

syssi, 2022-07-13:
Take your time! Caring about your family is important.

syssi, 2022-07-18:
This could also be the root cause: https://github.com/esphome/esphome/pull/3594

magnetus26, 2022-07-20:
> 5. Yes! "esphome logs config.yaml" will ask something like: Do you want to connect to the ESP via ttyUSB0 or network? Please select "ttyUSB0".

I finally  installed Phyton andesphome in my laptop.
should I run the command whithout changing anything? ( esphome logs config.yaml) or do I need to change something in the command?
for how long should I let the log to run?, until it fails? if so it could take 15min or up to 3 or 4 days

syssi, 2022-07-21:
Yes. Just run `esphome logs config.yaml` and try to capture the log until it fails. I assume the last few lines of the log are the important ones.

magnetus26, 2022-07-21:
I get the following error
[image]

syssi, 2022-07-21:
You have to place the configuration yaml of the ESPHome node you want to monitor into the same folder your are executing `esphome` / the CLI tool.

magnetus26, 2022-07-21:
I did
[image]

[image]

[image]

magnetus26, 2022-07-21:
ok installed Git and getting the port selection, ran it and stop it but dont see the log file
[image]

[image]

syssi, 2022-07-21:
Did you select `[1]` by typing the number "1" and press return?

magnetus26, 2022-07-21:
i did and it start loggin but if i stop it Ctrl + c I wont see any files...
does it save the logs or just whatever is in the screen?

syssi, 2022-07-21:
The output is written to the screen only. Please google something like "windows write command output to file". At linux you can do something like:

```
esphome logs config.yaml | tee output.log
```

magnetus26, 2022-07-21:
found it 
esphome logs config.yaml > output.log

[image]

let me know if im in the right track and will run it again
[log.txt](https://github.com/syssi/esphome-jk-bms/files/9160408/log.txt)

magnetus26, 2022-07-21:
ok thaks will monitor HA and see when it stops there and send you the logs.
gaot to run now have to take care of my mom and then to a Dr. appointment
73 de KP4MI

magnetus26, 2022-07-21:
IDK if this happened because the computer went to sleep or thats the actual error

[output.txt](https://github.com/syssi/esphome-jk-bms/files/9162686/output.txt)

[image]

the esp32 stopped working around 4pm my local time
[image]

and its not pinging
[image]

also looks like the uart got stuck too
[log1.txt](https://github.com/syssi/esphome-jk-bms/files/9162731/log1.txt)
[image]

Ctrl + c wont do anything

magnetus26, 2022-07-22:
Windows sucks, first computer went to sleep and stopped the Wi-Fi, now it decided to get and update and reboot so my command prompt stop. I'm trying again. what version of Linux are you using? I have an older laptop and will like to install Linux, cant be anything heavy due o the processor, its an AMD E2 Vision 2 cores only.

syssi, 2022-07-22:
Please try Ubuntu Mate or Xubuntu.

magnetus26, 2022-07-22:
ok this time the Laptop was on with no interruptions.
ESP32 is connected to a USB 3 port

[output.log](https://github.com/syssi/esphome-jk-bms/files/9167807/output.log)

[image]

[image]

syssi, 2022-07-22:
Okay, it looks like the ESP gets stuck here and doesn't recover:

```
[00:10:58][W][wifi_esp32:628]: Event: Disconnected ssid='ciscosb24' bssid=A8:B4:56:1F:72:10 reason='Beacon Timeout'
[00:10:58][W][wifi:120]: WiFi Connection lost... Reconnecting...
[00:10:58][W][wifi:572]: Restarting WiFi adapter...
[00:11:03]E (10158659) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[00:11:03]E (10158659) task_wdt:  - loopTask (CPU 1)
[00:11:03]E (10158659) task_wdt: Tasks currently running:
[00:11:03]E (10158659) task_wdt: CPU 0: IDLE
[00:11:03]E (10158659) task_wdt: CPU 1: IDLE
[00:11:03]E (10158659) task_wdt: Aborting.
[00:11:03]
[00:11:03]abort() was called at PC 0x40127b34 on core 0
[00:11:03]
[00:11:03]Backtrace:0x40081d4e:0x3ffbe440 0x40090d75:0x3ffbe460 0x40096e1e:0x3ffbe480 0x40127b34:0x3ffbe4f0 0x40082a8d:0x3ffbe510 0x401a73f3:0x3ffbced0 0x40127e39:0x3ffbcef0 0x40092118:0x3ffbcf10
```

syssi, 2022-07-22:
I'm back at home and will try to reproduce the issue. The bad thing: It's somewhere deep in the ESP-IDF/ESPHome stack. It doesn't look like an issue of this component.

valdiks, 2022-08-05:
I have issue with connection over ble too.
esp32ble + bms JK02, no connection after some time.

valdiks, 2022-08-05:
Some more information about BLE issue: https://github.com/jblance/mpp-solar/issues/120

syssi, 2022-08-05:
There is a 50:50 chance this issue is unrelated to the mppsolar issue. This is the background:

1. This implementation cannot get stuck. It should recover always. 
2. The ESP should be still available if the BLE connections get stuck but it looks like it crashs hard (offline) and doesn't recover.
3. IMO the ESP gets stuck because the BLE stack is unstable.
4. mppsolar gets probably stuck because of the implementation and the unstable BLE stack of the Raspberry Pi.

syssi, 2022-08-07:
1. Test: `esp32-ble-example.yaml` died after a few seconds because of a watchdog exception.
2. Test: `esp32-ble-example.yaml` + `throttle: 10s` got stuck after 5 hours  (no WiFi, no exception, reboot because of `no api connection)

```
[23:28:21][E][api:127]: No client connected to API. Rebooting...
[23:28:21][I][app:127]: Forcing a reboot...
[23:28:21][D][esp-idf:000]: I (906502) wifi:
[23:28:21][D][esp-idf:000]: state: run -> init (0)
[23:28:21][D][esp-idf:000]: 
[23:28:21]
[23:28:21][D][esp-idf:000]: I (906504) wifi:
[23:28:21][D][esp-idf:000]: pm stop, total sleep time: 614490580 us / 900691810 us
[23:28:21]
[23:28:21][D][esp-idf:000]: 
[23:28:21]
[23:28:21][D][esp-idf:000]: W (906506) wifi:
[23:28:21][D][esp-idf:000]: <ba-del>idx
[23:28:21][D][esp-idf:000]: 
[23:28:21]
[23:28:21][D][esp-idf:000]: I (906517) wifi:
[23:28:21][D][esp-idf:000]: new:<3,0>, old:<3,0>, ap:<255,255>, sta:<3,0>, prof:1
[23:28:21][D][esp-idf:000]: 
[23:28:21]
[23:28:21][D][esp-idf:000]: W (906535) wifi:
[23:28:21][D][esp-idf:000]: hmac tx: ifx0 stop, discard

[...]

[23:28:29][C][ble_client:035]: BLE Client:
[23:28:29][C][ble_client:036]:   Address: c8:47:8c:f2:7e:03
[23:28:29][C][mdns:084]: mDNS:
[23:28:29][C][mdns:085]:   Hostname: jk-bms-test
[23:28:29][C][ota:085]: Over-The-Air Updates:
[23:28:29][C][ota:086]:   Address: jk-bms-test.local:3232
[23:28:29][C][api:138]: API Server:
[23:28:29][C][api:139]:   Address: jk-bms-test.local:6053
[23:28:29][C][api:143]:   Using noise encryption: NO
[23:28:31][W][jk_bms_ble:267]: [c8:47:8c:f2:7e:03] Not connected
[23:28:33][D][esp32_ble_tracker:726]: Found device A4:C1:38:A1:9F:81 RSSI=-85
[23:28:33][D][esp32_ble_tracker:747]:   Address Type: PUBLIC
[23:28:33][D][esp32_ble_tracker:749]:   Name: 'ATC_A19F81'
[23:28:36][W][jk_bms_ble:267]: [c8:47:8c:f2:7e:03] Not connected
[23:28:38][D][ble_client:047]: Found device at MAC address [C8:47:8C:F2:7E:03]
```

3. Test: WLAN disabled. BLE only.

magnetus26, 2022-08-07:
what are you trying to accomplish with test 3? just ble logs via uart?

syssi, 2022-08-07:
I've removed the `wifi` section and the WiFi dependent components (`ota` and `api`). If the setup is stable in this szenario there is no memory leak at the external component and we have to find the root cause between the BLE and WiFi stack.

magnetus26, 2022-08-07:
roger that. thaks for tour hard work

syssi, 2022-08-12:
Test 3 logs good until now. No BLE connection drops, reconnects or restarts since 5 days of monitoring.

magnetus26, 2022-08-12:
so now we have to find the root cause between the BLE and WiFi stack. i dont know how to do that but im willing to test your code if necessary

syssi, 2022-08-12:
The journey doesn't stop here. I will let the test running for another 5 days. As next step I will re-enable WiFi again and try to find out how to trigger the issue.

jakehdk, 2022-09-07:
The suspense is killing me.. ;)
Any news regarding this? 

I am also having the same problem 2-4 times a day. 
It recovers itself so I am in no hurry, but I am also curious about what can cause this. :)

syssi, 2022-09-07:
It's super hard to debug. No news yet unfortunately.

syssi, 2022-09-12:
Please reflash your device using ESPHome 2022.8.3 and try again.

jakehdk, 2022-09-14:
Just reflashed with 2022.8.3 last night and unfortunately the dropouts still happens.
[image]

syssi, 2022-09-14:
@jakehdk Could you provide your configuration yaml? Do you have log captures of the dropout (recorded via serial line)? Please `throttle` the sensor updates to `10s` or more. Does this increase the stability?

```
jk_bms_ble:
  - ble_client_id: client0
    protocol_version: ${protocol_version}
    throttle: 10
    id: bms0
```

As next step. Please add the `debug` component to your setup:

```
debug:
  update_interval: 5s

text_sensor:
  - platform: debug
    device:
      name: "Device Info"

sensor:
  - platform: debug
    free:
      name: "Heap Free"
    block:
      name: "Heap Max Block"
    loop_time:
      name: "Loop Time"
```

I would like to track the free memory.

jakehdk, 2022-09-18:
Unfortunately I have no way of logging serial data directly, since the battery is located in a shed by itself. But I have now added the debug components - lets see how it goes at throttle 5. :)

yaml:
[esphome_JK_config.txt](https://github.com/syssi/esphome-jk-bms/files/9594044/esphome_JK_config.txt)

jakehdk, 2022-09-22:
Just had a dropout today after running without any drops for almost 4 days.
Seems like heap max block (whatever that is?) spikes just around that same time.

[image]

Let's see if it the the same pattern next time.

syssi, 2022-09-22:
Thanks for your support! Your analysis is very helpful.

syssi, 2022-09-23:
Do I read the grafana graph correctly: The drop-out happens at 12 o''clock?

jakehdk, 2022-09-23:
> Do I read the grafana graph correctly: The drop-out happens at 12 o''clock?

Yep.

woodie2, 2022-09-28:
same here with throttle 5s: after a while, the esp only sends debug messages: [jk_bms_ble:341] MAC: not connected
so wifi, and mqtt is working, but esp should send a new request to bms, to get connection again.

is it not possible to count ' not connected' state and after a while simply do a ble reconnection?

syssi, 2022-09-28:
The implementation reconnects of the BLE connection gets lost. It's possible this takes some minutes because the BMS doesn't accept / cannot handle multiple connections. The BMS is only ready to connect if the red LED is blinking.

woodie2, 2022-09-28:
Yes I know, but while the ESP shows 'not connected' I saw my BMS's blinking for new connection. I also saw my both BMS in Android App, so there was no connection to the BMSs. After restart of esp, both BMS can't be found in Andorid, and was connect again to the esp.

woodie2, 2022-09-28:
should be there a debug message with' try reconnect' or where to see that?

syssi, 2022-09-28:
We are talking about this "early exit" of the update loop: https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/jk_bms_ble.cpp#L341-L344

The warning is printed periodically to the log if there is no connection to the BMS:

```
  if (this->node_state != espbt::ClientState::ESTABLISHED) {
```

No BLE disconnect the `node_state` is changed to `IDLE`:

https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/jk_bms_ble.cpp#L136-L142

As next the the BLE tracker should discover the BMS again and reconnect:

https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/jk_bms_ble.cpp#L143-L149

In general this state machine is working fine.

woodie2, 2022-09-28:
ok I think it can help to get also a message in case of ESP_GATTC_DISCONNECT_EVT and ESP_GATTC_SEARCH_CMPL_EVT or?
In my case it doesn't tried again to reconnect.

syssi, 2022-09-28:
Good idea. I will add some info messages!

woodie2, 2022-10-11:
is it already on github to test with info messages?

syssi, 2022-11-18:
The BLE client implementation was improved. Please update to ESPHome `>2022.11.0` and try again.

magnetus26, 2022-11-18:
mine has been working good for a few months now. anyhow im dowloading `2022.11.1` and will report aftet a few days of testing
 btw im using `2022.10.2` atm

ans-gw, 2022-12-18:
> Hi all, I m the user wrote this:[#119 (reply in thread)](https://github.com/syssi/esphome-jk-bms/discussions/119#discussioncomment-3069605) even if I set 4 daily Esp32 restart,I keep on having the problem: Esp32 ramdomly loose connection...today 2 times.

The problem is fixed updating Esphome.

syssi, 2022-12-18:
It looks like the issue is resolved. I will close this issue now.

closed 2022-12-18 by syssi