syssi/esphome-jk-bms #139: Security concerns/question

J4nsen, 2022-07-10:
Hi,

I have some security concerns regarding theJK-BMS.

Am I right that the pins one needs to use in the offical app are somewhat snake oil? I assume that they are sent in the messages to the app and only verified inside the app, but not by the BMS itself?

Do you know if a constant connection from an esp via BT-LE to the BMS prevents attackers from exploting this behaviour?

syssi, 2022-07-10:
Your assumption is correct. This is my assumption (never verified): If you connect to the BMS and request the device info the BMS returns a frame containing the name of the BMS and the current password. If you change a restricted settings in the android app the entered/required password is compared to the value from the device info frame (locally). No real security here. ;-)

TL&DR: No password required to control the BMS.

> Do you know if a constant connection from an esp via BT-LE to the BMS prevents attackers from exploting this behaviour?

I don't have much knowledge about possible attacks of BLE connections. The ESP connection to the BMS isn't rock solid. An attacker could connect to the BMS on every "connection reset".

syssi, 2022-07-11:
The advantage of a permanent BLE connection: The BLE module doesn't advertise the BMS if an connection is active. Your neighbor won't see the device permanently anymore at the list of nearby bluetooth devices. ;-)

J4nsen, 2022-07-11:
Thanks for your answer. This BL connection thingy without any real authentication/authorization makes me somewhat uncomfortable. 

Do you know a way to disable bluetooth in the BMS or is there a way to make it read-only?

I like the idea of "hiding" the device with a permanent connection by the ESP. Even if it more security through obscurity...

syssi, 2022-07-11:
If you open the BMS you could pull the RST pin of the bluetooth module to GND to disable the module permanently. Did you see the issue about wired communication with the BMS using the serial lines of the BLE module? This is possible but not implemented yet.

syssi, 2022-07-12:
I re-open the issue as reminder to extend the docs.

syssi, 2023-04-14:
How to occupy the BLE module to stop advertising the BMS using BLE: https://github.com/syssi/esphome-jk-bms/blob/main/yaml-snippets/esp32-ble-block-traffic.yaml