syssi/esphome-jk-bms #140: Error after adding switch esp32-ble

channaq4, 2022-07-12:
I have the following error after adding switch and I'm using VMware to compile "esp-idf"
(without switch programe is working) please advise on this issue.

"INFO Reading configuration /config/esphome/jk-bms-ble.yaml...
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
INFO Generating C++ source...
INFO Compiling app...
Processing jk-bms-ble (board: wemos_d1_mini32; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf 3.40302.0 (4.3.2) 
 - tool-cmake 3.16.4 
 - tool-ninja 1.7.1 
 - toolchain-riscv32-esp 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32s2 8.4.0+2021r2-patch2
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <noise-c> 0.1.4
|   |-- <libsodium> 1.10018.1
Compiling /data/jk-bms-ble/.pioenvs/jk-bms-ble/src/main.o
Linking /data/jk-bms-ble/.pioenvs/jk-bms-ble/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/jk-bms-ble/.pioenvs/jk-bms-ble/src/main.o:(.literal._Z5setupv+0x1b0): undefined reference to `vtable for esphome::jk_bms_ble::JkSwitch'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/jk-bms-ble/.pioenvs/jk-bms-ble/src/main.o:(.literal._Z5setupv+0x1b4): undefined reference to `vtable for esphome::jk_bms_ble::JkSwitch'
collect2: error: ld returned 1 exit status
*** [/data/jk-bms-ble/.pioenvs/jk-bms-ble/firmware.elf] Error 1
========================= [FAILED] Took 22.93 seconds ========================="

here is my code

substitutions:
  name: jk-bms-ble
  external_components_source: github://syssi/esphome-jk-bms@main

esphome:
  name: ${name}

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf
    version: latest

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
  password: ""
  
logger:

api:
  encryption:
    key: ""

esp32_ble_tracker:

ble_client:
  - mac_address: !secret jk_mac_address
    id: client0

jk_bms_ble:
  - ble_client_id: client0
    throttle: 5s
    id: bms0

binary_sensor:
  - platform: jk_bms_ble
    balancing:
      name: "${name} balancing"
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"

sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    average_cell_voltage:
      name: "${name} average cell voltage"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
    cell_voltage_8:
      name: "${name} cell voltage 8"
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"   
    temperature_sensor_1:
      name: "${name} temperature sensor 1"
    temperature_sensor_2:
      name: "${name} temperature sensor 2"
    state_of_charge:
      name: "${name} state of charge"
    capacity_remaining:
      name: "${name} capacity remaining"
    balancing_current:
      name: "${name} balancing current"      

switch:
  platform: jk_bms_ble
  charging:
     name: "${name} charging"
#   discharging:
#     name: "${name} discharging"

syssi, 2022-07-12:
A clean build folder is required:

```
esphome clean config.yaml
esphome run config.yaml
```

should build without issues.

channaq4, 2022-07-13:
yes now its working :) thanks

closed 2022-07-13 by syssi