syssi/esphome-jk-bms #147: BLE JK_B2A8S20P with Firmware 11.17 only cell voltages are working

senfkorn, 2022-08-04:
Hi,

I've received my JK BMS today and directly connected it via your nice library to my home assistant server.

The cell voltages and the configuration data are working flawlessly.
Unfortunately, everything else is 0 or some random numbers.

Could you be so kind and have a look at my configuration? Have I done something wrong?

Here a screenshot of the sensors:

[image]

Here a log of the esp device:

[logs_pv-battery-bms_upload.txt](https://github.com/syssi/esphome-jk-bms/files/9259845/logs_pv-battery-bms_upload.txt)

And finaly here my yaml file:
[pv-battery-bms.txt](https://github.com/syssi/esphome-jk-bms/files/9259867/pv-battery-bms.txt)

Thanks in advance

Andy

syssi, 2022-08-04:
Thanks for providing a `VERY_VERBOSE` log. It includes all required informations to improve the frame decoder. I will have a look!

syssi, 2022-08-04:
```
[14:36:04][I][jk_bms_ble:1061]:   Vendor ID: JK_B2A8S20P
[14:36:04][I][jk_bms_ble:1062]:   Hardware version: 11.XW
[14:36:04][I][jk_bms_ble:1063]:   Software version: 11.17
[14:36:04][I][jk_bms_ble:1064]:   Uptime: 10800 s
[14:36:04][I][jk_bms_ble:1065]:   Power on count: 1
[14:36:04][I][jk_bms_ble:1066]:   Device name: JK_B2A8S20P
[14:36:04][I][jk_bms_ble:1067]:   Device passcode: 1234
[14:36:04][I][jk_bms_ble:1068]:   Manufacturing date: 220804
[14:36:04][I][jk_bms_ble:1069]:   Serial number: 2062015232
[14:36:04][I][jk_bms_ble:1070]:   Passcode: 0000
[14:36:04][I][jk_bms_ble:1071]:   User data: Input Userdata
[14:36:04][I][jk_bms_ble:1072]:   Setup passcode: 123456
```

syssi, 2022-08-04:
Please update your configuration yaml to:

```
substitutions:
  name: jk-bms
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@add-jk02-v11-support
```

and flash again. The protocol has changed a bit. In the past the maximum supported cells was 24. Now space for 32 cells is reserved / used:

```
0x55 0xAA 0xEB 0x90 Header
0x02                Record type (cell info)
0x8B                Frame counter
0xD7 0x0C           Voltage cell 01
0xDB 0x0C           Voltage cell 02
0xD5 0x0C           Voltage cell 03
0xFF 0x0C           Voltage cell 04
0x00 0x0D           Voltage cell 05
0xD5 0x0C           Voltage cell 06
0xD4 0x0C           Voltage cell 07
0xE0 0x0C           Voltage cell 08
0x00 0x00           Voltage cell 09
0x00 0x00           Voltage cell 10
0x00 0x00           Voltage cell 11
0x00 0x00           Voltage cell 12
0x00 0x00           Voltage cell 13
0x00 0x00           Voltage cell 14
0x00 0x00           Voltage cell 15
0x00 0x00           Voltage cell 16
0x00 0x00           Voltage cell 17
0x00 0x00           Voltage cell 18
0x00 0x00           Voltage cell 19
0x00 0x00           Voltage cell 20
0x00 0x00           Voltage cell 21
0x00 0x00           Voltage cell 22
0x00 0x00           Voltage cell 23
0x00 0x00           Voltage cell 24
0x00 0x00           Voltage cell 25?
0x00 0x00           Voltage cell 26?
0x00 0x00           Voltage cell 27?
0x00 0x00           Voltage cell 28?
0x00 0x00           Voltage cell 29?
0x00 0x00           Voltage cell 30?
0x00 0x00           Voltage cell 31?
0x00 0x00           Voltage cell 32?
0xFF 0x00 0x00 0x00 Enabled cells (8)

0xE2 0x0C           Average Cell Voltage
0x2B 0x00           Delta Cell Voltage
0x03                Max voltage cell
0x05                Min voltage cell
0x33 0x00           Resistance Cell 01
0x33 0x00           Resistance Cell 02
0x32 0x00           Resistance Cell 03
0x32 0x00           Resistance Cell 04
0x32 0x00           Resistance Cell 05
0x32 0x00           Resistance Cell 06
0x31 0x00           Resistance Cell 07
0x32 0x00           Resistance Cell 08
0x00 0x00           Resistance Cell 09
0x00 0x00           Resistance Cell 10
0x00 0x00           Resistance Cell 11
0x00 0x00           Resistance Cell 12
0x00 0x00           Resistance Cell 13
0x00 0x00           Resistance Cell 14
0x00 0x00           Resistance Cell 15
0x00 0x00           Resistance Cell 16
0x00 0x00           Resistance Cell 17
0x00 0x00           Resistance Cell 18
0x00 0x00           Resistance Cell 19
0x00 0x00           Resistance Cell 20
0x00 0x00           Resistance Cell 21
0x00 0x00           Resistance Cell 22
0x00 0x00           Resistance Cell 23
0x00 0x00           Resistance Cell 24
0x00 0x00           Resistance Cell 25
0x00 0x00           Resistance Cell 26
0x00 0x00           Resistance Cell 27
0x00 0x00           Resistance Cell 28
0x00 0x00           Resistance Cell 29
0x00 0x00           Resistance Cell 30
0x00 0x00           Resistance Cell 31
0x00 0x00           Resistance Cell 32
0x69 0x01
0x00 0x00
0x00 0x00
0x0D 0x67
0x00 0x00
0x00 0x00
0x00 0x00
0x00 0x00
0x00 0x00
0x29 0x01
0x2A 0x01
0x00 0x00
0x00 0x00
0x1D 0xF8
0x02 0x45
0xB2 0x0F
0x01 0x00
0xA0 0x86
0x01 0x00
0x00 0x00
0x00 0x00
0x2E 0x00
0x00 0x00
0x64 0x00
0x00 0x00
0xED 0x2A
0x00 0x00
0x01 0x01
0x00 0x00
0x00 0x00
0x00 0x00
0x00 0x00
0x00 0x00
0x00 0x00
0x00 0x00
0x07 0x00
0x01 0x00
0x00 0x00
0x9F 0x03
0x00 0x00
0x00 0x00
0xA7 0xB3
0x40 0x40
0xA3 0x00
0x00 0x00
0x4E 0x0A
0xD8 0xE1
0x00 0x01
0x00 0x01
0x00 0x05
0x00 0x00
0x44 0xAD
0x01 0x00
0x00 0x00
0x00 0x00
0x00 0x00
0x00 0x00
0x00 0x00
0x00 0x00
0x00 0x00
0x00 0x00
0x00 0x00
0x00 0x00
0x00 0x00
0x00 0x00
0x00 0x00
0x00 0x00
0x00 0x00
0x00 0x00
0x00 0x00
0x00 0x00
0x00 0xFE
0xFF 0x7F
0xDC 0x1F
0x01 0x00
0x00 0x00
0x00 0x00
0x00
0xF4          CRC
```

senfkorn, 2022-08-04:
Hei,

many thanks, you are crazy fast. 

It seems to work very well. I've also poked a little bit around in the bytes and I think the mosfet temperature is also at a wrong position. In the source it is at byte 134 but I think in my case it is at byte 144. So not behind Temperature 1 and 2. Instead 2 bytes in front of the total voltage.

Again, many thanks for this fast solution. 

PS you should post somewhere a "by me a coffee" or paypal link, so that I could throw you a dime in the hat, for that amazing performance :)

Andy

[image]

syssi, 2022-08-04:
Thanks for your feedback and providing more details about your device. I will improve the feature branch ASAP. There is a buy me a coffee link as icon/badge at the top of the README.

syssi, 2022-08-05:
Could you do me a favor? Could you verify the position of the `system alarm` bytes? I'm talking about:

https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/jk_bms_ble.cpp#L455-L476

You could change the cell count from 8S to 7S. This will raise an error. Is this error reported properly by this component or do we need to adjust the position of the `system alarm bitmask`? If you aren't sure please provide another `VERY_VERBOSE` log including a system alarm. If you don't want to change the cell count you could cool down one of the temperature probes.

senfkorn, 2022-08-05:
Sorry for my late response.

Here are two responses:

//OK
[16:01:11][VV][jk_bms_ble:367]: Cell info frame (300 bytes):
[16:01:11][VV][jk_bms_ble:368]:   55.AA.EB.90.02.38.10.0E.0C.0E.5D.0D.0D.0E.0D.0E.BD.0D.0F.0E.38.0D.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.00.00.D3.0D.D8.00.00.07.33.00.33.00.33.00.33.00.33.00.32.00.33.00.33.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.68.01.00.00.00.00 (150)
[16:01:11][VV][jk_bms_ble:369]:   9B.6E.00.00.00.00.00.00.00.00.00.00.34.01.3A.01.10.00.00.00.D0.07.01.64.A0.86.01.00.A0.86.01.00.00.00.00.00.C1.00.00.00.64.00.00.00.74.89.01.00.00.01.00.00.00.00.00.00.00.00.00.00.00.00.02.00.07.00.01.00.00.00.9F.03.00.00.00.00.A7.B3.40.40.00.00.CC.00.0F.0B.C0.E1.00.01.00.01.00.05.00.00.42.05.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DC.1F.01.00.00.00.00.00.00.8B (150)

// Fault 
[16:02:42][VV][jk_bms_ble:367]: Cell info frame (300 bytes):
[16:02:42][VV][jk_bms_ble:368]:   55.AA.EB.90.02.B8.FD.0D.FA.0D.5A.0D.FD.0D.FC.0D.9C.0D.FC.0D.38.0D.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.7F.00.00.00.D7.0D.A3.00.03.02.33.00.33.00.33.00.33.00.33.00.32.00.33.00.33.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.68.01.00.00.00.00 (150)
[16:02:42][VV][jk_bms_ble:369]:   E4.60.00.00.00.00.00.00.00.00.00.00.34.01.3A.01.10.00.00.00.4C.F8.02.64.A0.86.01.00.A0.86.01.00.00.00.00.00.C1.00.00.00.64.00.00.00.CF.89.01.00.00.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.07.00.01.00.00.00.9F.03.00.00.00.00.A7.B3.40.40.A8.00.00.00.B0.09.C3.DF.00.01.00.01.00.05.00.00.D0.08.0B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DC.1F.01.00.00.00.00.00.00.E5 (150)

Here a corresponding screenshot of the app while in error state

[image]

syssi, 2022-08-05:
```
POS  OK  ERR
0  55  55
1  AA  AA
2  EB  EB
3  90  90
4  02  02
5  38  B8
6  10  FD
7  0E  0D
8  0C  FA
9  0E  0D
10  5D  5A
11  0D  0D
12  0D  FD
13  0E  0D
14  0D  FC
15  0E  0D
16  BD  9C
17  0D  0D
18  0F  FC
19  0E  0D
20  38  38
21  0D  0D
22  00  00
23  00  00
24  00  00
25  00  00
26  00  00
27  00  00
28  00  00
29  00  00
30  00  00
31  00  00
32  00  00
33  00  00
34  00  00
35  00  00
36  00  00
37  00  00
38  00  00
39  00  00
40  00  00
41  00  00
42  00  00
43  00  00
44  00  00
45  00  00
46  00  00
47  00  00
48  00  00
49  00  00
50  00  00
51  00  00
52  00  00
53  00  00
54  00  00
55  00  00
56  00  00
57  00  00
58  00  00
59  00  00
60  00  00
61  00  00
62  00  00
63  00  00
64  00  00
65  00  00
66  00  00
67  00  00
68  00  00
69  00  00
70  FF  7F
71  00  00
72  00  00
73  00  00
74  D3  D7
75  0D  0D
76  D8  A3
77  00  00
78  00  03
79  07  02
80  33  33
81  00  00
82  33  33
83  00  00
84  33  33
85  00  00
86  33  33
87  00  00
88  33  33
89  00  00
90  32  32
91  00  00
92  33  33
93  00  00
94  33  33
95  00  00
96  00  00
97  00  00
98  00  00
99  00  00
100  00  00
101  00  00
102  00  00
103  00  00
104  00  00
105  00  00
106  00  00
107  00  00
108  00  00
109  00  00
110  00  00
111  00  00
112  00  00
113  00  00
114  00  00
115  00  00
116  00  00
117  00  00
118  00  00
119  00  00
120  00  00
121  00  00
122  00  00
123  00  00
124  00  00
125  00  00
126  00  00
127  00  00
128  00  00
129  00  00
130  00  00
131  00  00
132  00  00
133  00  00
134  00  00
135  00  00
136  00  00
137  00  00
138  00  00
139  00  00
140  00  00
141  00  00
142  00  00
143  00  00
144  68  68
145  01  01
146  00  00
147  00  00
148  00  00
149  00  00
150  9B  E4
151  6E  60
152  00  00
153  00  00
154  00  00
155  00  00
156  00  00
157  00  00
158  00  00
159  00  00
160  00  00
161  00  00
162  34  34
163  01  01
164  3A  3A
165  01  01
166  10  10
167  00  00
168  00  00 <-- No errors!?
169  00  00 <-- No errors!?
170  D0  4C
171  07  F8
172  01  02
173  64  64
174  A0  A0
175  86  86
176  01  01
177  00  00
178  A0  A0
179  86  86
180  01  01
181  00  00
182  00  00
183  00  00
184  00  00
185  00  00
186  C1  C1
187  00  00
188  00  00
189  00  00
190  64  64
191  00  00
192  00  00
193  00  00
194  74  CF
195  89  89
196  01  01
197  00  00
198  00  00
199  01  01
200  00  00
201  00  00
202  00  00
203  00  00
204  00  00
205  00  00
206  00  00
207  00  00
208  00  00
209  00  00
210  00  00
211  00  00
212  02  00
213  00  00
214  07  07
215  00  00
216  01  01
217  00  00
218  00  00
219  00  00
220  9F  9F
221  03  03
222  00  00
223  00  00
224  00  00
225  00  00
226  A7  A7
227  B3  B3
228  40  40
229  40  40
230  00  A8
231  00  00
232  CC  00
233  00  00
234  0F  B0
235  0B  09
236  C0  C3
237  E1  DF
238  00  00
239  01  01
240  00  00
241  01  01
242  00  00
243  05  05
244  00  00
245  00  00
246  42  D0
247  05  08
248  0B  0B
249  00  00
250  00  00
251  00  00
252  00  00
253  00  00
254  00  00
255  00  00
256  00  00
257  00  00
258  00  00
259  00  00
260  00  00
261  00  00
262  00  00
263  00  00
264  00  00
265  00  00
266  00  00
267  00  00
268  00  00
269  00  00
270  00  00
271  00  00
272  00  00
273  00  00
274  00  00
275  00  00
276  00  00
277  00  00
278  00  00
279  00  00
280  00  00
281  00  00
282  00  00
283  00  00
284  00  00
285  00  00
286  00  00
287  FE  FE
288  FF  FF
289  7F  7F
290  DC  DC
291  1F  1F
292  01  01
293  00  00
294  00  00
295  00  00
296  00  00
297  00  00
298  00  00
299  8B  E5
```

Either your frame does not contain an error or the error bitmask position has changed. Could you double check?

senfkorn, 2022-08-06:
Hi,
honestly, I've looked for the error_bits for an hour now. I think I'm blind or there somewhere else.

Here again a bytestream with a different error (overvoltage) maybe you have better eyes

POS	OK	ERR
0	55	55
1	AA	AA
2	EB	EB
3	90	90
4	02	02
5	D2	BF
6	DA	D8
7	0D	0D
8	DB	D6
9	0D	0D
10	C0	BD
11	0D	0D
12	DC	DB
13	0D	0D
14	DD	DA
15	0D	0D
16	C0	B5
17	0D	0D
18	DF	D7
19	0D	0D
20	DC	D8
21	0D	0D
22	00	00
23	00	00
24	00	00
25	00	00
26	00	00
27	00	00
28	00	00
29	00	00
30	00	00
31	00	00
32	00	00
33	00	00
34	00	00
35	00	00
36	00	00
37	00	00
38	00	00
39	00	00
40	00	00
41	00	00
42	00	00
43	00	00
44	00	00
45	00	00
46	00	00
47	00	00
48	00	00
49	00	00
50	00	00
51	00	00
52	00	00
53	00	00
54	00	00
55	00	00
56	00	00
57	00	00
58	00	00
59	00	00
60	00	00
61	00	00
62	00	00
63	00	00
64	00	00
65	00	00
66	00	00
67	00	00
68	00	00
69	00	00
70	FF	FF
71	00	00
72	00	00
73	00	00
74	D5	D0
75	0D	0D
76	21	26
77	00	00
78	03	04
79	05	05
80	33	32
81	00	00
82	33	33
83	00	00
84	33	33
85	00	00
86	33	32
87	00	00
88	33	32
89	00	00
90	32	32
91	00	00
92	33	32
93	00	00
94	33	32
95	00	00
96	00	00
97	00	00
98	00	00
99	00	00
100	00	00
101	00	00
102	00	00
103	00	00
104	00	00
105	00	00
106	00	00
107	00	00
108	00	00
109	00	00
110	00	00
111	00	00
112	00	00
113	00	00
114	00	00
115	00	00
116	00	00
117	00	00
118	00	00
119	00	00
120	00	00
121	00	00
122	00	00
123	00	00
124	00	00
125	00	00
126	00	00
127	00	00
128	00	00
129	00	00
130	00	00
131	00	00
132	00	00
133	00	00
134	00	00
135	00	00
136	00	00
137	00	00
138	00	00
139	00	00
140	00	00
141	00	00
142	00	00
143	00	00
144	4F	4F
145	01	01
146	00	00
147	00	00
148	00	00
149	00	00
150	A6	83
151	6E	6E
152	00	00
153	00	00
154	00	00
155	00	00
156	00	00
157	00	00
158	00	00
159	00	00
160	00	00
161	00	00
162	24	24
163	01	01
164	2E	2E
165	01	01
166	00	10
167	00	00
168	00	00
169	00	00
170	A8	96
171	07	F8
172	01	02
173	62	63
174	A9	9F
175	82	86
176	01	01
177	00	00
178	A0	A0
179	86	86
180	01	01
181	00	00
182	00	00
183	00	00
184	00	00
185	00	00
186	FC	FD
187	00	00
188	00	00
189	00	00
190	64	64
191	00	00
192	00	00
193	00	00
194	78	0B
195	CD	CE
196	01	01
197	00	00
198	01	00
199	01	01
200	00	00
201	00	00
202	00	00
203	00	00
204	00	00
205	00	00
206	00	00
207	00	00
208	00	00
209	00	00
210	00	00
211	00	00
212	00	00
213	00	00
214	07	07
215	00	00
216	01	01
217	00	00
218	00	00
219	00	00
220	9F	9F
221	03	03
222	00	00
223	00	00
224	00	00
225	00	00
226	A7	A7
227	B3	B3
228	40	40
229	40	40
230	00	A5
231	00	00
232	2C	00
233	01	00
234	10	0D
235	0B	0B
236	D1	CB
237	DF	E1
238	00	00
239	01	01
240	00	00
241	01	01
242	00	00
243	05	05
244	00	00
245	00	00
246	70	29
247	AD	B3
248	0D	0D
249	00	00
250	00	00
251	00	00
252	00	00
253	00	00
254	00	00
255	00	00
256	00	00
257	00	00
258	00	00
259	00	00
260	00	00
261	00	00
262	00	00
263	00	00
264	00	00
265	00	00
266	00	00
267	00	00
268	00	00
269	00	00
270	00	00
271	00	00
272	00	00
273	00	00
274	00	00
275	00	00
276	00	00
277	00	00
278	00	00
279	00	00
280	00	00
281	00	00
282	00	00
283	00	00
284	00	00
285	00	00
286	00	00
287	FE	FE
288	FF	FF
289	7F	7F
290	DC	DC
291	1F	1F
292	01	01
293	00	00
294	00	00
295	00	00
296	00	00
297	00	00
298	00	00
299	EA	3A

Screenshot without error:

[image]

Screenshot with error:

[image]

syssi, 2022-08-06:
Could you provide 20 frames without errors and 20 frames including the same error? It should be easier to see a pattern here instead of comparing two frames.

senfkorn, 2022-08-06:
Here are the two files:

[OK.txt](https://github.com/syssi/esphome-jk-bms/files/9275492/OK.txt)
[ERR.txt](https://github.com/syssi/esphome-jk-bms/files/9275493/ERR.txt)

Hope it helps.

syssi, 2022-08-06:
[OK.txt](https://github.com/syssi/esphome-jk-bms/files/9275540/OK.txt)
[ERR.txt](https://github.com/syssi/esphome-jk-bms/files/9275541/ERR.txt)

```
cat OK.txt ERR.txt | datamash -W transpose | sed 's#\t##g'
```

```
............................OK..............................  ...........................ERR.............................
 1 2 3 4 5 6 7 8 9101112131415161718192021222324252627282930  1 2 3 4 5 6 7 8 9101112131415161718192021222324252627282930

555555555555555555555555555555555555555555555555555555555555 555555555555555555555555555555555555555555555555555555555555
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
EBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEB EBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEBEB
909090909090909090909090909090909090909090909090909090909090 909090909090909090909090909090909090909090909090909090909090
020202020202020202020202020202020202020202020202020202020202 020202020202020202020202020202020202020202020202020202020202
858F99A3ADB7C1CBD5DFE9F3FD07111B252F39434D57616B757F89939DA7 67717B858F99A3ADB7C1CBD5DFE9F3FD07111B252F39434D57616B757F89
8F8F8F908F90908F8F908F8F909090908F8F908F908F8F90908F90908F8F 8F8F908F8F8F908F8F8F8F8F8F8F8F8F8F8F8F8F8F8F8F8F8F8D8F8F8F8F
0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C 0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C
D6D5D5D6D6D6D5D6D5D5D5D5D5D5D6D5D5D5D5D5D6D6D4D5D5D5D5D5D5D6 D4D5D5D4D5D5D5D4D5D5D4D4D4D5D4D5D5D4D4D5D4D4D4D4D5D5D4D4D4D5
0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C 0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C
A09F9FA09F9FA0A1A09F9FA0A0A0A0A09F9FA0A0A0A0A09FA0A09F9FA09F 9F9F9F9D9F9F9F9F9F9F9F9F9F9F9F9F9F9F9F9F9D9F9D9F9D9D9F9D9F9F
0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C 0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C
E0E0E0E1E0E1E0E0E0E1E1E1E0E0E0E1E0E0E0E1E0E1E0E0E1E1E1E0E0E0 DFDFDFDFDFE0E0DFDFE0DFDFDFDFDFDFDFE0DFDFE0DFDFDFDFDFDFDFDFDF
0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C 0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C
E0E1E3E1E0E1E0E0E0E0E0E3E0E1E1E3E1E1E3E0E1E1E0E0E0E0E0E0E0E1 DFE0E0DFE0E0E0E0DFE0DFE0E0E0E0DFDFDFDFDFE0E0DFDFDFDFE0DFDFDF
0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C 0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C
8F908F908F908F90908F8F8F8F8F90908F908F8F90908F8F8F90908F908F 8F8F8F8F8D8F8F8F8F8F8F8D8F8F8F8F8F8F8F8F8D8F8F8D8D8F8F8D8D8F
0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C 0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C
8F8F8F8F908D8F8F8F8F8D8F908F8F8F908F908F8F8F8D908F8F8F8F8F8F 8F8D8F8D8F8D8F8F8D8D8F8D8D8F8D8F8D8D8F8F8D8F8D8D8D8D8D8D8D8F
0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C 0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C
E1E1E3E0E1E3E1E1E0E3E1E1E3E1E1E1E1E0E3E1E3E0E0E1E0E1E1E0E1E1 E0E0E0E0E0E0E0E0E0E0DFE0E0E0E0E0DFE0E0E0DFE0E0DFE0DFE0DFDFE0
0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C 0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C
000000000000000000000000000000000000000000000000000000000000 000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000 000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000 000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000 000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000 000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000 000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000 000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000 000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000 000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000 000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000 000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000 000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000 000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000 000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000 000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000 000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000 000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000 000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000 000000000000000000000000000000000000000000000000000000000000
[... cut ...]
000000000000000000000000000000000000000000000000000000000000 000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000 000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000 000000000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000000000 000000000000000000000000000000000000000000000000000000000000
7731A0A2486CCFEE17DA96DBCCFA82CEB2BD80CF98F38F8D6BF4E468B686 3C1A405E03DC6D4B0D1293E8650B4F866FB619948F7B8C274EB1D0D276D2
```

syssi, 2022-08-06:
This line looks like the best match:

```
000000000000000000000000000000000000000000000000000000000000 101010101010101010101010101010101010101010101010101010101010
```

The error of the `ERR.txt` is a `Cell overvoltage` error again. Correct?

senfkorn, 2022-08-06:
Jop, its cell overvoltage again

syssi, 2022-08-06:
I assume byte 166 and 167 is the error bitmask. I will try to prepare something tomorrow to proof it.

senfkorn, 2022-08-06:
Ok, I'll try it, as soon as it is available

syssi, 2022-08-07:
I've moved the position of the error bitmask and added a condition to guess the frame version / format. Please update your ESPHome node and try to verify every sensor measurement (plausible or not). In best case the device suppot is complete now.

senfkorn, 2022-08-07:
Seems to work,

[image]

PR #148 (2022-08-07): Add JK-B2A8S20P firmware 11.17 support (BLE, JK02_32S)
This PR introduces support for slightly changed frames because it the firmware supports/reserves space for more cells (24S vs. 32S) now. Most registers are shifted by 8\*2 bytes (cell voltages) and 8\*2 bytes (cell resistances). May be the number of temperature probes was increased too.

- [x] Try to auto-detect the new frame format

Fixes: #147
closed 2022-08-07 by syssi