syssi/esphome-jk-bms #149: No balance current data in non BLE version

valdiks, 2022-08-05:
Hi,

Many thanks for your software!

I've tested esp32-example.yaml and esp32-ble-example.yaml.

The BLE version shows balancing_current but this parameter is not available in version without BLE.
Is it safe to add code to the esp32-example.yaml ?

```
    balancing_current:
      name: "${name} balancing current"
```

syssi, 2022-08-05:
It's safe but the sensor won't provide measurements because the UART-TTL protocol doesn't provide this measurement. This value is available via JK02 only.

valdiks, 2022-08-05:
Not available over UART-TTL, but in the same time BLE does not work :D (works but connection is lost after some time).

syssi, 2022-08-05:
You are right. The situation isn't perfect. We have to make the best of it. The most advanced solution is to open the BMS and wire somes cables to the BLE module (RX, TXD lines). The BLE module can be disabled by pulling the RST pin to GND. This solution will have a lot of benefits:

1. The BLE connection is unprotected at the moment. We pin number isn't required the change settings.
2. The BLE connection isn't stable because of bugs at the underlying stack.
3. The JK02 protocol is the only way to change settings (it's read-write). The TTL-UART connection is read-only.
4. The JK02 payload provides most measurements. The TTL-UART payload is incomplete.

At the moment I recommend the UART-TTL connection because it's rock solid. The balancer current isn't important. Try to use the incomplete payload of the UART-TTL protocol for some days. I bet you won't miss much. The missing measurements aren't very important / valuable.

closed 2022-08-07 by syssi