syssi/esphome-jk-bms #150: Can't get the "switches" to work properly using BLE

Sunbuzz, 2022-08-07:
Hi, I am not getting the switches to work.

I am using esphome-jk-bms to control 2 JK-B2A24S20P.

The actual "switches" saves it state, i.e. if I reload the page in homeassistant, the state of the switches persists, but it doesn't change the the function **in the first BMS** only in the second. for example in the screenshot below I flipped the switch discharging to off-position, but it still says discharging: on on BMS #1

Here is a portion of my .yaml file:

```
esp32_ble_tracker:

ble_client:
  - mac_address: ${mac_address_1}
    id: client1
  - mac_address: ${mac_address_2}
    id: client2
    

jk_bms_ble:
  - ble_client_id: client1
    protocol_version: ${protocol_version}
    throttle: 5s
    id: bms1
  - ble_client_id: client2
    protocol_version: ${protocol_version}
    throttle: 5s
    id: bms2

switch:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms1
    charging:
      name: "${name}-1 charging"
    discharging:
      name: "${name}-1 discharging"
    balancer:
      name: "${name}-1 balancer"
  - platform: jk_bms_ble
    jk_bms_ble_id: bms2
    charging:
      name: "${name}-2 charging"
    discharging:
      name: "${name}-2 discharging"
    balancer:
      name: "${name}-2 balancer"

```

[image]

syssi, 2022-08-07:
Could you provide a log with log level `DEBUG`?

syssi, 2022-08-07:
I know what's going on: The switch state is part of the `settings frame`. The settings are retrieved once on startup of the ESP / at the beginning of the BLE connection. Please press the `retrieve settings` button to retrieve the settings and update/sync the switch state. You could retrieve/press the button periodically but your BMS will beep on every request.

Sunbuzz, 2022-08-07:
Do you still need the log?

Is there no way of saving a settings frame per ble connection?

Sunbuzz, 2022-08-07:
I think I understand what you mean, after a restart of the esp32, I can only change BMS 1 (instead of BMS 2) and in the logs, it seams it was BMS 1 that first made BLE connection.
```
[21:16:40][I][jk_bms_ble:1065]: Device info frame (300 bytes):
[21:16:40][I][jk_bms_ble:1117]:   Vendor ID: JK-B2A24S20P
[21:16:40][I][jk_bms_ble:1118]:   Hardware version: 10.XW
[21:16:40][I][jk_bms_ble:1119]:   Software version: 10.07
[21:16:41][I][jk_bms_ble:1120]:   Uptime: 10878000 s
[21:16:41][I][jk_bms_ble:1121]:   Power on count: 8
[21:16:41][I][jk_bms_ble:1122]:   Device name: JK-BMS Upper
[21:16:41][I][jk_bms_ble:1123]:   Device passcode: 1234
[21:16:41][I][jk_bms_ble:1124]:   Manufacturing date: 220330
[21:16:41][I][jk_bms_ble:1125]:   Serial number: 2012601068
[21:16:41][I][jk_bms_ble:1126]:   Passcode: 0000
[21:16:41][I][jk_bms_ble:1127]:   User data: Input Userdata
[21:16:41][I][jk_bms_ble:1128]:   Setup passcode: 123456
```

after that this information did not appear for the second BMS.

Sunbuzz, 2022-08-07:
I am unable to press `retrieve` on both BMS's it only works on one of them:
```
[21:39:30][D][button:013]: 'jk-bms-1 retrieve device info' Pressed.
[21:39:30][D][jk_bms_ble:1154]: Write register: AA.55.90.EB.97.00.00.00.00.00.00.00.00.00.00.00.00.00.00.11 (20)
[21:39:30][I][jk_bms_ble:1065]: Device info frame (300 bytes):
[21:39:30][I][jk_bms_ble:1117]:   Vendor ID: JK-B2A24S20P
[21:39:30][I][jk_bms_ble:1118]:   Hardware version: 10.XW
[21:39:30][I][jk_bms_ble:1119]:   Software version: 10.07
[21:39:30][I][jk_bms_ble:1120]:   Uptime: 10879329 s
[21:39:30][I][jk_bms_ble:1121]:   Power on count: 8
[21:39:30][I][jk_bms_ble:1122]:   Device name: JK-BMS Upper
[21:39:30][I][jk_bms_ble:1123]:   Device passcode: 1234
[21:39:30][I][jk_bms_ble:1124]:   Manufacturing date: 220330
[21:39:30][I][jk_bms_ble:1125]:   Serial number: 2012601068
[21:39:30][I][jk_bms_ble:1126]:   Passcode: 0000
[21:39:30][I][jk_bms_ble:1127]:   User data: Input Userdata
[21:39:30][I][jk_bms_ble:1128]:   Setup passcode: 123456
```
and:
```
[21:39:35][D][button:013]: 'jk-bms-2 retrieve device info' Pressed.
[21:39:35][D][jk_bms_ble:1154]: Write register: AA.55.90.EB.97.00.00.00.00.00.00.00.00.00.00.00.00.00.00.11 (20)
```
with no respons.

syssi, 2022-08-08:
Did you buy the second BMS recently? It's may be another firmware version as the first one? If you remove the working/controllable BMS from the Yaml are you able to control the second one? In other words: Are you able to control both BMS in general If only one BLE connection is part of the configuration?

Sunbuzz, 2022-08-08:
I bought the BMS at the same time, and they share the same firmware, and yes they work separat.
I am trying to look in the logs, and i can only see initialization on one BMS, (sometimes BMS 1, other times BMS 2) and that is the BMS works with switches and buttons. (i.e. writes)

but it is reading from both BMS's just fine - so I don't know what is going on...

syssi, 2022-08-08:
Are you able to consume/retrieve the logs via a serial connection? I would be happy about some logs showing the startup of the ESP because if the BLE connection is established the component performs some `BLE writes` (like the switch entities). Without these commands the BMS wouldn't provide sensor values periodically (by BLE notification). I assume both instances are able to talk via the select BLE connection to a specific BMS (on start up). I would like to see a proof of this assumption at the logs. May be the switch (and button) entities aren't assigned to the correct JkBmsBle instance.

Sunbuzz, 2022-08-08:
I will give you a complete serial log as soon as I get back home (later this afternoon)

Sunbuzz, 2022-08-08:
Here is the captured serial log, I toggled the switched in at the and of the log.

[Serial Saved Output.txt](https://github.com/syssi/esphome-jk-bms/files/9281874/Serial.Saved.Output.txt)

syssi, 2022-08-08:
Alright. In general we are able to talk to two devices using different BLE client instances:

```
# jk-bms-2
[I][jk_bms_ble:162]: Request device info
[D][jk_bms_ble:1154]: Write register: AA.55.90.EB.97.00.00.00.00.00.00.00.00.00.00.00.00.00.00.11 (20)
[D][ble_client:047]: Found device at MAC address [C8:47:8C:E1:E1:9C]
[I][ble_client:085]: Attempting BLE connection to c8:47:8c:e1:e1:9c
[I][jk_bms_ble:1065]: Device info frame (300 bytes):
[I][jk_bms_ble:1117]:   Vendor ID: JK-B2A24S20P
[I][jk_bms_ble:1118]:   Hardware version: 10.XW
[I][jk_bms_ble:1119]:   Software version: 10.07
[I][jk_bms_ble:1120]:   Uptime: 10860900 s
[I][jk_bms_ble:1121]:   Power on count: 2
[I][jk_bms_ble:1122]:   Device name: JK-BMS Lower
[I][jk_bms_ble:1123]:   Device passcode: 1234
[I][jk_bms_ble:1124]:   Manufacturing date: 220403
[I][jk_bms_ble:1125]:   Serial number: 2012601069
[I][jk_bms_ble:1126]:   Passcode: 0000
[I][jk_bms_ble:1127]:   User data: Input Userdata
[I][jk_bms_ble:1128]:   Setup passcode: 123456
[I][jk_bms_ble:296]: Request status notification
[D][jk_bms_ble:1154]: Write register: AA.55.90.EB.96.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10 (20)
[I][jk_bms_ble:802]: Settings frame (300 bytes):
[D][jk_bms_ble:828]:   Unknown6: 0.600000
[I][jk_bms_ble:830]:   Cell UVP: 2.820000 V
[...]
[I][jk_bms_ble:912]:   Nominal battery capacity: 258.000000 Ah
[D][number:036]: 'jk-bms-2 total battery capacity': Sending state 258.000000
[...]
[D][text_sensor:067]: 'jk-bms-2 total runtime formatted': Sending state '125d 16h'
```

```
# jk-bms-1
[I][jk_bms_ble:162]: Request device info
[D][jk_bms_ble:1154]: Write register: AA.55.90.EB.97.00.00.00.00.00.00.00.00.00.00.00.00.00.00.11 (20)
[I][jk_bms_ble:1065]: Device info frame (300 bytes):
[I][jk_bms_ble:1117]:   Vendor ID: JK-B2A24S20P
[I][jk_bms_ble:1118]:   Hardware version: 10.XW
[I][jk_bms_ble:1119]:   Software version: 10.07
[I][jk_bms_ble:1120]:   Uptime: 10933800 s
[I][jk_bms_ble:1121]:   Power on count: 8
[I][jk_bms_ble:1122]:   Device name: JK-BMS Upper
[I][jk_bms_ble:1123]:   Device passcode: 1234
[I][jk_bms_ble:1124]:   Manufacturing date: 220330
[I][jk_bms_ble:1125]:   Serial number: 2012601068
[I][jk_bms_ble:1126]:   Passcode: 0000
[I][jk_bms_ble:1127]:   User data: Input Userdata
[I][jk_bms_ble:1128]:   Setup passcode: 123456
[I][jk_bms_ble:296]: Request status notification
[D][jk_bms_ble:1154]: Write register: AA.55.90.EB.96.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10 (20)
[I][jk_bms_ble:802]: Settings frame (300 bytes):
[D][jk_bms_ble:828]:   Unknown6: 0.600000
[I][jk_bms_ble:830]:   Cell UVP: 2.820000 V
[...]
[I][jk_bms_ble:912]:   Nominal battery capacity: 258.000000 Ah
[D][number:036]: 'jk-bms-1 total battery capacity': Sending state 258.000000
[...]
[D][text_sensor:067]: 'jk-bms-1 total runtime formatted': Sending state '126d 13h'
```

I would debug the switch entities as next step.

Sunbuzz, 2022-08-08:
Let me know if you need anything more from me.

syssi, 2022-08-08:
Let's assume all switches/buttons are assigned to the bring JkBmsBle instance. Could try do the following experiment:

1. Turn off `switch.jk_bms_1_discharging` and `switch.jk_bms_2_discharging`
2. One of the BMS should react (at the log it was jk-bms-1). You could monitor the `binary_sensor.jk_bms_*_discharging` entities. One BMS doesn't react (probably jk-bms-2).
3. Turn discharging `on` of the BMS with didn't react (jk-bms-2).
4. Please check the binary sensor again. Do we control always just the one or are the commands/instructions of the second set of controls routed into the nowhere?

Sunbuzz, 2022-08-08:
I would say the second one is routed into nowhere:

```

[14:23:35][D][switch:017]: 'jk-bms-1 discharging' Turning OFF.
[14:23:35][D][jk_bms_ble:1154]: Write register: AA.55.90.EB.1E.04.00.00.00.00.00.00.00.00.00.00.00.00.00.9C (20)
[14:23:35][D][switch:037]: 'jk-bms-1 discharging': Sending state OFF
[14:23:37][D][switch:017]: 'jk-bms-2 discharging' Turning OFF.
[14:23:37][D][jk_bms_ble:1154]: Write register: AA.55.90.EB.1E.04.00.00.00.00.00.00.00.00.00.00.00.00.00.9C (20)
[14:23:37][D][switch:037]: 'jk-bms-2 discharging': Sending state OFF

...

[14:23:40][D][binary_sensor:036]: 'jk-bms-1 discharging': Sending state OFF

...

[14:23:47][D][switch:013]: 'jk-bms-2 discharging' Turning ON.
[14:23:47][D][jk_bms_ble:1154]: Write register: AA.55.90.EB.1E.04.01.00.00.00.00.00.00.00.00.00.00.00.00.9D (20)
[14:23:47][D][switch:037]: 'jk-bms-2 discharging': Sending state ON
[14:23:48][D][switch:013]: 'jk-bms-1 discharging' Turning ON.
[14:23:48][D][jk_bms_ble:1154]: Write register: AA.55.90.EB.1E.04.01.00.00.00.00.00.00.00.00.00.00.00.00.9D (20)
[14:23:48][D][switch:037]: 'jk-bms-1 discharging': Sending state ON

...

[14:23:50][D][binary_sensor:036]: 'jk-bms-1 discharging': Sending state ON
```

syssi, 2022-08-08:
I assume we hit this issue: https://github.com/esphome/esphome/pull/3544

It was fixed at ESPHome 2022.6.2. Could you update your ESPHome installation to the latest release and try again?

Sunbuzz, 2022-08-08:
Yes!!! That fixed it! Now it is working as it should.

Thank you!

syssi, 2022-08-08:
Perfect! Let's close this issue. :-)

closed 2022-08-08 by syssi