syssi/esphome-jk-bms #152: Multiple JK BMS

RoySalisbury, 2022-08-11:
If I have 4 (or more) JK BMS devices, I assume that I would need an ESP32 for each BLE connection.  Is this true, and if not, then how would I setup multiple in the .yaml file.  

If it is true, any chance of adding support for more than a one-to-one relationship?

syssi, 2022-08-11:
The BLE stack of an ESP32 is limited to 3 BLE client connections:

> A maximum of three devices is supported due to limitations in the ESP32 BLE stack. If you wish to connect more devices, use additional ESP32 boards.
See https://esphome.io/components/ble_client.html

An example configuration to talk to two devices would look like this:

```
substitutions:
  name: jk-bms
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address_bms0: C8:47:AA:AA:AA:AA
  mac_address_bms1: C8:47:BB:BB:BB:BB

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.2.0

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf
    version: latest

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
logger:

# If you don't use Home Assistant please remove this `api` section and uncomment the `mqtt` component!
api:

# The MQTT component is ESP-IDF compatible since ESPHome version 2022.4.0. If
# ESPHome suggests to use the arduino framework instead because of missing ESP-IDF
# framework support you should update your setup.
# mqtt:
#   broker: !secret mqtt_host
#   username: !secret mqtt_username
#   password: !secret mqtt_password
#   id: mqtt_client

esp32_ble_tracker:

ble_client:
  - mac_address: ${mac_address_bms0}
    id: client0
  - mac_address: ${mac_address_bms1}
    id: client1

jk_bms_ble:
  - ble_client_id: client0
    throttle: 5s
    id: bms0
  - ble_client_id: client1
    throttle: 5s
    id: bms1

sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    state_of_charge:
      name: "${name} state of charge"

  - platform: jk_bms_ble
    jk_bms_ble_id: bms1
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    state_of_charge:
      name: "${name} state of charge"
```

Please try to identify/understand the pattern how to wire different components using IDs as reference.

RoySalisbury, 2022-08-11:
Ok.. That makes complete sense.  Thanks for the example.  I'll give it a shot.

closed 2022-08-11 by RoySalisbury