syssi/esphome-jk-bms #155: failure "xtensa-esp32-elf-gcc -- broken"

atze09, 2022-08-16:
I have issues after running the run command:

`pi@raspberrypi:~/esphome-jk-bms $ esphome run esp32-example.yaml
INFO Reading configuration esp32-example.yaml...
INFO Cloning https://github.com/syssi/esphome-jk-bms.git@main
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing jk-bms (board: wemos_d1_mini32; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf 3.40302.0 (4.3.2) 
 - tool-cmake 3.16.9 
 - tool-ninja 1.10.2 
 - toolchain-riscv32-esp 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32s2 8.4.0+2021r2-patch2
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.30.2") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/pi/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/pi/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/pi/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -- broken
-- Configuring incomplete, errors occurred!
See also "/home/pi/esphome-jk-bms/.esphome/build/jk-bms/.pioenvs/jk-bms/CMakeFiles/CMakeOutput.log".
See also "/home/pi/esphome-jk-bms/.esphome/build/jk-bms/.pioenvs/jk-bms/CMakeFiles/CMakeError.log".

fatal: Kein Git-Repository (oder irgendeines der Elternverzeichnisse): .git
CMake Error at /home/pi/.platformio/packages/tool-cmake/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/home/pi/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/pi/esphome-jk-bms/.esphome/build/jk-bms/.pioenvs/jk-bms/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/home/pi/.platformio/packages/tool-ninja/ninja cmTC_2030c && [1/2] Building C object CMakeFiles/cmTC_2030c.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_2030c
    FAILED: cmTC_2030c 
    : && /home/pi/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address   CMakeFiles/cmTC_2030c.dir/testCCompiler.c.obj  -o cmTC_2030c   && :
    /home/pi/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/pi/.platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: error loading plugin: /home/pi/.platformio/packages/toolchain-xtensa-esp32/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: Kann die Shared-Object-Datei nicht öffnen: Datei oder Verzeichnis nicht gefunden
    collect2: error: ld returned 1 exit status
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /home/pi/.platformio/packages/framework-espidf/tools/cmake/project.cmake:311 (__project)
  CMakeLists.txt:3 (project)

================================================================ [FAILED] Took 39.14 seconds `

Did i made something wrong?

syssi, 2022-08-16:
You did hit this issue: https://github.com/syssi/esphome-jk-bms#known-issues

Raspberry Pi & ESP-IDF: If the project doesn't compile because of Error: Could not find the package with 'platformio/toolchain-esp32ulp @ ~1.22851.0' requirements for your system 'linux_aarch64' please use a host with another processor architecture (f.e. x86). The toolchain isn't linux_aarch64 (ARM64) compatible at the moment.

syssi, 2022-08-16:
This is a short guide how to install esphome and compile a configuration YAML manually on debian based host: 

https://raw.githubusercontent.com/syssi/esphome-seplos-bms/main/install.gif

atze09, 2022-08-16:
> IDF

awesome thank you! i will try!

closed 2022-08-16 by atze09