syssi/esphome-jk-bms #160: My first connection JKBMS with ESPHome

proasnet, 2022-08-29:
Hi, I am sorry, that I write my asks here. I am totaly new user with ESPHome and I am trying connect ESP32 with HA and my JKBMS JK-B2A24S15P. I have two BMSs, the same, for two batteries.
I found that JKBMS is in external components after install ESPHome to HA.
jk_bms
jk_bms_ble
jk_modbus

So, is needed use this?
```
external_components:
  - source: github://syssi/esphome-jk-bms@main
```
I have uploaded firmware to ESP32, basic by USB cable from HA, but I dont know, what is next step. Is needed config some in yaml of sensor?
Is possible connect to BMS not by UART, but with BT integrated in ESP32?
In the manual how it install, I not found about this.
Thank you very much for help me.

syssi, 2022-08-29:
The docoumentation is incomplete. I will try to provide all needed details here. Feel free to ask more questions!

1. Please use the example configuration as starting point of you want to interface your BMS via UART: https://github.com/syssi/esphome-jk-bms/blob/main/esp32-example.yaml
2. Please use the `ble` example configuration if you want to interface your BMS via BLE: https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-example.yaml

If you want to connect to two devices via BLE please start here: https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-example-multiple-devices.yaml

You have to change to MAC addresses to the addresses of your devices:

https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-example-multiple-devices.yaml#L7-L8

proasnet, 2022-08-29:
Thank you very much. I am interesting about multiple device with BLE.
Where I can define passwords to bms?

syssi, 2022-08-29:
There is no passwords required. See https://github.com/syssi/esphome-jk-bms/issues/139

proasnet, 2022-08-29:
Thank you Sebastian, now it working, I see both bms in entities.
Please, where is possible see cells voltage for each cell?
Thanks

syssi, 2022-08-29:
This example configuration contains *all* supported entities:

https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-example.yaml#L60-L270

You will find cell voltages etc. there. The "multiple device BLE example" is reduced/incomplete. If you add too much sensors / entities the ESP32 will crash because of limited resources.

proasnet, 2022-08-29:
Thanks, for BLE, is only limited number? Or specific names?

syssi, 2022-08-29:
You could try to increase the number of sensors / entities step by step. At some point the ESP32 won't boot anymore because of a stack overflow. If you hit this limitation please let me know.

syssi, 2022-08-29:
Feel free to close the issue if there are no further questions.

closed 2022-08-29 by proasnet

proasnet, 2022-09-22:
Hi @syssi,
connection to both JK BMS working, but I noticed in HA that there are way too many unavailable connections. Is there any way to keep a constant connection without disconnecting regularly? Such as keep alive etc..
The ESP is about 1m from the BMS outlet and nothing blocks the BLE signal.
Thank you very much.

syssi, 2022-09-22:
What's the distance between the ESP and your WiFi accesspoint?

proasnet, 2022-09-22:
About 1,5 meters. AP is mikrotik 
[](https://mikrotik.com/product/RBcAPL-2nD-307)

syssi, 2022-09-22:
Are you able to log the serial output of the ESP? I would like to know what's happening (reboots vs. WiFi disconnects).

proasnet, 2022-09-22:
I have ESP connected and powered via USB cable to HA host computer.  How I can record a log? I am beginner with ESP and I have not any additional hardware.

proasnet, 2022-09-22:
Is possible use LOG via /dev/ttyACM0 from ESP Home started from JKBMS node?
Here is realtime logging with save to file.

syssi, 2022-09-22:
I don't understand your question.

syssi, 2022-09-22:
You could use any terminal program to read the log messages from your serial-to-usb device.

proasnet, 2022-09-22:
I meant this
[image]

proasnet, 2022-09-22:
At the exact time when the JKBMS entities were unavailable, I found this in the log:
```
[12:35:23][D][api:102]: Accepted ::FFFF:192.168.0.249
[12:35:23][D][api.connection:861]: Home Assistant 2022.9.5 (::FFFF:192.168.0.249): Connected successfully
```
It looks like a disconnect from the network, not from BLE.

syssi, 2022-09-22:
Do you see a reason for the disconnect a few lines above?

proasnet, 2022-09-22:
I can't see.
[logs_jk_bms_logs.txt](https://github.com/syssi/esphome-jk-bms/files/9625542/logs_jk_bms_logs.txt)

syssi, 2022-09-22:
Could you check the logbook of your accesspoint?

proasnet, 2022-09-22:
Not possible at the moment. I'm connected remotely and can't access the access point's web interface. Only to the Mikrotik router, I have the AP configured via CAPsMAN, and there the log of individual points is not available

syssi, 2022-09-23:
To be explicit: This ESPHome configuration is designed to keep the network connection and the API connection to Home Assistant up and running (permanently). If the device goes unavailable the ESP did crash hard or the WiFi connection is unstable.

As first try you could reduce the log level (of the `jk_bms_ble` component) to safe some resources by not writing every detail into the log / to the serial line. Please don't change the overall log level because some verbosity would still be nice for debugging the issue.

proasnet, 2022-09-23:
Well thank you.
Where can I find this setting in ESP Home please?

syssi, 2022-09-23:
This is the default behavior of ESPHome.

proasnet, 2022-09-23:
I am sorry, I am not understand. I am not expert with HA and ESP Home. Is any way, where I can set Log level of the ESP Home or a component? I dont know, what is in my case server or client, I have ESP Home installed in HA as addon, and HA as HASOS on x86 generic.
Thank you for your patience, I know that this topic already goes beyond your project and concerns ESP Home as such.

syssi, 2022-09-23:
This is the documentation of the `logger` component: https://esphome.io/components/logger.html

This configuration sets the overall log level to `DEBUG` and decreases the log level of the `jk_bms_ble`  component to `INFO`: 

```
logger:
  level: DEBUG
  logs:
    jk_bms_ble: INFO
```

proasnet, 2022-09-23:
Okay, thanks, INFO logging working. Now I will monitor ESP behavior towards the network.
I'll let you know how it works.