syssi/esphome-jk-bms #162: Lolin Wemos D1 Mini v4 no communication to BMS

JesperThorup, 2022-09-02:
Hi,

I have put the esp8266-example.yaml code on my Lolin Wemos D1 Mini v4 and connected it like this:

# RS485-TTL jack (4 Pin, JST 1.25mm pitch)
┌─── ─────── ────┐
│                                       │
│ O   O   O   O                  │
│GND  RX  TX VBAT│
└────────────────┘
  │   │   │
  │   │   └─── GPIO3 (`rx_pin`)
  │   └─────── GPIO1 (`tx_pin`)
  └─────────── GND

But all I get is unknown sensor values.
The Wemos shows as online in ESPHome..

All I get in the ESPHome log is:

```
[18:30:49][I][app:102]: ESPHome version 2022.8.1 compiled on Sep  2 2022, 18:29:45
[18:30:49][I][app:104]: Project syssi.esphome-jk-bms version 1.2.0
[18:30:49][C][wifi:491]: WiFi:
[18:30:49][C][wifi:353]:   Local MAC: 34:94:54:61:B3:B8
[18:30:49][C][wifi:354]:   SSID: [redacted]
[18:30:49][C][wifi:355]:   IP Address: 10.0.0.119
[18:30:49][C][wifi:356]:   BSSID: [redacted]
[18:30:49][C][wifi:358]:   Hostname: 'jk-bms'
[18:30:49][C][wifi:360]:   Signal strength: -48 dB ▂▄▆█
[18:30:49][C][wifi:364]:   Channel: 11
[18:30:49][C][wifi:365]:   Subnet: 255.255.255.0
[18:30:49][C][wifi:366]:   Gateway: 10.0.0.1
[18:30:49][C][wifi:367]:   DNS1: 0.0.0.0
[18:30:49][C][wifi:368]:   DNS2: 0.0.0.0
[18:30:49][C][logger:275]: Logger:
[18:30:49][C][logger:276]:   Level: DEBUG
[18:30:49][C][logger:277]:   Log Baud Rate: 0
[18:30:49][C][logger:278]:   Hardware UART: UART0
[18:30:49][C][uart.arduino_esp8266:102]: UART Bus:
[18:30:49][C][uart.arduino_esp8266:103]:   TX Pin: GPIO1
[18:30:49][C][uart.arduino_esp8266:104]:   RX Pin: GPIO3
[18:30:49][C][uart.arduino_esp8266:106]:   RX Buffer Size: 384
[18:30:49][C][uart.arduino_esp8266:108]:   Baud Rate: 115200 baud
[18:30:49][C][uart.arduino_esp8266:109]:   Data Bits: 8
[18:30:49][C][uart.arduino_esp8266:110]:   Parity: NONE
[18:30:49][C][uart.arduino_esp8266:111]:   Stop bits: 1
[18:30:49][C][uart.arduino_esp8266:113]:   Using hardware serial interface.
[18:30:49][C][jk_modbus:096]: JkModbus:
[18:30:49][C][jk_bms:485]: JkBms:
```
But I'm not sure the log output are working correctly when using the UART ??

syssi, 2022-09-02:
Could you provide a photo of your wiring? The configuration of the ESP looks good.

JesperThorup, 2022-09-02:
[image0.jpeg]

Sendt fra min iPhone

JesperThorup, 2022-09-02:
Sure...

[image]
[image]

syssi, 2022-09-03:
Bad news: Your wiring looks good and there are no obvious reasons why it shouldn't work out of the box. Where did you buy your BMS and how old it is?

JesperThorup, 2022-09-03:
On the JK official store. I don’t rhinsk it’s to Old but maybee to new..
[image]

syssi, 2022-09-03:
Your BMS version is pretty new. Could you try to ask your seller the GPS/UART-TTL port is enabled or not?

JesperThorup, 2022-09-05:
I have tried, but it does not seem like the understand..

syssi, 2022-11-07:
Where did you buy the device? It's likely the UART-TTL port isn't enabled/equipped.

bierlair, 2022-11-07:
> Where did you buy the device? It's likely the UART-TTL port isn't enabled/equipped.

I'm curious about this statement.

Is there a way to find out that this GPS Port is enabled or not? I am having similar issues with 3 devices, JK-B2A24S20P, 10.X HW / 10.8 SW, and I can't read anything from the GPS port either.

There is voltage though:

GND to VBAT: 56V (my Battery Voltage)
GND to RX: 3.3V
GND to TX: 3.3V

Should that not be an indicator that this port is enabled and alive?

syssi, 2022-11-07:
I don't have a good answer. I simply know some sellers are trying to make sure the GPS port is working. There are other sellers where the communication port don't work for unknown reason.

If you need/want CAN bus support you have to pay a few bucks more because in this case one additional chip (opto-coupler or converter?) is soldered to the mainboard. My be this also applies to devices without GPS/UART-TTL communication. I could provide some photos of the required opto-coupler between the BMS MCU and the GPS port. If you open your BMS you could do some checks. I think your idea is correct: If the TX/RX pin provides some logic level voltages it's likely they are connected properly to something.

I don't know the UART protocol is disabled at some firmware versions. Most sellers are unresponsive or are unable to provide useful answers.

bierlair, 2022-11-07:
Interesting. I ordered those 3 BMS with CAN support and I did pay a little extra for this. Now it turns out I don't need CAN support (yet) and I simply want to read via the GPS/UART-TTL port.

Now I am starting to wonder if ordering CAN support would render the GPS port useless for UART-TTL communication...

syssi, 2022-11-07:
I don't think so. A BMS supporting CAN should support both not either or.

robotz0mbie, 2022-11-23:
hi!
I have the same issue using a recent bms but the same ESP (LOLIN 8266 D1 mini)
[image] (BMS)
[image] (wiring)

Everything is showing up in HA / (sensors etc) but all unknown values.
Enabling the UART debug I see:
[19:41:26][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
repeated over and over, I guess this is asking the BMS for data but getting nothing back?
my BMS is the non CAN version, but I understood the GPS/UART-TTL port was enabled by default. I also have another BMS which is the CAN version (going onto another battery pack), but this one isnt set up yet!
I wonder if it could be an issue with this ESP? I have some ESP32on the way to check.
Would it be possible to check the port on the BMS is working by connecting using a terminal from a PC and a USB<->serial converter? 
Good luck and will be watching to see if you have any progress!

syssi, 2022-11-23:
Long story short: Your wiring looks good. Enabling the debug output of the `uart ` component is perfect! Your assumption is correct too: You can the see outgoing traffic (`>>>`) and there are no responses (`<<<`) from the BMS.

Please be careful if you attach a USB-to-TTL converter. Make sure the logic level is 3.3V instead of 5V. I would try to use an isolated connection. There is always the risk to destroy the BMS or the USB port.

To rule out a faulty ESP you could use another one. It doesn't matter much it's an ESP8266 or ESP32.

robotz0mbie, 2022-11-23:
thanks, the idea of damage is a little scary so I might skip that test for now hah!
I will try another ESP / ESP32.
I bought the BMS from Hankzor store so it seems the GPS/TTL port should be working I guess. Is there any requirement to reboot the BMS for it to realise something is connected to that port?

syssi, 2022-11-24:
No. If the BMS is on (indicated by the red LED) the BMS should respond to requests.

bierlair, 2022-11-24:
I have been told by my vendor, since I ordered CAN support in my JK BMS, there would not be support for RS485/TTL. Apparently they have checked this with JK and that's the feedback. Apparently you have to specify which support you woud like to have (TTL, CAN, both) or you'll get it shipped with none, except Bluetooth.
This is ridiculous, but it is what it is.

wattwetter, 2022-12-11:
Hi, same here. Drastically reduce the number of sensors, then it works fine.

syssi, 2023-02-23:
I will close this issue because it cannot be solved software-wise. Please feel free to re-open the issue.

closed 2023-02-23 by syssi