syssi/esphome-jk-bms #165: Description of UART/GPS and Display port

proasnet, 2022-09-05:
Hi, please, is possible add to docs some informations about theese ports?
GPS as UART, is TTL level 5v or 3,3V level?
Display port, K+ a K- is for button
pins A and B is as data for display? This is UART? Or RS485?
Thank you very much.

syssi, 2022-09-05:
1. The logic level of the GPS/UART-TTL port is 3.3V.
2. Yes K+ and K- are the power button pins: https://github.com/syssi/esphome-jk-bms/blob/main/docs/display-port.md
3. As far as I know the A/B pins of the display port a RS485 (differential signal) but I'm not 100% sure.

syssi, 2022-09-05:
Please keep in mind the protocol of the display port isn't reverse engineered yet / isn't supported as data source yet.

closed 2022-09-05 by proasnet