syssi/esphome-jk-bms #166: Wemos C3 Mini compile error

JesperThorup, 2022-09-05:
Trying to use the Lolin Wemos C3 Mini to use the bluetooth, but i'm getting this error on compile:

`/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /data/jk-bms/.pioenvs/jk-bms/src/esphome/components/ble_client/ble_client.o: in function `esphome::ble_client::BLEClient::connect()':
/config/esphome/.esphome/build/jk-bms/src/esphome/components/ble_client/ble_client.cpp:87: undefined reference to `esp_ble_gattc_open'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /data/jk-bms/.pioenvs/jk-bms/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.o: in function `esphome::esp32_ble_tracker::ESP32BLETracker::start_scan_(bool)':
/config/esphome/.esphome/build/jk-bms/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:231: undefined reference to `esp_ble_gap_set_scan_params'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /config/esphome/.esphome/build/jk-bms/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:234: undefined reference to `esp_ble_gap_start_scanning'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /data/jk-bms/.pioenvs/jk-bms/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.o: in function `esphome::esp32_ble_tracker::ESP32BLETracker::loop()':
/config/esphome/.esphome/build/jk-bms/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:98: undefined reference to `esp_ble_gap_stop_scanning'
collect2: error: ld returned 1 exit status
*** [/data/jk-bms/.pioenvs/jk-bms/firmware.elf] Error 1`

Does enyone know what that means?

syssi, 2022-09-05:
Could you provide your configuration yaml? I would like to know the board section of your configuration to be able to reproduce the issue.

JesperThorup, 2022-09-05:
@syssi sure thing...

`substitutions:
  name: jk-bms
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: C8:47:FF:FF:FF:FF
  # Defaults to "JK02". Please use "JK04" if you have some old JK-BMS version (f.e. JK-B2A16S hw 3.0, sw. 3.3.0)
  protocol_version: JK04

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.2.0

esp32:
  board: esp32-c3-devkitm-1
  framework:
#  The arduino platform crashs
#    type: arduino
#    version: latest
    type: esp-idf
    version: latest

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: "XXXXXX"
  password: "XXXXXXX"
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 10.0.0.119
    # Set this to the IP address of the router. Often ends with .1
    gateway: 10.0.0.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0
  output_power: 8.5

ota:
logger:

api:

# This feature is only available with frameworks ['arduino'] <= 2022.3.1
# https://github.com/esphome/esphome/pull/2930: Add mqtt for idf >= 2022.4.0
#
# mqtt:
#   broker: !secret mqtt_host
#   username: !secret mqtt_username
#   password: !secret mqtt_password
#   id: mqtt_client

esp32_ble_tracker:

ble_client:
  - mac_address: ${mac_address}
    id: client0

jk_bms_ble:
  - ble_client_id: client0
    protocol_version: ${protocol_version}
    throttle: 5s
    id: bms0

binary_sensor:
  - platform: jk_bms_ble
    balancing:
      name: "${name} balancing"

sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    average_cell_voltage:
      name: "${name} average cell voltage"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
    cell_voltage_8:
      name: "${name} cell voltage 8"
    cell_voltage_9:
      name: "${name} cell voltage 9"
    cell_voltage_10:
      name: "${name} cell voltage 10"
    cell_voltage_11:
      name: "${name} cell voltage 11"
    cell_voltage_12:
      name: "${name} cell voltage 12"
    cell_voltage_13:
      name: "${name} cell voltage 13"
    cell_voltage_14:
      name: "${name} cell voltage 14"
    cell_voltage_15:
      name: "${name} cell voltage 15"
    cell_voltage_16:
      name: "${name} cell voltage 16"
    cell_voltage_17:
      name: "${name} cell voltage 17"
    cell_voltage_18:
      name: "${name} cell voltage 18"
    cell_voltage_19:
      name: "${name} cell voltage 19"
    cell_voltage_20:
      name: "${name} cell voltage 20"
    cell_voltage_21:
      name: "${name} cell voltage 21"
    cell_voltage_22:
      name: "${name} cell voltage 22"
    cell_voltage_23:
      name: "${name} cell voltage 23"
    cell_voltage_24:
      name: "${name} cell voltage 24"
    cell_resistance_1:
      name: "${name} cell resistance 1"
    cell_resistance_2:
      name: "${name} cell resistance 2"
    cell_resistance_3:
      name: "${name} cell resistance 3"
    cell_resistance_4:
      name: "${name} cell resistance 4"
    cell_resistance_5:
      name: "${name} cell resistance 5"
    cell_resistance_6:
      name: "${name} cell resistance 6"
    cell_resistance_7:
      name: "${name} cell resistance 7"
    cell_resistance_8:
      name: "${name} cell resistance 8"
    cell_resistance_9:
      name: "${name} cell resistance 9"
    cell_resistance_10:
      name: "${name} cell resistance 10"
    cell_resistance_11:
      name: "${name} cell resistance 11"
    cell_resistance_12:
      name: "${name} cell resistance 12"
    cell_resistance_13:
      name: "${name} cell resistance 13"
    cell_resistance_14:
      name: "${name} cell resistance 14"
    cell_resistance_15:
      name: "${name} cell resistance 15"
    cell_resistance_16:
      name: "${name} cell resistance 16"
    cell_resistance_17:
      name: "${name} cell resistance 17"
    cell_resistance_18:
      name: "${name} cell resistance 18"
    cell_resistance_19:
      name: "${name} cell resistance 19"
    cell_resistance_20:
      name: "${name} cell resistance 20"
    cell_resistance_21:
      name: "${name} cell resistance 21"
    cell_resistance_22:
      name: "${name} cell resistance 22"
    cell_resistance_23:
      name: "${name} cell resistance 23"
    cell_resistance_24:
      name: "${name} cell resistance 24"
    total_voltage:
      name: "${name} total voltage"
    total_runtime:
      name: "${name} total runtime"
    balancing_current:
      name: "${name} balancing current"
    power_tube_temperature:
      name: "${name} power tube temperature"

text_sensor:
  - platform: jk_bms_ble
    operation_status:
      name: "${name} operation status"
    total_runtime_formatted:
      name: "${name} total runtime formatted"

switch:
  - platform: jk_bms_ble
    balancer:
      name: "${name} balancer"

  - platform: ble_client
    ble_client_id: client0
    name: "${name} enable bluetooth connection"`

JesperThorup, 2022-09-05:
@syssi Got it working on another board and connected using bluetooth, but all values are 0...

[21:28:07][I][jk_bms_ble:296]: Request status notification
[21:28:07][D][jk_bms_ble:1154]: Write register: AA.55.90.EB.96.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10 (20)
[21:28:07][C][jk_bms_ble.switch:010]: JkBmsBle Switch 'jk-bms balancer'
[21:28:07][C][jk_bms_ble.switch:010]:   Icon: 'mdi:seesaw'
[21:28:07][C][ble_switch:035]: BLE Client Switch 'jk-bms enable bluetooth connection'
[21:28:07][C][ble_switch:035]:   Icon: 'mdi:bluetooth'
[21:28:07][C][esp32_ble_tracker:717]: BLE Tracker:
[21:28:07][C][esp32_ble_tracker:718]:   Scan Duration: 300 s
[21:28:07][C][esp32_ble_tracker:719]:   Scan Interval: 320.0 ms
[21:28:07][C][esp32_ble_tracker:720]:   Scan Window: 30.0 ms
[21:28:07][C][esp32_ble_tracker:721]:   Scan Type: ACTIVE
[21:28:07][C][ble_client:035]: BLE Client:
[21:28:07][C][ble_client:036]:   Address: c8:47:8c:ea:68:14
[21:28:07][C][mdns:094]: mDNS:
[21:28:07][C][mdns:095]:   Hostname: jk-bms
[21:28:07][C][ota:085]: Over-The-Air Updates:
[21:28:07][C][ota:086]:   Address: 10.0.0.119:3232
[21:28:07][C][api:138]: API Server:
[21:28:07][C][api:139]:   Address: 10.0.0.119:6053
[21:28:07][C][api:143]:   Using noise encryption: NO
[21:28:07][I][jk_bms_ble:974]: Settings frame (300 bytes):
[21:28:07][D][jk_bms_ble:1001]:   Unknown6: 58 02 00 00 (0.000000)
[21:28:07][I][jk_bms_ble:1010]:   Cell count: 0
[21:28:07][I][jk_bms_ble:1013]:   Power off voltage: 0.000000 V
[21:28:07][D][jk_bms_ble:1024]:   Unknown74: 0.000000
[21:28:07][I][jk_bms_ble:1032]:   Start balance voltage: -nan V
[21:28:07][D][jk_bms_ble:1035]:   Unknown102: -nan
[21:28:07][I][jk_bms_ble:1038]:   Trigger Delta Voltage: 0.000000 V
[21:28:07][I][jk_bms_ble:1041]:   Max. balance current: 0.000000 A
[21:28:07][D][switch:037]: 'jk-bms balancer': Sending state ON
[21:28:07][I][jk_bms_ble:1045]:   ADC Vref: unknown V
[21:28:08][D][sensor:127]: 'jk-bms cell voltage 1': Sending state 0.00000 V with 3 decimals of accuracy
[21:28:08][D][sensor:127]: 'jk-bms cell voltage 2': Sending state 0.00000 V with 3 decimals of accuracy
[21:28:08][D][sensor:127]: 'jk-bms cell voltage 3': Sending state 0.00000 V with 3 decimals of accuracy
[21:28:08][D][sensor:127]: 'jk-bms cell voltage 4': Sending state 0.00000 V with 3 decimals of accuracy
[21:28:08][D][sensor:127]: 'jk-bms cell voltage 5': Sending state 0.00000 V with 3 decimals of accuracy
[21:28:08][D][sensor:127]: 'jk-bms cell voltage 6': Sending state 0.00000 V with 3 decimals of accuracy
[21:28:08][D][sensor:127]: 'jk-bms cell voltage 7': Sending state 0.00000 V with 3 decimals of accuracy
[21:28:08][D][sensor:127]: 'jk-bms cell voltage 8': Sending state 0.00000 V with 3 decimals of accuracy
[21:28:08][D][sensor:127]: 'jk-bms cell voltage 9': Sending state 0.00000 V with 3 decimals of accuracy
[21:28:08][D][sensor:127]: 'jk-bms cell voltage 10': Sending state 0.00000 V with 3 decimals of accuracy
[21:28:08][D][sensor:127]: 'jk-bms cell voltage 11': Sending state 0.00000 V with 3 decimals of accuracy
[21:28:08][D][sensor:127]: 'jk-bms cell voltage 12': Sending state 0.00000 V with 3 decimals of accuracy
[21:28:08][D][sensor:127]: 'jk-bms cell voltage 13': Sending state 0.00000 V with 3 decimals of accuracy
[21:28:08][D][sensor:127]: 'jk-bms cell voltage 14': Sending state 0.00000 V with 3 decimals of accuracy
[21:28:08][D][sensor:127]: 'jk-bms cell voltage 15': Sending state 0.00000 V with 3 decimals of accuracy
[21:28:08][D][sensor:127]: 'jk-bms cell voltage 16': Sending state 0.00000 V with 3 decimals of accuracy
[21:28:08][D][sensor:127]: 'jk-bms min cell voltage': Sending state 0.00000 V with 3 decimals of accuracy
[21:28:08][D][sensor:127]: 'jk-bms max cell voltage': Sending state 0.00000 V with 3 decimals of accuracy
[21:28:08][D][sensor:127]: 'jk-bms max voltage cell': Sending state 1.00000  with 0 decimals of accuracy
[21:28:08][D][sensor:127]: 'jk-bms min voltage cell': Sending state 23.00000  with 0 decimals of accuracy
[21:28:08][D][sensor:127]: 'jk-bms total voltage': Sending state 0.00000 V with 2 decimals of accuracy
[21:28:08][D][sensor:127]: 'jk-bms average cell voltage': Sending state 0.00000 V with 3 decimals of accuracy
[21:28:08][D][sensor:127]: 'jk-bms delta cell voltage': Sending state 0.00000 V with 3 decimals of accuracy
[21:28:08][D][jk_bms_ble:737]: Unknown210: 00 03 00 00 (always 0x00 0x00 0x00 0x00)
[21:28:08][D][jk_bms_ble:740]: Unknown214: 3C FE A3 01 (0xFF 0xFF: 24 cells?)
[21:28:08][D][jk_bms_ble:743]: Unknown218: 00
[21:28:08][D][jk_bms_ble:746]: Unknown219: 00
[21:28:08][D][binary_sensor:036]: 'jk-bms balancing': Sending state OFF
[21:28:08][D][text_sensor:067]: 'jk-bms operation status': Sending state 'Idle'
[21:28:08][D][jk_bms_ble:754]: Unknown221: 00
[21:28:08][D][sensor:127]: 'jk-bms balancing current': Sending state 0.00000 A with 2 decimals of accuracy
[21:28:08][D][jk_bms_ble:761]: Unknown226: 00 00 00 00 00 00 00 (always 0x00...0x00?)
[21:28:08][D][jk_bms_ble:765]: Unknown233: 00 00 00 00 (0.000000)
[21:28:08][D][jk_bms_ble:769]: Unknown237: 00 00 00 00 (always 0x40 0x00 0x00 0x00?)
[21:28:08][D][sensor:127]: 'jk-bms total runtime': Sending state 0.00000 s with 0 decimals of accuracy
[21:28:08][D][text_sensor:067]: 'jk-bms total runtime formatted': Sending state ''
[21:28:08][D][jk_bms_ble:783]: Unknown290: 00 00 00 00 (always 0x00 0x00 0x00 0x00?)
[... 17 lines cut ...]
[... cut ...]
[21:28:13][D][jk_bms_ble:765]: Unknown233: 00 00 00 00 (0.000000)
[21:28:13][D][jk_bms_ble:769]: Unknown237: 00 00 00 00 (always 0x40 0x00 0x00 0x00?)
[21:28:13][D][sensor:127]: 'jk-bms total runtime': Sending state 0.00000 s with 0 decimals of accuracy
[21:28:13][D][text_sensor:067]: 'jk-bms total runtime formatted': Sending state ''
[21:28:13][D][jk_bms_ble:783]: Unknown290: 00 00 00 00 (always 0x00 0x00 0x00 0x00?)
[21:28:13][D][jk_bms_ble:786]: Unknown294: 00 00 00 00
[21:28:13][D][jk_bms_ble:789]: Unknown298: 00

syssi, 2022-09-05:
It looks like you have flashed the example for the Heltec smart balancer. Please use the `esp32-ble-example.yaml` instead to talk to a JK-BMS.

syssi, 2022-09-05:
If you are using the `esp32-ble-example.yaml` already please increase the log level by:

```
logger:
  level: VERY_VERBOSE
  logs:
    scheduler: DEBUG
    component: DEBUG
    esp32_ble_tracker: DEBUG
    sensor: DEBUG
```

and provide another log capture. I'm interested in the 300 bytes raw frames (the response of the BMS).

JesperThorup, 2022-09-05:
@syssi 
This is the output with more logging:

https://pastebin.com/6ykSwRWK

syssi, 2022-09-05:
Please use `protocol_version: JK02` and try again.

syssi, 2022-09-05:
It looks like you have used the `esp32-ble-jk04-example.yaml` is starting point. This is the example configuration for hardware version 4 of the JK-BMS. Please try `esp32-ble-example.yaml` as starting point.

JesperThorup, 2022-09-05:
It works!! :) Thanks..

syssi, 2022-09-06:
If you want to use the ESP32-C3 you have to change the board section slightly to re-enable BLE 4.2 feature of the ESP-IDF framework:

```
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
    version: latest
    sdkconfig_options:
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
```

closed 2022-09-06 by syssi