syssi/esphome-jk-bms #171: JK-B1A20S15P V10.07 is not working with BLE

woodie2, 2022-09-14:
my other JK-B1A24S15P with sw 8 is working, but the 
JK-B1A20S15P V10.07 doesn't work. it only makes beep all few seconds.

with serial connection it works, but I want to have also BLE like on my other JK :)

thats the log:
[15:20:36][I][jk_bms_ble:296]: Request status notification
[15:20:36][D][jk_bms_ble:1154]: Write register: AA.55.90.EB.96.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10 (20)
[15:20:39][W][jk_bms_ble:291]: [20:21:11:28:18:dd] Not connected
[15:20:41][I][jk_bms_ble:296]: Request status notification
[15:20:41][D][jk_bms_ble:1154]: Write register: AA.55.90.EB.96.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10 (20)
[15:20:44][W][jk_bms_ble:291]: [20:21:11:28:18:dd] Not connected
[15:20:46][I][jk_bms_ble:296]: Request status notification
[15:20:46][D][jk_bms_ble:1154]: Write register: AA.55.90.EB.96.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10 (20)
[15:20:49][W][jk_bms_ble:291]: [20:21:11:28:18:dd] Not connected
[15:20:51][I][jk_bms_ble:296]: Request status notification
[15:20:51][D][jk_bms_ble:1154]: Write register: AA.55.90.EB.96.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10 (20)

syssi, 2022-09-14:
It looks like the BLE communication has changed a bit. Do you own an Android device? I would like to provide some instructions how to dump the BLE traffic between your JK-BMS and the Android app. If you are able to provide this traffic capture we should be able to improve the component to handle your BMS model.

woodie2, 2022-09-14:
Yes i did the BLE traffic capture already, i will upload it soon.

woodie2, 2022-09-14:
the MAC of my newer JK is 20:21:11:28:18:DD, it doesn't start with C8:47 like the older ones. perhaps that is the difference?

syssi, 2022-09-14:
Yes. I assume the bluetooth module manufacturer has changed. We could proof this if you are willing to open the BMS. ;-)

woodie2, 2022-09-14:
[btsnoop_hci.log](https://github.com/syssi/esphome-jk-bms/files/9567450/btsnoop_hci.log)

syssi, 2022-09-15:
Good job! I will provide some details about the traffic asap.

syssi, 2022-09-15:
```
Service UUID  0xffe0
Char UUID     0xffe1
Client Char Config 0x2902 -> Notification: True

# Device info request + response

--> Write command (0x52)
    Write Request (0x12)
    Value: aa5590eb9700f42af61e98d0e99b64a86aafd62a
    
<-- 55aaeb9003744a4b2d4231413230533135500000000031302e585700000031302e30370000007cc6e9000f0000004a4b2d4231413230533135500000000031323334000000000000000000000000323230333132000031313232313039323736003030303000496e70757420557365726461746100003132333435360000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000072aa5590ebc8010100000000000000000000000044 (320)

# ?

--> Write command (0x52)
    Write Request (0x12)
    Value: aa5590eb9600c8887d97e4dace50456b704b1ed9
    
<--

55aaeb90017458020000280a0000b80b0000de0d0000480d00000500000000000000000000000000000000000000b0090000a08601001e0000003c000000a08601002c0100003c0000003c000000e8030000bc02000058020000bc02000058020000000000002800000084030000bc02000010000000010000000100000001000000a0860100dc050000480d00000000000000000000

000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000081

aa5590ebc8010100000000000000000000000044

55aaeb900274de0cd50cd80ce10ce00cdd0cd80cdd0ce00cde0cde0cdb0cde0ce00cd90cd90c00000000000000000000000000000000ffff0000dc0c0c00040c4e004b004a00460046004900440040004500470052005000470046004000410000000000000000000000000000000000000000000000c4cd0000d5ed0100a0f6fffffa00ec00f200000000000061187b0100a0860100
36000000ce0153006400d305fbc6e90001018f06000000000000000000000000070001000000e80300000c00b22a3f4000000000e204000000000001000500009c6c150100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000004
55aaeb900275dd0cde0cde0cde0cdb0ce00ce10cde0cdb0cd30ce00ce20cdc0cd50cd80ce20c00000000000000000000000000000000ffff0000dd0c0c00040c4e004b004a00460046004900440040004500470052005000470046004000410000000000000000000000000000000000000000000000cfcd0000f0ed0100a0f6fffffa00ec00f200000000000061187b0100a0860100
36000000ce0153006400d305fcc6e90001018f06000000000000000000000000070001000000e80300000c00b22a3f4000000000e20400000000000100050000a16c150100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000039
55aaeb900276db0cd50ce00ce20cdc0cdc0cdd0ce00cde0cd50ce20cdc0cd30cdd0ce00cde0c00000000000000000000000000000000ffff0000dc0c0f000b094e004b004a00460046004900440040004500470052005000470046004000410000000000000000000000000000000000000000000000c6cd0000dbed0100a0f6fffffa00ec00f200000000000061187b0100a0860100
36000000cf0153006400d305fcc6e90001018f06000000000000000000000000070001000000e80300000c00b22a3f4000000000e20400000000000100050000a66c150100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000022
55aaeb900277e00cd80cd80ce00ce10ce10cd80cd80cdd0cde0ce00cdd0cde0cdd0cd70cdb0c00000000000000000000000000000000ffff0000dd0c0f000b094e004b004a00460046004900440040004500470052005000470046004000410000000000000000000000000000000000000000000000cccd0000eaed0100a0f6fffffa00ec00f200000000000061177b0100a0860100
36000000cf0153006400d305fdc6e90001018f06000000000000000000000000070001000000e80300000c00b22a3f4000000000e20400000000000100050000ab6c15010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000003f

# ?

--> Write command (0x52)
    Write Request (0x12)
    Value: aa5590eb96004a7626d7c79b03be391b8fa3b62c
    
<--

55aaeb90017858020000280a0000b80b0000de0d0000480d00000500000000000000000000000000000000000000b0090000a08601001e0000003c000000a08601002c0100003c0000003c000000e8030000bc02000058020000bc02000058020000000000002800000084030000bc02000010000000010000000100000001000000a0860100dc050000480d00000000000000000000

000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000085

aa5590ebc8010100000000000000000000000044

# ?

--> Write command (0x52)
    Write Request (0x12)
    Value: aa5590eb960074cba0a3a2692c9f02907ad060a4

<--

55aaeb900278db0cd50cdd0ce10cde0cdc0cd80cde0cde0cd70cdd0ce20cde0cdb0cdd0ce00c00000000000000000000000000000000ffff0000dc0c0f000b094e004b004a00460046004900440040004500470052005000470046004000410000000000000000000000000000000000000000000000c4cd0000fc160200a0f6fffffa00ec00f200000000000061177b0100a0860100
36000000d00153006400d305fec6e90001018f06000000000000000000000000070001000000e80300000c00b22a3f4000000000e20400000000000100050000b56c15010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000 [... cut ...]
55aaeb90027cdd0ce00cd90cde0cdd0ce00cde0cd80cd90cde0ce10cd90cd30cdd0cde0cd80c00000000000000000000000000000000ffff0000dc0c0f000b094e004b004a00460046004900440040004500470052005000470046004000410000000000000000000000000000000000000000000000bfcd0000ef160200d8f5fffffa00ec00f200000000000061157b0100a0860100
36000000d10153006400d30500c7e90001018f06000000000000000000000000070001000000e80300000d00b22a3f4000000000e20400000000000100050000cc6c1501000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000b9
```

woodie2, 2022-09-16:
ok and what does that tell us now? is this something completely new?

syssi, 2022-09-16:
It's work in progress / just my notes. I will ping you as soon I have a clue.

woodie2, 2022-09-16:
but you can see, that V10.07 is not always the same right now? you think that the bluetooth module has changed?

syssi, 2022-09-18:
Decoded contents of the `device info frame` (no difference to other BMS versions):

```
[11:35:41][I][jk_bms_ble:1098]:   Vendor ID: JK-B1A20S15P
[11:35:41][I][jk_bms_ble:1099]:   Hardware version: 10.XW
[11:35:41][I][jk_bms_ble:1100]:   Software version: 10.07
[11:35:41][I][jk_bms_ble:1101]:   Uptime: 15320700 s
[11:35:41][I][jk_bms_ble:1102]:   Power on count: 15
[11:35:41][I][jk_bms_ble:1103]:   Device name: JK-B1A20S15P
[11:35:41][I][jk_bms_ble:1104]:   Device passcode: 1234
[11:35:41][I][jk_bms_ble:1105]:   Manufacturing date: 220312
[11:35:41][I][jk_bms_ble:1106]:   Serial number: 1122109276
[11:35:41][I][jk_bms_ble:1107]:   Passcode: 0000
[11:35:41][I][jk_bms_ble:1108]:   User data: Input Userdata
[11:35:41][I][jk_bms_ble:1109]:   Setup passcode: 123456
```

syssi, 2022-09-18:
Settings frame (no difference to other BMS versions):

```
[11:37:33][VV][jk_bms_ble:784]:   55.AA.EB.90.01.74.58.02.00.00.28.0A.00.00.B8.0B.00.00.DE.0D.00.00.48.0D.00.00.05.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.B0.09.00.00.A0.86.01.00.1E.00.00.00.3C.00.00.00.A0.86.01.00.2C.01.00.00.3C.00.00.00.3C.00.00.00.E8.03.00.00.BC.02.00.00.58.02.00.00.BC.02.00.00.58.02.00.00.00.00.00.00.28.00.00.00.84.03.00.00.BC.02.00.00.10.00.00.00.01.00.00.00.01.00.00.00.01.00.00.00.A0.86.01.00.DC.05.00.00.48.0D.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 
[11:37:33][VV][jk_bms_ble:785]:   00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.81 (140)
[11:37:33][D][jk_bms_ble:809]:   Unknown6: 0.600000
[11:37:33][I][jk_bms_ble:811]:   Cell UVP: 2.600000 V
[11:37:33][D][number:012]: 'jk-bms cell voltage undervoltage protection': Sending state 2.600000
[11:37:33][I][jk_bms_ble:815]:   Cell UVPR: 3.000000 V
[11:37:33][D][number:012]: 'jk-bms cell voltage undervoltage recovery': Sending state 3.000000
[11:37:33][I][jk_bms_ble:819]:   Cell OVP: 3.550000 V
[11:37:33][D][number:012]: 'jk-bms cell voltage overvoltage protection': Sending state 3.550000
[11:37:33][I][jk_bms_ble:823]:   Cell OVPR: 3.400000 V
[11:37:33][D][number:012]: 'jk-bms cell voltage overvoltage recovery': Sending state 3.400000
[11:37:33][I][jk_bms_ble:827]:   Balance trigger voltage: 0.005000 V
[11:37:33][D][number:012]: 'jk-bms balance trigger voltage': Sending state 0.005000
[11:37:33][I][jk_bms_ble:835]:   Power off voltage: 2.480000 V
[11:37:33][D][number:012]: 'jk-bms power off voltage': Sending state 2.480000
[11:37:33][I][jk_bms_ble:839]:   Max. charge current: 100.000008 A
[11:37:33][D][number:012]: 'jk-bms max charge current': Sending state 100.000008
[11:37:33][I][jk_bms_ble:843]:   Charge OCP delay: 30.000000 s
[11:37:33][I][jk_bms_ble:845]:   Charge OCP recovery delay: 60.000000 s
[11:37:33][I][jk_bms_ble:847]:   Max. discharge current: 100.000008 A
[11:37:33][D][number:012]: 'jk-bms max discharge current': Sending state 100.000008
[11:37:33][I][jk_bms_ble:851]:   Discharge OCP recovery delay: 300.000000 s
[11:37:33][I][jk_bms_ble:853]:   Discharge OCP recovery delay: 60.000000 s
[11:37:33][I][jk_bms_ble:855]:   SCP recovery time: 60.000000 s
[11:37:33][I][jk_bms_ble:857]:   Max. balance current: 1.000000 A
[11:37:33][D][number:012]: 'jk-bms max balance current': Sending state 1.000000
[11:37:33][I][jk_bms_ble:861]:   Charge OTP: 70.000000 °C
[11:37:33][I][jk_bms_ble:863]:   Charge OTP recovery: 60.000000 °C
[11:37:33][I][jk_bms_ble:865]:   Discharge OTP: 70.000000 °C
[11:37:33][I][jk_bms_ble:867]:   Discharge OTP recovery: 60.000000 °C
[11:37:33][I][jk_bms_ble:869]:   Charge UTP: 0.000000 °C
[11:37:33][I][jk_bms_ble:871]:   Charge UTP recovery: 4.000000 °C
[11:37:33][I][jk_bms_ble:873]:   MOS OTP: 90.000000 °C
[11:37:33][I][jk_bms_ble:875]:   MOS OTP recovery: 70.000000 °C
[11:37:33][I][jk_bms_ble:877]:   Cell count: 16.000000
[11:37:33][D][number:012]: 'jk-bms cell count': Sending state 16.000000
[11:37:33][I][jk_bms_ble:881]:   Charge switch: on
[11:37:33][I][jk_bms_ble:885]:   Discharge switch: on
[11:37:33][I][jk_bms_ble:889]:   Balancer switch: on
[11:37:33][I][jk_bms_ble:893]:   Nominal battery capacity: 100.000008 Ah
[11:37:33][D][number:012]: 'jk-bms total battery capacity': Sending state 100.000008
[11:37:33][D][jk_bms_ble:897]:   Unknown134: 1.500000
[11:37:33][I][jk_bms_ble:899]:   Start balance voltage: 3.400000 V
[11:37:33][D][number:012]: 'jk-bms balance starting voltage': Sending state 3.400000
[11:37:33][I][jk_bms_ble:931]:   Con. wire resistance 1: 0.000000 Ohm
[11:37:33][I][jk_bms_ble:931]:   Con. wire resistance 2: 0.000000 Ohm
[11:37:33][I][jk_bms_ble:931]:   Con. wire resistance 3: 0.000000 Ohm
[11:37:33][I][jk_bms_ble:931]:   Con. wire resistance 4: 0.000000 Ohm
[11:37:33][I][jk_bms_ble:931]:   Con. wire resistance 5: 0.000000 Ohm
[11:37:33][I][jk_bms_ble:931]:   Con. wire resistance 6: 0.000000 Ohm
[11:37:33][I][jk_bms_ble:931]:   Con. wire resistance 7: 0.000000 Ohm
[11:37:33][I][jk_bms_ble:931]:   Con. wire resistance 8: 0.000000 Ohm
[11:37:33][I][jk_bms_ble:931]:   Con. wire resistance 9: 0.000000 Ohm
[11:37:33][I][jk_bms_ble:931]:   Con. wire resistance 10: 0.000000 Ohm
[11:37:33][I][jk_bms_ble:931]:   Con. wire resistance 11: 0.000000 Ohm
[11:37:33][I][jk_bms_ble:931]:   Con. wire resistance 12: 0.000000 Ohm
[11:37:33][I][jk_bms_ble:931]:   Con. wire resistance 13: 0.000000 Ohm
[11:37:33][I][jk_bms_ble:931]:   Con. wire resistance 14: 0.000000 Ohm
[11:37:33][I][jk_bms_ble:931]:   Con. wire resistance 15: 0.000000 Ohm
[11:37:34][I][jk_bms_ble:931]:   Con. wire resistance 16: 0.000000 Ohm
[11:37:34][I][jk_bms_ble:931]:   Con. wire resistance 17: 0.000000 Ohm
[11:37:34][I][jk_bms_ble:931]:   Con. wire resistance 18: 0.000000 Ohm
[11:37:34][I][jk_bms_ble:931]:   Con. wire resistance 19: 0.000000 Ohm
[11:37:34][I][jk_bms_ble:931]:   Con. wire resistance 20: 0.000000 Ohm
[11:37:34][I][jk_bms_ble:931]:   Con. wire resistance 21: 0.000000 Ohm
[11:37:34][I][jk_bms_ble:931]:   Con. wire resistance 22: 0.000000 Ohm
[11:37:34][I][jk_bms_ble:931]:   Con. wire resistance 23: 0.000000 Ohm
[11:37:34][I][jk_bms_ble:931]:   Con. wire resistance 24: 0.000000 Ohm
```

syssi, 2022-09-18:
Cell info frame (no difference to other BMS versions):

```
[11:37:34][VV][jk_bms_ble:360]: Cell info frame (version 2, 300 bytes):
[11:37:34][VV][jk_bms_ble:361]:   55.AA.EB.90.02.74.DE.0C.D5.0C.D8.0C.E1.0C.E0.0C.DD.0C.D8.0C.DD.0C.E0.0C.DE.0C.DE.0C.DB.0C.DE.0C.E0.0C.D9.0C.D9.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.DC.0C.0C.00.04.0C.4E.00.4B.00.4A.00.46.00.46.00.49.00.44.00.40.00.45.00.47.00.52.00.50.00.47.00.46.00.40.00.41.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.C4.CD.00.00.D5.ED.01.00.A0.F6.FF.FF.FA.00.EC.00.F2.00.00.00.00.00.00.61.18.7B.01.00.A0.86.01.00 (150)
[11:37:34][VV][jk_bms_ble:362]:   36.00.00.00.CE.01.53.00.64.00.D3.05.FB.C6.E9.00.01.01.8F.06.00.00.00.00.00.00.00.00.00.00.00.00.07.00.01.00.00.00.E8.03.00.00.0C.00.B2.2A.3F.40.00.00.00.00.E2.04.00.00.00.00.00.01.00.05.00.00.9C.6C.15.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.04 (150)
[11:37:34][D][sensor:127]: 'jk-bms cell voltage 1': Sending state 3.29400 V with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell resistance 1': Sending state 0.07800 Ω with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell voltage 2': Sending state 3.28500 V with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell resistance 2': Sending state 0.07500 Ω with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell voltage 3': Sending state 3.28800 V with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell resistance 3': Sending state 0.07400 Ω with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell voltage 4': Sending state 3.29700 V with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell resistance 4': Sending state 0.07000 Ω with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell voltage 5': Sending state 3.29600 V with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell resistance 5': Sending state 0.07000 Ω with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell voltage 6': Sending state 3.29300 V with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell resistance 6': Sending state 0.07300 Ω with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell voltage 7': Sending state 3.28800 V with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell resistance 7': Sending state 0.06800 Ω with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell voltage 8': Sending state 3.29300 V with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell resistance 8': Sending state 0.06400 Ω with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell voltage 9': Sending state 3.29600 V with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell resistance 9': Sending state 0.06900 Ω with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell voltage 10': Sending state 3.29400 V with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell resistance 10': Sending state 0.07100 Ω with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell voltage 11': Sending state 3.29400 V with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell resistance 11': Sending state 0.08200 Ω with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell voltage 12': Sending state 3.29100 V with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell resistance 12': Sending state 0.08000 Ω with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell voltage 13': Sending state 3.29400 V with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell resistance 13': Sending state 0.07100 Ω with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell voltage 14': Sending state 3.29600 V with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell resistance 14': Sending state 0.07000 Ω with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell voltage 15': Sending state 3.28900 V with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell resistance 15': Sending state 0.06400 Ω with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell voltage 16': Sending state 3.28900 V with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell resistance 16': Sending state 0.06500 Ω with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell voltage 17': Sending state 0.00000 V with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell resistance 17': Sending state 0.00000 Ω with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell voltage 18': Sending state 0.00000 V with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell resistance 18': Sending state 0.00000 Ω with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell voltage 19': Sending state 0.00000 V with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell resistance 19': Sending state 0.00000 Ω with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell voltage 20': Sending state 0.00000 V with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell resistance 20': Sending state 0.00000 Ω with 3 decimals of accuracy
[11:37:34][D][sensor:127]: 'jk-bms cell voltage 21': Sending state 0.00000 V with 3 decimals of accuracy
[... cut ...]
[11:37:34][D][jk_bms_ble:556]: Unknown168: 8F.06.00.00.00.00.00.00.00.00.00.00.00.00.07.00.01.00.00.00.E8.03.00.00.0C.00.B2.2A.3F.40.00.00.00.00.E2.04.00.00.00.00.00.01.00.05.00.00 (46)                                                                                                                                                     
[11:37:34][D][jk_bms_ble:570]: Unknown189: 03 00
[11:37:34][D][jk_bms_ble:574]: Unknown192: 0C
[11:37:34][D][jk_bms_ble:576]: Unknown193: 00 B2 (0x00 0x8D)
[11:37:34][D][jk_bms_ble:578]: Unknown195: 2A 3F (0x21 0x40)
```

syssi, 2022-09-18:
At the moment I don't see any differences to other BMS versions. It's still possible your BMS behaves a bit special if the BLE connection is establed but I would like to rule out other issues beforehand.

1. Please use the `esp32-ble-example.yaml` as starting point. Change the MAC address at the top of the file and flash the exmaple configuration to your ESP. I want to make sure we are talking to a single BMS for now.
2. Make sure the status LED of the `20:21:11:28:18:dd` BMS is blinking. This indicates the BMS is able to accept a new bluetooth connection. The bluetooth module is unable to hande two connections (f.e. android app + ESPHome). If you close the android app the connection to the BMS isn't disconnected. You have to force close the android app.
3. Let ESPHome connect to the BMS. The status LED should be solid now. If the BMS receives an instruction/command from the ESP it will beep.
4. Please provide a capture of the ESPHome log again.

If it doesn't work out of the box please increase the log level of some components:

```
logger:
  level: VERY_VERBOSE
  logs:
    scheduler: DEBUG
    component: DEBUG
    esp32_ble_tracker: DEBUG
    sensor: DEBUG
```

Reflash the ESP and provide another log. This time the log should include the complete transmitted and received raw data. I assume the assembly of partial frames doesn't work for you because the MTU of your BMS is different.

woodie2, 2022-09-18:
[log.txt](https://github.com/syssi/esphome-jk-bms/files/9593903/log.txt)

I did it, but it is the same. The BMS is beeping on log entry, but no solid LED, always blinking.

syssi, 2022-09-18:
Are you able to read the ESPHome log via a serial connection? There are some early log messages missing I'm interested in. I cannot find the `Request device info` message and I would like to see the setup of the BLE connection.

woodie2, 2022-09-18:
[log.txt](https://github.com/syssi/esphome-jk-bms/files/9594294/log.txt)

ok now better? :)

syssi, 2022-09-18:
Good job! This is the important part of your log:

```
[17:25:56][D][ble_client:047]: Found device at MAC address [20:21:11:28:18:DD]
[17:25:56][I][ble_client:086]: Attempting BLE connection to 20:21:11:28:18:dd
[17:25:57][V][ble_client:126]: [20:21:11:28:18:dd] ESP_GATTC_CONNECT_EVT
[17:25:57][V][ble_client:116]: [20:21:11:28:18:dd] ESP_GATTC_OPEN_EVT
[17:25:57][V][ble_client:144]: cfg_mtu status 0, mtu 247
[17:25:57][V][ble_client:169]: [20:21:11:28:18:dd] ESP_GATTC_SEARCH_CMPL_EVT
[17:25:57][I][ble_client:171]: Service UUID: 0xFFE0
[17:25:57][I][ble_client:172]:   start_handle: 0x1  end_handle: 0xffff
[17:25:57][I][ble_client:384]:  characteristic 0xFFE1, handle 0x3, properties 0xc
[17:25:57][I][ble_client:384]:  characteristic 0xFFE1, handle 0x5, properties 0x12
[17:25:57][V][ble_client:419]:    descriptor 0x2902, handle 0x6
[17:25:57][D][esp32_ble_tracker:222]: Starting scan...
[17:25:57][W][ble_client:182]: No descriptor found for notify of handle 0x3
[17:25:57][I][jk_bms_ble:162]: Request device info
[17:25:57][D][jk_bms_ble:1154]: Write register: AA.55.90.EB.97.00.00.00.00.00.00.00.00.00.00.00.00.00.00.11 (20)
[17:25:59][D][esp32_ble_tracker:731]: Found device 1B:D1:46:7C:6A:75 RSSI=-70
[17:25:59][D][esp32_ble_tracker:752]:   Address Type: RANDOM
[17:26:00][I][jk_bms_ble:296]: Request status notification
[17:26:00][D][jk_bms_ble:1154]: Write register: AA.55.90.EB.96.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10 (20)
```

Compared to my BMS model it's a big difference:

```
[09:52:57][W][jk_bms_ble:291]: [c8:47:8c:f2:7e:03] Not connected
[09:52:58][D][sensor:127]: 'Heap Free': Sending state 108028.00000 B with 0 decimals of accuracy
[09:52:58][D][sensor:127]: 'Heap Max Block': Sending state 83968.00000 B with 0 decimals of accuracy
[09:52:58][D][sensor:127]: 'Loop Time': Sending state 17.00000 ms with 0 decimals of accuracy
[09:52:59][D][ble_client:047]: Found device at MAC address [C8:47:8C:F2:7E:03]
[09:52:59][I][ble_client:086]: Attempting BLE connection to c8:47:8c:f2:7e:03
[09:53:01][I][ble_client:171]: Service UUID: 0x1800
[09:53:01][I][ble_client:172]:   start_handle: 0x1  end_handle: 0x9
[09:53:01][I][ble_client:384]:  characteristic 0x2A00, handle 0x3, properties 0xa
[09:53:01][I][ble_client:384]:  characteristic 0x2A01, handle 0x5, properties 0xa
[09:53:01][I][ble_client:384]:  characteristic 0x2A02, handle 0x7, properties 0x2
[09:53:01][I][ble_client:384]:  characteristic 0x2A04, handle 0x9, properties 0x2
[09:53:01][I][ble_client:171]: Service UUID: 0x1801
[09:53:01][I][ble_client:172]:   start_handle: 0xa  end_handle: 0xd
[09:53:01][I][ble_client:384]:  characteristic 0x2A05, handle 0xc, properties 0x22
[09:53:01][I][ble_client:171]: Service UUID: 0xFFE0
[09:53:01][I][ble_client:172]:   start_handle: 0xe  end_handle: 0x13
[09:53:01][I][ble_client:384]:  characteristic 0xFFE2, handle 0x10, properties 0x4
[09:53:01][I][ble_client:384]:  characteristic 0xFFE1, handle 0x12, properties 0x1c
[09:53:01][I][ble_client:171]: Service UUID: 0x180A
[09:53:01][I][ble_client:172]:   start_handle: 0x14  end_handle: 0x26
[09:53:01][I][ble_client:384]:  characteristic 0x2A29, handle 0x16, properties 0x2
[09:53:01][I][ble_client:384]:  characteristic 0x2A24, handle 0x18, properties 0x2
[09:53:01][I][ble_client:384]:  characteristic 0x2A25, handle 0x1a, properties 0x2
[09:53:01][I][ble_client:384]:  characteristic 0x2A27, handle 0x1c, properties 0x2
[09:53:01][I][ble_client:384]:  characteristic 0x2A26, handle 0x1e, properties 0x2
[09:53:01][I][ble_client:384]:  characteristic 0x2A28, handle 0x20, properties 0x2
[09:53:01][I][ble_client:384]:  characteristic 0x2A23, handle 0x22, properties 0x2
[09:53:01][I][ble_client:384]:  characteristic 0x2A2A, handle 0x24, properties 0x2
[09:53:01][I][ble_client:384]:  characteristic 0x2A50, handle 0x26, properties 0x2
[09:53:01][I][ble_client:171]: Service UUID: 0x180F
[09:53:01][I][ble_client:172]:   start_handle: 0x27  end_handle: 0x2a
[09:53:01][I][ble_client:384]:  characteristic 0x2A19, handle 0x29, properties 0x12
[09:53:01][I][ble_client:171]: Service UUID: F000FFC0-0451-4000-B000-000000000000
[09:53:01][I][ble_client:172]:   start_handle: 0x2b  end_handle: 0x33
[09:53:01][I][ble_client:384]:  characteristic F000FFC1-0451-4000-B000-000000000000, handle 0x2d, properties 0x1c
[09:53:01][I][ble_client:384]:  characteristic F000FFC2-0451-4000-B000-000000000000, handle 0x31, properties 0x1c
[09:53:01][D][esp32_ble_tracker:222]: Starting scan...
[09:53:01][I][jk_bms_ble:162]: Request device info
[09:53:01][D][jk_bms_ble:1154]: Write register: AA.55.90.EB.97.00.00.00.00.00.00.00.00.00.00.00.00.00.00.11 (20)
[09:53:01][I][jk_bms_ble:1065]: Device info frame (320 bytes):
[09:53:01][I][jk_bms_ble:1117]:   Vendor ID: JK-BD6A17S6P
[09:53:01][I][jk_bms_ble:1118]:   Hardware version: 7.X
[09:53:01][I][jk_bms_ble:1119]:   Software version: 7.1.0H
[09:53:01][I][jk_bms_ble:1120]:   Uptime: 19409100 s
[09:53:01][I][jk_bms_ble:1121]:   Power on count: 14
[09:53:01][I][jk_bms_ble:1122]:   Device name: JK-BD6A17S6P
[09:53:01][I][jk_bms_ble:1123]:   Device passcode: 1234
[09:53:01][I][jk_bms_ble:1124]:   Manufacturing date: 210609
[09:53:01][I][jk_bms_ble:1125]:   Serial number: 1060706180
[09:53:01][I][jk_bms_ble:1126]:   Passcode: 0000
[09:53:01][I][jk_bms_ble:1127]:   User data: Input Userdata
[09:53:01][I][jk_bms_ble:1128]:   Setup passcode: 123456
[09:53:02][I][jk_bms_ble:296]: Request status notification
[09:53:02][D][jk_bms_ble:1154]: Write register: AA.55.90.EB.96.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10 (20)
[09:53:02][I][jk_bms_ble:802]: Settings frame (320 bytes):
[09:53:02][D][jk_bms_ble:828]:   Unknown6: 0.600000
[09:53:02][I][jk_bms_ble:830]:   Cell UVP: 2.650000 V
```
```

syssi, 2022-09-18:
Some more notes for myself to describe the traffic of yout btnsoop capture for a better comparison:

1. Sent Write Request (0x12) to
   Service UUID: 0xffe0 
   Characteristic UUID: 0xffe1
   UUID: Client Characteristic Configuration: 0x2902
   using handle **0x6**
   
The BMS will respond with a "write response 0x13" at handle 0x6.

2. Sent Write Command (0x52) to
   Service UUID: 0xffe0 
   UUID: 0xffe1
   using handle **0x3**
   Payload: aa5590eb9700f42af61e98d0e99b64a86aafd62a

The BMS will respond with 3 notifications (handle **0x5**):

2.1 55aaeb9003744a4b2d4231413230533135500000000031302e585700000031302e30370000007cc6e9000f0000004a4b2d4231413230533135500000000031323334000000000000000000000000323230333132000031313232313039323736003030303000496e70757420557365726461746100003132333435360000000000000000000000000000000000000000000000000000
2.2 000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000072
2.3 aa5590ebc8010100000000000000000000000044

This response should be assembled in 1-2 JK02 frames. The purpose of 2.3 is unknown.

3. Sent Write Commaind (0x52) to
   Service UUID: 0xffe0 
   UUID: 0xffe1
   using handle **0x3**
   Payload: aa5590eb9600c8887d97e4dace50456b704b1ed9

The BMS will respond with a settings frame (2-3 notifications = 300 + 20 bytes) and cell info frames (2 notifications = 300 bytes) on handle **0x5**.

syssi, 2022-09-18:
I've pushed a small change. Please update your configuration YAML to:

```
substitutions:
  name: jk-bms
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@add-new-ble-module-support
```

flash again. If it doesn't work out of the box I would be happy about another esphome log capture (starting with `Found device at MAC address ...`).

woodie2, 2022-09-19:
```
[12:48:15][V][ble_client:108]: gattc registered app id 1
[12:48:15][D][switch:037]: 'jk-bms enable bluetooth connection': Sending state ON
[12:48:15][D][ble_client:047]: Found device at MAC address [20:21:11:28:18:DD]
[12:48:15][D][esp32_ble_tracker:731]: Found device 21:4C:3C:68:21:8C RSSI=-85
[12:48:15][D][esp32_ble_tracker:752]:   Address Type: RANDOM
[12:48:15][I][ble_client:086]: Attempting BLE connection to 20:21:11:28:18:dd
[12:48:15][V][ble_client:126]: [20:21:11:28:18:dd] ESP_GATTC_CONNECT_EVT
[12:48:15][V][ble_client:116]: [20:21:11:28:18:dd] ESP_GATTC_OPEN_EVT
[12:48:16][V][ble_client:144]: cfg_mtu status 0, mtu 247
[12:48:16][V][ble_client:169]: [20:21:11:28:18:dd] ESP_GATTC_SEARCH_CMPL_EVT
[12:48:16][I][ble_client:171]: Service UUID: 0xFFE0
[12:48:16][I][ble_client:172]:   start_handle: 0x1  end_handle: 0xffff
[12:48:16][I][ble_client:384]:  chaGuru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[12:48:16]
[12:48:16]Core  1 register dump:
[12:48:16]PC      : 0x400dd2e4  PS      : 0x00060030  A0      : 0x800d392a  A1      : 0x3ffcde40
←[33mWARNING Decoded 0x400dd2e4: esphome::jk_bms_ble::JkBmsBle::gattc_event_handler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at C:\Users\Cavin\Dropbox\BMS\esphome-jk-bms-main\.esphome\build\jk-bms/src/esphome/components/jk_bms_ble/jk_bms_ble.cpp:153
 (inlined by) esphome::jk_bms_ble::JkBmsBle::gattc_event_handler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at C:\Users\Cavin\Dropbox\BMS\esphome-jk-bms-main\.esphome\build\jk-bms/src/esphome/components/jk_bms_ble/jk_bms_ble.cpp:129←[0m
[12:48:16]A2      : 0x3ffced60  A3      : 0x00000003  A4      : 0x00000003  A5      : 0x0000000a
[12:48:16]A6      : 0x00000001  A7      : 0x00000000  A8      : 0x800dd2e1  A9      : 0x3ffcde00
[12:48:16]A10     : 0x00000000  A11     : 0x00002902  A12     : 0x0000ffe1  A13     : 0x0000ffff
[12:48:16]A14     : 0x3ffcde18  A15     : 0x00000003  SAR     : 0x00000016  EXCCAUSE: 0x0000001c
[12:48:16]EXCVADDR: 0x00000012  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000
[12:48:16]
[12:48:16]Backtrace:0x400dd2e1:0x3ffcde40 0x400d3927:0x3ffcde80 0x400d3d00:0x3ffcdf10 0x400d4a46:0x3ffcdf30 0x400d5bd5:0x3ffcdf50 0x401c0799:0x3ffce050 0x401c0885:0x3ffce070 0x400e7d7a:0x3ffce090 0x400eee39:0x3ffce0d0 0x400d3f23:0x3ffce190
←[33mWARNING Found stack trace! Trying to decode it←[0m
←[33mWARNING Decoded 0x400dd2e1: esphome::jk_bms_ble::JkBmsBle::gattc_event_handler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at C:\Users\Cavin\Dropbox\BMS\esphome-jk-bms-main\.esphome\build\jk-bms/src/esphome/components/jk_bms_ble/jk_bms_ble.cpp:153
 (inlined by) esphome::jk_bms_ble::JkBmsBle::gattc_event_handler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at C:\Users\Cavin\Dropbox\BMS\esphome-jk-bms-main\.esphome\build\jk-bms/src/esphome/components/jk_bms_ble/jk_bms_ble.cpp:129←[0m
←[33mWARNING Decoded 0x400d3927: esphome::ble_client::BLEClient::gattc_event_handler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at C:\Users\Cavin\Dropbox\BMS\esphome-jk-bms-main\.esphome\build\jk-bms/src/esphome/components/ble_client/ble_client.cpp:205←[0m
←[33mWARNING Decoded 0x400d3d00: esphome::ble_client::BLEClient::gattc_event_handler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at C:\Users\Cavin\Dropbox\BMS\esphome-jk-bms-main\.esphome\build\jk-bms/src/esphome/components/ble_client/ble_client.cpp:98 (discriminator 1)←[0m
←[33mWARNING Decoded 0x400d4a46: esphome::esp32_ble_tracker::ESP32BLETracker::real_gattc_event_handler_(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at C:\Users\Cavin\Dropbox\BMS\esphome-jk-bms-main\.esphome\build\jk-bms/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:309 (discriminator 2)←[0m
←[33mWARNING Decoded 0x400d5bd5: esphome::esp32_ble_tracker::ESP32BLETracker::loop() at C:\Users\Cavin\Dropbox\BMS\esphome-jk-bms-main\.esphome\build\jk-bms/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:62←[0m
←[33mWARNING Decoded 0x401c0799: esphome::Component::call_loop() at C:\Users\Cavin\Dropbox\BMS\esphome-jk-bms-main\.esphome\build\jk-bms/src/esphome/core/component.cpp:75←[0m
←[33mWARNING Decoded 0x401c0885: esphome::Component::call() at C:\Users\Cavin\Dropbox\BMS\esphome-jk-bms-main\.esphome\build\jk-bms/src/esphome/core/component.cpp:97←[0m
←[33mWARNING Decoded 0x400e7d7a: esphome::Application::setup() at C:\Users\Cavin\Dropbox\BMS\esphome-jk-bms-main\.esphome\build\jk-bms/src/esphome/core/application.cpp:52 (discriminator 2)←[0m
←[33mWARNING Decoded 0x400eee39: setup() at C:\Users\Cavin\Dropbox\BMS\esphome-jk-bms-main\.esphome\build\jk-bms/src/main.cpp:2865 (discriminator 4)←[0m
←[33mWARNING Decoded 0x400d3f23: esphome::loop_task(void*) at C:\Users\Cavin\Dropbox\BMS\esphome-jk-bms-main\.esphome\build\jk-bms/src/esphome/components/esp32/core.cpp:72←[0m
[12:48:17]
[12:48:17]
[12:48:17]ELF file SHA256: 41d8f665ca436837
[12:48:17]
[12:48:17]Rebooting...
```

woodie2, 2022-09-19:
no beep from the bms with that version
and after that I need to reset the bms, otherwise it is no longer visible, even for the android app

syssi, 2022-09-19:
Sad news. The bad thing:

```
[17:25:57][I][ble_client:384]:  characteristic 0xFFE1, handle 0x3, properties 0xc
[17:25:57][I][ble_client:384]:  characteristic 0xFFE1, handle 0x5, properties 0x12
[17:25:57][V][ble_client:419]:    descriptor 0x2902, handle 0x6
```

The BLE module of your BMS advertises two characteristics with the same UUID (0xffe1). This isn't good practice and I'm a bit confused how to handle this. The BMS expects the notification subscription via handle `0x6`. The notifications will arrive via handle `0x5` later on. Handle `0x3` is used to issue commands/settings.

I will ping you as soon another try is ready.

woodie2, 2022-09-19:
perhaps a decompile of the apk show use the usage, I will try it

syssi, 2022-09-19:
I've hardcoded handle `0x6` now. Please recompile & flash the configuration again and give it a try. I hope the BLE module doesn't get stuck this time again.

syssi, 2022-09-19:
Hmm.. it looks like using the descriptor 0x2902 to subscribe the notifications was a good idea compared to line 225 the `BleService.java`.

woodie2, 2022-09-19:
[14:59:44][V][ble_client:108]: gattc registered app id 1
[14:59:44][D][switch:037]: 'jk-bms enable bluetooth connection': Sending state ON
[14:59:44][D][ble_client:047]: Found device at MAC address [20:21:11:28:18:DD]
[14:59:44][I][ble_client:086]: Attempting BLE connection to 20:21:11:28:18:dd
[14:59:44][V][ble_client:126]: [20:21:11:28:18:dd] ESP_GATTC_CONNECT_EVT
[14:59:44][V][ble_client:116]: [20:21:11:28:18:dd] ESP_GATTC_OPEN_EVT
[14:59:45][V][ble_client:144]: cfg_mtu status 0, mtu 247
[14:59:45][V][ble_client:169]: [20:21:11:28:18:dd] ESP_GATTC_SEARCH_CMPL_EVT
[14:59:45][I][ble_client:171]: Service UUID: 0xFFE0
[14:59:45][I][ble_client:172]:   start_handle: 0x1  end_handle: 0xffff
[14:59:45][I][ble_client:384]:  characteristic 0xFFE1, handle 0x3, properties 0xc
[14:59:45][I][ble_client:384]:  characteristic 0xFFE1, handle 0x5, properties 0x12
[14:59:45][V][ble_client:419]:    descriptor 0x2902, handle 0x6
[14:59:45][D][esp32_ble_tracker:222]: Starting scan...
[14:59:45][W][ble_client:182]: No descriptor found for notify of handle 0x6
[14:59:45][I][jk_bms_ble:161]: Request device info
[14:59:45][D][jk_bms_ble:1153]: Write register: AA.55.90.EB.97.00.00.00.00.00.00.00.00.00.00.00.00.00.00.11 (20)
[14:59:46][I][jk_bms_ble:295]: Request status notification
[14:59:46][D][jk_bms_ble:1153]: Write register: AA.55.90.EB.96.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10 (20)

syssi, 2022-09-19:
I've pushed another change. ;-) This time we try to subscribe via handle 0x5. Please try again (recompile + flash).

woodie2, 2022-09-19:
`[15:12:53][D][ble_client:047]: Found device at MAC address [20:21:11:28:18:DD]
[15:12:53][I][ble_client:086]: Attempting BLE connection to 20:21:11:28:18:dd
[15:12:54][V][ble_client:126]: [20:21:11:28:18:dd] ESP_GATTC_CONNECT_EVT
[15:12:54][V][ble_client:116]: [20:21:11:28:18:dd] ESP_GATTC_OPEN_EVT
[15:12:54][W][jk_bms_ble:290]: [20:21:11:28:18:dd] Not connected
[15:12:54][V][ble_client:144]: cfg_mtu status 0, mtu 247
[15:12:54][V][ble_client:169]: [20:21:11:28:18:dd] ESP_GATTC_SEARCH_CMPL_EVT
[15:12:54][I][ble_client:171]: Service UUID: 0xFFE0
[15:12:54][I][ble_client:172]:   start_handle: 0x1  end_handle: 0xffff
[15:12:54][I][ble_client:384]:  characteristic 0xFFE1, handle 0x3, properties 0xc
[15:12:54][I][ble_client:384]:  characteristic 0xFFE1, handle 0x5, properties 0x12
[15:12:54][V][ble_client:419]:    descriptor 0x2902, handle 0x6
[15:12:54][D][esp32_ble_tracker:222]: Starting scan...
[15:12:56][I][jk_bms_ble:161]: Request device info
[15:12:56][D][jk_bms_ble:1153]: Write register: AA.55.90.EB.97.00.00.00.00.00.00.00.00.00.00.00.00.00.00.11 (20)
[15:12:57][D][esp32_ble_tracker:731]: Found device 0B:09:60:12:0E:BE RSSI=-63
[15:12:57][D][esp32_ble_tracker:752]:   Address Type: RANDOM

`

syssi, 2022-09-19:
Still no solid LED + incoming data?

woodie2, 2022-09-19:
no, every 5 sec a beep

`[15:13:29][I][jk_bms_ble:295]: Request status notification
[15:13:29][D][jk_bms_ble:1153]: Write register: AA.55.90.EB.96.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10 (20)
[15:13:29][D][esp32_ble_tracker:731]: Found device 16:AD:C2:FC:BD:B0 RSSI=-67
[15:13:29][D][esp32_ble_tracker:752]:   Address Type: RANDOM
[15:13:34][I][jk_bms_ble:295]: Request status notification
[15:13:34][D][jk_bms_ble:1153]: Write register: AA.55.90.EB.96.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10 (20)
[15:13:39][I][jk_bms_ble:295]: Request status notification
[15:13:39][D][jk_bms_ble:1153]: Write register: AA.55.90.EB.96.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10 (20)
[15:13:44][I][jk_bms_ble:295]: Request status notification
[15:13:44][D][jk_bms_ble:1153]: Write register: AA.55.90.EB.96.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10 (20)`

syssi, 2022-09-19:
Nevertheless, i think we are one step further. We solved this warning:

```
[17:25:57][W][ble_client:182]: No descriptor found for notify of handle 0x3
...
[14:59:45][W][ble_client:182]: No descriptor found for notify of handle 0x6
```

As next step I will add some debug output to the "notification receive handler". This will take a bit.

syssi, 2022-09-19:
I've pushed another change which adds some debug output. Please give it a try.

woodie2, 2022-09-19:
```
[15:41:30][D][ble_client:047]: Found device at MAC address [20:21:11:28:18:DD]
[15:41:30][I][ble_client:086]: Attempting BLE connection to 20:21:11:28:18:dd
[15:41:30][V][ble_client:126]: [20:21:11:28:18:dd] ESP_GATTC_CONNECT_EVT
[15:41:30][V][ble_client:116]: [20:21:11:28:18:dd] ESP_GATTC_OPEN_EVT
[15:41:31][V][ble_client:144]: cfg_mtu status 0, mtu 247
[15:41:31][V][ble_client:169]: [20:21:11:28:18:dd] ESP_GATTC_SEARCH_CMPL_EVT
[15:41:31][I][ble_client:171]: Service UUID: 0xFFE0
[15:41:31][I][ble_client:172]:   start_handle: 0x1  end_handle: 0xffff
[15:41:31][I][ble_client:384]:  characteristic 0xFFE1, handle 0x3, properties 0xc
[15:41:31][I][ble_client:384]:  characteristic 0xFFE1, handle 0x5, properties 0x12
[15:41:31][V][ble_client:419]:    descriptor 0x2902, handle 0x6
[15:41:31][I][jk_bms_ble:161]: Request device info
[15:41:31][D][jk_bms_ble:1156]: Write register: AA.55.90.EB.97.00.00.00.00.00.00.00.00.00.00.00.00.00.00.11 (20)
[15:41:31][D][esp32_ble_tracker:731]: Found device 13:81:3B:39:31:CB RSSI=-55
[15:41:31][D][esp32_ble_tracker:752]:   Address Type: RANDOM
[15:41:31][D][jk_bms_ble:168]: Notification received: 55.AA.EB.90.03.58.4A.4B.2D.42.31.41.32.30.53.31.35.50.00.00.00.00.31.30.2E.58.57.00.00.00.31.30.2E.30.37.00.00.00.D4.3A.F0.00.15.00.00.00.4A.4B.2D.42.31.41.32.30.53.31.35.50.00.00.00.00.31.32.33.34.00.00.00.00.00.00.00.00.00.00.00.00.32.32.30.33.31.32.00.00.31.31.32.32.31.30.39.32.37.36.00.30.30.30.30.00.49.6E.70.75.74.20.55.73.65.72.64.61.74.61.00.00.31.32.33.34.35.36.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (150)
[15:41:31][D][jk_bms_ble:168]: Notification received: 00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.2F (150)
[15:41:31][D][jk_bms_ble:168]: Notification received: AA.55.90.EB.C8.01.01.00.00.00.00.00.00.00.00.00.00.00.00.44 (20)
[15:41:45][D][jk_bms_ble:168]: Notification received: 55.AA.EB.90.02.61.F9.0C.F9.0C.FD.0C.FC.0C.F7.0C.FD.0C.FD.0C.F9.0C.F8.0C.F3.0C.00.0D.FD.0C.F2.0C.F2.0C.F9.0C.FD.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.F9.0C.0F.00.03.0C.4F.00.4D.00.4C.00.47.00.48.00.4B.00.46.00.43.00.47.00.49.00.53.00.52.00.4C.00.4B.00.43.00.43.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.97.CF.00.00.4A.98.02.00.80.F3.FF.FF.DA.00.CE.00.DB.00.00.00.00.00.00.5E.05.72.01.00.A0.86.01.00 (150)
[15:41:45][D][jk_bms_ble:168]: Notification received: 38.00.00.00.3C.8C.56.00.64.00.87.08.02.3B.F0.00.01.01.8D.06.00.00.00.00.00.00.00.00.00.00.00.00.07.00.01.00.00.00.E8.03.00.00.10.00.C3.3D.41.40.00.00.00.00.E2.04.00.00.00.00.00.01.00.05.00.00.A7.62.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.B2 (150)
[15:41:45][D][jk_bms_ble:168]: Notification received: 55.AA.EB.90.02.62.FB.0C.F1.0C.FC.0C.01.0D.F9.0C.F9.0C.F7.0C.FC.0C.FE.0C.F0.0C.FD.0C.F9.0C.F1.0C.FC.0C.FC.0C.FB.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.F9.0C.0F.00.03.0C.4F.00.4D.00.4C.00.47.00.48.00.4B.00.46.00.43.00.47.00.49.00.53.00.52.00.4C.00.4B.00.43.00.43.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.96.CF.00.00.46.98.02.00.80.F3.FF.FF.DA.00.CE.00.DB.00.00.00.00.00.00.5E.04.72.01.00.A0.86.01.00 (150)
[15:41:45][D][jk_bms_ble:168]: Notification received: 38.00.00.00.3C.8C.56.00.64.00.87.08.03.3B.F0.00.01.01.8D.06.00.00.00.00.00.00.00.00.00.00.00.00.07.00.01.00.00.00.E8.03.00.00.10.00.C3.3D.41.40.00.00.00.00.E2.04.00.00.00.00.00.01.00.05.00.00.AD.62.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.B3 (150)
[15:41:46][D][jk_bms_ble:168]: Notification received: 55.AA.EB.90.02.63.FD.0C.F9.0C.F7.0C.FD.0C.FD.0C.FE.0C.F7.0C.F8.0C.FB.0C.F7.0C.00.0D.F6.0C.F1.0C.F4.0C.F6.0C.F8.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.FA.0C.0F.00.03.0C.4F.00.4D.00.4C.00.47.00.48.00.4B.00.46.00.43.00.47.00.49.00.53.00.52.00.4C.00.4B.00.43.00.43.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.9D.CF.00.00.5E.98.02.00.80.F3.FF.FF.DA.00.CE.00.DB.00.00.00.00.00.00.5E.04.72.01.00.A0.86.01.00 (150)
[15:41:46][D][jk_bms_ble:168]: Notification received: 38.00.00.00.3C.8C.56.00.64.00.87.08.03.3B.F0.00.01.01.8D.06.00.00.00.00.00.00.00.00.00.00.00.00.07.00.01.00.00.00.E8.03.00.00.10.00.C3.3D.41.40.00.00.00.00.E2.04.00.00.00.00.00.01.00.05.00.00.B2.62.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.D2 (150)
[15:41:46][D][jk_bms_ble:168]: Notification received: 55.AA.EB.90.02.64.F9.0C.F9.0C.FD.0C.FE.0C.F7.0C.FD.0C.FD.0C.FD.0C.F9.0C.F0.0C.00.0D.FD.0C.F6.0C.F3.0C.00.0D.FE.0C.00.00.00.00.00.00.00.00.00.00.00.00.00.0 [... cut ...]
[15:41:50][D][jk_bms_ble:168]: Notification received: 38.00.00.00.40.8C.56.00.64.00.87.08.07.3B.F0.00.01.01.8D.06.00.00.00.00.00.00.00.00.00.00.00.00.07.00.01.00.00.00.E8.03.00.00.0F.00.C3.3D.41.40.00.00.00.00.E2.04.00.00.00.00.00.01.00.05.00.00.D9.62.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.28 (150)
```

syssi, 2022-09-19:
Yay! This looks good. I've an idea what needs to be changed next.

syssi, 2022-09-19:
I've pushed another change. The notifications should be decoded now.

woodie2, 2022-09-19:
```
[16:45:48][D][esp32_ble_tracker:731]: Found device 0B:0D:7C:2C:F8:01 RSSI=-54
[16:45:48][D][esp32_ble_tracker:752]:   Address Type: RANDOM
[16:45:48][I][ble_client:086]: Attempting BLE connection to 20:21:11:28:18:dd
[16:45:48][V][ble_client:126]: [20:21:11:28:18:dd] ESP_GATTC_CONNECT_EVT
[16:45:48][V][ble_client:116]: [20:21:11:28:18:dd] ESP_GATTC_OPEN_EVT
[16:45:48][V][ble_client:144]: cfg_mtu status 0, mtu 247
[16:45:48][V][ble_client:169]: [20:21:11:28:18:dd] ESP_GATTC_SEARCH_CMPL_EVT
[16:45:48][I][ble_client:171]: Service UUID: 0xFFE0
[16:45:48][I][ble_client:172]:   start_handle: 0x1  end_handle: 0xffff
[16:45:48][I][ble_client:384]:  characteristic 0xFFE1, handle 0x3, properties 0xc
[16:45:48][I][ble_client:384]:  characteristic 0xFFE1, handle 0x5, properties 0x12
[16:45:48][V][ble_client:419]:    descriptor 0x2902, handle 0x6
[16:45:48][D][esp32_ble_tracker:222]: Starting scan...
[16:45:48][I][jk_bms_ble:161]: Request device info
[16:45:48][D][jk_bms_ble:1156]: Write register: AA.55.90.EB.97.00.00.00.00.00.00.00.00.00.00.00.00.00.00.11 (20)
[16:45:49][D][jk_bms_ble:168]: Notification received (notify handle 5, char handle 3): 55.AA.EB.90.03.D5.4A.4B.2D.42.31.41.32.30.53.31.35.50.00.00.00.00.31.30.2E.58.57.00.00.00.31.30.2E.30.37.00.00.00.E4.48.F0.00.15.00.00.00.4A.4B.2D.42.31.41.32.30.53.31.35.50.00.00.00.00.31.32.33.34.00.00.00.00.00.00.00.00.00.00.00.00.32.32.30.33.31.32.00.00.31.31.32.32.31.30.39.32.37.36.00.30.30.30.30.00.49.6E.70.75.74.20.55.73.65.72.64.61.74.61.00.00.31.32.33.34.35.36.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.0
[16:45:49][D][jk_bms_ble:168]: Notification received (notify handle 5, char handle 3): 00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.0
[16:45:49][I][jk_bms_ble:1067]: Device info frame (300 bytes):
[16:45:49][VV][jk_bms_ble:1068]:   55.AA.EB.90.03.D5.4A.4B.2D.42.31.41.32.30.53.31.35.50.00.00.00.00.31.30.2E.58.57.00.00.00.31.30.2E.30.37.00.00.00.E4.48.F0.00.15.00.00.00.4A.4B.2D.42.31.41.32.30.53.31.35.50.00.00.00.00.31.32.33.34.00.00.00.00.00.00.00.00.00.00.00.00.32.32.30.33.31.32.00.00.31.31.32.32.31.30.39.32.37.36.00.30.30.30.30.00.49.6E.70.75.74.20.55.73.65.72.64.61.74.61.00.00.31.32.33.34.35.36.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00
[16:45:49][VV][jk_bms_ble:1069]:   00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.CA (140)
[16:45:49][I][jk_bms_ble:1119]:   Vendor ID: JK-B1A20S15P
[16:45:49][I][jk_bms_ble:1120]:   Hardware version: 10.XW
[16:45:49][I][jk_bms_ble:1121]:   Software version: 10.07
[16:45:49][I][jk_bms_ble:1122]:   Uptime: 15747300 s
[16:45:49][I][jk_bms_ble:1123]:   Power on count: 21
[16:45:49][I][jk_bms_ble:1124]:   Device name: JK-B1A20S15P
[16:45:49][I][jk_bms_ble:1125]:   Device passcode: 1234
[16:45:49][I][jk_bms_ble:1126]:   Manufacturing date: 220312
[16:45:49][I][jk_bms_ble:1127]:   Serial number: 1122109276
[16:45:49][I][jk_bms_ble:1128]:   Passcode: 0000
[16:45:49][I][jk_bms_ble:1129]:   User data: Input Userdata
[16:45:49][I][jk_bms_ble:1130]:   Setup passcode: 123456
[16:45:49][D][jk_bms_ble:168]: Notification received (notify handle 5, char handle 3): AA.55.90.EB.C8.01.01.00.00.00.00.00.00.00.00.00.00.00.00.44 (20)
[16:45:50][I][jk_bms_ble:298]: Request status notification
[16:45:50][D][jk_bms_ble:1156]: Write register: AA.55.90.EB.96.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10 (20)
[16:45:50][D][jk_bms_ble:168]: Notification received (notify handle 5, char handle 3): 55.AA.EB.90.01.D5.58.02.00.00.28.0A.00.00.B8.0B.00.00.DE.0D.00.00.48.0D.00.00.05.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.B0.09.00.00.A0.86.01.00.1E.00.00.00.3C.00.00.00.A0.86.01.00.2C.01.00.00.3C.00.00.00.3C.00.00.00.E8.03.00.00.BC.02.00.00.58.02.00.00.BC.02.00.00.58.02.00.00.00.00.00.00.28.00.00.00.84.03.00.00.BC.02.00.00.10.00.00.00.01.00.00.00.01.00.00.00.01.00.00.00.A0.86.01.00.DC.05.00.00.48.0D.00.00.0
[16:45:50][D][jk_bms_ble:168]: Notification received (notify handle 5, char handle 3): 00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.0
[16:45:50][I][jk_bms_ble:804]: Settings frame (300 bytes):
[16:45:51][VV][jk_bms_ble:805]:   55.AA.EB.90.01.D5.58.02.00.00.28.0A.00.00.B8.0B.00.00.DE.0D.00.00.48.0D.00.00.05.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.B0.09.00.00.A0.86.01.00.1E.00.00.00.3C.00.00.00.A0.86.01.00.2C.01.00.00.3C.00.00.00.3C.00.00.0 [... cut ...] 00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.E2 (140)
[16:45:51][D][jk_bms_ble:830]:   Unknown6: 0.600000
[16:45:51][I][jk_bms_ble:832]:   Cell UVP: 2.600000 V
[16:45:51][D][number:012]: 'jk-bms cell voltage undervoltage protection': Sending state 2.600000
[16:45:51][I][jk_bms_ble:836]:   Cell UVPR: 3.000000 V
[16:45:51][D][number:012]: 'jk-bms cell voltage undervoltage recovery': Sending state 3.000000
[16:45:51][I][jk_bms_ble:840]:   Cell OVP: 3.550000 V
```

syssi, 2022-09-19:
I can see a properly decoded `device info` frame and a `settings` frame. Do you see a stream of `cell info` frames too?

woodie2, 2022-09-19:
```
[16:45:56][D][jk_bms_ble:168]: Notification received (notify handle 5, char handle 3): 55.AA.EB.90.02.DF.0C.0D.08.0D.07.0D.0A.0D.0C.0D.0B.0D.0A.0D.08.0D.08.0D.08.0D.0A.0D.07.0D.0B.0D.08.0D.08.0D.0A.0D.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.09.0D.0B.00.07.09.4F.00.4D.00.4C.00.47.00.48.00.4B.00.46.00.43.00.47.00.49.00.53.00.52.00.4C.00.4B.00.43.00.43.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.94.D0.00.00.93.D0.00.00.E8.03.00.00.D6.00.CB.00.D8.00.00.00.00.00.00.5D.D
[16:45:56][D][jk_bms_ble:168]: Notification received (notify handle 5, char handle 3): 38.00.00.00.53.91.56.00.64.00.87.08.E2.49.F0.00.01.01.8D.06.00.00.00.00.00.00.00.00.00.00.00.00.07.00.01.00.00.00.E8.03.05.00.02.00.C3.3D.41.40.00.00.00.00.E2.04.00.00.00.00.00.01.00.05.00.00.63.F7.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.0
[16:45:56][VV][jk_bms_ble:381]: Cell info frame (version 2, 300 bytes):
[16:45:56][VV][jk_bms_ble:382]:   55.AA.EB.90.02.DF.0C.0D.08.0D.07.0D.0A.0D.0C.0D.0B.0D.0A.0D.08.0D.08.0D.08.0D.0A.0D.07.0D.0B.0D.08.0D.08.0D.0A.0D.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.09.0D.0B.00.07.09.4F.00.4D.00.4C.00.47.00.48.00.4B.00.46.00.43.00.47.00.49.00.53.00.52.00.4C.00.4B.00.43.00.43.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.94.D0.00.00.93.D0.00.00.E8.03.00.00.D6.00.CB.00.D8.00.00.00.00.00.00.5D.D9.6D.01.00.A0.86.01.00 (150)
[16:45:56][VV][jk_bms_ble:383]:   38.00.00.00.53.91.56.00.64.00.87.08.E2.49.F0.00.01.01.8D.06.00.00.00.00.00.00.00.00.00.00.00.00.07.00.01.00.00.00.E8.03.05.00.02.00.C3.3D.41.40.00.00.00.00.E2.04.00.00.00.00.00.01.00.05.00.00.63.F7.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.57 (150)
[16:45:56][D][sensor:127]: 'jk-bms cell voltage 1': Sending state 3.34000 V with 3 decimals of accuracy
[16:45:56][D][sensor:127]: 'jk-bms cell resistance 1': Sending state 0.07900 Ω with 3 decimals of accuracy
[16:45:56][D][sensor:127]: 'jk-bms cell voltage 2': Sending state 3.33600 V with 3 decimals of accuracy
[16:45:56][D][sensor:127]: 'jk-bms cell resistance 2': Sending state 0.07700 Ω with 3 decimals of accuracy
[16:45:56][D][sensor:127]: 'jk-bms cell voltage 3': Sending state 3.33500 V with 3 decimals of accuracy
[16:45:56][D][sensor:127]: 'jk-bms cell resistance 3': Sending state 0.07600 Ω with 3 decimals of accuracy
[16:45:56][D][sensor:127]: 'jk-bms cell voltage 4': Sending state 3.33800 V with 3 decimals of accuracy
[16:45:56][D][sensor:127]: 'jk-bms cell resistance 4': Sending state 0.07100 Ω with 3 decimals of accuracy
[16:45:56][D][sensor:127]: 'jk-bms cell voltage 5': Sending state 3.34000 V with 3 decimals of accuracy
[16:45:56][D][sensor:127]: 'jk-bms cell resistance 5': Sending state 0.07200 Ω with 3 decimals of accuracy
[16:45:57][D][sensor:127]: 'jk-bms cell voltage 6': Sending state 3.33900 V with 3 decimals of accuracy
[16:45:57][D][sensor:127]: 'jk-bms cell resistance 6': Sending state 0.07500 Ω with 3 decimals of accuracy
[16:45:57][D][sensor:127]: 'jk-bms cell voltage 7': Sending state 3.33800 V with 3 decimals of accuracy
[16:45:57][D][sensor:127]: 'jk-bms cell resistance 7': Sending state 0.07000 Ω with 3 decimals of accuracy
[16:45:57][D][sensor:127]: 'jk-bms cell voltage 8': Sending state 3.33600 V with 3 decimals of accuracy
[16:45:57][D][sensor:127]: 'jk-bms cell resistance 8': Sending state 0.06700 Ω with 3 decimals of accuracy
[16:45:57][D][sensor:127]: 'jk-bms cell voltage 9': Sending state 3.33600 V with 3 decimals of accuracy
[16:45:57][D][sensor:127]: 'jk-bms cell resistance 9': Sending state 0.07100 Ω with 3 decimals of accuracy
[16:45:57][D][sensor:127]: 'jk-bms cell voltage 10': Sending state 3.33600 V with 3 decimals of accuracy
[16:45:57][D][sensor:127]: 'jk-bms cell resistance 10': Sending state 0.07300 Ω with 3 decimals of accuracy
[16:45:57][D][sensor:127]: 'jk-bms cell voltage 11': Sending state 3.33800 V with 3 decimals of accuracy
[16:45:57][D][sensor:127]: 'jk-bms cell resistance 11': Sending state 0.08300 Ω with 3 decimals of accuracy
[16:45:57][D][sensor:127]: 'jk-bms cell voltage 12': Sending state 3.33500 V with 3 decimals of accuracy
[16:45:57][D][sensor:127]: 'jk-bms cell resistance 12': Sending state 0.08200 Ω with 3 decimals of accuracy
[16:45:57][D][sensor:127]: 'jk-bms cell voltage 13': Sending state 3.33900 V with 3 decimals of accuracy
[16:45:57][D][sensor:127]: 'jk-bms cell resistance 13': Sending state 0.07600 Ω with 3 decimals of accuracy
[16:45:57][D][sensor:127]: 'jk-bms cell voltage 14': Sending state 3.33600 V with 3 decimals of accuracy
[16:45:57][D][sensor:127]: 'jk-bms cell resistance 14': Sending state 0.07500 Ω with 3 decimals of accuracy
```

woodie2, 2022-09-19:
i think all datas are received now :)

woodie2, 2022-09-19:
should I now try to use two jk bms? the old and the new one together with one esp? or does the new code only work with new new bms?

syssi, 2022-09-19:
Perfect! Now we will need one or two iterations for polishing the change. I've reverted the hardcoded handle (`0x5`). We are using `0x3` again. In best case the BMS doesn't care about the exact handle and we are still able to receive the notification stream. Please recompile + flash another time.

syssi, 2022-09-19:
If the handle doesn't need to be hardcoded we have a solution which should work for both BMS versions:

https://github.com/syssi/esphome-jk-bms/pull/176/files

syssi, 2022-09-19:
I don't see multiple

```
[17:15:53][I][jk_bms_ble:162]: Request device info
[17:15:53][D][jk_bms_ble:1154]: Write register: AA.55.90.EB.97.00.00.00.00.00.00.00.00.00.00.00.00.00.00.11 (20)
```

blocks but the log capture means the BMS doesn't respond. Correct?

woodie2, 2022-09-19:
```
[17:16:10][I][jk_bms_ble:296]: Request status notification
[17:16:10][D][jk_bms_ble:1154]: Write register: AA.55.90.EB.96.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10 (20)
[17:16:15][I][jk_bms_ble:296]: Request status notification
[17:16:15][D][jk_bms_ble:1154]: Write register: AA.55.90.EB.96.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10 (20)
```

sorry I shortened it... Yes the BMS doesn't response

syssi, 2022-09-19:
Alright. I will take some rest and try to implement a generic solution which works for both BMS.

woodie2, 2022-09-19:
Thank you and good luck :)

syssi, 2022-09-25:
I've pushed an updated version to the feature branch. It should support/handle both BLE modules now.

syssi, 2022-09-27:
I'm aware of this issue but I don't know the root cause yet.

closed 2022-09-27 by syssi

woodie2, 2022-09-27:
> I'm aware of this issue but I don't know the root cause yet.

this issue is also by using one one bms?
I used until now the solution from https://forum.drbacke.de/viewtopic.php?t=982 there it was an update frequency of less than 1 second working. perhabs you can look in there sourcecode for an idea :)

PR #176 (2022-09-27): Add new BLE module support
Fixes #171

```
substitutions:
  name: jk-bms
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@add-new-ble-module-support
```