syssi/esphome-jk-bms #177: connecton to MQTT di a crash and reboot

preussal, 2022-09-19:
i use the mosquitto MQTT and working fine with tasmota and other Programms.

But the i flash the JK-BMS to a esp then i get always this.

```
~/GIT/esphome-jk-bms$ ./test-esp8266.sh 
INFO Reading configuration esp8266-example-faker.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing jk-bms (board: d1_mini; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
---------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- AsyncMqttClient-esphome @ 0.8.6
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|-- ArduinoJson @ 6.18.5
RAM:   [====      ]  43.5% (used 35640 bytes from 81920 bytes)
Flash: [====      ]  41.4% (used 432225 bytes from 1044464 bytes)
===================================================== [SUCCESS] Took 1.32 seconds =====================================================
INFO Successfully compiled program.
Found multiple options, please choose one:
  [1] /dev/ttyUSB0 (USB2.0-Ser!)
  [2] Over The Air (jk-bms.local)
(number): 1
esptool.py v3.3.1
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: c8:c9:a3:5a:d1:f6
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00000000 to 0x0006afff...
Compressed 436384 bytes to 293172...
Wrote 436384 bytes (293172 compressed) at 0x00000000 in 6.9 seconds (effective 507.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[12:17:29]rll\xb0r$\xe2n

                        \xc4
                            l\xc7
                                 b|\x8e\x90\x92rb\xc4
                                                     b\xf2nnlnnb
                                                                bp\x8c$blrlp\xf2n0

                                                                                  \x90
                                                                                      l
                                                                                       \x8c

                                                                                           b
                                                                                            n\xe2n\xc7
                                                                                                      \xc4
                                                                                                          b\xc4\xf2nn'l\x8cl`\x90nn
                                                                                                                                   $`nr\x8e\x92\x92n

             rr\x92`p\xf2n0

[... 623 lines cut ...]
[12:17:38]3fffff90:  feefeffe feefeffe feefeffe 3fff0304  
[12:17:38]3fffffa0:  3fffdad0 00000000 3fff02f0 40224020  
WARNING Decoded 0x40224020: loop_wrapper() at core_esp8266_main.cpp
[12:17:38]<<<stack<<<
[12:17:38]
[12:17:38]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[12:17:38]
[12:17:38] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[12:17:38]
[12:17:38]load 0x4010f000, len 3460, room 16 
[12:17:38]tail 4
[12:17:38]chksum 0xcc
[12:17:38]load 0x3fff20b8, len 40, room 4 
[12:17:38]tail 4
[12:17:38]chksum 0xc9
[12:17:38]csum 0xc9
[12:17:38]v0006a8a0
[12:17:38]~ld
[12:17:38][I][logger:243]: Log initialized
[12:17:38][C][ota:461]: There have been 3 suspected unsuccessful boot attempts.
[12:17:38][I][app:029]: Running\xfe\xa1\xff[C][wifi:037]: Setting up WiFi...
[12:17:38][D][wifi:373]: Starting scan...
[12:17:38]NWh)[I][jk_bms:062]: Status frame received
```
and then reboot loop, to the save mode.

syssi, 2022-09-19:
If I try to flash the `esp8266-example.yaml` without changes I receive an out of memory exception. To solve this issue I've removed some sensors from the configuration. The diff looks like this:

```diff
diff --git a/esp8266-example-debug.yaml b/esp8266-example-debug.yaml
index a1f935f..6302926 100644
--- a/esp8266-example-debug.yaml
+++ b/esp8266-example-debug.yaml
@@ -1,7 +1,6 @@
 <<: !include esp8266-example.yaml
 
 logger:
-  baud_rate: 0
   level: DEBUG
 
 uart:
diff --git a/esp8266-example.yaml b/esp8266-example.yaml
index 45a9df6..257a1cc 100644
--- a/esp8266-example.yaml
+++ b/esp8266-example.yaml
@@ -2,8 +2,8 @@ substitutions:
   name: jk-bms
   device_description: "Monitor and control a JK-BMS via UART-TTL"
   external_components_source: github://syssi/esphome-jk-bms@main
-  tx_pin: GPIO1
-  rx_pin: GPIO3
+  tx_pin: GPIO4
+  rx_pin: GPIO5
 
 esphome:
   name: ${name}
@@ -25,7 +25,6 @@ wifi:
 
 ota:
 logger:
-  baud_rate: 0
 
 mqtt:
   broker: !secret mqtt_host
@@ -148,90 +147,6 @@ sensor:
       name: "${name} capacity remaining"
     capacity_remaining_derived:
       name: "${name} capacity remaining derived"
-    temperature_sensors:
-      name: "${name} temperature sensors"
-    charging_cycles:
-      name: "${name} charging cycles"
-    total_charging_cycle_capacity:
-      name: "${name} total charging cycle capacity"
-    battery_strings:
-      name: "${name} battery strings"
-    errors_bitmask:
-      name: "${name} errors bitmask"
-    operation_mode_bitmask:
-      name: "${name} operation mode bitmask"
-    total_voltage_overvoltage_protection:
-      name: "${name} total voltage overvoltage protection"
-    total_voltage_undervoltage_protection:
-      name: "${name} total voltage undervoltage protection"
-    cell_voltage_overvoltage_protection:
-      name: "${name} cell voltage overvoltage protection"
-    cell_voltage_overvoltage_recovery:
-      name: "${name} cell voltage overvoltage recovery"
-    cell_voltage_overvoltage_delay:
-      name: "${name} cell voltage overvoltage delay"
-    cell_voltage_undervoltage_protection:
-      name: "${name} cell voltage undervoltage protection"
-    cell_voltage_undervoltage_recovery:
-      name: "${name} cell voltage undervoltage recovery"
-    cell_voltage_undervoltage_delay:
-      name: "${name} cell voltage undervoltage delay"
-    cell_pressure_difference_protection:
-      name: "${name} cell pressure difference protection"
-    discharging_overcurrent_protection:
-      name: "${name} discharging overcurrent protection"
-    discharging_overcurrent_delay:
-      name: "${name} discharging overcurrent delay"
-    charging_overcurrent_protection:
-      name: "${name} charging overcurrent protection"
-    charging_overcurrent_delay:
-      name: "${name} charging overcurrent delay"
-    balance_starting_voltage:
-      name: "${name} balance starting voltage"
-    balance_opening_pressure_difference:
-      name: "${name} balance opening pressure difference"
-    power_tube_temperature_protection:
-      name: "${name} power tube temperature protection"
-    power_tube_temperature_recovery:
-      name: "${name} power tube temperature recovery"
-    temperature_sensor_temperature_protection:
-      name: "${name} temperature sensor temperature protection"
-    temperature_sensor_temperature_recovery:
-      name: "${name} temperature sensor temperature recovery"
-    temperature_sensor_temperature_difference_protection:
-      name: "${name} temperature sensor temperature difference protection"
-    charging_high_temperature_protection:
-      name: "${name} charging high temperature protection"
-    discharging_high_temperature_protection:
-      name: "${name} discharging high temperature protection"
-    charging_low_temperature_protection:
-      name: "${name} charging low temperature protection"
-    charging_low_temperature_recovery:
-      name: "${name} charging low temperature recovery"
-    discharging_low_temperature_protection:
-      name: "${name} discharging low temperature protection"
-    discharging_low_temperature_recovery:
-      name: "${name} discharging low temperature recovery"
-    total_battery_capacity_setting:
-      name: "${name} total battery capacity setting"
-    current_calibration:
-      name: "${name} current calibration"
-    device_address:
-      name: "${name} device address"
-    sleep_wait_time:
-      name: "${name} sleep wait time"
-    alarm_low_volume:
-      name: "${name} alarm low volume"
-    manufacturing_date:
-      name: "${name} manufacturing date"
-    total_runtime:
-      name: "${name} total runtime"
-#    start_current_calibration:
-#      name: "${name} start current calibration"
-    actual_battery_capacity:
-      name: "${name} actual battery capacity"
-#    protocol_version:
-#      name: "${name} protocol version"
 
 text_sensor:
   - platform: jk_bms
```

Could you try the same? In my case it's stable as soon the total number of sensors is reduced. You issue looks a bit different:

```
WARNING Exception type: Access to invalid address: STORE (wild pointer?)
```

Could you tell me the GPIOs you are using for UART? I assume it isn't `GPIO1` and `GPIO3` anymore because you are able to show the serial output of the logger.

preussal, 2022-09-23:
Hi Syssi,
perfect now is working fine.

I use GPIO1 and GPIO3 with the name TX and RX on the PCB

git diff 
```
diff --git a/esp8266-example-debug.yaml b/esp8266-example-debug.yaml
index a1f935f..6302926 100644
--- a/esp8266-example-debug.yaml
+++ b/esp8266-example-debug.yaml
@@ -1,7 +1,6 @@
 <<: !include esp8266-example.yaml
 
 logger:
-  baud_rate: 0
   level: DEBUG
 
 uart:
diff --git a/esp8266-example.yaml b/esp8266-example.yaml
index 45a9df6..540b101 100644
--- a/esp8266-example.yaml
+++ b/esp8266-example.yaml
@@ -10,7 +10,8 @@ esphome:
   comment: ${device_description}
   project:
     name: "syssi.esphome-jk-bms"
-    version: 1.2.0
+    version: latest
+ #   version: 1.2.0
 
 esp8266:
   board: d1_mini
@@ -25,12 +26,15 @@ wifi:
 
 ota:
 logger:
-  baud_rate: 0
+  baud_rate: 115200
+  level: DEBUG
+
 
 mqtt:
   broker: !secret mqtt_host
   username: !secret mqtt_username
   password: !secret mqtt_password
+  port: 1883
   id: mqtt_client
 
 uart:
@@ -148,90 +152,6 @@ sensor:
       name: "${name} capacity remaining"
     capacity_remaining_derived:
       name: "${name} capacity remaining derived"
-    temperature_sensors:
-      name: "${name} temperature sensors"
-    charging_cycles:
-      name: "${name} charging cycles"
-    total_charging_cycle_capacity:
-      name: "${name} total charging cycle capacity"
-    battery_strings:
-      name: "${name} battery strings"
-    errors_bitmask:
-      name: "${name} errors bitmask"
-    operation_mode_bitmask:
-      name: "${name} operation mode bitmask"
-    total_voltage_overvoltage_protection:
-      name: "${name} total voltage overvoltage protection"
-    total_voltage_undervoltage_protection:
-      name: "${name} total voltage undervoltage protection"
[... 105 lines cut ...]
[264631.467720] usb 3-2: Product: USB2.0-Ser!
[264631.522498] usbcore: registered new interface driver usbserial_generic
[264631.522522] usbserial: USB Serial support registered for generic
[264631.524438] usbcore: registered new interface driver ch341
[264631.524454] usbserial: USB Serial support registered for ch341-uart
[264631.524482] ch341 3-2:1.0: ch341-uart converter detected
[264631.535806] ch341-uart ttyUSB0: break control not supported, using simulated break
[264631.535941] usb 3-2: ch341-uart converter now attached to ttyUSB0
[264632.091643] input: BRLTTY 6.4 Linux Screen Driver Keyboard as /devices/virtual/input/input42
[264632.097726] usb 3-2: usbfs: interface 0 claimed by ch341 while 'brltty' sets config #1
[264632.100877] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[264632.100899] ch341 3-2:1.0: device disconnected
```

and with Disabled BRTTY
```
[264860.460662] usb 3-2: USB disconnect, device number 8
[264864.350242] usb 3-2: new full-speed USB device number 9 using xhci_hcd
[264864.511546] usb 3-2: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.54
[264864.511553] usb 3-2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[264864.511557] usb 3-2: Product: USB2.0-Ser!
[264864.521594] ch341 3-2:1.0: ch341-uart converter detected
[264864.534622] ch341-uart ttyUSB0: break control not supported, using simulated break
[264864.534738] usb 3-2: ch341-uart converter now attached to ttyUSB0
```

preussal, 2022-09-23:
and the MQTT Data

```
jk-bms/sensor/jk-bms_cell_voltage_1/state 3.528
jk-bms/sensor/jk-bms_cell_voltage_2/state 3.527
jk-bms/sensor/jk-bms_cell_voltage_3/state 3.535
jk-bms/sensor/jk-bms_cell_voltage_4/state 3.541
jk-bms/sensor/jk-bms_cell_voltage_5/state 3.544
jk-bms/sensor/jk-bms_cell_voltage_6/state 3.555
jk-bms/sensor/jk-bms_cell_voltage_7/state 3.563
jk-bms/sensor/jk-bms_cell_voltage_8/state 3.536
jk-bms/sensor/jk-bms_cell_voltage_9/state 3.556
jk-bms/sensor/jk-bms_cell_voltage_10/state 3.563
jk-bms/sensor/jk-bms_cell_voltage_11/state 3.535
jk-bms/sensor/jk-bms_cell_voltage_12/state 3.560
jk-bms/sensor/jk-bms_cell_voltage_13/state 3.562
jk-bms/sensor/jk-bms_cell_voltage_14/state 3.560
jk-bms/sensor/jk-bms_cell_voltage_15/state 3.562
jk-bms/sensor/jk-bms_cell_voltage_16/state 3.556
jk-bms/sensor/jk-bms_min_cell_voltage/state 3.527
jk-bms/sensor/jk-bms_max_cell_voltage/state 3.563
jk-bms/sensor/jk-bms_max_voltage_cell/state 7
jk-bms/sensor/jk-bms_min_voltage_cell/state 2
jk-bms/sensor/jk-bms_delta_cell_voltage/state 0.036
jk-bms/sensor/jk-bms_average_cell_voltage/state 3.549
jk-bms/sensor/jk-bms_power_tube_temperature/state 35
jk-bms/sensor/jk-bms_temperature_sensor_1/state 31
jk-bms/sensor/jk-bms_temperature_sensor_2/state 32
jk-bms/sensor/jk-bms_total_voltage/state 56.78
jk-bms/sensor/jk-bms_current/state -0.65
jk-bms/sensor/jk-bms_power/state -36.91
jk-bms/sensor/jk-bms_charging_power/state 0.00
jk-bms/sensor/jk-bms_discharging_power/state 36.91
jk-bms/sensor/jk-bms_capacity_remaining/state 95
jk-bms/sensor/jk-bms_errors/state (null)
jk-bms/sensor/jk-bms_operation_mode/state Charging enabled;Discharging enabled;Balancer enabled
jk-bms/sensor/jk-bms_capacity_remaining_derived/state 266
jk-bms/sensor/jk-bms_battery_type/state Ternary Lithium
jk-bms/sensor/jk-bms_password/state **********
jk-bms/sensor/jk-bms_device_type/state Battery 
jk-bms/sensor/jk-bms_total_runtime_formatted/state 54d 5h
jk-bms/sensor/jk-bms_software_version/state 10.XW_S10.07___
jk-bms/sensor/jk-bms_manufacturer/state Battery #2

```

syssi, 2022-09-23:
This part of the diff doesn't look good to me:

```
 ota:
 logger:
-  baud_rate: 0
+  baud_rate: 115200
+  level: DEBUG
+
```

If you set the `baud_rate` of the `logger` component the logger starts to write log data to `GPIO1/GPIO3`. In other words: You are writing log data into your BMS if the `uart` component is also attached to `GPIO1/GPIO3`. Please set the `baud_rate` to `0` if the BMS is attached to `GPIO1/3`.

preussal, 2022-09-25:
Hi Syssi,

yes i removed this.
this is my working config

```
substitutions:
  name: jk-bms2
  device_description: "Monitor and control a JK-BMS via UART-TTL"
  external_components_source: github://syssi/esphome-jk-bms@main
  tx_pin: GPIO1
  rx_pin: GPIO3

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
#    version: latest
    version: 1.2.0

esp8266:
  board: d1_mini

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
#  manual_ip:
#    # Set this to the IP of the ESP
#    static_ip: 192.168.1.240
#    # Set this to the IP address of the router. Often ends with .1
#    gateway: 192.168.1.1
#    # The subnet of the network. 255.255.255.0 works for most home networks.
#    subnet: 255.255.255.0

ota:
logger:
  baud_rate: 0
  level: INFO

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  port: 1883
  id: mqtt_client

uart:
  id: uart0
  baud_rate: 115200
  rx_buffer_size: 384
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}

jk_modbus:
  id: modbus0
  uart_id: uart0
  rx_timeout: 50ms

jk_bms:
  id: bms0
  jk_modbus_id: modbus0

binary_sensor:
  - platform: jk_bms
    balancing:
[... 205 lines cut ...]
Well after about 1 hour he didn't do it anymore and grilled the ESP. And unfortunately also the TTL output of the JK BMS.

Please take it in your documentary that people have to be careful. Use only 3.3V and best install a digital insulation between the ESP and the BMS.
The BMS has installed a digital insulation, but to change it you have to open the BMS.

Now I'm measuring between the TX and TX pins in the off condition 4 mega ohm. Before that, it was around 60 kilos.
Here I grilled the digital insulation.

Here is the board I built. While the ESP was hung on the BMS via USB power.

[image]
[image]

First of all, a DC-DC Step Down Converter
DC-DC Buck Step Down 09V-1220V (80max) TO 5V 3A
[image]

Then a small DC-DC step down for 3.3 volts
[image]

A digital insulation is definitely coming to the next version.

But now I have to wait 3 months before I get back to my battery. In the meantime, I can get the parts for the two BMS.

[image]

preussal, 2022-09-25:
and here the last output from the grilled ESP
```
jk-bms/sensor/jk-bms_cell_voltage_1/state 3.310
jk-bms/sensor/jk-bms_cell_voltage_2/state 3.307
jk-bms/sensor/jk-bms_cell_voltage_3/state 3.307
jk-bms/sensor/jk-bms_cell_voltage_4/state 3.311
jk-bms/sensor/jk-bms_cell_voltage_5/state 3.312
jk-bms/sensor/jk-bms_cell_voltage_6/state 3.307
jk-bms/sensor/jk-bms_cell_voltage_7/state 3.307
jk-bms/sensor/jk-bms_cell_voltage_8/state 3.310
jk-bms/sensor/jk-bms_cell_voltage_9/state 3.312
jk-bms/sensor/jk-bms_cell_voltage_10/state 3.312
jk-bms/sensor/jk-bms_cell_voltage_11/state 3.308
jk-bms/sensor/jk-bms_cell_voltage_12/state 3.307
jk-bms/sensor/jk-bms_cell_voltage_13/state 3.310
jk-bms/sensor/jk-bms_cell_voltage_14/state 3.311
jk-bms/sensor/jk-bms_cell_voltage_15/state 3.312
jk-bms/sensor/jk-bms_cell_voltage_16/state 3.312
jk-bms/sensor/jk-bms_min_cell_voltage/state 3.307
jk-bms/sensor/jk-bms_max_cell_voltage/state 3.312
jk-bms/sensor/jk-bms_min_voltage_cell/state 2
jk-bms/sensor/jk-bms_max_voltage_cell/state 5
jk-bms/sensor/jk-bms_delta_cell_voltage/state 0.005
jk-bms/sensor/jk-bms_average_cell_voltage/state 3.310
jk-bms/sensor/jk-bms_power_tube_temperature/state 32
jk-bms/sensor/jk-bms_temperature_sensor_1/state 30
jk-bms/sensor/jk-bms_temperature_sensor_2/state 29
jk-bms/sensor/jk-bms_total_voltage/state 52.95
jk-bms/sensor/jk-bms_current/state -17.15
jk-bms/sensor/jk-bms_power/state -908.09
jk-bms/sensor/jk-bms_charging_power/state 0.00
jk-bms/sensor/jk-bms_discharging_power/state 908.09
jk-bms/sensor/jk-bms_capacity_remaining/state 85
jk-bms/sensor/jk-bms_capacity_remaining_derived/state 238
jk-bms/sensor/jk-bms_temperature_sensors/state 2
jk-bms/sensor/jk-bms_charging_cycles/state 16
jk-bms/sensor/jk-bms_total_charging_cycle_capacity/state 4608
jk-bms/sensor/jk-bms_battery_strings/state 16
jk-bms/sensor/jk-bms_errors_bitmask/state 0
jk-bms/sensor/jk-bms_operation_mode_bitmask/state 3
jk-bms/sensor/jk-bms_total_battery_capacity_setting/state 280
jk-bms/sensor/jk-bms_device_address/state 1
jk-bms/sensor/jk-bms_sleep_wait_time/state 10
jk-bms/sensor/jk-bms_alarm_low_volume/state 20
jk-bms/sensor/jk-bms_total_runtime/state 1331
jk-bms/sensor/jk-bms_actual_battery_capacity/state 256
jk-bms/sensor/jk-bms_operation_mode/state Charging enabled;Discharging enabled
jk-bms/sensor/jk-bms_battery_type/state Ternary Lithium
jk-bms/sensor/jk-bms_device_type/state Battery 
jk-bms/sensor/jk-bms_software_version/state 10.XW_S10.07___
jk-bms/sensor/jk-bms_manufacturer/state Battery #2
jk-bms/sensor/jk-bms_total_runtime_formatted/state 55d 11h
jk-bms/binary_sensor/jk-bms_charging/state ON
jk-bms/binary_sensor/jk-bms_charging_switch/state ON
jk-bms/binary_sensor/jk-bms_discharging/state ON
jk-bms/binary_sensor/jk-bms_discharging_switch/state ON
jk-bms/binary_sensor/jk-bms_balancing/state OFF
jk-bms/binary_sensor/jk-bms_balancing_switch/state ON
jk-bms/binary_sensor/jk-bms_dedicated_charger_switch/state OFF
jk-bms/debug [W][jk_modbus:050]: Invalid header
jk-bms/status offline
```

syssi, 2022-09-25:
Sorry for your loss! Are you able to replace the optical isolator on your BMS? I could point you to the location and provide some measurements if needed. Could you explain your schematics of the power supply in some more words? I would like to understand: What's the VIN of the big DC/DC converter? How did you power the big converter?

syssi, 2022-09-25:
Could you measure the resistance between B- and GND of the UART-TTL port? It should be 0 Ohm.

preussal, 2022-09-26:
Hi,
i measure the resistance Between RX and GND and TX and GND.
Both have >4Mega Ohm 

I did that with both BMS. BMS switched off. Separated from the battery and then measured.

Now I switched to the ES32 and read the data Bluetooth.

**BMS 1**
```
jk-bms1-ble/sensor/cell_voltage_1/state 3.456
jk-bms1-ble/sensor/cell_resistance_1/state 0.064
jk-bms1-ble/sensor/cell_voltage_2/state 3.456
jk-bms1-ble/sensor/cell_resistance_2/state 0.062
jk-bms1-ble/sensor/cell_voltage_3/state 3.456
jk-bms1-ble/sensor/cell_resistance_3/state 0.063
jk-bms1-ble/sensor/cell_voltage_4/state 3.456
jk-bms1-ble/sensor/cell_resistance_4/state 0.064
jk-bms1-ble/sensor/cell_voltage_5/state 3.457
jk-bms1-ble/sensor/cell_resistance_5/state 0.063
jk-bms1-ble/sensor/cell_voltage_6/state 3.456
jk-bms1-ble/sensor/cell_resistance_6/state 0.069
jk-bms1-ble/sensor/cell_voltage_7/state 3.457
jk-bms1-ble/sensor/cell_resistance_7/state 0.070
jk-bms1-ble/sensor/cell_voltage_8/state 3.456
jk-bms1-ble/sensor/cell_resistance_8/state 0.065
jk-bms1-ble/sensor/cell_voltage_9/state 3.457
jk-bms1-ble/sensor/cell_resistance_9/state 0.066
jk-bms1-ble/sensor/cell_voltage_10/state 3.457
jk-bms1-ble/sensor/cell_resistance_10/state 0.066
jk-bms1-ble/sensor/cell_voltage_11/state 3.454
jk-bms1-ble/sensor/cell_resistance_11/state 0.071
jk-bms1-ble/sensor/cell_voltage_12/state 3.452
jk-bms1-ble/sensor/cell_resistance_12/state 0.068
jk-bms1-ble/sensor/cell_voltage_13/state 3.457
jk-bms1-ble/sensor/cell_resistance_13/state 0.065
jk-bms1-ble/sensor/cell_voltage_14/state 3.457
jk-bms1-ble/sensor/cell_resistance_14/state 0.064
jk-bms1-ble/sensor/cell_voltage_15/state 3.456
jk-bms1-ble/sensor/cell_resistance_15/state 0.063
jk-bms1-ble/sensor/cell_voltage_16/state 3.456
jk-bms1-ble/sensor/cell_resistance_16/state 0.062
jk-bms1-ble/sensor/min_cell_voltage/state 3.452
jk-bms1-ble/sensor/max_cell_voltage/state 3.457
jk-bms1-ble/debug [I][jk_bms_ble:445]: Enabled cells: FF FF 00 00
jk-bms1-ble/sensor/average_cell_voltage/state 3.456
jk-bms1-ble/sensor/delta_cell_voltage/state 0.004
jk-bms1-ble/sensor/max_voltage_cell/state 3
jk-bms1-ble/sensor/min_voltage_cell/state 11
jk-bms1-ble/debug [I][jk_bms_ble:474]: Wire resistance warning: 00 00 00 00
jk-bms1-ble/sensor/total_voltage/state 55.29
jk-bms1-ble/sensor/current/state -0.38
jk-bms1-ble/sensor/power/state -20.85
jk-bms1-ble/sensor/charging_power/state 0.00
jk-bms1-ble/sensor/discharging_power/state 20.85
jk-bms1-ble/sensor/temperature_sensor_1/state 30
jk-bms1-ble/sensor/temperature_sensor_2/state 31
jk-bms1-ble/sensor/power_tube_temperature/state 35
jk-bms1-ble/sensor/errors_bitmask/state 0
jk-bms1-ble/sensor/errors/state (null)
jk-bms1-ble/sensor/balancing_current/state 0.00
jk-bms1-ble/sensor/state_of_charge/state 98
jk-bms1-ble/sensor/capacity_remaining/state 277
jk-bms1-ble/sensor/total_battery_capacity_setting/state 280
jk-bms1-ble/sensor/charging_cycles/state 13
jk-bms1-ble/sensor/total_charging_cycle_capacity/state 3805.640
jk-bms1-ble/sensor/total_runtime/state 5210590
```

**BMS 2**
```
jk-bms2-ble/sensor/cell_voltage_1/state 3.457
jk-bms2-ble/sensor/cell_resistance_1/state 0.062
jk-bms2-ble/sensor/cell_voltage_2/state 3.457
jk-bms2-ble/sensor/cell_resistance_2/state 0.061
jk-bms2-ble/sensor/cell_voltage_3/state 3.457
jk-bms2-ble/sensor/cell_resistance_3/state 0.060
jk-bms2-ble/sensor/cell_voltage_4/state 3.456
jk-bms2-ble/sensor/cell_resistance_4/state 0.060
jk-bms2-ble/sensor/cell_voltage_5/state 3.457
jk-bms2-ble/sensor/cell_resistance_5/state 0.059
jk-bms2-ble/sensor/cell_voltage_6/state 3.455
jk-bms2-ble/sensor/cell_resistance_6/state 0.060
jk-bms2-ble/sensor/cell_voltage_7/state 3.455
jk-bms2-ble/sensor/cell_resistance_7/state 0.061
jk-bms2-ble/sensor/cell_voltage_8/state 3.456
jk-bms2-ble/sensor/cell_resistance_8/state 0.064
jk-bms2-ble/sensor/cell_voltage_9/state 3.457
jk-bms2-ble/sensor/cell_resistance_9/state 0.064
jk-bms2-ble/sensor/cell_voltage_10/state 3.456
jk-bms2-ble/sensor/cell_resistance_10/state 0.062
jk-bms2-ble/sensor/cell_voltage_11/state 3.457
jk-bms2-ble/sensor/cell_resistance_11/state 0.059
jk-bms2-ble/sensor/cell_voltage_12/state 3.456
jk-bms2-ble/sensor/cell_resistance_12/state 0.059
jk-bms2-ble/sensor/cell_voltage_13/state 3.455
jk-bms2-ble/sensor/cell_resistance_13/state 0.059
jk-bms2-ble/sensor/cell_voltage_14/state 3.456
jk-bms2-ble/sensor/cell_resistance_14/state 0.060
jk-bms2-ble/sensor/cell_voltage_15/state 3.455
jk-bms2-ble/sensor/cell_resistance_15/state 0.059
jk-bms2-ble/sensor/cell_voltage_16/state 3.455
jk-bms2-ble/sensor/cell_resistance_16/state 0.059
jk-bms2-ble/sensor/min_cell_voltage/state 3.455
jk-bms2-ble/sensor/max_cell_voltage/state 3.457
jk-bms2-ble/debug [I][jk_bms_ble:445]: Enabled cells: FF FF 00 00
jk-bms2-ble/sensor/average_cell_voltage/state 3.456
jk-bms2-ble/sensor/delta_cell_voltage/state 0.004
jk-bms2-ble/sensor/max_voltage_cell/state 3
jk-bms2-ble/sensor/min_voltage_cell/state 13
jk-bms2-ble/debug [I][jk_bms_ble:474]: Wire resistance warning: 00 00 00 00
jk-bms2-ble/sensor/total_voltage/state 55.30
jk-bms2-ble/sensor/current/state -0.43
jk-bms2-ble/sensor/power/state -24.00
jk-bms2-ble/sensor/charging_power/state 0.00
jk-bms2-ble/sensor/discharging_power/state 24.00
jk-bms2-ble/sensor/temperature_sensor_1/state 30
jk-bms2-ble/sensor/temperature_sensor_2/state 31
jk-bms2-ble/sensor/power_tube_temperature/state 34
jk-bms2-ble/sensor/errors_bitmask/state 0
jk-bms2-ble/sensor/errors/state (null)
jk-bms2-ble/sensor/balancing_current/state 0.00
jk-bms2-ble/sensor/state_of_charge/state 98
jk-bms2-ble/sensor/capacity_remaining/state 277
jk-bms2-ble/sensor/total_battery_capacity_setting/state 280
jk-bms2-ble/sensor/charging_cycles/state 17
jk-bms2-ble/sensor/total_charging_cycle_capacity/state 4846.222
jk-bms2-ble/sensor/total_runtime/state 4951060
```

I adjusted the Yaml that not all sensors are queried.

Now I only have to configure Telegraf to write the data from the MQTT to InfluxDB.

preussal, 2022-09-26:
i uses the esp32-ble-exampe.yaml to create my config with the Sensors what i need.

I think that was even more than 6 mega ohm between RX and GND.
I can't reach the BMS now because I have already closed the rack.
I'll look at the TTL problem when I'm back home.

[image]

preussal, 2022-09-26:
Thank you. I make a design copy from the off-grid garage.

the display ports are working all the time.
 I first tried it with the JK TTL to rs485 adapter.  but apparently I have those with 9200 baud.  the functions don't.  then I bought some TTL adapters and then only got data from the lower  BMS.  I may have already grilled the upper BMS while testing.
but after I didn't get much further with the Python script from .  I decided to do something similar to what I built for my jdb BMS.  an esp12 as Modbus to TTL over network.  but then I came across your project.
I think it's just the optocoupler that's broken.  you can replace it, maybe even a two resistors.  but I need time for that.  which I don't have right now.  and smb soldering is no fun without the right tools either.

preussal, 2022-09-26:
but I still have a question.  can you make the mttq output json compliant.  I can't get Telegraf to write the data to the influxdb.  somehow he doesn't want to parse that properly with the topic

syssi, 2022-09-26:
I think so too. You have fried one of these components:

[image]

syssi, 2022-09-26:
I use the official MQTT component from ESPHome. IMO the output isn't JSON because each measurement is available on a dedicated topic as int, float or text. You have to tell telegraf the correct `data_format` per topic.

syssi, 2022-09-26:
This is what I'm talking about:

```
[[inputs.mqtt_consumer]]
  name_override = "my_integer"
  topics = ["/XXX"]
  data_format = "value"
  data_type = "integer"

[[inputs.mqtt_consumer]]
  name_override = "my_string"
  topics = ["/SSS"]
  data_format = "value"
  data_type = "string"
```

preussal, 2022-09-26:
I totally forgot that I use different databases for different inputs.
Therefore, the data all went into the false database. :(

```
[[inputs.mqtt_consumer]]
  # Override mqtt_consumer to 
  name_override = "jk-bms1"

  servers = ["tcp://127.0.0.1:1883"]
  qos = 2
  topics = ["jk-bms1-ble/sensor/#"]

  data_format = "value"
  data_type = "float"

  persistent_session = false

[[outputs.influxdb]]
 # Only accept measurements with
  namepass = ["jk-bms1"]
 
  urls = ["http://127.0.0.1:8086"]
  database = "jk-bms1"
  skip_database_creation = true
  username = "XYZ"
  password = "123"
```

```
[[inputs.mqtt_consumer]]
  # Override mqtt_consumer to 
  name_override = "jk-bms2"

  servers = ["tcp://127.0.0.1:1883"]
  qos = 2
  topics = ["jk-bms2-ble/sensor/#"]

  data_format = "value"
  data_type = "float"

  persistent_session = false

[[outputs.influxdb]]
  # Only accept measurements with
  namepass = ["jk-bms2"]

  urls = ["http://127.0.0.1:8086"]
  database = "jk-bms2"
  skip_database_creation = true
  username = "XYZ"
  password = "123"
```

syssi, 2022-09-26:
If I remember correctly you can find warnings/errors at the telegraf log if telegraf is unable to interpret a received value/measurement.

preussal, 2022-09-26:
Do you have an idea how I can overwrite the topic in Telegraf.

I have this key at Show Series in the InfluxDB

```
jk-bms1,host=powermeterpi4.th.secretisland.de,topic=jk-bms1-ble/sensor/average_cell_voltage/state
jk-bms1,host=powermeterpi4.th.secretisland.de,topic=jk-bms1-ble/sensor/balancing_current/state
jk-bms1,host=powermeterpi4.th.secretisland.de,topic=jk-bms1-ble/sensor/capacity_remaining/state
jk-bms1,host=powermeterpi4.th.secretisland.de,topic=jk-bms1-ble/sensor/cell_resistance_1/state
jk-bms1,host=powermeterpi4.th.secretisland.de,topic=jk-bms1-ble/sensor/cell_resistance_10/state
```
i want 
```
jk-bms1,host=powermeterpi4.th.secretisland.de,topic=average_cell_voltage
jk-bms1,host=powermeterpi4.th.secretisland.de,topic=balancing_current
jk-bms1,host=powermeterpi4.th.secretisland.de,topic=capacity_remaining
jk-bms1,host=powermeterpi4.th.secretisland.de,topic=cell_resistance_1
jk-bms1,host=powermeterpi4.th.secretisland.de,topic=cell_resistance_10
```

preussal, 2022-09-26:
I'm just saying that as a sensor, and the sensor day is the name of the sensor

```
[[inputs.mqtt_consumer]]
  # Override mqtt_consumer to 
  name_override = "jk-bms1"

  servers = ["tcp://127.0.0.1:1883"]
  persistent_session = false
  qos = 2
  topics = ["jk-bms1-ble/sensor/#"]

  data_format = "value"
  data_type = "float"

[[inputs.mqtt_consumer.topic_parsing]]
  data_format = "value"
  data_type = "float"

  topic = "jk-bms1-ble/sensor/+/state"
  tags = "_/_/sensor/_"
  measurement = "_/_/measurement/_"
  fields = "_/_/_/state"

[[outputs.influxdb]]
  # Only accept measurements with
  namepass = ["jk-bms1"]

  urls = ["http://127.0.0.1:8086"]
  database = "jk-bms1"
  skip_database_creation = true
  username = ""
  password = ""
```
InfluxDB
```
jk-bms1,host=powermeterpi4.th.secretisland.de,sensor=average_cell_voltage,topic=jk-bms1-ble/sensor/average_cell_voltage/state
jk-bms1,host=powermeterpi4.th.secretisland.de,sensor=balancing_current,topic=jk-bms1-ble/sensor/balancing_current/state
jk-bms1,host=powermeterpi4.th.secretisland.de,sensor=capacity_remaining,topic=jk-bms1-ble/sensor/capacity_remaining/state
jk-bms1,host=powermeterpi4.th.secretisland.de,sensor=cell_resistance_1,topic=jk-bms1-ble/sensor/cell_resistance_1/state
jk-bms1,host=powermeterpi4.th.secretisland.de,sensor=cell_resistance_10,topic=jk-bms1-ble/sensor/cell_resistance_10/state
jk-bms1,host=powermeterpi4.th.secretisland.de,sensor=cell_resistance_11,topic=jk-bms1-ble/sensor/cell_resistance_11/state

```
And then it's nice in the Grafana

[image]

syssi, 2022-09-26:
I assume you are looking for something like this:

```
[[processors.rename]]
  namepass = "cpu"
  [[processors.rename.field]]
    from = "usage_idle"
    to = "idle"
```

But I must admit my telegraf knowledge is pretty limited.

preussal, 2022-09-26:
> I assume you are looking for something like this:
> 
> ```
> [[processors.rename]]
>   namepass = "cpu"
>   [[processors.rename.field]]
>     from = "usage_idle"
>     to = "idle"
> ```
> 
> But I must admit my telegraf knowledge is pretty limited.

i try this. but not working i got a solution

syssi, 2022-09-26:
It looks like your solution is better / more generic.

closed 2022-10-09 by syssi