syssi/esphome-jk-bms #179: Splitting the Codebase into the reading and Esphome part?

kinsi55, 2022-09-23:
My ESP handles many tasks, and one of them is supposed to be geting / publishing and using the values from the BMS. With how the classes of this component are structured, as far as I can tell, using just the part that handles reading the values isnt easily possible.

I think splitting it into two codebases coulkd make sense?

syssi, 2022-09-23:
Do you want to use this implementation without using ESPHome?

kinsi55, 2022-09-23:
Yeah that was the idea, integrating this component / reading the BMS with my other code running on the same ESP

syssi, 2022-09-23:
I must admit my C++ skills are too limited to prepare and maintain a library. The target group for this project is so small that it is hardly worth the effort.

kinsi55, 2022-09-23:
Understandable, no worries. I was just mainly trying to not reinvent the wheel. I guess I will have to dig through the code and try to figure out the communication to do that afterall :D

syssi, 2022-09-23:
Feel free to copy the important parts and ask questions if something isn't obvious.

closed 2022-09-23 by syssi