syssi/esphome-jk-bms #186: Jkbms-ble branch have no more switch working

Bandit-17, 2022-10-09:
I just updated last ESPhome dev and now gor error while compile the Jkbms using the BLE branch.

Here is the log of compilation

[issue jkbmsble.txt](https://github.com/syssi/esphome-jk-bms/files/9741987/issue.jkbmsble.txt)

syssi, 2022-10-09:
Are you sure you are using the most recent version of ESPHome? It doesn't look like. 

I tried this to reproduce the issue:

```
$ pip3 install esphome -U
Collecting esphome
  Downloading esphome-2022.9.4-py2.py3-none-any.whl (2.4 MB)

$ mkdir test
$ cd test
$ wget https://raw.githubusercontent.com/syssi/esphome-jk-bms/main/esp32-ble-example.yaml
$ echo -e "wifi_ssid: ssid\nwifi_password: password\nmqtt_host: host\nmqtt_username: username\nmqtt_password: password" > secrets.yaml
$ ~/.local/bin/esphome run esp32-ble-example.yaml
Indexing .pioenvs/jk-bms/bootloader/esp-idf/spi_flash/libspi_flash.a
Linking .pioenvs/jk-bms/bootloader.elf
Building .pioenvs/jk-bms/bootloader.bin
Linking .pioenvs/jk-bms/firmware.elf
RAM:   [==        ]  16.0% (used 52456 bytes from 327680 bytes)
Flash: [=======   ]  67.2% (used 1232441 bytes from 1835008 bytes)
Building .pioenvs/jk-bms/firmware.bin
esp32_create_combined_bin([".pioenvs/jk-bms/firmware.bin"], [".pioenvs/jk-bms/firmware.elf"])
Wrote 0x13cfc0 bytes to file /tmp/test/.esphome/build/jk-bms/.pioenvs/jk-bms/firmware-factory.bin, ready to flash to offset 0x0
================================================================== [SUCCESS] Took 190.26 seconds ==================================================================

INFO Successfully compiled program.
INFO Resolving IP address of jk-bms.local
```

Bandit-17, 2022-10-09:
(base) PS D:\HOMEASSISTANT\homeassistant\PVbrain> pip3 install esphome -U
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: esphome in c:\users\sougnoux\appdata\roaming\python\python39\site-packages (2022.10.0.dev0)
Requirement already satisfied: aioesphomeapi==10.13.0 in c:\users\sougnoux\appdata\roaming\python\python39\site-packages (from esphome) (10.13.0)
Requirement already satisfied: tzdata>=2021.1 in c:\programdata\anaconda3\lib\site-packages (from esphome) (2022.1)
Requirement already satisfied: esptool==3.3.1 in c:\programdata\anaconda3\lib\site-packages (from esphome) (3.3.1)
Requirement already satisfied: esphome-dashboard==20220925.0 in c:\users\sougnoux\appdata\roaming\python\python39\site-packages (from esphome) (20220925.0)
Requirement already satisfied: colorama==0.4.5 in c:\programdata\anaconda3\lib\site-packages (from esphome) (0.4.5)
Requirement already satisfied: tornado==6.1 in c:\programdata\anaconda3\lib\site-packages (from esphome) (6.1)
Requirement already satisfied: pyserial==3.5 in c:\programdata\anaconda3\lib\site-packages (from esphome) (3.5)
Requirement already satisfied: platformio==6.0.2 in c:\programdata\anaconda3\lib\site-packages (from esphome) (6.0.2)
Requirement already satisfied: click==8.1.3 in c:\programdata\anaconda3\lib\site-packages (from esphome) (8.1.3)
Requirement already satisfied: kconfiglib==13.7.1 in c:\programdata\anaconda3\lib\site-packages (from esphome) (13.7.1)
Requirement already satisfied: paho-mqtt==1.6.1 in c:\programdata\anaconda3\lib\site-packages (from esphome) (1.6.1)
Requirement already satisfied: tzlocal==4.2 in c:\programdata\anaconda3\lib\site-packages (from esphome) (4.2)
Requirement already satisfied: PyYAML==6.0 in c:\programdata\anaconda3\lib\site-packages (from esphome) (6.0)
Requirement already satisfied: zeroconf==0.39.1 in c:\users\sougnoux\appdata\roaming\python\python39\site-packages (from esphome) (0.39.1)
Requirement already satisfied: voluptuous==0.13.1 in c:\programdata\anaconda3\lib\site-packages (from esphome) (0.13.1)
Requirement already satisfied: protobuf<4.0,>=3.12.2 in c:\programdata\anaconda3\lib\site-packages (from aioesphomeapi==10.13.0->esphome) (3.19.1)
Requirement already satisfied: noiseprotocol<1.0,>=0.3.1 in c:\programdata\anaconda3\lib\site-packages (from aioesphomeapi==10.13.0->esphome) (0.3.1)
Requirement already satisfied: reedsolo<=1.5.4,>=1.5.3 in c:\programdata\anaconda3\lib\site-packages (from esptool==3.3.1->esphome) (1.5.4)
Requirement already satisfied: bitstring>=3.1.6 in c:\programdata\anaconda3\lib\site-packages (from esptool==3.3.1->esphome) (3.1.9)
Requirement already satisfied: ecdsa>=0.16.0 in c:\programdata\anaconda3\lib\site-packages (from esptool==3.3.1->esphome) (0.18.0)
Requirement already satisfied: cryptography>=2.1.4 in c:\programdata\anaconda3\lib\site-packages (from esptool==3.3.1->esphome) (3.4.8)
Requirement already satisfied: aiofiles==0.8.* in c:\programdata\anaconda3\lib\site-packages (from platformio==6.0.2->esphome) (0.8.0)
Requirement already satisfied: marshmallow==3.* in c:\programdata\anaconda3\lib\site-packages (from platformio==6.0.2->esphome) (3.17.0)
Requirement already satisfied: ajsonrpc==1.* in c:\programdata\anaconda3\lib\site-packages (from platformio==6.0.2->esphome) (1.2.0)
Requirement already satisfied: semantic-version==2.10.* in c:\programdata\anaconda3\lib\site-packages (from platformio==6.0.2->esphome) (2.10.0)
Requirement already satisfied: uvicorn==0.17.* in c:\programdata\anaconda3\lib\site-packages (from platformio==6.0.2->esphome) (0.17.6)
Requirement already satisfied: starlette==0.20.* in c:\programdata\anaconda3\lib\site-packages (from platformio==6.0.2->esphome) (0.20.4)
Requirement already satisfied: wsproto==1.1.* in c:\programdata\anaconda3\lib\site-packages (from platformio==6.0.2->esphome) (1.1.0)
Requirement already satisfied: bottle==0.12.* in c:\programdata\anaconda3\lib\site-packages (from platformio==6.0.2->esphome) (0.12.21)
Requirement already satisfied: pyelftools<1,>=0.27 in c:\programdata\anaconda3\lib\site-packages (from platformio==6.0.2->esphome) (0.28)
Requirement already satisfied: tabulate==0.8.* in c:\programdata\anaconda3\lib\site-packages (from platformio==6.0.2->esphome) (0.8.9)
Requirement already satisfied: requests==2.* in c:\programdata\anaconda3\lib\site-packages (from platformio==6.0.2->esphome) (2.27.1)
Requirement already satisfied: packaging>=17.0 in c:\programdata\anaconda3\lib\site-packages (from marshmallow==3.*->platformio==6.0.2->esphome) (21.3)
Requirement already satisfied: certifi>=2017.4.17 in c:\programdata\anaconda3\lib\site-packages (from requests==2.*->platformio==6.0.2->esphome) (2021.10.8)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in c:\programdata\anaconda3\lib\site-packages (from requests==2.*->platformio==6.0.2->esphome) (1.26.9)
Requirement already satisfied: idna<4,>=2.5 in c:\programdata\anaconda3\lib\site-packages (from requests==2.*->platformio==6.0.2->esphome) (3.3)
Requirement already satisfied: charset-normalizer~=2.0.0 in c:\programdata\anaconda3\lib\site-packages (from requests==2.*->platformio==6.0.2->esphome) (2.0.4)
Requirement already satisfied: anyio<5,>=3.4.0 in c:\programdata\anaconda3\lib\site-packages (from starlette==0.20.*->platformio==6.0.2->esphome) (3.5.0)
[... cut ...]
Requirement already satisfied: cffi>=1.12 in c:\programdata\anaconda3\lib\site-packages (from cryptography>=2.1.4->esptool==3.3.1->esphome) (1.15.0)
Requirement already satisfied: pycparser in c:\programdata\anaconda3\lib\site-packages (from cffi>=1.12->cryptography>=2.1.4->esptool==3.3.1->esphome) (2.21)
Requirement already satisfied: six>=1.9.0 in c:\programdata\anaconda3\lib\site-packages (from ecdsa>=0.16.0->esptool==3.3.1->esphome) (1.16.0)
Requirement already satisfied: pyparsing!=3.0.5,>=2.0.2 in c:\programdata\anaconda3\lib\site-packages (from packaging>=17.0->marshmallow==3.*->platformio==6.0.2->esphome) (3.0.4)

SeByDocKy, 2022-10-09:
Got also the error in compilation.... (last dev installed)

SeByDocKy, 2022-10-09:
Fixed .... need first to uninstall esphome (pip uninstall esphome), reinstall last dev.... compiling fine then.... False alert :)

closed 2022-10-09 by Bandit-17